openwrt/openwrt #12829: parse_elf_properties failed when elf_ppnt->p_type == PT_GNU_PROPERTY for ld-linux-ld-linux-aarch64.so.1

gmzhuo, 2023-06-07:
### Describe the bug

When elf_ppnt->p_type == PT_GNU_PROPERTY for ld-linux-ld-linux-aarch64.so.1, "if (phdr->p_filesz > sizeof(note))" would be true, then failed with return -ENOEXEC;
The patch target/linux/generic/pending-5.15/130-binfmt_elf-dynamically-allocate-note.data-in-parse_e.patch changed union note 
`
 	union {
 		struct elf_note nhdr;
-		char data[NOTE_DATA_SZ];
+		char *data;
 	} note;
`

but still check the phdr->p_filesz with code:
`
/* If the properties are crazy large, that's too bad (for now): */
	if (phdr->p_filesz > sizeof(note))
		return -ENOEXEC;
`

So if phdr->p_filesz bigger(unfortunately, it is) than "struct elf_note nhdr", it failed.

### OpenWrt version

Any version with patch 130-binfmt_elf-dynamically-allocate-note.data-in-parse_e.patch if pnt->p_type == PT_GNU_PROPERTY 

### OpenWrt target/subtarget

Any target if there's a shared library which pnt->p_type == PT_GNU_PROPERTY 

### Device

qihoo 360t7

### Image kind

Self-built image

### Steps to reproduce

I build the rootfs with yocto for 360t7, kernel source from openwrt, then device failed to boot.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-06-07:
Sizeof(* char)  pointer is 4 or 8 octets.

gmzhuo, 2023-06-07:
> Sizeof(* char) pointer is 4 or 8 octets.

(Sizeof(union note) = sizeof(struct elf_note)) < (phdr->p_filesz = 32 for ld.so build from yocto)

Ansuel, 2023-06-07:
the old union size was actually `sizeof(char) * NOTE_DATA_SZ`

brada4, 2023-06-07:
Now it is sizeof(pointer) ie 1k less