openwrt/openwrt #12843: [openwrt-23.05] iptables fails to display DNAT rules (was: luci-app-upnp:"There are no active redirects",but miniupnpd works fine.)

wtfr-dot, 2023-06-07:
I built the latest OpenWrt-23.05, which includes dual WAN and Passwall support. I disabled nftables and compiled fw3 because mwan3 is incompatible with fw4 and Passwall. The debug logs of miniupnpd and the firewall both indicated that they were working fine. However, UPnP did not display any entries in the Luci interface. The same configuration worked perfectly in OpenWrt-22.03.

OpenWrt version information from system `/etc/openwrt_release`

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05-SNAPSHOT'
DISTRIB_REVISION='r23058-e2e7dd4450'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 23.05-SNAPSHOT r23058-e2e7dd4450'
DISTRIB_TAINTS='no-all mklibs'
[image]

wtfr-dot, 2023-06-08:
I installed ‘iptables-legacy’ and I disabled nft as much as possible.

brada4, 2023-06-09:
> mwan3 is incompatible with fw4

Not really, just that package dependencies pull iptables-legacy and makes mess. Once iptables command is iptables-nft it should work fine.

wtfr-dot, 2023-06-10:
> > mwan3 is incompatible with fw4
> 
> Not really, just that package dependencies pull iptables-legacy and makes mess. Once iptables command is iptables-nft it should work fine.

I mean that mwan3 is incompatible with passwall when I chose nftables. I built OpenWrt 23.05, which includes nftables, mwan3, and passwall. It's basically working, but with a few bugs,but UPnP works perfectly. I have verified that mwan3 causes passwall to work unstably. 
Here is the link to my issue: https://github.com/xiaorouji/openwrt-passwall/discussions/2601. 
I also had another issue earlier, which can be found here: https://github.com/xiaorouji/openwrt-passwall/issues/2330.

brada4, 2023-06-10:
> /usr/share/passwall/nftables.sh: not found

Seems nftables support is incomplete by at least that script. If there is a way to force it to iptables(-nft) in presence if nft and fw4?

wtfr-dot, 2023-06-10:
> > /usr/share/passwall/nftables.sh: not found
> 
> Seems nftables support is incomplete by at least that script. If there is a way to force it to iptables(-nft) in presence if nft and fw4?

This bug had fix yesterday.
As of now, all bugs in Passwall have been fixed. MWAN3 operates on iptables while Passwall operates on nftables. After restarting the firewall, Passwall doesn't have TCP and UDP connection records. Upon investigation, it was found that the issue was caused by MWAN3. After uninstalling MWAN3, the network returned to normal.

brada4, 2023-06-10:
Thus there is no need for iptables-legacy anymore.