openwrt/openwrt #12849: kernel package is not in the Package lists

rvandegrift, 2023-06-08:
### Describe the bug

I recently ran into the issue described in the FAQ about satisfying kernel dependencies, except I was running a stable release (22.03.3).  This is described in detail in the forum at [1], but in short I was seeing errors like this:
```
root@gateway:~# opkg update
...
root@gateway:~# opkg install kmod-ipt-ipset
Unknown package 'kmod-ipt-ipset'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.161-1-978cadea017a004ae5b73fcf48befb86) for kmod-nf-reject
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-reject found, but incompatible with the architectures configured
...
```

In my case, the `kernel` package was not installed.  I don't know I ended up in this state- I haven't touched this box since the 22.03.3 release.  The fix was simple: I needed to grab and manually install the platforms `kernel` package.  But it took me a lot of digging to figure that out.

It seems like `opkg` could solve this itself - but `kernel` is not in the package list at [2].  That seems to only exist so opkg can have a kernel abi version to depend on, so I'd guess it'd be harmless to add.  And at least useful in some cases.

I didn't survey other targets or releases, so it's possible this is unique to rockchip/armv8 or 22.03.3.  I'm filing this as a bug, but if it's really a feature request, apologies.

Thanks,
Ross

[1] - https://forum.openwrt.org/t/on-22-03-3-stable-getting-cannot-satisfy-dependencies-kernel-is-not-compatible-errors/161953
[2] - https://downloads.openwrt.org/releases/22.03.3/targets/rockchip/armv8/packages/Packages 

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R4S

### Image kind

Official downloaded image

### Steps to reproduce

My router was missing just the package `kernel`.  I don't know how I got there, but I guess you could reproduce this with `opkg --nodeps remove kernel`.

From that state, `opkg install kmod-nf-reject` will produce the error.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-06-08:
The empty "kernel" package is included in both ext4 and squashfs images? Look in /rom/usr/lib/opkg/.

jow-, 2023-06-08:
This is intentional. The actual running kernel image cannot be upgraded and the `kernel` package you installed is an empty meta package. This empty package is withheld from the repo indexes to ensure that it can't be upgraded through opkg.

closed 2023-06-08 by jow-