openwrt/openwrt #12866: openssl fails to compile for mips64_octeonplus

MoetaYuko, 2023-06-10:
### Describe the bug

```
2023-06-09T21:08:06.2660787Z mips64-openwrt-linux-musl-gcc  -I. -Iinclude -fPIC -pthread -mabi=64 -mips3 -Wa,--noexecstack -Wall -O3 -Os -pipe -mno-branch-likely -march=octeon+ -mabi=64 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -ffile-prefix-map=/builder/build_dir/target-mips64_octeonplus_64_musl/openssl-1.1.1u=openssl-1.1.1u -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -ffunction-sections -fdata-sections -znow -zrelro -DOPENSSL_USE_NODELETE -DOPENSSL_PIC -DOPENSSL_BN_ASM_MONT -DSHA1_ASM -DSHA256_ASM -DSHA512_ASM -DAES_ASM -DPOLY1305_ASM -DOPENSSLDIR="\"/etc/ssl\"" -DENGINESDIR="\"/usr/lib/engines-1.1\"" -DNDEBUG -DOPENSSL_PREFER_CHACHA_OVER_GCM -DOPENSSL_SMALL_FOOTPRINT -I/builder/staging_dir/toolchain-mips64_octeonplus_64_gcc-11.2.0_musl/usr/include -I/builder/staging_dir/toolchain-mips64_octeonplus_64_gcc-11.2.0_musl/include/fortify -I/builder/staging_dir/toolchain-mips64_octeonplus_64_gcc-11.2.0_musl/include -MMD -MF apps/app_rand.d.tmp -MT apps/app_rand.o -c -o apps/app_rand.o apps/app_rand.c
2023-06-09T21:08:06.2758313Z cc1: error: '-mips3' conflicts with the other architecture options, which specify a mips64r2 processor
2023-06-09T21:08:06.2814831Z Assembler messages:
2023-06-09T21:08:06.2815441Z Error: -mips3 conflicts with the other architecture options, which imply -mips64r2
2023-06-09T21:08:06.2821676Z make[5]: *** [Makefile:707: apps/app_rand.o] Error 1
2023-06-09T21:08:06.2822838Z make[5]: Leaving directory '/builder/build_dir/target-mips64_octeonplus_64_musl/openssl-1.1.1u'
2023-06-09T21:08:06.2829411Z make[4]: *** [Makefile:174: all] Error 2
2023-06-09T21:08:06.2830486Z make[4]: Leaving directory '/builder/build_dir/target-mips64_octeonplus_64_musl/openssl-1.1.1u'
2023-06-09T21:08:06.2837223Z make[3]: *** [Makefile:401: /builder/build_dir/target-mips64_octeonplus_64_musl/openssl-1.1.1u/.built] Error 2
2023-06-09T21:08:06.2839388Z make[3]: Leaving directory '/builder/feeds/base/package/libs/openssl'
2023-06-09T21:08:06.2844307Z time: package/feeds/base/openssl/compile#1.75#0.40#2.80
2023-06-09T21:08:06.2854597Z     ERROR: package/feeds/base/openssl failed to build.
2023-06-09T21:08:06.2869331Z make[2]: *** [package/Makefile:120: package/feeds/base/openssl/compile] Error 1
2023-06-09T21:08:06.2872324Z make[2]: *** Waiting for unfinished jobs....
```

```
2023-06-09T21:21:19.9467743Z make[5]: Entering directory '/builder/build_dir/target-mips64_octeonplus_64_musl/openssl-3.0.9'
2023-06-09T21:21:20.0692181Z make[5]: Leaving directory '/builder/build_dir/target-mips64_octeonplus_64_musl/openssl-3.0.9'
2023-06-09T21:21:20.0932848Z make[5]: Entering directory '/builder/build_dir/target-mips64_octeonplus_64_musl/openssl-3.0.9'
2023-06-09T21:21:20.0938950Z mips64-openwrt-linux-musl-gcc  -I. -Iinclude -Iapps/include  -fPIC -pthread -mabi=64 -mips3 -Wa,--noexecstack -Wall -O3 -Os -pipe -mno-branch-likely -march=octeon+ -mabi=64 -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -ffile-prefix-map=/builder/build_dir/target-mips64_octeonplus_64_musl/openssl-3.0.9=openssl-3.0.9 -ffunction-sections -fdata-sections -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -znow -zrelro -DOPENSSL_USE_NODELETE -DOPENSSL_PIC -DOPENSSLDIR="\"/etc/ssl\"" -DENGINESDIR="\"/usr/lib/engines-3\"" -DMODULESDIR="\"/usr/lib/ossl-modules\"" -DOPENSSL_BUILDING_OPENSSL -DNDEBUG -DOPENSSL_PREFER_CHACHA_OVER_GCM -I/builder/staging_dir/toolchain-mips64_octeonplus_64_gcc-12.3.0_musl/usr/include -I/builder/staging_dir/toolchain-mips64_octeonplus_64_gcc-12.3.0_musl/include/fortify -I/builder/staging_dir/toolchain-mips64_octeonplus_64_gcc-12.3.0_musl/include -MMD -MF apps/lib/libapps-lib-app_libctx.d.tmp -MT apps/lib/libapps-lib-app_libctx.o -c -o apps/lib/libapps-lib-app_libctx.o apps/lib/app_libctx.c
2023-06-09T21:21:20.1107967Z cc1: error: '-mips3' conflicts with the other architecture options, which specify a mips64r2 processor
2023-06-09T21:21:20.1169438Z Assembler messages:
2023-06-09T21:21:20.1170140Z Error: -mips3 conflicts with the other architecture options, which imply -mips64r2
2023-06-09T21:21:20.1179488Z make[5]: *** [Makefile:2613: apps/lib/libapps-lib-app_libctx.o] Error 1
2023-06-09T21:21:20.1180150Z make[5]: Leaving directory '/builder/build_dir/target-mips64_octeonplus_64_musl/openssl-3.0.9'
2023-06-09T21:21:20.1180623Z make[4]: *** [Makefile:1590: build_sw] Error 2
2023-06-09T21:21:20.1181879Z make[4]: Leaving directory '/builder/build_dir/target-mips64_octeonplus_64_musl/openssl-3.0.9'
2023-06-09T21:21:20.1182542Z make[3]: *** [Makefile:434: /builder/build_dir/target-mips64_octeonplus_64_musl/openssl-3.0.9/.built] Error 2
2023-06-09T21:21:20.1183130Z make[3]: Leaving directory '/builder/feeds/base/package/libs/openssl'
2023-06-09T21:21:20.1193816Z time: package/feeds/base/openssl/compile#8.14#1.09#10.33
2023-06-09T21:21:20.1223976Z     ERROR: package/feeds/base/openssl failed to build.
2023-06-09T21:21:20.1248688Z make[2]: *** [package/Makefile:120: package/feeds/base/openssl/compile] Error 1
2023-06-09T21:21:20.1250148Z make[2]: *** Waiting for unfinished jobs....
```

### OpenWrt version

22.03 & 23.05 & master

### OpenWrt target/subtarget

mips64_octeonplus

### Device

sdk

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

MoetaYuko, 2023-06-10:
cc @hauke @AuthorReflex

AuthorReflex, 2023-06-10:
```
+++ openssl-3.0.8/Configure     2023-05-30 15:31:57.000000000 +0300
--- openssl-3.0.9/Configure     2023-02-07 16:43:33.000000000 +0300
@@ -1423,7 +1423,7 @@ if ($target =~ /^mingw/ && `$config{CC}
         }

 if ($target =~ /linux.*-mips/ && !$disabled{asm}
-        && !grep { $_ =~ /-m(ips|arch=)/ } (@{$config{CFLAGS}})) {
+        && !grep { $_ !~ /-m(ips|arch=)/ } (@{$config{CFLAGS}})) {
         # minimally required architecture flags for assembly modules
         my $value;
         $value = '-mips2' if ($target =~ /mips32/);
```
probably is a OpenSSL typo/mistake
with this patch, compiled wo errors
i mean on 1.1.1u is same problrm

commit 51f57e7c (2023-06-14): openssl: passing cflags to configure

openssl sets additional cflags in its configuration script. We need to
make it aware of our custom cflags to avoid adding conflicting cflags.

Fixes: #12866
PR #12899 (2023-06-14): openssl: passing cflags to configure
Fixes: #12866

cc @hauke @AuthorReflex
commit e9d2ff80 (2023-06-17): openssl: passing cflags to configure

openssl sets additional cflags in its configuration script. We need to
make it aware of our custom cflags to avoid adding conflicting cflags.

Fixes: #12866

(cherry picked from commit 51f57e7c2dd2799e34036ec74b3436bf490fade0)
PR #12903 (2023-06-17): [23.05] openssl: passing cflags to configure
openssl sets additional cflags in its configuration script. We need to make it aware of our custom cflags to avoid adding conflicting cflags.

Fixes: #12866

(cherry picked from commit 51f57e7c2dd2799e34036ec74b3436bf490fade0)
closed 2023-06-14 by openwrt-bot