openwrt/openwrt #12868: pppoe can't handle reconnect

M1cha, 2023-06-10:
### Describe the bug

On my router, whenever the pppoe connection drops(either because the ISP disconnected me or because I hit "restart" in luci), it'll never recover until I do `/etc/init.d/network restart`.

In that state, the interface is reported like this. The error can be different depending on the disconnect reason but will not change by trying to restart the interface:
`ubus call network.interface status '{ "interface" : "wan" }'`:
```json
{
        "up": false,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "proto": "pppoe",
        "device": "eth0",
        "data": {

        },
        "errors": [
                {
                        "subsystem": "pppoe",
                        "code": "USER_REQUEST"
                }
        ]
}
```

A interface restart in luci seems to trigger this command:
```bash
ubus call network.interface up '{ "interface" : "wan" }
```

But, that doesn't trigger any log, visible to `logread`. It also doesn't cause a call to `/lib/netifd/proto/ppp.sh` which I've verified by adding `echo "$@" >> /tmp/ppp-proto.args` to that file.
I'm not sure what to test or debug since I'm not able to provide you with a reproducible environment, so any pointers in the right direction would be very helpful.

### OpenWrt version

r19803-9a599fee93

### OpenWrt target/subtarget

aarch64_generic

### Device

FriendlyElec NanoPi R4S

### Image kind

Official downloaded image

### Steps to reproduce

Unfortunately I don't know. I tried to reproduce this using the qemu x86_64 image and a pppoe-server but there it seems to work just fine.

### Actual behaviour

ppp interface stays down forever

### Expected behaviour

automatic restart with a new connection attempt.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-06-10:
You can run ifup script via hotplug down event.

M1cha, 2023-06-10:
> You can run ifup script via hotplug down event.

What exactly do you mean by that? `ifup` makes a call to `ubus call network.interface status` and as I described, that doesn't work. There's no way of getting ppp working again excapt restarting the whole network manager (netifd).

brada4, 2023-06-10:
Make a hotplug script that runs `logger` and check if any event is generated on 'disconnect'

SuperKali, 2024-11-07:
The issue is still present

brada4, 2024-11-07:
You need to debug pppd, namely flip options that seemingly cause disconnect.