openwrt/openwrt #12874: [23.05-rc1] uhttpd no longer supports TLS 1.3

n12c, 2023-06-10:
### Describe the bug

TLS 1.3 can no longer be negotiated. This worked fine in 22.03.

### OpenWrt version

r23069-e2701e0f33

### OpenWrt target/subtarget

ipq806x/generic

### Device

TP-Link Archer C2600

### Image kind

Official downloaded image

### Steps to reproduce

```
$ curl -vvvk --tlsv1.3 https://192.168.1.1
```

### Actual behaviour

```
*   Trying 192.168.1.3:443...
* Connected to 192.168.1.3 (192.168.1.3) port 443 (#0)
* ALPN: offers h2,http/1.1
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS alert, handshake failure (552):
* OpenSSL/3.0.9: error:0A000410:SSL routines::sslv3 alert handshake failure
* Closing connection 0
curl: (35) OpenSSL/3.0.9: error:0A000410:SSL routines::sslv3 alert handshake failure
```

### Expected behaviour

The TLS handshake should be successful.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

orangepizza, 2023-06-11:
guess wolfssl >  mbedtls caused this?

n12c, 2023-06-11:
> Are you running any TLS-guard? Den sön 11 juni 2023 03:19orangepizza ***@***.***> skrev:
> […](#)

Nope

ynezz, 2023-06-12:
> guess wolfssl > mbedtls caused this?

Yes, likely 4bcc3fd3d222b

> Official downloaded image

IMO if you need TLS v1.3 then you unfortunately can't use official images, some manual fiddling is needed, you basically need  to get rid of `libustream-mbedtls` and install `libustream-openssl` instead (or `libustream-wolfssl` if you're short of flash space).

> * OpenSSL/3.0.9: error:0A000410:SSL routines::sslv3 alert handshake failure

You apparently have OpenSSL already installed, so maybe you just need to install `libustream-openssl` and remove `libustream-mbedtls` (and probably restarting `uhttpd`) ?

ynezz, 2023-06-12:
> TLS 1.3 should be more and more the default, and towards eventually the only version available.

I [completely agree](https://github.com/openwrt/packages/pull/19008#issuecomment-1304789737) and IMO the only solution at this point of time is to move forward with [have openssl on targets with big flash (~128MB) and use mbedtls on devices with small flash](https://openwrt.org/meetings/20221130#cryptographic_library).

> TP-Link Archer C2600

I'm afraid, that above mentioned proposal to use OpenSSL by default on devices with suitable flash space wouldn't help in your case as it seems, that this device has only 32MB of flash. I don't remember exactly why ~128MB is needed, but there was probably some reason for this.

dfateyev, 2023-06-12:
> I'm afraid, that above mentioned proposal to use OpenSSL by default on devices with suitable flash space wouldn't help in your case as it seems, that this device has only 32MB of flash. I don't remember exactly why ~128MB is needed, but there was probably some reason for this.

IMO, the ~128MB requirements is just a new milestone for what is considered _"big flash targets"._
I have been successfully  using TLSv1.3 with `libopenssl1.1` on several [16MB devices](https://openwrt.org/toh/totolink/x5000r) since 21.02 releases, and yes: openssl with CLI takes 1-2MB on flash, otherwise there are no issues beside the limited space.

orangepizza, 2023-06-12:
mbedtls supports tls 1.3 in 3.x version (as client in 3.1 and server 3.2. as mbedtls 2.28 branch are supported only until end of 2024, I think this is right timeing for move to mbedtls 3.x version

ynezz, 2023-06-13:
> mbedtls 2.28 branch are supported only until end of 2024

[It says](https://github.com/Mbed-TLS/mbedtls/blob/development/BRANCHES.md#current-branches) "maintained until at least the end of 2024" so I interpret it differently. 

> I think this is right timeing for move to mbedtls 3.x version

Well, looking at https://github.com/Mbed-TLS/mbedtls/blob/development/BRANCHES.md#maintained-branches I can see following:

> Currently, the only supported LTS branch is: [mbedtls-2.28](https://github.com/Mbed-TLS/mbedtls/tree/mbedtls-2.28).
> Users are urged to always use the latest version of a maintained branch.

orangepizza, 2023-06-13:
from https://github.com/Mbed-TLS/mbedtls/blob/development/BRANCHES.md#maintained-branches

Every major version will become an LTS branch when the next major version is released. We may occasionally create LTS branches from other releases at our discretion. When a new LTS branch is created, it usually remains supported for three years.

as mbedtls 3.0.0 released in 2021 July, and 2.28 branch was created in 2021 dec so LTS version end of 2024 expected EOL time for such branch. old lts branch (2.1) start in 2015 Sep til 2018 Aug.

maybe they will create another LTS branch based on 3.X, but as about 1.5 year left I think it's time to start fiddle about (p.s just changing version number and hash was not enough to work)

(why replying to mail doesn't fill this web page?)

orangepizza, 2023-06-14:
found that mbedtls 1.3 removed all the legacy ciphers (rc4 entirely, 3DES ciphersuites in TLS options.  I think it may touch a lot of things (at least libustream/ maybe hostapd?)

neheb, 2023-06-15:
Right. In TLS. Still available as a cipher.

hhasert, 2023-06-16:
I tried the release, but Chrome was constantly complaining about XBR errors for Luci and I could not successfully configure it due to errors. Would this also be caused by the new ssl implementation ? The experience is so horrible I went back to 22.03

Neustradamus, 2023-06-18:
To follow

ynezz, 2023-06-19:
> I tried the release, but Chrome was constantly complaining about XBR errors for Luci and I could not successfully configure it due to errors.

@hhasert can you share more details about the issues? Ideally creating a separate ticket in https://github.com/openwrt/luci/issues

hhasert, 2023-06-19:
I upgraded my TP-Link Archer C7 V2 and I use https for Luci. I updated nettfid as well and then I broke so I had to reset the config. This causes the certs I use to be reverted to the default which make Chrome complain about the connection not being secure, which I normally overrule and then things work ok. The problem was that after I did that I had a lot of XBR issues in chrome, some pages would just not load even after refreshing and I had to again allow the unsecure certificates again before I could load the page. The problem was that it was unworkable with the default certs, which was not the case with earlier releases.

ynezz, 2023-06-19:
> The problem was that after I did that I had a lot of XBR issues in chrome

@hhasert I particularly interested  in this, what exactly is that XBR issue in Chrome?

orangepizza, 2023-06-19:
I think that may chrome refuse to give cookie from old page because of cert error override. IIRC chrome will do different things when you override invalid certs, like not doing cache / not allow reading old cookie/

hhasert, 2023-06-29:
I just installed rc2 and although I had to reboot to get lan working and I still have the issue with the 
[87765.218511] irq 23: nobody cared (try booting with the "irqpoll" option)
[87765.225331] CPU: 0 PID: 317 Comm: irq/23-keys Not tainted 5.15.118 #0
shutting down all wifi, but the problem with Chrome using the certs I trust did not give me the problems before, so it must have been chrome itself doing things wrong.

gjvons, 2023-07-30:
I just upgraded from 22.03 to 23.05-rc2, and also hit this issue: unable to connect via HTTPS using both firefox 160 and microsoft edge (both with default settings). Device: nanopi R4S, client Windows 11.
Firefox spins for a while and then reports "Secure connection failed ... PR_END_OF_FILE_ERROR". The developer tools show that after 30secs, there's a timeout in the setup of the TLS connection. Edge just says "site took too long to respond".
When I replace mbedtls20201210 with openssl (both libustream and luci-ssl), everything works again.

Neustradamus, 2023-09-28:
Linked to:
- https://github.com/openwrt/openwrt/pull/12917

quantumpacket, 2023-10-15:
> Should be more of a need actually - <1.3 version are severely affected by security-related, exploitable bugs, so TLS1.3 should be the ONLY TLS option actually from a security-perspective, imho.

Absolutely agree, reducing security for everyone by downgrading to a less secure TLS version makes no sense, especially for something as critical as a network device. :confused:

To confirm, to get back to TLS1.3 I would need to remove -> install these packages? Are there any other packages I should also alter?

```
libmbedtls* -> libwolfssl*
libustream-mbedtls* -> libustream-wolfssl*
px5g-mbedtls -> px5g-wolfssl
wpad-basic-mbedtls -> wpad-basic-wolfssl
```

neheb, 2023-10-15:
either wolfssl or openssl, yes.

quantumpacket, 2023-10-15:
@neheb It seems `luci-ssl` may depend on `libmbedtls`? https://github.com/openwrt/openwrt/pull/12917#issuecomment-1594462754 is this still the case?

gstrauss, 2024-07-19:
FYI: The patches I wrote in (#10303) for hostapd to support mbedtls 2.28 were also written to support mbedtls 3.x.  However, replacing mbedtls 2.28 with mbedtls 3.x has implications for everything else using mbedtls 2.28 since the migration from mbedtls 2.x to mbedtls 3.x is *not straightforward* and often requires code changes to those other packages.

@ynezz @neheb is there an open issue to migrate openwrt packages that depend on mbedlts to use mbedtls 3.x?  mbedtls 3.6.0 was added to Debian experimental back in April this year (2024), so presumably more packages support mbedtls 3.x than did in 2021 when mbedtls 3.0 was first released.

orangepizza, 2024-07-19:
https://github.com/openwrt/openwrt/pull/15092
already updates in  master at Apr 29, but few (less than 10) packages in package repo are left broken)
look https://downloads.openwrt.org/snapshots/faillogs/x86_64/packages/ but keep mind might not all of them are failed because of mbedtls

gstrauss, 2024-07-19:
Thanks, @orangepizza (and thanks for fixing that bug, hidden by an ifdef, in my hostapd mbedtls patch).  I hadn't searched openwrt for closed issues for mbedtls 3.x.  If I get a chance, I'll submit patches for some of those packages broken with mbedtls 3.x.  First one submitted upstream: https://github.com/lorabasics/basicstation/pull/198

gstrauss, 2024-10-24:
https://github.com/umurmur/umurmur/pull/190 for umurmur to support mbedtls 3.x

gstrauss, 2024-10-24:
https://github.com/mwarning/KadNode/pull/130 for KadNode to support mbedtls 3.x

gstrauss, 2024-10-25:
https://github.com/zhaojh329/ssl added support for mbedtls-3.x back in May, so the remaining change needed is to update the `src/ssl` submodule in the https://github.com/zhaojh329/libuhttpd repository to point to tip of https://github.com/zhaojh329/ssl master branch (and use that in openwrt)

gstrauss, 2024-10-25:
https://github.com/zhaojh329/libuhttpd/pull/27 is waiting on zhaojh329, and then https://github.com/zhaojh329/libuhttpd/pull/26 requests a new tag and release of libuhttpd.

gstrauss, 2024-10-25:
@orangepizza, @ynezz patches have been submitted to upstream repositories for what I think are the remaining openwrt packages which use mbedtls and have not yet been updated to support mbedtls 3.x.  Would you like these submitted as patches to openwrt to avoid waiting for upstream maintainers to respond?  (After all this time since mbedtls-3.x was released, these upstream maintainers have not updated their software to support mbedtls-3.x, so I do not expect the turnaround to be quick.)

orangepizza, 2024-10-25:
As those packages were broken for like half a year, I don't think there are much users using it on trunk: it may just enough to arrive 25.x be enough

Neustradamus, 2024-10-25:
It is important to have TLS 1.3 support with the next OpenWrt version and current too.

gstrauss, 2024-10-26:
This issue is for uhttpd, which lost support for TLSv1.3 when the underlying dependency was changed from wolfssl to mbedtls 2.28.x.

The issue was addressed in https://github.com/openwrt/openwrt/pull/15092 which was merged 28 April 2024 and updated mbedtls to 3.6.0, the first version of mbedtls with TLSv1.3 support for both client and server.

I think that this issue can be closed unless there is something an action to be taken such as backporting a fix to 23.05.

Other packages mentioned in this PR are off-topic, and as @orangepizza noted, do not appear to be widely used.