openwrt/openwrt #12876: [22.03] ipq40xx: packet steering bug

Payti, 2023-06-10:
### Describe the bug

Hello,
I'm user OpenWrt 22.03-SNAPSHOT r20043-1f32774ded on ZTE MF289F.
When set 'packet steering' option in 'Global  Network Options' ZTE MF289F stiil use 1 CPU core for download stream and reduce download speed to max 880Mbit/s (with active 'program flow offloading' firewall option).
Please correct this bug.

Thanks and Regards.
Tom

### OpenWrt version

OpenWrt 22.03-SNAPSHOT r20043-1f32774ded

### OpenWrt target/subtarget

ipq40xx

### Device

ZTE MF289F

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-06-11:
Please upgrade to a supported release and re-test.

jmv2009, 2023-06-13:
Moreover, run cat /proc/interrupts. And divide the tx and rx interrupts over the processors with echo 8> /proc/irq/xx/smp_affinity. Or use irqbalance.

brada4, 2023-06-13:
Also one might need to set queues on adapter to enable either steering or multiple irqs or queues using `ethtool -l/-L`

Payti, 2023-06-14:
Yes, but it is not, like on screen above.

brada4, 2023-06-14:
it is one 64-byte acknowledgement per 4 packets received going back, so the reduction in speed matches gigabit CPU switch.

Payti, 2023-06-15:
On original firmware when load to ZTE MF289F is full download speed - 950Mbit/s

brada4, 2023-06-15:
According to spec sheet switch included supports 2 CPU ports, thus should be able to do gigabit over NAT, but if SoC or DTS does not wire them right you get what you see. I.e. extract DTS from factory uograde image and check whats missing.

jmv2009, 2023-06-16:
I think a reasonable explanation is that if openwrt uses only one of the PSGMII/RGMII interfaces to the cpu, the aggregated upstream/downstream bandwidth drops from 2Gbit/s to 1 GBit/s.
[https://github.com/Deoptim/atheros/blob/master/IPQ4019-datasheet.pdf](url)

Payti, 2023-06-17:
Could be possible write code to OpenWRT kernel/driver that activates the hardware nat - HNAT in this SoC?

brada4, 2023-06-17:
The problem is that OpenWRT does not support second gigabit port (called ser-des in docs) attached to SoC whcih keeps the router strictly under gigabit.
The flow offload is handled by Linux framework, it will only lower CPU load in case SoC nat gets better instrumented over time, again your CPU handles that completely in software.
Another problem is that 1 core cannot handle all 16+4 ?? queues, so irqbalance was of immense help, it is sort of next door from steering, just that queues are per IRQ that can/should be distributed to multiple cores. 
You can try PR installing irqbalance on this router (or all based on same SoC) in default packages to save confusion for your co-users?

brada4, 2023-06-26:
Wait for improvements with dsa switch to support 2 system ports. TCP ack backscatter is 1-10% as you see. No wsy around it.