openwrt/openwrt #12882: Several different UBNT device not working in 23.05rc1, file system seems write protected

dnordenberg, 2023-06-11:
### Describe the bug

Tried several differens UBNT 5AC models, all behaves the same. Problem is also discussed in a forum topic here https://forum.openwrt.org/t/powerbeam-m5-xw-configuration-loss-after-reboot/141925/8
This is probably not a openwrt bug in itself but something caused by the kernel upgrade to 5.15

Here is a dmesg bootlog from a fresh install on a nanobeam 5AC (one with a WA board but problem is seen on a XC one too).
(https://textbin.net/eo6yuvznph)
As you can see, it has a large amout of "jffs2: Newly-erased block contained word..." indicating that it can't save anything to the filesystem. On reboot, everything is reverted back to defaults.
On the IRC channel a similar bug was mentioned as a possible cause to this one too (https://github.com/openwrt/openwrt/commit/83b5fbddf28e943e8c90b4099a1e36d158f5995c)

### OpenWrt version

23.05rc1

### OpenWrt target/subtarget

ath79/generic

### Device

Ubiquiti NanoBeam AC Gen1 (WA), Gen2 (XC)

### Image kind

Official downloaded image

### Steps to reproduce

Install image, look at dmesg after at boot. Try to do a change, save, reboot, see if change is reverted.

### Actual behaviour

Change is reverted

### Expected behaviour

Changes are saved to the filesystem and saved on reboot.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dnordenberg, 2023-06-11:
Maybe I should add that openwrt 22.03.4 and prior works as expected. Forum post indicating that the problem has been here since last november at least. I first discovered it a few months later but didn't not report anything by then since I didn't know if there was a work in progress causing this but now when 23.05 was branched off and problem is still there I thought that possibility was unlikely :(

DragonBluep, 2023-06-11:
This problem is a regression of https://github.com/openwrt/openwrt/issues/7770. It's a spi driver issue. And this commit https://github.com/openwrt/openwrt/commit/83b5fbddf28e943e8c90b4099a1e36d158f5995c also looks suspicious. Macronix usually uses the same ID for multiple products.

dnordenberg, 2023-06-11:
It was just suggested on the IRC channel that the issue was probably similar. I don't even know if this device uses macronix, they are ultrasound welded together so you have to open them with a saw :(

DragonBluep, 2023-06-11:
You can query the Flash device ID from the system log:
```
[    0.384192] spi-nor spi0.0: mx25l6405d (8192 Kbytes)
```

dnordenberg, 2023-06-11:
aha :)
[    0.442259] spi-nor spi0.0: mx25l12805d (16384 Kbytes)

DragonBluep, 2023-06-11:
> aha :) [ 0.442259] spi-nor spi0.0: mx25l12805d (16384 Kbytes)

It's a 16 MiB Macronix Flash, mx25l12805d.

I removed the macronix patch and built this firmware. Maybe you can try to see if it works properly. But please be careful, there is a very small chance that it will brick your device.
[ubnt_nanobeam-ac.tar.gz](https://github.com/openwrt/openwrt/files/11714822/ubnt_nanobeam-ac.tar.gz)

dnordenberg, 2023-06-11:
Oki thanks, will try later when I'm back home. But that patch did not do anything to mx25l12805d as far as I can see? Also this patch was already shipped in 22.03 which I already know works?

danitool, 2023-06-11:
Image generation should be disabled for these Ubiquiti devices in 23.05 until solved. They will cause soft bricks, only recoverable via TFTP (uart serial interface not required), which is a bit tricky because the ethernet link is flaky in bootloader mode on some devices.

dnordenberg, 2023-06-11:
Yes, uploading new image does nothing, TFTP back to stock fw is the only way back and from there flash openwrt again. I never had problems with TFTP on these, wasn't it on the M5s that this could be flaky? I mostly have experience with the more recent 5AC generation...

danitool, 2023-06-11:
The issue while recovering via TFTP and bad ethernet gigabit links happened to me on two Nanobeam AC. After the recovering the link was 100% reliable when running stock or OpenWrt.

About the flash issue, I was scrutinizing some kernel commits without any luck. Apparently all is ok, but the old flash bug is there again. I suspect the bootloader is also involved.

luizluca, 2023-06-11:
> This problem is a regression of #7770

In what way? #7770 was discovered because TP-Link WR2543ND might have a HW problem with the SPI big-bang protocol. When we dropped fast-flash read (reading/writting flash from/to memory mapped address), the device was returning sometimes random data.

The patch was applied in OpenWrt since 4.19 (19.07) and it is still there in 5.15 (23.05) as a patch. It will be dropped for 6.1 as it is in upstream kernel since 5.18 (https://github.com/torvalds/linux/commit/d08de0259dfe172caf073b921c6b27ff089605a9). If it was related to that patch, it should have broken devices since 21.07.

I will try snapshot tomorrow in a TP-Link WR2543ND testing #12864 (preliminar for DSA migration) and I'll return if the SPI IO is broken.

dnordenberg, 2023-06-11:
> About the flash issue, I was scrutinizing some kernel commits without any luck. Apparently all is ok, but the old flash bug is there again. I suspect the bootloader is also involved.

There are not so many more commits to suspect in micronix.c I guess, it is not touched very often so problem is probably elsewhere :(

@luizluca I guess we are not talking about the same patch now? I guess the one @DragonBluep is referring to is https://github.com/openwrt/openwrt/commit/83b5fbddf28e943e8c90b4099a1e36d158f5995c ? Not crystal clear but he is referring to it as the "macronix patch" so I don't think he means your patch for #7770.

dnordenberg, 2023-06-11:
@DragonBluep Your image did not boot :( it was stuck as flashing two signal LEDs and turning ethernet on/off so I guess the bootloader was not happy

rany2, 2023-06-11:
> Yes, uploading new image does nothing, TFTP back to stock fw is the only way back and from there flash openwrt again. I never had problems with TFTP on these, wasn't it on the M5s that this could be flaky? I mostly have experience with the more recent 5AC generation...

This sounds like the jffs2 partition is corrupted. Is it possible to get it to failsafe mode and then running firstboot to reformat it?

danitool, 2023-06-11:
> This sounds like the jffs2 partition is corrupted. Is it possible to get it to failsafe mode and then running firstboot to reformat it?

No, you can't recover the device with Openwrt failsafe. UBNT TFTP recovery is mandatory in this case.

dnordenberg, 2023-06-11:
> This sounds like the jffs2 partition is corrupted. Is it possible to get it to failsafe mode and then running firstboot to reformat it?

It is not corrupted, it is just always in "factory" state since it can't be written to at all. Firstboot does nothing since it is always in firstboot state.

dnordenberg, 2023-06-11:
> The issue while recovering via TFTP and bad ethernet gigabit links happened to me on two Nanobeam AC. After the recovering the link was 100% reliable again when running stock or OpenWrt.

Oki, strange. I spent quite some time in the past on the ubnt forum to and never heard of people having TFTP issues like that, the ones who do have problems with it had broken ethernet ports because of lightning/ESD or something. Sure you didn't see the same ethernet flickering as I did trying the image @DragonBluep provided? It is a strange way to signal something but I guess that is bootloaders way of telling a major fault that you can also see remotely. But once I hold down reset long enough to put it in TFTP mode (the alternating LEDs or ping TTL=128) it took the transferred image without problems.

DragonBluep, 2023-06-12:
> Oki thanks, will try later when I'm back home. But that patch did not do anything to mx25l12805d as far as I can see? Also this patch was already shipped in 22.03 which I already know works?

I removed this one, sadly it doesn't work for you. https://github.com/openwrt/openwrt/blob/9519a625499a963508755b7fc08fb825b3d8cfde/target/linux/generic/pending-6.1/499-mtd-spi-nor-disable-16-bit-sr-for-macronix.patch#L27

> > This sounds like the jffs2 partition is corrupted. Is it possible to get it to failsafe mode and then running firstboot to reformat it?
> 
> No, you can't recover the device with Openwrt failsafe. UBNT TFTP recovery is mandatory in this case.

Usually, users can enter failsafe mode by pressing any button. Then, the device can be repaired by executing the `mtd` command via ssh.

luizluca, 2023-06-12:
> @luizluca I guess we are not talking about the same patch now? I guess the one @DragonBluep is referring to is [83b5fbd](https://github.com/openwrt/openwrt/commit/83b5fbddf28e943e8c90b4099a1e36d158f5995c) ? Not crystal clear but he is referring to it as the "macronix patch" so I don't think he means your patch for #7770.

My patch was the only one in #7770. Maybe that issue it is simply not related. In a quick search, I found this other issue with similar results:

https://github.com/openwrt/openwrt/issues/8287

It looks like something similar is not new for UBNT devices. The fix is related to block protection:

https://github.com/openwrt/openwrt/pull/3536

Maybe that was lost between 22.3 and 23.5. I would suggest bisecting the tree.

DragonBluep, 2023-06-12:
> #3536
> 
> Maybe that was lost between 22.3 and 23.5. I would suggest bisecting the tree.

@luizluca This patch was already dropped when adding the 5.10 kernel https://github.com/openwrt/openwrt/commit/d6b785d47739a84611088ab26256fa5808d06a9d.

luizluca, 2023-06-12:
> > #3536
> > 
> > Maybe that was lost between 22.3 and 23.5. I would suggest bisecting the tree.
> 
> @luizluca This patch was already dropped when adding the 5.10 kernel https://github.com/openwrt/openwrt/commit/d6b785d47739a84611088ab26256fa5808d06a9d.
> 

Yes, I didn't mean that exactly patch but something related to block protection that changed. If someone fixed that in the past and the issue was similar, it might have the same cause. I would focus on patches that touched that area.

As I suggested, bisecting the git might give a clue. If you get lucky and it wasn't a major kernel version change, you can easily pinpoint it. If not, it is much harder. First some openwrt patch might be the cause, but you can compare with the set of patches from the working version. If it was an upstream change, you'll need to start from the last openwrt that worked and backports changes that might have an effect. I did that for some DSA issues and it took me quite some time. However, reading/writing flash in an old device might have a much shorter history between 5.10 and 5.15.

marek26340, 2023-08-23:
Hi, might as well chip in my 2 cents too. Seems like I just got stumped by this same exact issue while tinkering with my NS5ACL for more than 2 hours before I found this issue thread. Same exact issue appeared for me on 23.05.0-rc3 straight away. No amout of reflashing, factory resetting or anything that I tried had any effect. Not even downgrading back to 22. Only a TFTP recovery of the original firmware 8.7.0 has fixed the issue for me. 
I'm currently trying out 22.03.5 and so far so good. Just wanted to try the new 23.05 firmware and I got a bit burned to say the least - I'll just wait for a stable release...

dnordenberg, 2023-08-24:
You can't sysupgrade downgrade to 22.03 directly because the new firmware can't be stored before flashing, kind of catch 22. TFTP stock fw and start over is the one way back.

marek26340, 2023-08-24:
Oh! My bad, thanks for explaining. TFTP'd stock and all is well on 22.03.5.
Well anyway, the rest of my comment still applies - I was getting these errors in the log:
```
mount_root: jffs2 not ready yet, using temporary tmpfs overlay
jffs2: Newly-erased block contained word 0x19852003 at offset 0x........ (repeats multiple times with different offsets)
```

My comment probably should've just ended up on the forum. But, I'd like to also say that I'd be happy to try and install a firmware image for testing purposes.

But, while I had the 23.05.0-rc3 firmware up and running, I ran some iperf3 tests with my Galaxy Note 10, 80MHz bandwidth, WPA3. DL to phone was about 265Mb/s tops. I was only able to get about 230Mb/s tops with 22 and it was dropping the speed to about 160Mb/s every couple of seconds (diagnosed that a long time ago - blamed it on ath10k high amount of interrupts every time this happened, it also felt like the whole device froze every time the WiFi chip or ath10k sent the interrupt avalanche). So I'd like to say that 23 might've fixed that issue, but I ain't running any further testing until this issue gets fixed.
The faulty images should also get pulled from the downloads website.

aanon4, 2023-10-15:
I dont know if anyone found a better solution to this problem, but I had to set ```MTD_SPI_NOR_SWP_DISABLE=y``` to get this working on my various UBNT devices.

danitool, 2023-10-16:
> I dont know if anyone found a better solution to this problem, but I had to set `MTD_SPI_NOR_SWP_DISABLE=y` to get this working on my various UBNT devices.

Thanks for sharing your solution.

I was looking for a similar fix. Maybe backporting this commit:
https://github.com/torvalds/linux/commit/18d7d01a0a0eb32b78149c8259bf49504d5fa4e0

and setting  SNOR_F_NO_WP new property in the dts file could also fix it?

Leo-PL, 2023-12-23:
> > I dont know if anyone found a better solution to this problem, but I had to set `MTD_SPI_NOR_SWP_DISABLE=y` to get this working on my various UBNT devices.
> 
> Thanks for sharing your solution.
> 
> I was looking for a similar fix. Maybe backporting this commit: [torvalds/linux@18d7d01](https://github.com/torvalds/linux/commit/18d7d01a0a0eb32b78149c8259bf49504d5fa4e0)
> 
> and setting SNOR_F_NO_WP new property in the dts file could also fix it?

I tried fix in #14237, but your proposal looks way better. Will try - it makes much more sense.

closed 2023-12-24 by openwrt-bot

hauke, 2023-12-24:
Maybe extend the code to make the chip writable if it contains any writable partition. It should at least print an warning message for such cases.

hauke, 2023-12-24:
I merged your change for now, if you have a better solution we can replace this. 
Some extension looking at the partitions or a device tree setting should first go into upstream Linux, we can then later backport it to OpenWrt.

dnordenberg, 2023-12-24:
> I think we need a similar device tree knob to enable this fix selectively, or ensure that unlock done at mounting jffs2 during boot is effective - it puzzles me, why sysupgrade works, but writing to rootfs_data does not.

Thanks for your fix :)
Not sure what you mean with sysupgrade works? For many of us it seems to fail so downgrade to 22.04 wasn't even possible, after reboot device was still on 23.05. We had to TFTP the device back to stock FW before loading 22.04. So I don't think sysupgrade can store anything persistantly either but maybe it was not what you meant?

Leo-PL, 2023-12-26:
> Thanks for your fix :) Not sure what you mean with sysupgrade works? For many of us it seems to fail so downgrade to 22.04 wasn't even possible, after reboot device was still on 23.05. We had to [...]

I think it may depend on the device itself. On my Loco M5XW, sysupgrade worked, but writing to rootfs_data did not. I tested 23.05.2 release and had latest XW firmware loaded beforehand.

fblaese, 2024-03-19:
Is the workaround supposed to be included in the current snapshot?
I tried to restore a Nanostation 5AC Loco via TFTP today, but observed the same non-persistent overlay-fs behavior (including sysupgrade not working) with the current snapshot image. OpenWrt 22.03.6 works fine.

As it seems that all OpenWrt 23.02 releases for this device are broken to a point where recovering requires TFTP via the bootloader, the images should probably be removed from the downloads page, so no further devices get bricked.

Leo-PL, 2024-03-19:
Yes, this should work with 23.05-SNAPSHOT. I agree with removal of 23.05.2 images for these devices.

KaneTW, 2024-05-14:
I think this issue still persists with snapshot? I just installed a snapshot (downloaded, not self-built) and it did the same thing. Went back to 22. for now.

marek26340, 2024-05-14:
> I think this issue still persists with snapshot? I just installed a snapshot (downloaded, not self-built) and it did the same thing. Went back to 22. for now.

Hmm? I installed the latest stable release I could find and the issue seems to be fixed. 23.05.3 is working perfectly for me. (NS5ACL)

KaneTW, 2024-05-14:
Weird. Maybe I selected the wrong file when flashing it.

I'll try again at some point, but it's a headache to TFTP restore it so I'll do it later sometimes.

KaneTW, 2024-05-14:
Weird. Maybe I selected the wrong file. (NanoBeam AC Gen 2)

I'll try it later again sometimes.

dnordenberg, 2024-05-14:
> Du får inte e-post ofta från ***@***.******@***.***>. Se varför det här är viktigt.<https://aka.ms/LearnAboutSenderIdentification>
> EXTERN AVSÄNDARE: E-postmeddelandet har sitt ursprung utanför kommunen. Klicka inte på några länkar eller öppna några bilagor om du inte litar på avsändaren och vet att innehållet är säkert.
> EXTERNAL SENDER: Do not click any links or open any attachments unless you trust the sender and know the content is safe.

So the fix was maybe applied to the 23.xx branch only and not to the upcoming 24.xx? I thought I saw it applied to main too but I may be wrong...

Leo-PL, 2024-05-14:
No, such fixes are always applied to main first. And 24 wasn't even branched off.

dnordenberg, 2024-05-14:
Sorry I mean what will become 24.xx...

aaron13223, 2024-06-30:
So i was trying to get add device support for the UAP LR V2 and I was successful in booting it up and installing it too. (Despite Ubiquiti disabling flashing on 4.X, I was able to downgrade to 3.X and bricked my device but that allowed me to flash my own openwrt build on it)

Anyway, it works but I still get the jffs2 error on boot everytime so no changes are persisted (Even with the "MTD_SPI_NOR_SWP_DISABLE=y" ):

```
[   78.439816] jffs2_scan_eraseblock(): End of filesystem marker found at 0x0
[   78.446836] jffs2_build_filesystem(): unlocking the mtd device...
[   78.457297] done.
[   78.465540] jffs2_build_filesystem(): erasing all blocks after the end marker...
[   79.277053] jffs2: Newly-erased block contained word 0x461a6cdd at offset 0x001c0000
[   79.319797] jffs2: Newly-erased block contained word 0x8af0ce84 at offset 0x001b0000
[   79.340110] jffs2: Newly-erased block contained word 0x5826882b at offset 0x001a0000
[   79.360128] jffs2: Newly-erased block contained word 0xdeef3228 at offset 0x00190000
[   79.380103] jffs2: Newly-erased block contained word 0xc3dc8b0a at offset 0x00180000
[   79.400108] jffs2: Newly-erased block contained word 0xa4480ac5 at offset 0x00170000
[   79.420120] jffs2: Newly-erased block contained word 0x9cc700be at offset 0x00160000
[   79.440080] jffs2: Newly-erased block contained word 0x817ed557 at offset 0x00150000
[   79.467277] jffs2: Newly-erased block contained word 0x989972c7 at offset 0x00140000
[   79.487277] jffs2: Newly-erased block contained word 0xda215671 at offset 0x00130000
[   79.509773] jffs2: Newly-erased block contained word 0xd4d903e at offset 0x00120000
[   79.530089] jffs2: Newly-erased block contained word 0x3c632b18 at offset 0x00110000
[   79.550123] jffs2: Newly-erased block contained word 0x47300391 at offset 0x00100000
[   79.570130] jffs2: Newly-erased block contained word 0x5b3fef1f at offset 0x000f0000
[   79.590155] jffs2: Newly-erased block contained word 0xdeadc0de at offset 0x000e0000
[   79.610107] jffs2: Newly-erased block contained word 0xc6beeddb at offset 0x000d0000
[   79.637319] jffs2: Newly-erased block contained word 0xe5e34c36 at offset 0x000c0000
[   79.657285] jffs2: Newly-erased block contained word 0x93a7833 at offset 0x000b0000
[   79.679780] jffs2: Newly-erased block contained word 0xdeadc0de at offset 0x000a0000
[   79.700113] jffs2: Newly-erased block contained word 0xdeadc0de at offset 0x00090000
[   79.720109] jffs2: Newly-erased block contained word 0xf64418ac at offset 0x00080000
[   79.740115] jffs2: Newly-erased block contained word 0x7d9b902 at offset 0x00070000
[   79.760113] jffs2: Newly-erased block contained word 0xc396f9e4 at offset 0x00060000
[   79.780141] jffs2: Newly-erased block contained word 0xd5d6ce85 at offset 0x00050000
[   79.800123] jffs2: Newly-erased block contained word 0xe00f8db7 at offset 0x00040000
[   79.827295] jffs2: Newly-erased block contained word 0x455a56ed at offset 0x00030000
[   79.847284] jffs2: Newly-erased block contained word 0x34d3f41a at offset 0x00020000
[   79.869781] jffs2: Newly-erased block contained word 0xb51e503 at offset 0x00010000
[   79.890137] jffs2: Newly-erased block contained word 0xdeadc0de at offset 0x00000000
[   79.907608] done.
[   79.909601] jffs2: notice: (1216) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
[   80.092877] overlayfs: upper fs does not support tmpfile.
```

DragonBluep, 2024-06-30:
@aaron13223 Maybe you can try these patches: https://github.com/openwrt/openwrt/pull/14425

aaron13223, 2024-07-01:
@DragonBluep Just tried it out and still the same issue. The jffs2 erases the block right when the ethernet is registered as up. The fix you proposed added a time delay so the ethernet took longer than usual to come up but well it still erased the block :/

I also had one error compiling so I had to change:

```
WARN_ONCE("max_message_size %u\n", sp->max_message_size);
```

to:

```
WARN_ONCE(1, "max_message_size %u\n", sp->max_message_size);
```

The compile error was:

```
drivers/spi/spi-ath79.c: In function 'ath79_spi_max_message_size':
drivers/spi/spi-ath79.c:115:46: error: passing argument 4 of 'warn_slowpath_fmt' makes pointer from integer without a cast [-Werror=int-conversion]
  115 |         WARN_ONCE("max_message_size %u\n", sp->max_message_size);
      |                                            ^~~~~~~~~~~~~~
      |                                              |
      |                                              u32 {aka unsigned int}
./include/asm-generic/bug.h:91:62: note: in definition of macro 'WARN_printf'
   91 |                 warn_slowpath_fmt(FILE__, LINE, taint, arg);      \
      |                                                              ^
./include/linux/once_lite.h:31:25: note: in expansion of macro 'WARN'
   31 |                         func(VA_ARGS);                              \
      |                         ^~
./include/asm-generic/bug.h:150:9: note: in expansion of macro 'DO_ONCE_LITE_IF'
  150 |         DO_ONCE_LITE_IF(condition, WARN, 1, format)
      |         ^~~~~~~~~~~
drivers/spi/spi-ath79.c:115:9: note: in expansion of macro 'WARN_ONCE'
  115 |         WARN_ONCE("max_message_size %u\n", sp->max_message_size);
      |         ^~~~~~~
./include/asm-generic/bug.h:87:36: note: expected 'const char *' but argument is of type 'u32' {aka 'unsigned int'}
   87 |                        const char *fmt, ...);
      |                        ~~~~~~~~~~^
```

But anyway I don't think this matter that much - still have all settings wiped n each boot :p

DragonBluep, 2024-07-01:
You are testing the wrong patch... Just pull this branch. And you need to change the dts. https://github.com/DragonBluep/openwrt/tree/fix-wr2543n-spi

aaron13223, 2024-07-01:
The mtd partitions feels wrong, they add up to more than 8MB, what do you think?

cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00060000 00010000 "u-boot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 00740000 00010000 "firmware"
mtd3: 00240000 00010000 "kernel"
mtd4: 00500000 00010000 "rootfs"
mtd5: 00240000 00010000 "rootfs_data"
mtd6: 00040000 00010000 "cfg"
mtd7: 00010000 00010000 "art"

kernel, rootfs and rootfs_data should be part of "firmware" right? But the sizes don't add up considering the flash is 8MB. 🤔

aaron13223, 2024-07-02:
For anyone who comes across this problem, it has nothing to do with any bugs in openwrt or linux kernel whatsoever!!!

This is just Ubiquiti being an A** lol. They have been trying to block install of openwrt for a while and this is just that. 

I compiled my own u-boot for this device using pepe2k's u-boot_mod (It's old yes but so is the device and the versions aren't that far apart) - and viola! Everything just works without any issues!

I might compile a newer version of u-boot, tho I am not sure about that right now, only if I am just too bored lol since this is such an old device.

So yeah, all this while, it was the u-boot playing around with the flash (somehow locking it down) :/

Leo-PL, 2024-07-02:
Does it have support built in for this one? I was looking to port u-boot_mod for XW boards too, because how PITA it was to replace the flash for 16MB variant on them.

danitool, 2024-07-02:
> For anyone who comes across this problem, it has nothing to do with any bugs in openwrt or linux kernel whatsoever!!!
> 
> This is just Ubiquiti being an A** lol. They have been trying to block install of openwrt for a while and this is just that.

If true, then also old OpenWrt versions should also fail, but this isn't the case. However maybe partially true.

aaron13223, 2024-07-03:
> If true, then also old OpenWrt versions should also fail, but this isn't the case. However maybe partially true.

It's possible it's to do with specific versions of u-boot? I figured out the protect command on the original u-boot but it still never erased anything. So a perpetual state of the flash being locked? I tried a lot of fixes from when this problem appeared in openwrt to now - none of them worked except changing the u-boot.

I was able to flash another APLRV2 on firmware 4.X which had a newer version of u-boot to test this but that guy refuses to boot saying "Bad magic number" :/

aaron13223, 2024-07-03:
> Does it have support built in for this one? I was looking to port u-boot_mod for XW boards too, because how PITA it was to replace the flash for 16MB variant on them, and I could not use trusty W25Q128, and had to exactly order the Macronix parts...

Not inbuilt, I had to add it but it was fairly easy since similar hardware was already added in, just had to tweak a couple of files and compile an ancient openwrt toolchain. Even tho I got it working I am still facing one issue. The u-boot_mod does not support the AR8032 ethernet controller by default so I might have to spend quite some time adding this for the ethernet to work :/ Definitely above my pay grade tho lol.

pepitosbazzegutisbazzeguti, 2024-07-22:
> Yes, this should work with 23.05-SNAPSHOT. I agree with removal of 23.05.2 images for these devices.

I have 23.05 installed and I have the persistency memory problem. How do I install the 23.5.4 to correct it, if I can´t install due to the persistency problem? (I have a NANOBEAM AC GEN2)

LipkeGu, 2024-07-24:
i have installed the mentioned snapshot version and still have the write protected filesystem problem (Nanostation m2 XW)

LipkeGu, 2024-07-24:
> > Yes, this should work with 23.05-SNAPSHOT. I agree with removal of 23.05.2 images for these devices.
> 
> I have 23.05 installed and I have the persistency memory problem. How do I install the 23.5.4 to correct it, if I can´t install due to the persistency problem? (I have a NANOBEAM AC GEN2)

Leo-PL, 2024-07-24:
Which flash does your device use? I've had problems on my NSM5 XW when modding, and I had specifically order a Macronix chip. Winbond ones I use everywhere else didn't work.

LipkeGu, 2024-07-31:
> Which flash does your device use?
Firmware versions of freifunk shows macronix. The Firmware "Falter 1.2.3" seems not affected and the image keep its settings.

Leo-PL, 2024-08-01:
Yes. I did that, but I had to use Macronix chip exactly, Winbond didn't work at all. Of course, device tree will need modding for extra space.

aaron13223, 2024-08-01:
> Yes. I did that, but I had to use Macronix chip exactly, Winbond didn't work at all. Of course, device tree will need modding for extra space.

Are you suggesting that a new Macronix chip works while the original Macronix chip doesn’t? Just want to make sure if that’s what you mean before I order them since I have multiple winbond ones. (Tbf both old macronix chip and winbond ones work perfectly fine with a modded u-boot)

LipkeGu, 2024-08-04:
@Leo-PL Is there any chance that we can add a "Nanostation M XW (16/64) device profile" ? or add an option to produce a 16MB image? 

With an Hint: "This Profile/option can be used for nanostations m xw with 16MB flash chips")? This would be nice for Freifunk communities but how/where can i let solder in such suiteable chip?

Leo-PL, 2024-08-04:
OpenWrt core team is against upstreaming support for such one-off mods - however I'm thinking about a solution using a cleverly crafted device tree. See here: https://github.com/openwrt/openwrt/pull/12174 - this is done for TP-links, but can be done for UBNTs just as well. Maybe you can help get it some traction. Sadly, this solution was proposed to Linux maintainers, and they said no as well. We need to improve that.