openwrt/openwrt #12886: [23.05-rc1] ath10k_ahb Wifi Bug on ZTE MF 289F

tango38317, 2023-06-12:
### Describe the bug

Hi,
i have flashed 23.05-rc1 on my ZTE MF289F and now i have issues with WLAN. On this device i seen only 5GHZ Wlan device, but not the 2,4 GHZ device.
Kernel log:
`[   18.971532] ath10k_ahb a000000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   18.971602] ath10k_ahb a000000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   18.983435] ath10k_ahb a000000.wifi: firmware ver 10.4b-ct-4019-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 6b2b5c5b
[   19.043011] ath10k_ahb a000000.wifi: Loading BDF type 0
[   19.189522] ath10k_ahb a000000.wifi: board_file api 2 bmi_id 0:20 crc32 d140cd7d
[   21.517514] ath10k_ahb a000000.wifi: failed to receive control response completion, polling..
[   22.557323] ath10k_ahb a000000.wifi: ctl_resp never came in (-110)
[   22.557706] ath10k_ahb a000000.wifi: failed to connect to HTC: -110
[   22.616399] ------------[ cut here ]------------
[   22.616450] WARNING: CPU: 2 PID: 9 at kernel/workqueue.c:3084 __flush_work+0x2d8/0x310
[   22.620107] Modules linked in: ath10k_pci(+) ath10k_core ath xt_state xt_conntrack wireguard usb_wwan qmi_wwan pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211 libchacha20poly1305 iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables curve25519_neon cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables usbserial usbnet slhc sch_cake poly1305_arm nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c hwmon crc_ccitt compat chacha_neon cdc_wdm sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic
[   22.620579]  act_skbedit act_mirred act_gact ip6_gre ip_gre gre ifb ip6_tunnel tunnel6 ip_tunnel vxlan udp_tunnel ip6_udp_tunnel sha512_arm ghash_arm_ce cmac leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom gpio_button_hotplug mii crc32c_generic
[   22.715428] CPU: 2 PID: 9 Comm: kworker/u8:0 Not tainted 5.15.113 #0
[   22.737638] Hardware name: Generic DT based system
[   22.744063] Workqueue: ath10k_wq ath10k_core_register_work [ath10k_core]
[   22.748673] [<c030d488>] (unwind_backtrace) from [<c030986c>] (show_stack+0x10/0x14)
[   22.755525] [<c030986c>] (show_stack) from [<c0602da0>] (dump_stack_lvl+0x40/0x4c)
[   22.763251] [<c0602da0>] (dump_stack_lvl) from [<c032274c>] (__warn+0x8c/0x100)
[   22.770628] [<c032274c>] (__warn) from [<c0322828>] (warn_slowpath_fmt+0x68/0x78)
[   22.777834] [<c0322828>] (warn_slowpath_fmt) from [<c033c9a8>] (__flush_work+0x2d8/0x310)
[   22.785474] [<c033c9a8>] (__flush_work) from [<c033cb9c>] (__cancel_work_timer+0xe0/0x1d0)
[   22.793631] [<c033cb9c>] (__cancel_work_timer) from [<bf4ce350>] (ath10k_htc_stop_hl+0x18/0x4c [ath10k_core])
[   22.801831] [<bf4ce350>] (ath10k_htc_stop_hl [ath10k_core]) from [<bf4d7010>] (ath10k_htt_tx_free+0x14/0x48 [ath10k_core])
[   22.811782] [<bf4d7010>] (ath10k_htt_tx_free [ath10k_core]) from [<bf4cae54>] (ath10k_core_start+0xb0c/0x15b8 [ath10k_core])
[   22.822720] [<bf4cae54>] (ath10k_core_start [ath10k_core]) from [<bf4cc75c>] (ath10k_core_register_work+0x638/0xb74 [ath10k_core])
[   22.834088] [<bf4cc75c>] (ath10k_core_register_work [ath10k_core]) from [<c033b608>] (process_one_work+0x210/0x478)
[   22.845634] [<c033b608>] (process_one_work) from [<c033c184>] (worker_thread+0x4c/0x598)
[   22.855958] [<c033c184>] (worker_thread) from [<c0342b08>] (kthread+0x148/0x164)
[   22.864292] [<c0342b08>] (kthread) from [<c0300110>] (ret_from_fork+0x14/0x24)
[   22.871668] Exception stack(0xc1061fb0 to 0xc1061ff8)
[   22.878699] 1fa0:                                     00000000 00000000 00000000 00000000
[   22.883829] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   22.891986] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[   22.900189] ---[ end trace 93d8523097d51500 ]---
[   22.906579] ------------[ cut here ]------------
[   22.911419] WARNING: CPU: 2 PID: 9 at kernel/workqueue.c:3084 __flush_work+0x2d8/0x310
[   22.916027] Modules linked in: ath10k_pci(+) ath10k_core ath xt_state xt_conntrack wireguard usb_wwan qmi_wwan pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211 libchacha20poly1305 iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables curve25519_neon cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables usbserial usbnet slhc sch_cake poly1305_arm nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c hwmon crc_ccitt compat chacha_neon cdc_wdm sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic
[   22.916501]  act_skbedit act_mirred act_gact ip6_gre ip_gre gre ifb ip6_tunnel tunnel6 ip_tunnel vxlan udp_tunnel ip6_udp_tunnel sha512_arm ghash_arm_ce cmac leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom gpio_button_hotplug mii crc32c_generic
[   23.011343] CPU: 2 PID: 9 Comm: kworker/u8:0 Tainted: G        W         5.15.113 #0
[   23.033561] Hardware name: Generic DT based system
[   23.041369] Workqueue: ath10k_wq ath10k_core_register_work [ath10k_core]
[   23.045977] [<c030d488>] (unwind_backtrace) from [<c030986c>] (show_stack+0x10/0x14)
[   23.052833] [<c030986c>] (show_stack) from [<c0602da0>] (dump_stack_lvl+0x40/0x4c)
[   23.060559] [<c0602da0>] (dump_stack_lvl) from [<c032274c>] (__warn+0x8c/0x100)
[   23.067936] [<c032274c>] (__warn) from [<c0322828>] (warn_slowpath_fmt+0x68/0x78)
[   23.075140] [<c0322828>] (warn_slowpath_fmt) from [<c033c9a8>] (__flush_work+0x2d8/0x310)
[   23.082782] [<c033c9a8>] (__flush_work) from [<c033cb9c>] (__cancel_work_timer+0xe0/0x1d0)
[   23.090940] [<c033cb9c>] (__cancel_work_timer) from [<bf4ce360>] (ath10k_htc_stop_hl+0x28/0x4c [ath10k_core])
[   23.099107] [<bf4ce360>] (ath10k_htc_stop_hl [ath10k_core]) from [<bf4d7010>] (ath10k_htt_tx_free+0x14/0x48 [ath10k_core])
[   23.109089] [<bf4d7010>] (ath10k_htt_tx_free [ath10k_core]) from [<bf4cae54>] (ath10k_core_start+0xb0c/0x15b8 [ath10k_core])
[   23.120027] [<bf4cae54>] (ath10k_core_start [ath10k_core]) from [<bf4cc75c>] (ath10k_core_register_work+0x638/0xb74 [ath10k_core])
[   23.131398] [<bf4cc75c>] (ath10k_core_register_work [ath10k_core]) from [<c033b608>] (process_one_work+0x210/0x478)
[   23.142942] [<c033b608>] (process_one_work) from [<c033c184>] (worker_thread+0x4c/0x598)
[   23.153267] [<c033c184>] (worker_thread) from [<c0342b08>] (kthread+0x148/0x164)
[   23.161601] [<c0342b08>] (kthread) from [<c0300110>] (ret_from_fork+0x14/0x24)
[   23.168976] Exception stack(0xc1061fb0 to 0xc1061ff8)
[   23.176009] 1fa0:                                     00000000 00000000 00000000 00000000
[   23.181136] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   23.189295] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[   23.197484] ---[ end trace 93d8523097d51501 ]---
[   23.204055] ath10k_ahb a000000.wifi: could not init core (-110)
[   23.208999] ath10k_ahb a000000.wifi: could not probe fw (-110)
[   24.170752] ath10k_ahb a800000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   24.170825] ath10k_ahb a800000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   24.184187] ath10k_ahb a800000.wifi: firmware ver 10.4b-ct-4019-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 6b2b5c5b
[... 16 lines cut ...]
r23069-e2701e0f33

### OpenWrt target/subtarget

ipq40xx/generic

### Device

ZTE MF289F

### Image kind

Official downloaded image

### Steps to reproduce

Flash Openwrt 23.05 RC1 on affected Device

### Actual behaviour

No 2,4 GHZ Wlan Device after Flashing Openwrt 23.05 RC1, 22.03 working without issues.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

tango38317, 2023-06-12:
Tried iwinfo, shows only one device:
iwinfo
wlan0     ESSID: unknown
          Access Point: C8:EA:F8:9E:A2:66
          Mode: Client  Channel: unknown (unknown)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 0 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11ac/n
          Hardware: embedded [Qualcomm Atheros IPQ4019]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

MayorBug, 2023-06-12:
Can confirm, I have the same issue

fisch2, 2023-06-17:
I was also affected by this issue a few weeks ago on snapshot. I've been using ASU frequently to upgrade my device to the next snapshot and one day I wasn't able to, because of an invalid package selection, which seems to have been caused by this commit: 8217f02a1c5e2396f083793575de2e79811842e8 . After I fixed my package selection and upgraded, 2.4GHz Wifi broke, so I switched back to 22.03.

tango38317, 2023-06-22:
Hi,
for a test i have rolled back the changes in makefile and generic.mk for ZTE_MF289F 
added in package/firmware/ipq-wifi/Makefile: 
zte_mf289f 
$(eval $(call generate-ipq-wifi-package,zte_mf289f,ZTE MF289F))

in target/linux/ipq40xx/image/generic.mk
added: 
DEVICE_PACKAGES += ipq-wifi-zte_mf289f ath10k-firmware-qca9984-ct
removed:
DEVICE_PACKAGES += ath10k-firmware-qca9984-ct

at the affected lines, tried to build a firmware for ZTE-MF289F and now 2,4 GHZ Wlan is working.

br
Chris

ynezz, 2023-06-22:
> Hi, for a test i have rolled back the changes in makefile and generic.mk for ZTE_MF289F added in package/firmware/ipq-wifi/Makefile: zte_mf289f (eval(call generate-ipq-wifi-package,zte_mf289f,ZTE [...]

Well, I've just unpacked `board-2.bin` from [ath10k-board-qca4019_20230515-1_arm_cortex-a7_neon-vfpv4.ipk](https://downloads.openwrt.org/releases/23.05.0-rc1/packages/arm_cortex-a7_neon-vfpv4/base/ath10k-board-qca4019_20230515-1_arm_cortex-a7_neon-vfpv4.ipk) and compared that with [board-zte_mf289f.qca4019](https://github.com/openwrt/firmware_qca-wireless/blob/main/board-zte_mf289f.qca4019) and they're the same, so I'm quite puzzled:

```shell
175139f44d2ae01c2b23e093d571e962  bus=ahb,bmi-chip-id=0,bmi-board-id=20,variant=zte,mf289f.bin
175139f44d2ae01c2b23e093d571e962  board-zte_mf289f.qca4019
```

ynezz, 2023-06-22:
> which seems to have been caused by this commit: https://github.com/openwrt/openwrt/commit/8217f02a1c5e2396f083793575de2e79811842e8

@chunkeey you've probably more context, can you please take a look?

gkourachanis, 2023-06-30:
Hey guys. I created a pull request that fixes this issue. ( https://github.com/openwrt/openwrt/pull/13019 ).
Hopefully it gets merged before 23.05.0 gets released.
( btw sorry for the rebase, I just wanted to sign my commit messages )

commit 75505c5e (2023-07-04): ipq-wifi: fix upstream board-2.bin ZTE M289F snafu

The upstream board-2.bin file in the linux-firmware.git
repository for the QCA4019 contains a packed board-2.bin
for this device for both 2.4G and 5G wifis. This isn't
something that the ath10k driver supports.

Until this feature either gets implemented - which is
very unlikely -, or the upstream boardfile is mended
(both, the original submitter and ath10k-firmware
custodian have been notified). OpenWrt will go back
and use its own bespoke boardfile. This unfortunately
means that 2.4G and on some revisions the 5G WiFi is
not available in the [... cut]
commit 894b2086 (2023-07-04): ipq-wifi: fix upstream board-2.bin ZTE M289F snafu

The upstream board-2.bin file in the linux-firmware.git
repository for the QCA4019 contains a packed board-2.bin
for this device for both 2.4G and 5G wifis. This isn't
something that the ath10k driver supports.

Until this feature either gets implemented - which is
very unlikely -, or the upstream boardfile is mended
(both, the original submitter and ath10k-firmware
custodian have been notified). OpenWrt will go back
and use its own bespoke boardfile. This unfortunately
means that 2.4G and on some revisions the 5G WiFi is
not available in the [... cut]
commit 80a6b0a9 (2023-07-04): ipq-wifi: fix upstream board-2.bin ZTE M289F snafu

The upstream board-2.bin file in the linux-firmware.git
repository for the QCA4019 contains a packed board-2.bin
for this device for both 2.4G and 5G wifis. This isn't
something that the ath10k driver supports.

Until this feature either gets implemented - which is
very unlikely -, or the upstream boardfile is mended
(both, the original submitter and ath10k-firmware
custodian have been notified). OpenWrt will go back
and use its own bespoke boardfile. This unfortunately
means that 2.4G and on some revisions the 5G WiFi is
not available in the [... cut]
closed 2023-07-04 by openwrt-bot