openwrt/openwrt #12888: [23.05-rc1] ramips/mt7621 Disabling the LAN Led is not possible on Netgear WAX202

MichiMunich, 2023-06-12:
### Describe the bug

Since 23.05-rc1 it is no longer possible to disable the LAN Leds on the Netgear WAX202. 
Under "/sys/class/leds/" I can see differences depending on the version.
With 22.03.x this was still possible and under /sys/class/leds there was also e.g. "green:lan1". The "green:lanX" directories no longer exist under 23.05-rc1. 
Below are the differences of the directories between 22.03.5 and 23.05-rc1.

Output 23.05-rc1 version:

ls /sys/class/leds/
blue:net      blue:wifi5g   green:power   green:wifi5g  mt76-phy1     orange:lan2   orange:power
blue:wifi2g   green:net     green:wifi2g  mt76-phy0     orange:lan1   orange:lan3

Output 22.03.5:

ls /sys/class/leds/
blue:net      blue:wifi5g   green:lan2    green:net     green:wifi2g  orange:lan1   orange:lan3
blue:wifi2g   green:lan1    green:lan3    green:power   green:wifi5g  orange:lan2   orange:power

Unfortunately, the LAN LEDs cannot be deactivated via green:net.

### OpenWrt version

r23069-e2701e0f33

### OpenWrt target/subtarget

ramips/mt7621

### Device

Netgear WAX202

### Image kind

Official downloaded image

### Steps to reproduce

Upgrade from 22.03.5 with disabled LAN Leds to 23.05-rc1 and the LAN Leds are blinking

### Actual behaviour

LAN Led cannot be disabled and lights/flashes green as soon as a LAN cable is connected.

### Expected behaviour

LAN Led should be able to be deactivated again

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Djfe, 2023-06-27:
The devs disabled software LEDs in favor of the default hw functionality (which is setup by the bootloader). Because
- It's more performant (hw offloading, the switch controls the leds)
- It detects all ethernet frames on LAN (software missed offloaded frames that were sent between lan ports)

hw LEDs can be configured by software but the necessary code is currently being added to the Linux Kernel.
Once it's ready you will be able to deactivate them again in software.

I saw someone post some custom code (taken from oem image) just for disabling their led's but I cannot find it right now.

MichiMunich, 2023-06-29:
Thank you for the explanations. I will close the issue for now.

closed 2023-06-29 by MichiMunich