openwrt/openwrt #12892: Ubiquiti UNIFI AP AC PRO GEN2

Zarouka, 2023-06-12:
### Describe the bug

Good morning,
I have a problem installing openwrt on Ubiquiti UNIFI AP AC PRO GEN2. I got the AP AC PRO GEN2 with version 6.5.28. I get an invalid version error message, when I try to downgrade to 3.7.58.

### OpenWrt version

XXXXXXXXXXXXXXXXXX

### OpenWrt target/subtarget

XXXXXXXXXXXXXXXXXX

### Device

biquiti UNIFI AP AC PRO GEN2

### Image kind

Official downloaded image

### Steps to reproduce

``root@DELL:/home/dell007/Downloads# md5sum ubnt.bin 
5bb23b387defcbd1f6cda5819c7905e5  ubnt.bin
root@DELL:/home/dell007/Downloads# 

root@DELL:/home/dell007/Downloads# ./flash_openwrt.sh
+ IP=192.168.1.20
+ IP_WRT=192.168.1.1
+ HOST=ubnt@192.168.1.20
+ wping 192.168.1.20
+ ping -c 1 -W 1 192.168.1.20
+ sleep 10
+ iscp ubnt.bin ubnt@192.168.1.20:/tmp/
+ sshpass -p ubnt scp -o StrictHostKeyChecking=no -o UserKnownHostsFile=/dev/null -o LogLevel=error -oMacs=hmac-sha1 -O ubnt.bin ubnt@192.168.1.20:/tmp/
+ issh ubnt@192.168.1.20 -C fwupdate.real -m /tmp/ubnt.bin
+ sshpass -p ubnt ssh -oMacs=hmac-sha1 -o StrictHostKeyChecking=no -o UserKnownHostsFile=/dev/null -o PubkeyAuthentication=no -o LogLevel=error -o ServerAliveInterval=3 -o ServerAliveCountMax=3 ubnt@192.168.1.20 -C fwupdate.real -m /tmp/ubnt.bin
Invalid version 'BZ.qca956x.v3.7.58.6385.170508.0957'
+ wping 192.168.1.20
+ ping -c 1 -W 1 192.168.1.20
+ sleep 10
+ iscp openwrt.bin ubnt@192.168.1.20:/tmp/
+ sshpass -p ubnt scp -o StrictHostKeyChecking=no -o UserKnownHostsFile=/dev/null -o LogLevel=error -oMacs=hmac-sha1 -O openwrt.bin ubnt@192.168.1.20:/tmp/
+ issh ubnt@192.168.1.20 mtd write /tmp/openwrt.bin
+ sshpass -p ubnt ssh -oMacs=hmac-sha1 -o StrictHostKeyChecking=no -o UserKnownHostsFile=/dev/null -o PubkeyAuthentication=no -o LogLevel=error -o ServerAliveInterval=3 -o ServerAliveCountMax=3 ubnt@192.168.1.20 mtd write /tmp/openwrt.bin
ash: mtd: not found``

### Actual behaviour

xxxxxxxxxxxxxxxxxxxxxxxxxxxxx

### Expected behaviour

XXXXXXXXXXXXXXXXXXXXXXXXXXXXX

### Additional info

XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

### Diffconfig

```text
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Djfe, 2023-06-12:
maybe try running this tool instead :) 
https://codeberg.org/pitch/firmwareloader_uap_ac_pro/releases

Zarouka, 2023-06-12:
I launched but nothing happens

Djfe, 2023-06-12:
or here are German instructions on how to flash the device (just use the openwrt firmware instead)
https://forum.darmstadt.freifunk.net/t/unifi-ap-erstinstallation/790

downgrade doesn't appear to be necessary anymore but the wiki wasn't updated accordingly. I don't own the device myself though I know of atleast two persons who used the tool in my last post successfully on their AP AC Pro

Djfe, 2023-06-12:
> I launched but nothing happens

maybe read the instructions on how to use it
https://codeberg.org/pitch/firmwareloader_uap_ac_pro/src/branch/main/README.md

you need to download and rename the openwrt image and you need to be connected to the router already (static IPs in your system are set)

Zarouka, 2023-06-13:
perfect problem solved. Thanks a lot

closed 2023-06-13 by Zarouka

j-oss2023, 2024-02-27:
https://blog.christophersmart.com/2021/10/02/flashing-openwrt-onto-ubiquiti-unifi-ap-ac/

This works as well...