openwrt/openwrt #12914: [MT7530/MT7621] RX speed progressively gets worse with time

rany2, 2023-06-15:
### Describe the bug

Over time, the Ethernet performance gets progressively worse until the situation is unbearable. Sometimes it starts off extremely poorly from the get-go but other times it has a good and healthy start of 800Mbit/s only to degrade to ~480Mbit/s and worse.

The dmesg (even with CONFIG_DEVICE_DEBUG and CONFIG_PCI_DEBUG) are identical to each other from when it starts off healthy and when it starts off poorly. Completely power cycling the device does not make a difference either, the issue occurs seemingly at random. In both cases even if I was fortunate enough to have it working from the get-go, it will eventually break again and have abysmal performance. 

The temperature does not appear to be related, a measurement shows ~50 C so it is not throttling itself it seems.

NOTE: when I say "but other times it has a good" this only occurs maybe about 1 in 10 times but as previously stated, it doesn't matter; it will eventually regress typically under an hour.

NOTE2: the device is a dumb AP and no other services besides hostapd, dawn, and uhttpd are running. 

### OpenWrt version

r23376-bc3af3b858

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Self-built image

### Steps to reproduce

1. Run iperf3 over extended period of time
2. Watch performance start off decentish at ~800Mbit/s (or poorly if you weren't lucky at bootup)
3. Try again in a few minutes/hours
4. Note difference in bitrate 

### Expected behaviour

Performance shouldn't just drop out of nowhere.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rany2, 2023-06-15:
@brada4 

```
root@asus:~# ifconfig eth0
eth0      Link encap:Ethernet  HWaddr FC:34:97:0B:39:70  
          inet6 addr: fe80::fe34:97ff:fe0b:3970/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:2026  Metric:1
          RX packets:9540015 errors:0 dropped:0 overruns:0 frame:0
          TX packets:971968 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:14714527840 (13.7 GiB)  TX bytes:243472624 (232.1 MiB)
          Interrupt:21 

root@asus:~# ethtool -S eth0
NIC statistics:
     tx_bytes: 243477840
     tx_packets: 971989
     tx_skip: 0
     tx_collisions: 0
     rx_bytes: 14714538001
     rx_packets: 9540035
     rx_overflow: 0
     rx_fcs_errors: 0
     rx_short_errors: 0
     rx_long_errors: 0
     rx_checksum_errors: 0
     rx_flow_control_packets: 0
     rx_xdp_redirect: 0
     rx_xdp_pass: 0
     rx_xdp_drop: 0
     rx_xdp_tx: 0
     rx_xdp_tx_errors: 0
     tx_xdp_xmit: 0
     tx_xdp_xmit_errors: 0
     p06_TxDrop: 0
     p06_TxCrcErr: 0
     p06_TxUnicast: 9536051
     p06_TxMulticast: 3285
     p06_TxBroadcast: 699
     p06_TxCollision: 0
     p06_TxSingleCollision: 0
     p06_TxMultipleCollision: 0
     p06_TxDeferred: 0
     p06_TxLateCollision: 0
     p06_TxExcessiveCollistion: 0
     p06_TxPause: 0
     p06_TxPktSz64: 0
     p06_TxPktSz65To127: 157697
     p06_TxPktSz128To255: 104018
     p06_TxPktSz256To511: 241250
     p06_TxPktSz512To1023: 24126
     p06_Tx1024ToMax: 9012944
     p06_TxBytes: 14752698141
     p06_RxDrop: 0
     p06_RxFiltering: 14
     p06_RxUnicast: 967368
     p06_RxMulticast: 3974
     p06_RxBroadcast: 647
     p06_RxAlignErr: 0
     p06_RxCrcErr: 0
     p06_RxUnderSizeErr: 0
     p06_RxFragErr: 0
     p06_RxOverSzErr: 0
     p06_RxJabberErr: 0
     p06_RxPause: 11518
     p06_RxPktSz64: 11518
     p06_RxPktSz65To127: 732209
     p06_RxPktSz128To255: 94775
     p06_RxPktSz256To511: 25461
     p06_RxPktSz512To1023: 16643
     p06_RxPktSz1024ToMax: 102901
     p06_RxBytes: 244214992
     p06_RxCtrlDrop: 0
     p06_RxIngressDrop: 0
     p06_RxArlDrop: 0
```

rany2, 2023-06-15:
Maybe the ethtool stats are wrong? There is no way there are no packets dropped.

rany2, 2023-06-15:
Here are the ethtool stats for the only port connected to the device:
```
root@asus:~# ethtool -S lan2
NIC statistics:
     tx_packets: 1197734
     tx_bytes: 251891696
     rx_packets: 14841316
     rx_bytes: 24801706385
     TxDrop: 0
     TxCrcErr: 0
     TxUnicast: 1241843
     TxMulticast: 2125
     TxBroadcast: 334
     TxCollision: 0
     TxSingleCollision: 0
     TxMultipleCollision: 0
     TxDeferred: 0
     TxLateCollision: 0
     TxExcessiveCollistion: 0
     TxPause: 0
     TxPktSz64: 7712
     TxPktSz65To127: 995898
     TxPktSz128To255: 95430
     TxPktSz256To511: 24690
     TxPktSz512To1023: 17069
     Tx1024ToMax: 103503
     TxBytes: 260450409
     RxDrop: 90436
     RxFiltering: 0
     RxUnicast: 14927696
     RxMulticast: 3358
     RxBroadcast: 699
     RxAlignErr: 0
     RxCrcErr: 0
     RxUnderSizeErr: 0
     RxFragErr: 0
     RxOverSzErr: 0
     RxJabberErr: 0
     RxPause: 0
     RxPktSz64: 9533
     RxPktSz65To127: 257892
     RxPktSz128To255: 108847
     RxPktSz256To511: 241675
     RxPktSz512To1023: 27653
     RxPktSz1024ToMax: 14286153
     RxBytes: 25206069331
     RxCtrlDrop: 0
     RxIngressDrop: 90436
     RxArlDrop: 0
```

rany2, 2023-06-15:
When checking the `lan2` DSA port's ethtool stats I could confirm that the RX dropped stats went from 109398 to 118852 while doing a 10 second iperf test. So it seems like there a lot of failures when it reaches 480Mbit/s.

brada4, 2023-06-15:
That seems like flow control in the switch overreacting. In principle there shoud be some drops if you approach CPU gigabit port from 2 other gigabit ports via switch that should induce slowing down TCP connections and sending flow control packets to clients supporting that, but not in the clean environment.
Can you try installing irqbalance and/or enabling network packet steering? Unlikely to help but who knows.

rany2, 2023-06-15:
I should have noted this but I tried that. This is with irqbalance and packet steering enabled. It occurred with those two disabled as well.

brada4, 2023-06-15:
Lets disable all flow control in path:
`ethtool -a eth0`
`ethtool -A eth0 auto off tx off rx off`
also lanX port(s) involved. TCP should gracefully survive small drop rate on connection and slow down transmission per connection, not inflicting damage on low speed connections around.

rany2, 2023-06-15:
I just checked and I have realized that none of my devices support sending pause frames but somehow this is the only device that registers that it has received something on the system port?

brada4, 2023-06-15:
Switch ASIC may drop few packets from all ports receiving pause frame from invisible system ports. Most enterprise switches do not support any flow control at gigabit.

rany2, 2023-06-15:
> Lets disable all flow control in path: `ethtool -a eth0` `ethtool -A eth0 auto off tx off rx off` also lanX port(s) involved. TCP should gracefully survive small drop rate on connection and slow down transmission per connection, not inflicting damage on low speed connections around.

No luck with your suggestion, no impact unfortunately

ptpt52, 2023-06-15:
how to run the test?
```
iperf3 client --- lan1[Router]wan --- iperf3 server
```

hardware offload enable?

rany2, 2023-06-16:
Something I tried was adding a delay after gpio reset because this what OEM driver did, but it does not help; but the fact that it sometimes works on boot up only to degrade with time tells me something is broken in initialization

rany2, 2023-06-16:
I just noticed that https://github.com/openwrt/openwrt/blob/main/target/linux/generic/pending-6.1/723-net-mt7531-ensure-all-MACs-are-powered-down-before-r.patch is wrong. It should be PMCR_FORCE_LNK and that change impacts mt7530_setup not mt7531_setup.

rany2, 2023-06-16:
> I just noticed that https://github.com/openwrt/openwrt/blob/main/target/linux/generic/pending-6.1/723-net-mt7531-ensure-all-MACs-are-powered-down-before-r.patch is wrong. It should be PMCR_FORCE_LNK and that change impacts mt7530_setup not mt7531_setup so patch name is wrong as well.

OEM driver uses PMCR_FORCE_MODE actually

rany2, 2023-06-16:
it doesn't make a difference anyway, issue remains

brada4, 2023-06-16:
Ethrool -k offluads?

rany2, 2023-06-16:
```
# ethtool -k eth0 | grep -v ': off'
Features for eth0:
rx-checksumming: on
tx-checksumming: on
	tx-checksum-ipv4: on
	tx-checksum-ipv6: on
scatter-gather: on
	tx-scatter-gather: on
tcp-segmentation-offload: on
	tx-tcp-segmentation: on
	tx-tcp-ecn-segmentation: on
	tx-tcp-mangleid-segmentation: on
	tx-tcp6-segmentation: on
generic-segmentation-offload: on
generic-receive-offload: on
tx-vlan-offload: on
hw-tc-offload: on
```

(When I said "not using hardware offloading"  I was referring to flow offloading, just fyi)

brada4, 2023-06-17:
(i got that)

Disable offloads one by one (starting from bottom) and do quick measurement if ot magically starts working at some point.

rany2, 2023-06-17:
yeah no luck with that, issue remains..

rany2, 2023-06-17:
@dangowrt @nbd168 When under load for longer periods of time, this sometimes ends up happening. Could this help with understanding the issue?

```
[  197.095778] Unhandled kernel unaligned access[#1]:
[  197.100617] CPU: 3 PID: 271 Comm: napi/mtk_eth-6 Not tainted 5.15.117 #0
[  197.107309] $ 0   : 00000000 00000001 00000122 80b7ca9c
[  197.112543] $ 4   : fffffffe 00000000 81027bb8 00000000
[  197.117774] $ 8   : 0061d6ef 00000040 00000039 8155e3c0
[  197.122996] $12   : 00000b97 000034fc 00000000 00000000
[  197.128220] $16   : 80b1dc20 00000000 80b2c620 00150015
[  197.133442] $20   : 81027bb8 80b7ca98 80150015 8141df00
[  197.138669] $24   : 00000000 8032e3b8                  
[  197.143893] $28   : 81d7c000 81d7dcb8 00000000 801b1140
[  197.149115] Hi    : 00000000
[  197.151981] Lo    : 00000041
[  197.154844] epc   : 801b1374 ___slab_alloc.constprop.0+0x530/0x790
[  197.161032] ra    : 801b1140 ___slab_alloc.constprop.0+0x2fc/0x790
[  197.167196] Status: 1100fc02 KERNEL EXL 
[  197.171113] Cause : 40800014 (ExcCode 05)
[  197.175104] BadVA : 00000122
[  197.177967] PrId  : 0001992f (MIPS 1004Kc)
[  197.182041] Modules linked in: wireguard mt7915e mt76_connac_lib mt76 mac80211 libchacha20poly1305 cfg80211 poly1305_mips libcurve25519_generic libcrc32c hwmon compat cls_flower chacha_mips act_vlan cls_bpf act_bpf sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport ip_tunnel vxlan udp_tunnel ip6_udp_tunnel sha512_generic seqiv jitterentropy_rng drbg hmac cmac leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore nls_base usb_common crc32c_generic
[  197.232644] Process napi/mtk_eth-6 (pid: 271, threadinfo=769a9b6c, task=444f8b41, tls=00000000)
[  197.241319] Stack : 00000000 fffff7ff ff7f5530 81cea840 00000000 00000015 81416308 00000000
[  197.249681]         81416300 00000001 844e63c0 81d7dcf0 81d7dce8 81d7dce8 85bf8fd4 00000000
[  197.258039]         80150015 00000920 0000081e 00000920 00000a20 8141df00 00000a20 00000000
[  197.266393]         00000000 8089a3f0 808a0000 81d248b8 81d2c000 801b1aac 8155e210 00000a20
[  197.274754]         000005ee 00000013 00000a20 801b1f94 0000005f 81d248b8 81d2c000 800a4b74
[  197.283113]         ...
[  197.285556] Call Trace:
[  197.287986] [<801b1374>] ___slab_alloc.constprop.0+0x530/0x790
[  197.293821] [<801b1aac>] __slab_alloc.constprop.0.isra.0+0x3c/0x90
[  197.299990] [<801b1f94>] kmem_cache_alloc+0x494/0x578
[  197.305031] [<80519438>] build_skb+0x34/0x190
[  197.309380] [<804d9bf0>] mtk_napi_rx+0x8c8/0xeb0
[  197.313996] [<8053e6f4>] __napi_poll+0x70/0x1f8
[  197.318545] [<8053e924>] napi_threaded_poll+0xa8/0x108
[  197.323674] [<80053d64>] kthread+0x17c/0x1a0
[  197.327946] [<800034cc>] ret_from_kernel_thread+0x14/0x1c

[  197.334821] Code: 8e420008  8e430004  ac620004 <ac430000> 8fa30020  24020100  ae420004  24020122  ae420008 

[  197.346125] ---[ end trace bfba77fb6de4beaa ]---
[  197.350802] Kernel panic - not syncing: Fatal exception in interrupt

```

and

```
[   80.071102] CPU 2 Unable to handle kernel paging request at virtual address 744d2094, epc == 801b1be8, ra == 80519438                              
[   80.081760] Oops[#1]:                                                                                                                              
[   80.084050] CPU: 2 PID: 274 Comm: napi/mtk_eth-6 Not tainted 5.15.117 #0                   
[   80.090728] $ 0   : 00000000 00000001 744d2094 4d749420                                                                                            
[   80.095963] $ 4   : ac80cabf 00000a20 00000001 00000000                                                                                            
[   80.101184] $ 8   : 00091472 00000040 00000039 8155ffc0                                                                                            
[   80.106410] $12   : 00000b97 000034fc 00000000 00000000                                      
[   80.111632] $16   : 00000a20 8141da00 00000a20 00000000                                                      
[   80.116860] $20   : 744d203c 8089a3f0 808a0000 814418b8
[   80.122082] $24   : 00000000 8032e3b8                              
[   80.127307] $28   : 81dfe000 81dffd48 81d6e000 80519438                                                                                                                                                                       
[   80.132530] Hi    : 00000000                                                                                                                                                                  
[   80.135397] Lo    : 00000041                                            
[   80.138259] epc   : 801b1be8 kmem_cache_alloc+0xe8/0x578        
[   80.143587] ra    : 80519438 build_skb+0x34/0x190                                            
[   80.148294] Status: 1100fc03 KERNEL EXL IE                                                   
[   80.152477] Cause : 40800008 (ExcCode 02)                                                                                                                                                                                     
[   80.156470] BadVA : 744d2094                                                                                                                                                                  
[... cut ...]
[   80.297264] [<800034cc>] ret_from_kernel_thread+0x14/0x1c                                                                                                                                                                                                                  

[   80.304137] Code: 8e240078  02821021  7c0218a0 <8c420000> 00231c02  00821026  00433826  41656000  30a50001                                                                                                                                                                 

[   80.315460] ---[ end trace 4067ba48270f4645 ]---                                                                                    

```

rany2, 2023-06-17:
If it's useful:

```
$ ../../staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/bin/mipsel-openwrt-linux-gdb linux-ramips_mt7621/linux-5.15.117/drivers/net/ethernet/mediatek/mtk_eth_soc.o
GNU gdb (GDB) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=mipsel-openwrt-linux-musl".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from linux-ramips_mt7621/linux-5.15.117/drivers/net/ethernet/mediatek/mtk_eth_soc.o...
(gdb) l *mtk_napi_rx+0x8c8/0xeb0
0x89d0 is in mtk_napi_rx (drivers/net/ethernet/mediatek/mtk_eth_soc.c:2616).
2611		return tx_done;
2612	}
2613	
2614	static int mtk_napi_rx(struct napi_struct *napi, int budget)
2615	{
2616		struct mtk_eth *eth = container_of(napi, struct mtk_eth, rx_napi);
2617		const struct mtk_reg_map *reg_map = eth->soc->reg_map;
2618		int rx_done_total = 0;
2619	
2620		mtk_handle_status_irq(eth);
(gdb) 
```

rany2, 2023-06-17:
Nevermind it's a known issue: https://github.com/openwrt/openwrt/issues/9561

rany2, 2023-06-17:
I can't trigger those two kernel panics anymore, not sure what was up with those :/

brada4, 2023-06-17:
Unaligned access is most likely miscompilation ie to do with compiler parameters.
Second is failing kmem allocation typically due to oom. In ideal world driver could gracefully continue on such failure like drop packet, but doe not seem the case.

Openwrt does not have slab stats by default, with them you could be able to identify if net skb pool uses too much or some other memory pool grow to all memory. Things like slabtop, /proc/slabinfo etc.

rany2, 2023-06-17:
@brada4 I doubt it has to do with oom, the device has 256MB of RAM and these are the only services running:
```
USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root         1  0.1  0.5   1972  1328 ?        Ss   19:34   0:05 /sbin/procd
ubus       544  0.0  0.4   1536  1072 ?        S    19:35   0:02 /sbin/ubusd
root       545  0.0  0.2   1064   588 ttyS0    Ss+  19:35   0:00 /sbin/askfirst /usr/libexec/login.sh
root       580  0.0  0.3   1208   792 ?        S    19:35   0:00 /sbin/urngd
root       626  0.0  0.6   2076  1572 ?        S    19:35   0:01 /usr/sbin/haveged -F -w 1024 -d 32 -i 32 -v 1
logd       931  0.0  0.3   1488   924 ?        S    19:35   0:01 /sbin/logd -S 64
root       990  0.0  0.8   3460  2156 ?        S    19:35   0:00 /sbin/rpcd -s /var/run/ubus/ubus.sock -t 30
root      1135  0.0  0.4   1280  1028 ?        S    19:35   0:00 /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -s -g -p 22 -K 300 -T 3
root      1186  0.0  0.3   2916   988 ?        S    19:35   0:00 /sbin/ujail -t 5 -n hostapd -U network -G network -C /etc/capabilities/wpad.json -c -- /usr/sbin/hostapd -s -g /var/run/hostapd/global
root      1187  0.0  0.3   2916   988 ?        S    19:35   0:00 /sbin/ujail -t 5 -n wpa_supplicant -U network -G network -C /etc/capabilities/wpad.json -c -- /usr/sbin/wpa_supplicant -n -s -g /var/run/wpa_supplicant/global
network   1211  1.3  2.1   7036  5340 ?        S    19:35   1:01 /usr/sbin/hostapd -s -g /var/run/hostapd/global
network   1212  0.0  0.8   6588  2156 ?        S    19:35   0:00 /usr/sbin/wpa_supplicant -n -s -g /var/run/wpa_supplicant/global
root      1251  0.0  0.5   2176  1448 ?        S    19:35   0:02 /sbin/netifd
root      1520  0.0  0.3   1648   904 ?        S    19:35   0:00 /usr/sbin/uhttpd -f -h /www -r asus -x /cgi-bin -t 60 -T 30 -k 20 -A 1 -n 3 -N 100 -R -p 0.0.0.0:80 -p [::]:80
root      1703  0.0  0.4   2916  1204 ?        S    19:35   0:00 /sbin/ujail -t 5 -n umdns -S /etc/seccomp/umdns.json -u -l -- /usr/sbin/umdns
root      1712  0.0  0.4   1856  1216 ?        S    19:35   0:00 /usr/sbin/umdns
root      1810  2.6  0.8   3328  2108 ?        SL   19:35   1:57 /usr/sbin/dawn
root      1873  0.0  0.3   1356   864 ?        Sl   19:35   0:01 /usr/sbin/irqbalance -f -c 2 -t 10
root      2338  0.0  0.5   2916  1256 ?        S    19:35   0:00 /sbin/ujail -t 5 -n ntpd -U ntp -G ntp -C /etc/capabilities/ntpd.json -c -u -r /bin/ubus -r /usr/bin/env -r /usr/bin/jshn -r /usr/sbin/ntpd-hotplug -r /usr/share/libubox/jshn.sh -- /usr/sbin/ntpd -n -N -S /usr/sbin/ntpd-hotplug -p 10.146.109.1
ntp       2369  0.0  0.4   1380  1016 ?        S    19:35   0:00 /usr/sbin/ntpd -n -N -S /usr/sbin/ntpd-hotplug -p 10.146.109.1
root      3237  0.7  0.4   1300  1008 ?        S    20:49   0:00 /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -s -g -p 22 -K 300 -T 3 -2 9
root      3238  0.0  0.4   1384  1076 pts/0    Ss   20:49   0:00 -ash
root      3276  0.0  0.3   1864   952 pts/0    R+   20:50   0:00 ps aux
root      3277  0.0  0.3   1380   976 pts/0    S+   20:50   0:00 grep -Ev \[\S+\]$
```

and this is `free -h`:

```
 free -h
              total        used        free      shared  buff/cache   available
Mem:          243Mi        55Mi       165Mi       0.0Ki        22Mi       152Mi
Swap:            0B          0B          0B
``

rany2, 2023-06-17:
> Unaligned access is most likely miscompilation ie to do with compiler parameters.

I don't know, but I am doing a clean build every time so I don't think it is?

brada4, 2023-06-17:
It is in different place, catch kmem pool growing wild, like a runaway process could be detected with your used tools. I don't have clear recipe to add support for zoneinfo in a single rebuild, but it still will be chasing ghosts

rany2, 2023-06-18:
UPDATE: OpenWRT 21.02 does not have this issue, all subsequent versions do.

rany2, 2023-06-19:
Another observation, on 21.02 the ERR counter in /proc/interrupts is under 50 whereas on SNAPSHOT/23.05/22.03 it exceeds a million.

SNAPSHOT:

```
           CPU0       CPU1       CPU2       CPU3       
  8:   11560538   11548657   11624111   11519204  MIPS GIC Local   1  timer
  9:    6188256          0          0          0  MIPS GIC  63  IPI call
 10:          0   13486634          0          0  MIPS GIC  64  IPI call
 11:          0          0   11192002          0  MIPS GIC  65  IPI call
 12:          0          0          0    6396570  MIPS GIC  66  IPI call
 13:       8029          0          0          0  MIPS GIC  67  IPI resched
 14:          0      40426          0          0  MIPS GIC  68  IPI resched
 15:          0          0      54612          0  MIPS GIC  69  IPI resched
 16:          0          0          0       8949  MIPS GIC  70  IPI resched
 17:          0          0          0          0  MIPS GIC  19  1e000600.gpio-bank0, 1e000600.gpio-bank1, 1e000600.gpio-bank2
 19:         14          0          0          0  MIPS GIC  33  ttyS0
 20:          0          0          0          0  MIPS GIC  29  xhci-hcd:usb1
 21:    7802624          0          0          0  MIPS GIC  10  1e100000.ethernet
 22:    1809817          0          0          0  MIPS GIC  11  PCIe PME, aerdrv, mt7915e-hif
 23:   15054549          0          0          0  MIPS GIC  31  PCIe PME, aerdrv, mt7915e
 24:          3          0          0          0  MIPS GIC  30  mt7530
 25:          0          0          0          0    mt7530   1  mt7530-0:01
 26:          0          0          0          0    mt7530   2  mt7530-0:02
 27:          3          0          0          0    mt7530   3  mt7530-0:03
ERR:     998120
```

22.02:

```
           CPU0       CPU1       CPU2       CPU3       
  8:      44547      44511      44450      44505  MIPS GIC Local   1  timer
  9:      24594          0          0          0  MIPS GIC  63  IPI call
 10:          0      18399          0          0  MIPS GIC  64  IPI call
 11:          0          0      91160          0  MIPS GIC  65  IPI call
 12:          0          0          0      29323  MIPS GIC  66  IPI call
 13:       9580          0          0          0  MIPS GIC  67  IPI resched
 14:          0      26259          0          0  MIPS GIC  68  IPI resched
 15:          0          0      26170          0  MIPS GIC  69  IPI resched
 16:          0          0          0      39845  MIPS GIC  70  IPI resched
 17:          0          0          0          0  MIPS GIC  19  1e000600.gpio-bank0, 1e000600.gpio-bank1, 1e000600.gpio-bank2
 19:         14          0          0          0  MIPS GIC  33  ttyS0
 20:          0          0          0          0  MIPS GIC  29  xhci-hcd:usb1
 21:     220579          0          0          0  MIPS GIC  10  1e100000.ethernet
 23:      11272          0          0          0  MIPS GIC  11  mt7915e-hif
 24:      32708          0          0          0  MIPS GIC  31  mt7915e
 26:          0          0          0          0  1e000600.gpio  16  keys
 27:          0          0          0          0  1e000600.gpio  15  keys
ERR:         27
```

ThiloteE, 2023-06-19:
https://git-scm.com/docs/git-bisect

rany2, 2023-06-19:
@ThiloteE I am aware but there is no way I'm doing that here. Just too many changes....

rany2, 2023-06-19:
What I could say is that this started from the very first rc for 22.03, so 22.03 was always broken basically.

rany2, 2023-06-19:
Also I should mention that a bisect will be extremely time consuming because as stated in the original post, this doesn't occur all the time. Sometimes it starts off well at bootup only to degrade, sometimes the boot is solid and it only degrades later. So the testing will take AGES!

DragonBluep, 2023-06-20:
> @ptpt52 directly via router, and there is no hardware offloading enabled.
> 
> Edit: performance between switch ports is OK, getting gigabit speeds there

Running iperf3 on router side is not the right way. There has been a lot of discussion here:
https://forum.openwrt.org/t/mt7621-ethernet-performance-below-expectations/119901

By the way, the main change from 21.x to 22.x was to replace iptables with nftables, which seems to have led to significant performance degradation.

rany2, 2023-06-20:
@DragonBluep I do not have any firewall installed:

```
cfg80211              288282  4 mt7915e,mt76_connac_lib,mt76,mac80211
chacha_mips             5531  1 libchacha20poly1305
compat                  1602  2 mac80211,cfg80211
crc_ccitt               1806  0
gpio_button_hotplug     6834  0
hwmon                   7810  1 mt7915e
ip_tunnel              14022  0
leds_gpio               3090  0
ledtrig_usbport         2850  0
libchacha20poly1305     4678  1 wireguard
libcurve25519_generic   10454  1 wireguard
mac80211              547992  3 mt7915e,mt76_connac_lib,mt76
mt76                   48038  2 mt7915e,mt76_connac_lib
mt76_connac_lib        26464  1 mt7915e
mt7915e               101296  0
nls_base                5626  1 usbcore
poly1305_mips           2415  1 libchacha20poly1305
udp_tunnel              2937  2 wireguard,vxlan
usb_common              3166  3 xhci_plat_hcd,xhci_hcd,usbcore
usbcore               150574  5 ledtrig_usbport,xhci_plat_hcd,xhci_pci,xhci_mtk,xhci_hcd
vxlan                  37170  0
wireguard              53095  0
xhci_hcd              120200  3 xhci_plat_hcd,xhci_pci,xhci_mtk
xhci_mtk                5010  0
xhci_pci                3890  0
xhci_plat_hcd           5842  0
```

Regarding iperf, I do understand that however the network is in fact slower both via router and another PC on the same network. Also it doesn't mean that the perf would just drop out of nowhere.

As previously stated, this is a dumb AP with the bare minimum.

Spudz76, 2023-06-20:
Also of note 22.02 is fine and 22.03 is not whilst both are nftables, if that was even involved.

And to clarify it never drops packets it's just slower?  Is the RTT on ICMP ping noticeably higher during this also?  Do small packets such as ICMP ECHO or DNS seem unaffected while streaming/max-MTU packets have trouble?  I wonder if there is a magic/poison packet condition that triggers the problem, leading to apparent randomness but if such packet-of-death were figured out it could be reproduced to assist in bisect.

Probably the biggest clue is the interrupt ERR count, something basic system level driver must have changed.  I note the three `mt7530-0:` devices are only in the snapshot output.  What are the diffs between tag 22.02 and 22.03 for the relevant target/platform files, and could those be a smaller set to bisect?  Doesn't seem like it would be in common parts or there would be problems with many other targets.

Spudz76, 2023-06-20:
(seems weird to have a MTU:2026 btw)

Spudz76, 2023-06-20:
[Here are some musings](https://patchwork.ozlabs.org/project/netdev/cover/20181130075737.8041-1-gerg@kernel.org/) from when the mt7530 DSA driver became a thing, noting that there are quirks where packets of max-size just get silently lost... hmmmmmm

I wonder if issue disappears by setting MTU:1500 or 1492 or less, or gets worse due to fragging.  Still, 2026 seems wrong, unless things know how to do jumbos, and then why not 9000.

Spudz76, 2023-06-20:
The router I'm currently using, I don't note any problems but also have millions of interrupts ERR.  Have not tried to replicate with iperf.  I do sometimes/rarely get a slowdown with respect to Internet speeds but I have NAT and nftables stuff going and a bunch of other junk using CPU.

Also of note my MTU is 1504, but that other thread seemed to say it will tell you >1500 is illegal if you try to set it.  But it's definitely not 2026.

```
root@wonky:~# uptime
 05:33:36 up 40 days,  4:56,  load average: 2.03, 1.98, 1.99
root@wonky:~# cat /proc/interrupts
           CPU0       CPU1       CPU2       CPU3
  7:          0          0          0          0      MIPS   7  timer
  8:  347379294  347379276  347379267  347379261  MIPS GIC Local   1  timer
  9: 1013590807          0          0          0  MIPS GIC  63  IPI call
 10:          0 1315474950          0          0  MIPS GIC  64  IPI call
 11:          0          0 3254889250          0  MIPS GIC  65  IPI call
 12:          0          0          0 2764087623  MIPS GIC  66  IPI call
 13:   24159280          0          0          0  MIPS GIC  67  IPI resched
 14:          0 3038645992          0          0  MIPS GIC  68  IPI resched
 15:          0          0  122056870          0  MIPS GIC  69  IPI resched
 16:          0          0          0  498250549  MIPS GIC  70  IPI resched
 17:          0          0          0          0  MIPS GIC  19  1e000600.gpio-bank0, 1e000600.gpio-bank1, 1e000600.gpio-bank2
 19:         12          0          0          0  MIPS GIC  33  ttyS0
 20: 3797738205          0          0          0  MIPS GIC  29  xhci-hcd:usb1
 21:  592178944          0          0          0  MIPS GIC  10  1e100000.ethernet
 22:        563          0  771886119          0  MIPS GIC  11  mt7615e
 23:       2015          0          0  118339328  MIPS GIC  31  mt7615e
 25:          3          0          0          0  MIPS GIC  30  mt7530
 26:          3          0          0          0    mt7530   1  mt7530-0:01
 27:          0          0          0          0    mt7530   2  mt7530-0:02
 28:          0          0          0          0    mt7530   3  mt7530-0:03
 29:          0          0          0          0    mt7530   4  mt7530-0:04
 30:          0          0          0          0  1e000600.gpio  17  keys
 31:          0          0          0          0  1e000600.gpio  12  keys
ERR:   31408381
root@wonky:~# ifconfig eth0
eth0      Link encap:Ethernet  HWaddr 4E:54:55:25:57:68
          inet6 addr: fe80::4c54:55ff:fe25:5768/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1504  Metric:1
          RX packets:2675228781 errors:0 dropped:166 overruns:0 frame:0
          TX packets:5202119236 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:334482353348 (311.5 GiB)  TX bytes:7008190302937 (6.3 TiB)
          Interrupt:21
```

rany2, 2023-06-20:
@Spudz76 I had this issue with jumbo frames enabled and disabled, that mt7530 router in particular needs jumbo frames for WireGuard/VXLAN interfaces (so it ends up with the standard MTU of 1500 not 1450). I think this misconception that mt7530 has issues with jumbo frames occurred because the datasheet provided invalid information about jumbo frames and mt7530, they claimed it supported 9k when it actually supported 2048 MAX.

P.S. the reason your eth0 has a weird MTU of 1504 is because it needs an extra 4 bytes for DSA tag but the ports themselves would have the "actual" MTU.

rany2, 2023-06-20:
> I wonder if issue disappears by setting MTU:1500 or 1492 or less, or gets worse due to fragging. Still, 2026 seems wrong, unless things know how to do jumbos, and then why not 9000.

It does work fine with 2030 as MTU, that's not the issue. 

> Probably the biggest clue is the interrupt ERR count, something basic system level driver must have changed. I note the three mt7530-0: devices are only in the snapshot output. What are the diffs [...]

OpenWRT 22.03 had a massive refactor for mt7621's drivers and used a lot of the now upstreamed drivers. So this is most likely the case, however I have no idea which component might cause this issue. Perhaps @paraka could help as help and provide suggestions? He was responsible for most of the entire effort on that front

paraka, 2023-06-20:
I don't really know but @arinc9 is the expert in mt7530 so maybe he could help.

rany2, 2023-06-20:
So I guess you think increase in errors in /proc/interrupts is related to mt7530 and not the platform drivers?

paraka, 2023-06-20:
> So I guess you think increase in errors in /proc/interrupts is related to mt7530 and not the platform drivers?

I only meant that I have never  touched by myself nothing related with ethernet or mt7530 for these SoCs and @arinc9 is normally doing that. So I was only pinging him for help :) Interrupts for mt7530 were enabled recently also and there was not support before.

brada4, 2023-06-20:
ERR seems proportionate to uptime (i.e. timer) , probably some unhandled semi-hidden device on SoC. comparable to x86 SMI or so. Can you check if it grows on idle device (ssh connection allowed) same as on busy one, like for 1h

Spudz76, 2023-06-20:
On a very very idle copy of my main router.  This unit just sits around in between test flashes.  It is running a different build, 5.15 kernel vs 5.10 and some other hacks

But ERR is zero on this one.  Either 5.15 fixes something (snapshot should be 5.15 anyway by now?) or it needs traffic.

Also of note there is one more mt7530 interrupt compared to the OP device.  Guess the Linksys EA7300v1 has one more (physical) port vs the ASUS RT-AX53U?

MTU stuff makes sense I guess, I also have no knowledge of wireguard.  And am still of the old best practice of avoiding jumbos at all costs because everything never supports them right. :)
```
root@wonky2:~# uptime
 11:07:29 up 113 days, 18:00,  load average: 0.00, 0.00, 0.00
root@wonky2:~# cat /proc/interrupts
           CPU0       CPU1       CPU2       CPU3
  8:  982805130  982805139  982805130  982805123  MIPS GIC Local   1  timer
  9:  464644681          0          0          0  MIPS GIC  63  IPI call
 10:          0 1541131021          0          0  MIPS GIC  64  IPI call
 11:          0          0 1041173388          0  MIPS GIC  65  IPI call
 12:          0          0          0  961117599  MIPS GIC  66  IPI call
 13:      41601          0          0          0  MIPS GIC  67  IPI resched
 14:          0      42745          0          0  MIPS GIC  68  IPI resched
 15:          0          0      42075          0  MIPS GIC  69  IPI resched
 16:          0          0          0      51776  MIPS GIC  70  IPI resched
 17:          0          0          0          0  MIPS GIC  19  1e000600.gpio-bank0, 1e000600.gpio-bank1, 1e000600.gpio-bank2
 19:        114          0          0          0  MIPS GIC  33  ttyS0
 20:          0          0          0          0  MIPS GIC  29  xhci-hcd:usb1
 21:      11866          0          0          0  MIPS GIC  26  1e004000.crypto
 22:   34449992          0          0          0  MIPS GIC  10  1e100000.ethernet
 23:          5          0          0          0  MIPS GIC  30  mt7530
 24:          5          0          0          0    mt7530   1  mt7530-0:01
 25:          0          0          0          0    mt7530   2  mt7530-0:02
 26:          0          0          0          0    mt7530   3  mt7530-0:03
 27:          0          0          0          0    mt7530   4  mt7530-0:04
 28:          0          0          0          0  1e000600.gpio  17  keys
 29:          0          0          0          0  1e000600.gpio  12  keys
 30:         18          0          0          0  MIPS GIC  11  mt7615e
 31:         18          0          0          0  MIPS GIC  31  mt7615e
ERR:          0
```

brada4, 2023-06-20:
That is within variety of SoC configuration at production line.

arinc9, 2023-06-21:
> I just noticed that https://github.com/openwrt/openwrt/blob/main/target/linux/generic/pending-6.1/723-net-mt7531-ensure-all-MACs-are-powered-down-before-r.patch is wrong. It should be PMCR_FORCE_LNK and that change impacts mt7530_setup not mt7531_setup so patch name is wrong as well.

I'm a maintainer of the MT7530 DSA subdriver. Everything said above is utterly false. The PMCR_FORCE_LNK bit only exists on the MT7530 switch. The bit that has got the same function on the MT7531 switch is 31 and is called MT7531_FORCE_LNK. The SYS_CTRL_PHY_RST bit on the 0x7000 register doesn't exist on the MT7531 switch. Because of this, it's pointed out on the MT7531 Reference Manual for Development Board rev 1.0 to set all the MACs down using the MT7531_FORCE_LNK bit before setting SYS_CTRL_SW_RST and SYS_CTRL_REG_RST, hence the reason for the patch linked above. 

If there's anything remotely wrong here, it's the unnecessary setting of the SYS_CTRL_PHY_RST bit for the MT7531 switch.

I suggest you read a bit more before jumping to conclusions.

I suspect the regression reported here is caused by the changes on the MediaTek ethernet driver. I don't see a reason to investigate it as these SoCs were not intended to do IP packet forwarding at CPU let alone dealing with the iperf3 load, be it on the generating or the receiving side. The packet processing engine (PPE, marketed as Network Accelerator on the MediaTek website) is supposed to deal with forwarding packets.

Enable hardware offloading then try a download or upload test between two switch ports, each on separate networks. You should consistently get 933 Mbps.

If you really want to find the commit that causes the regression, the bisection should be done on the Linux repository, not OpenWrt. It should take about 10 tests to pinpoint the kernel minor version that introduces the regression.

You can either somehow make OpenWrt compile the Linux version of your choice, or just boot vanilla Linux with Buildroot as the filesystem on mainline U-Boot that can boot this without any issues.

I will try the latter on my MT7621 board, see if I also come across this regression.

rany2, 2023-06-21:
> Enable hardware offloading, which lets the PPE deal with forwarding packets, then try a download or upload test between two switch ports, each on separate networks. You should consistently get 933 Mbps.

Does this mean that I need to have nft-offload installed?

arinc9, 2023-06-21:
kmod-nf-flow which I believe compiles https://github.com/torvalds/linux/blob/master/net/netfilter/nf_flow_table_offload.c with dependencies.

rany2, 2023-06-21:
> I'm a maintainer of the MT7530 DSA subdriver. Everything said above is utterly false. The PMCR_FORCE_LNK bit only exists on the MT7530 switch.

I've noticed but never sent a correction. I looked at the patch and thought it changed mt7530_setup. My bad about that remark.

rany2, 2023-06-21:
> ~kmod-nf-flow~ kmod-nft-offload which I believe compiles https://github.com/torvalds/linux/blob/master/net/netfilter/nf_flow_table_offload.c with dependencies.
> 
> You can either do it using firewall4 which would mean enabling the hardware flow offloading option on the uci configuration. Or do it using nftables as documented here:
> 
> https://www.kernel.org/doc/html/latest/networking/nf_flowtable.html#hardware-offload

I'll try with this, I presume without a firewall all forwarding is CPU bound?

arinc9, 2023-06-21:
Without:
- A program to manipulate the netfilter framework (nftables),
- Netfilter nf_tables hardware flow offload module (NFT_FLOW_OFFLOAD, called the kmod-nft-offload package on OpenWrt),
- The 'offload' flag in your flowtable definition,
all forwarding is CPU bound.

rany2, 2023-06-21:
No luck with enabling hardware offload per https://www.kernel.org/doc/html/latest/networking/nf_flowtable.html#hardware-offload, I guess it has to be something else.

rany2, 2023-06-21:
@arinc9 maybe this information will help you but I dropped all mtk_eth_soc and mt7530 DSA patches and the issue remains. So if this issue is in the Ethernet/DSA driver I guess it had to be in mainline 5.15 as well..

arinc9, 2023-06-22:
I've done tests using iperf3 on my Unielec U7621-06. I've booted the initramfs images of 22.03.5 and 21.02.7.

Network 1:
router: 192.168.1.1/24
iperf3 server: 192.168.1.2/24

Network 2:
router: 192.168.2.1/24
iperf3 client: 192.168.2.2/24

Masquerading is being done on the network the iperf3 server is on, 192.168.1.0/24.

I enable hardware flow offloading at some point, I will leave it to you to figure out when, by looking at the iperf3 log below.

<details><summary>OpenWrt 22.03.5</summary>
<p>

```
$ iperf3 -c 192.168.1.2 -t 0 -R
Connecting to host 192.168.1.2, port 5201
Reverse mode, remote host 192.168.1.2 is sending
[  5] local 192.168.2.2 port 60118 connected to 192.168.1.2 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  54.2 MBytes   455 Mbits/sec                  
[  5]   1.00-2.00   sec  54.6 MBytes   458 Mbits/sec                  
[  5]   2.00-3.00   sec  44.3 MBytes   372 Mbits/sec                  
[  5]   3.00-4.00   sec  59.8 MBytes   501 Mbits/sec                  
[  5]   4.00-5.00   sec  58.0 MBytes   487 Mbits/sec                  
[  5]   5.00-6.00   sec  60.7 MBytes   509 Mbits/sec                  
[  5]   6.00-7.00   sec  68.3 MBytes   573 Mbits/sec                  
[  5]   7.00-8.00   sec  69.2 MBytes   580 Mbits/sec                  
[  5]   8.00-9.00   sec  69.7 MBytes   585 Mbits/sec                  
[  5]   9.00-10.00  sec  53.6 MBytes   450 Mbits/sec                  
[  5]  10.00-11.00  sec  54.3 MBytes   455 Mbits/sec                  
[  5]  11.00-12.00  sec  69.8 MBytes   585 Mbits/sec                  
[  5]  12.00-13.00  sec  68.5 MBytes   575 Mbits/sec                  
[  5]  13.00-14.00  sec  70.6 MBytes   592 Mbits/sec                  
[  5]  14.00-15.00  sec  70.5 MBytes   592 Mbits/sec                  
[  5]  15.00-16.00  sec  67.8 MBytes   569 Mbits/sec                  
[  5]  16.00-17.00  sec  46.9 MBytes   393 Mbits/sec                  
[  5]  17.00-18.00  sec  56.4 MBytes   473 Mbits/sec                  
[  5]  18.00-19.00  sec  50.5 MBytes   423 Mbits/sec                  
[  5]  19.00-20.00  sec  48.4 MBytes   406 Mbits/sec                  
[  5]  20.00-21.00  sec  50.3 MBytes   422 Mbits/sec                  
[  5]  21.00-22.00  sec  45.9 MBytes   385 Mbits/sec                  
[  5]  22.00-23.00  sec  50.8 MBytes   426 Mbits/sec                  
[  5]  23.00-24.00  sec  44.2 MBytes   371 Mbits/sec                  
[  5]  24.00-25.00  sec  49.8 MBytes   418 Mbits/sec                  
[  5]  25.00-26.00  sec  53.8 MBytes   451 Mbits/sec                  
[  5]  26.00-27.00  sec  47.6 MBytes   399 Mbits/sec                  
[  5]  27.00-28.00  sec  51.6 MBytes   433 Mbits/sec                  
[  5]  28.00-29.00  sec  43.8 MBytes   367 Mbits/sec                  
[  5]  29.00-30.00  sec   111 MBytes   931 Mbits/sec                  
[  5]  30.00-31.00  sec   112 MBytes   936 Mbits/sec                  
[  5]  31.00-32.00  sec   112 MBytes   936 Mbits/sec                  
[  5]  32.00-33.00  sec   112 MBytes   936 Mbits/sec                  
[  5]  33.00-34.00  sec   112 MBytes   936 Mbits/sec                  
[  5]  34.00-35.00  sec   112 MBytes   936 Mbits/sec                  
[  5]  35.00-36.00  sec   112 MBytes   936 Mbits/sec                  
[  5]  36.00-37.00  sec   112 MBytes   936 Mbits/sec                  
[  5]  37.00-38.00  sec   112 MBytes   936 Mbits/sec                  
[  5]  38.00-39.00  sec   112 MBytes   937 Mbits/sec                  
[  5]  39.00-40.00  sec   112 MBytes   936 Mbits/sec                  
[  5]  40.00-41.00  sec   112 MBytes   936 Mbits/sec                  
[  5]  41.00-42.00  sec   112 MBytes   936 Mbits/sec                  
[  5]  42.00-43.00  sec   112 MBytes   937 Mbits/sec                  
[  5]  43.00-44.00  sec   112 MBytes   937 Mbits/sec                  
[  5]  44.00-45.00  sec   112 MBytes   937 Mbits/sec                  
[  5]  45.00-46.00  sec   112 MBytes   937 Mbits/sec                  
[  5]  46.00-47.00  sec   112 MBytes   937 Mbits/sec                  
[... 81 lines cut ...]
[  5]  50.00-51.00  sec   112 MBytes   936 Mbits/sec                  
[  5]  51.00-52.00  sec   112 MBytes   936 Mbits/sec                  
[  5]  52.00-53.00  sec   112 MBytes   936 Mbits/sec                  
[  5]  53.00-54.00  sec   112 MBytes   936 Mbits/sec                  
[  5]  54.00-55.00  sec   112 MBytes   936 Mbits/sec
```

</p>
</details> 

I do see 21.02.7 performs better than 22.03.5 without hardware flow offloading. However, I don't acknowledge this as a valid issue. It's likely the changes on the MediaTek ethernet driver causes the performance degradation without hardware flow offloading. This does not necessarily mean it is an issue as I stated before, these SoCs were not intended to do IP packet forwarding at CPU. Hardware flow offloading performs the same on both versions, as it should. Use hardware flow offloading if you want to get full performance out of this device.

I enabled hardware flow offloading the conventional way for OpenWrt:

`/etc/config/firewall`
```
config defaults
        [...]
        option flow_offloading '1'
        option flow_offloading_hw '1'
```

I suspect the changes that cause this behaviour is a new feature for the newer SoCs to benefit from. Feel free to bisect the commits:

https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net-next.git/log/drivers/net/ethernet/mediatek

rany2, 2023-06-22:
@arinc9 No luck with enabling it via OpenWRT's config:

```
root@asus:~# nft list ruleset
table inet fw4 {
	flowtable ft {
		hook ingress priority filter
		devices = { lan1, lan2, lan3, wan }
		flags offload
		counter
	}

	chain input {
		type filter hook input priority filter; policy drop;
		iifname "lo" accept comment "!fw4: Accept traffic from loopback"
		ct state established,related accept comment "!fw4: Allow inbound established and related flows"
		tcp flags syn / fin,syn,rst,ack jump syn_flood comment "!fw4: Rate limit TCP syn packets"
		iifname "br-lan" jump input_lan comment "!fw4: Handle lan IPv4/IPv6 input traffic"
		jump handle_reject
	}

	chain forward {
		type filter hook forward priority filter; policy drop;
		meta l4proto { tcp, udp } flow add @ft
		ct state established,related accept comment "!fw4: Allow forwarded established and related flows"
		iifname "br-lan" jump forward_lan comment "!fw4: Handle lan IPv4/IPv6 forward traffic"
		jump handle_reject
	}
        ....

root@asus:~# cat /etc/config/firewall |head -40

config defaults
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option synflood_protect '1'
	option flow_offloading '1'
	option flow_offloading_hw '1'

config zone
	option name 'lan'
	list network 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'

config zone
	option name 'wan'
	list network 'wan'
	list network 'wan6'
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option masq '1'
	option mtu_fix '1'
```

The results are the same and the performance is just as inconsistent as they were in the initial post. Unfortunately I really don't know how to do a bisect on the kernel with OpenWRT (I don't think it's even possible because of the reliance of many device specific patches to get that thing booting).

Something to note is that my results are somehow much better than your results on OpenWRT 21.02. Without hardware offloading/any nft kernel modules I am maxing out at 936+Mbit/s and the CPU usage isn't that bad (htop says 30% usage on each CPU core).

# OpenWRT 21.02 (without any nft modules installed)

```
root@gw:~# iperf3 -c asus -R
Connecting to host asus, port 5201
Reverse mode, remote host asus is sending
[  5] local 10.146.80.1 port 57772 connected to 10.146.109.9 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   113 MBytes   947 Mbits/sec                  
[  5]   1.00-2.00   sec   106 MBytes   886 Mbits/sec                  
[  5]   2.00-3.00   sec   112 MBytes   939 Mbits/sec                  
[  5]   3.00-4.00   sec   110 MBytes   923 Mbits/sec                  
[  5]   4.00-5.00   sec   112 MBytes   940 Mbits/sec                  
[  5]   5.00-6.00   sec   113 MBytes   951 Mbits/sec                  
[  5]   6.00-7.00   sec   112 MBytes   940 Mbits/sec                  
[  5]   7.00-8.00   sec   113 MBytes   944 Mbits/sec                  
[  5]   8.00-9.00   sec   113 MBytes   952 Mbits/sec                  
[  5]   9.00-10.00  sec   113 MBytes   947 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.09 GBytes   937 Mbits/sec    0             sender
[  5]   0.00-10.00  sec  1.09 GBytes   937 Mbits/sec                  receiver

iperf Done.
```

Similar to your setup they are on different networks, meanwhile these are the results for OpenWRT SNAPSHOT with offloading enabled (no difference between IPv4 and IPv6, equally bad results):

# OpenWRT SNAPSHOT with nft-offload setup

```
root@gw:~# iperf3 -c dd1c:2285:baee:ffff::2
Connecting to host dd1c:2285:baee:ffff::2, port 5201
[  5] local dd1c:2285:baee::1 port 56034 connected to dd1c:2285:baee:ffff::2 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  29.4 MBytes   246 Mbits/sec    0    432 KBytes       
[  5]   1.00-2.00   sec  27.8 MBytes   233 Mbits/sec    0    360 KBytes       
[  5]   2.00-3.00   sec  27.5 MBytes   231 Mbits/sec    0    392 KBytes       
[  5]   3.00-4.00   sec  26.5 MBytes   222 Mbits/sec    0    385 KBytes       
[  5]   4.00-5.00   sec  27.0 MBytes   227 Mbits/sec    0    538 KBytes       
[  5]   5.00-6.00   sec  27.5 MBytes   231 Mbits/sec    0    414 KBytes       
[  5]   6.00-7.00   sec  27.5 MBytes   231 Mbits/sec    0    483 KBytes       
[  5]   7.00-8.00   sec  25.0 MBytes   210 Mbits/sec    0    352 KBytes       
[  5]   8.00-9.00   sec  26.2 MBytes   220 Mbits/sec    0    392 KBytes       
[  5]   9.00-10.00  sec  25.0 MBytes   210 Mbits/sec    0   7.27 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   269 MBytes   226 Mbits/sec    0             sender
[  5]   0.00-10.04  sec   266 MBytes   222 Mbits/sec                  receiver

iperf Done.
```

rany2, 2023-06-22:
Also note that this is very inconsistent on snapshot and the only factor that might change the result is a reboot, about 1 in 20 times I end up with decent performance even exceeding OpenWRT 21.02; but the thing is it's inconsistent and it degrades with time; so eventually it ends up running poorly.

Unfortunately I don't know what factors change to get it working properly besides just rebooting until I get something decent. It's all really odd...

rany2, 2023-06-22:
I just tried to enable hardware flow offloading in OpenWRT 21.02 and somehow the performance is just as bad as snapshot. Perhaps the performance regression is caused by some change in MTK-PPE?

arinc9, 2023-06-22:
I don't think anything's wrong with mtk_ppe_offload regarding this as I've already shown it works perfectly fine on my MT7621AT device. One thing to add is the official image from the OpenWrt firmware selector is enough by itself to do hardware flow offloading, no extra kmod package or anything else is needed.

I think the real issue here is that, for some reason, your device won't perform better with hardware flow offloading. I can see the rules for hardware flow offloading are properly set on the ruleset. Let's try a simpler ruleset anyway.

```
flush ruleset

table ip global {
	flowtable f {
		hook ingress priority 0
		devices = { wan, lan1, lan2, lan3 }
		flags offload
	}

	chain forward {
		type filter hook forward priority 0
		ip protocol { tcp, udp } flow offload @f
	}

	chain postrouting {
		type nat hook postrouting priority 0
		oifname "wan" masquerade
	}
}
```

With this, I was able to get full speed between wan and lan3 interfaces, on OpenWrt 22.03.5 and vanilla Linux next-20230622 with Buildroot. wan being the iperf3 server, lan3 being the iperf3 client.

[1 more reply]