openwrt/openwrt #12918: log page_pool_release_retry() stalled pool shutdown 14 inflight 23102 sec

ptpt52, 2023-06-15:
### Describe the bug

run on mt7986/mt7981
what may cause this log?
```
[23909.149146] page_pool_release_retry() stalled pool shutdown 14 inflight 23102 sec
[23969.453067] page_pool_release_retry() stalled pool shutdown 14 inflight 23162 sec
[24029.756999] page_pool_release_retry() stalled pool shutdown 14 inflight 23223 sec
[24090.060968] page_pool_release_retry() stalled pool shutdown 14 inflight 23283 sec
[24150.364950] page_pool_release_retry() stalled pool shutdown 14 inflight 23343 sec
[24210.668954] page_pool_release_retry() stalled pool shutdown 14 inflight 23404 sec
[24270.972964] page_pool_release_retry() stalled pool shutdown 14 inflight 23464 sec
[24331.277032] page_pool_release_retry() stalled pool shutdown 14 inflight 23524 sec
[24391.581046] page_pool_release_retry() stalled pool shutdown 14 inflight 23585 sec
[24451.885137] page_pool_release_retry() stalled pool shutdown 14 inflight 23645 sec
[24512.189227] page_pool_release_retry() stalled pool shutdown 14 inflight 23705 sec
[24572.493322] page_pool_release_retry() stalled pool shutdown 14 inflight 23766 sec
[24632.797437] page_pool_release_retry() stalled pool shutdown 14 inflight 23826 sec
[24693.101583] page_pool_release_retry() stalled pool shutdown 14 inflight 23886 sec
[24753.405749] page_pool_release_retry() stalled pool shutdown 14 inflight 23947 sec
[24813.709918] page_pool_release_retry() stalled pool shutdown 14 inflight 24007 sec
[24874.014103] page_pool_release_retry() stalled pool shutdown 14 inflight 24067 sec
[24934.318468] page_pool_release_retry() stalled pool shutdown 14 inflight 24128 sec
[24994.622533] page_pool_release_retry() stalled pool shutdown 14 inflight 24188 sec
[25054.926785] page_pool_release_retry() stalled pool shutdown 14 inflight 24248 sec
[25115.231022] page_pool_release_retry() stalled pool shutdown 14 inflight 24308 sec
[25175.535289] page_pool_release_retry() stalled pool shutdown 14 inflight 24369 sec
[25235.839571] page_pool_release_retry() stalled pool shutdown 14 inflight 24429 sec
[25296.143865] page_pool_release_retry() stalled pool shutdown 14 inflight 24489 sec
[25356.448166] page_pool_release_retry() stalled pool shutdown 14 inflight 24550 sec
[25416.752498] page_pool_release_retry() stalled pool shutdown 14 inflight 24610 sec
[25477.066829] page_pool_release_retry() stalled pool shutdown 14 inflight 24670 sec
[25537.361194] page_pool_release_retry() stalled pool shutdown 14 inflight 24731 sec
[25597.665552] page_pool_release_retry() stalled pool shutdown 14 inflight 24791 sec
[25657.969942] page_pool_release_retry() stalled pool shutdown 14 inflight 24851 sec
[25718.274327] page_pool_release_retry() stalled pool shutdown 14 inflight 24912 sec
[25778.578724] page_pool_release_retry() stalled pool shutdown 14 inflight 24972 sec
[25838.882719] page_pool_release_retry() stalled pool shutdown 14 inflight 25032 sec
[25899.186405] page_pool_release_retry() stalled pool shutdown 14 inflight 25093 sec
[25959.490048] page_pool_release_retry() stalled pool shutdown 14 inflight 25153 sec
[26019.793740] page_pool_release_retry() stalled pool shutdown 14 inflight 25213 sec
[26080.097459] page_pool_release_retry() stalled pool shutdown 14 inflight 25274 sec
[26140.401174] page_pool_release_retry() stalled pool shutdown 14 inflight 25334 sec
[26200.704901] page_pool_release_retry() stalled pool shutdown 14 inflight 25394 sec
[26261.008656] page_pool_release_retry() stalled pool shutdown 14 inflight 25455 sec
[26321.312416] page_pool_release_retry() stalled pool shutdown 14 inflight 25515 sec
[26381.616194] page_pool_release_retry() stalled pool shutdown 14 inflight 25575 sec
[26441.919983] page_pool_release_retry() stalled pool shutdown 14 inflight 25636 sec
[26502.223815] page_pool_release_retry() stalled pool shutdown 14 inflight 25696 sec
[26562.527593] page_pool_release_retry() stalled pool shutdown 14 inflight 25756 sec
[26622.831428] page_pool_release_retry() stalled pool shutdown 14 inflight 25816 sec
[26683.135274] page_pool_release_retry() stalled pool shutdown 14 inflight 25877 sec
[26743.439127] page_pool_release_retry() stalled pool shutdown 14 inflight 25937 sec
[26803.743003] page_pool_release_retry() stalled pool shutdown 14 inflight 25997 sec
[26864.046892] page_pool_release_retry() stalled pool shutdown 14 inflight 26058 sec
[26924.350764] page_pool_release_retry() stalled pool shutdown 14 inflight 26118 sec
[26984.654685] page_pool_release_retry() stalled pool shutdown 14 inflight 26178 sec
[27044.958594] page_pool_release_retry() stalled pool shutdown 14 inflight 26239 sec
[27105.262529] page_pool_release_retry() stalled pool shutdown 14 inflight 26299 sec
[27165.566458] page_pool_release_retry() stalled pool shutdown 14 inflight 26359 sec
[27225.870417] page_pool_release_retry() stalled pool shutdown 14 inflight 26420 sec
```

### OpenWrt version

xx

### OpenWrt target/subtarget

xx

### Device

xx

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-06-15:
https://github.com/openwrt/openwrt/issues/11318 please check if ethtool disabling rx vlan offload helps. (or bisect through offloads if not)

mrkiko, 2023-06-17:
If you'd like, you might be more specific about your workload and specific device you'reusing (if this applies).

ptpt52, 2023-06-18:
maybe the issue is fixed with latest code. close for now.

closed 2023-06-18 by ptpt52

mundschenk-at, 2024-01-27:
I am still seeing this on TUF-AX6000 with `OpenWrt SNAPSHOT r24870-ced3fbcda1` (custom build).

systemcrash, 2024-11-21:
Just got this on my device. Also mediatek. 

```json
	"model": {
		"id": "glinet,gl-mt6000",
		"name": "GL.iNet GL-MT6000"
	},
```

```
Tue Nov 19 20:26:36 2024 kern.warn kernel: [1228581.247516] page_pool_release_retry() stalled pool shutdown 6 inflight 505602 sec
Tue Nov 19 20:27:36 2024 kern.warn kernel: [1228641.502612] page_pool_release_retry() stalled pool shutdown 6 inflight 505662 sec
Tue Nov 19 20:28:36 2024 kern.warn kernel: [1228701.767975] page_pool_release_retry() stalled pool shutdown 6 inflight 505722 sec
Tue Nov 19 20:29:37 2024 kern.warn kernel: [1228762.013573] page_pool_release_retry() stalled pool shutdown 6 inflight 505783 sec
Tue Nov 19 20:30:37 2024 kern.warn kernel: [1228822.269191] page_pool_release_retry() stalled pool shutdown 6 inflight 505843 sec
Tue Nov 19 20:31:37 2024 kern.warn kernel: [1228882.525118] page_pool_release_retry() stalled pool shutdown 6 inflight 505903 sec
Tue Nov 19 20:32:38 2024 kern.warn kernel: [1228942.781118] page_pool_release_retry() stalled pool shutdown 6 inflight 505964 sec
Tue Nov 19 20:33:38 2024 kern.warn kernel: [1229003.036465] page_pool_release_retry() stalled pool shutdown 6 inflight 506024 sec
Tue Nov 19 20:34:38 2024 kern.warn kernel: [1229063.291561] page_pool_release_retry() stalled pool shutdown 6 inflight 506084 sec
Tue Nov 19 20:35:39 2024 kern.warn kernel: [1229123.546962] page_pool_release_retry() stalled pool shutdown 6 inflight 506145 sec
Tue Nov 19 20:36:39 2024 kern.warn kernel: [1229183.812437] page_pool_release_retry() stalled pool shutdown 6 inflight 506205 sec
Tue Nov 19 20:37:39 2024 kern.warn kernel: [1229244.068051] page_pool_release_retry() stalled pool shutdown 6 inflight 506265 sec
Tue Nov 19 20:38:40 2024 kern.warn kernel: [1229304.313873] page_pool_release_retry() stalled pool shutdown 6 inflight 506326 sec
Tue Nov 19 20:39:40 2024 kern.warn kernel: [1229364.569821] page_pool_release_retry() stalled pool shutdown 6 inflight 506386 sec
Tue Nov 19 20:40:40 2024 kern.warn kernel: [1229424.835541] page_pool_release_retry() stalled pool shutdown 6 inflight 506446 sec
Tue Nov 19 20:41:41 2024 kern.warn kernel: [1229485.080662] page_pool_release_retry() stalled pool shutdown 6 inflight 506507 sec
Tue Nov 19 20:42:41 2024 kern.warn kernel: [1229545.335917] page_pool_release_retry() stalled pool shutdown 6 inflight 506567 sec
Tue Nov 19 20:43:41 2024 kern.warn kernel: [1229605.591350] page_pool_release_retry() stalled pool shutdown 6 inflight 506627 sec
Tue Nov 19 20:44:42 2024 kern.warn kernel: [1229665.846904] page_pool_release_retry() stalled pool shutdown 6 inflight 506687 sec
Tue Nov 19 20:45:42 2024 kern.warn kernel: [1229726.102744] page_pool_release_retry() stalled pool shutdown 6 inflight 506748 sec
Tue Nov 19 20:46:42 2024 kern.warn kernel: [1229786.368945] page_pool_release_retry() stalled pool shutdown 6 inflight 506808 sec
Tue Nov 19 20:47:42 2024 kern.warn kernel: [1229846.614898] page_pool_release_retry() stalled pool shutdown 6 inflight 506868 sec
Tue Nov 19 20:48:43 2024 kern.warn kernel: [1229906.879946] page_pool_release_retry() stalled pool shutdown 6 inflight 506929 sec
Tue Nov 19 20:49:43 2024 kern.warn kernel: [1229967.125242] page_pool_release_retry() stalled pool shutdown 6 inflight 506989 sec
Tue Nov 19 20:50:43 2024 kern.warn kernel: [1230027.380570] page_pool_release_retry() stalled pool shutdown 6 inflight 507049 sec
Tue Nov 19 20:51:44 2024 kern.warn kernel: [1230087.636189] page_pool_release_retry() stalled pool shutdown 6 inflight 507110 sec
Tue Nov 19 20:52:44 2024 kern.warn kernel: [1230147.901955] page_pool_release_retry() stalled pool shutdown 6 inflight 507170 sec
Tue Nov 19 20:53:44 2024 kern.warn kernel: [1230208.147750] page_pool_release_retry() stalled pool shutdown 6 inflight 507230 sec
Tue Nov 19 20:54:45 2024 kern.warn kernel: [1230268.403974] page_pool_release_retry() stalled pool shutdown 6 inflight 507291 sec
Tue Nov 19 20:55:45 2024 kern.warn kernel: [1230328.659055] page_pool_release_retry() stalled pool shutdown 6 inflight 507351 sec
Tue Nov 19 20:56:45 2024 kern.warn kernel: [1230388.914158] page_pool_release_retry() stalled pool shutdown 6 inflight 507411 sec
Tue Nov 19 20:57:46 2024 kern.warn kernel: [1230449.169590] page_pool_release_retry() stalled pool shutdown 6 inflight 507472 sec
Tue Nov 19 20:58:46 2024 kern.warn kernel: [1230509.425183] page_pool_release_retry() stalled pool shutdown 6 inflight 507532 sec
Tue Nov 19 20:59:46 2024 kern.warn kernel: [1230569.681024] page_pool_release_retry() stalled pool shutdown 6 inflight 507592 sec
Tue Nov 19 21:00:47 2024 kern.warn kernel: [1230629.937123] page_pool_release_retry() stalled pool shutdown 6 inflight 507653 sec
Tue Nov 19 21:01:47 2024 kern.warn kernel: [1230690.202853] page_pool_release_retry() stalled pool shutdown 6 inflight 507713 sec
Tue Nov 19 21:02:47 2024 kern.warn kernel: [1230750.447930] page_pool_release_retry() stalled pool shutdown 6 inflight 507773 sec
Tue Nov 19 21:03:48 2024 kern.warn kernel: [1230810.703172] page_pool_release_retry() stalled pool shutdown 6 inflight 507834 sec
Tue Nov 19 21:04:48 2024 kern.warn kernel: [1230870.958618] page_pool_release_retry() stalled pool shutdown 6 inflight 507894 sec
```

Can't honestly say what aggravating factor might be.