openwrt/openwrt #12921: dnsmasq fails to start on an interface with a /31 subnet, with iprange start 1 limit 1

john-tho, 2023-06-16:
### Describe the bug

```
logread -e dnsmasq | tail
Fri Jun 16 14:54:01 2023 daemon.crit dnsmasq[1]: inconsistent DHCP range at line 43 of /var/etc/dnsmasq.conf.cfg01411c
Fri Jun 16 14:54:01 2023 daemon.crit dnsmasq[1]: FAILED to start up
Fri Jun 16 14:54:01 2023 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 7 crashes, 0 seconds since last crash

tail -n+41 /var/etc/dnsmasq.conf.cfg01411c | head -n3
dhcp-range=set:wifi,10.53.20.100,10.53.20.249,255.255.255.0,1h
dhcp-option=wifi,option:dns-server,0.0.0.0
dhcp-range=set:link_house,10.53.20.100,10.53.20.249,255.255.255.254,1h
# uses the $START,$END from previous dhcp config (wrong for the interface), which breaks dnsmasq config

uci show network.link_house
network.link_house=interface
network.link_house.device='br0.link_house'
network.link_house.proto='static'
network.link_house.ip6assign='60'
network.link_house.ip6hint='60'
network.link_house.ipaddr='10.53.100.0/31'

uci show dhcp.link_house
dhcp.link_house=dhcp
dhcp.link_house.interface='link_house'
dhcp.link_house.dhcpv4='server'
dhcp.link_house.dhcpv6='server'
dhcp.link_house.ra='server'
dhcp.link_house.ra_slaac='1'
dhcp.link_house.ra_flags='managed-config' 'other-config'
dhcp.link_house.dhcp_option='option:dns-server,0.0.0.0'
dhcp.link_house.start='1'
dhcp.link_house.limit='1'
dhcp.link_house.leasetime='1h'

```

23.05.0-rc1
```
ipcalc.sh 10.53.100.0 31 1 1
IP=10.53.100.0
NETMASK=255.255.255.254
BROADCAST=10.53.100.1
NETWORK=10.53.100.0
PREFIX=31
network (10.53.100.0/31) too small
```
Because this program no longer prints START and END for a /31, those from a previous interface are used, which don't match the network for the current interface.

suspect commit: https://github.com/openwrt/openwrt/commit/a40a96e54b0d047d14d99bb715ee2dd94458acb6 @yogo1212 @Ansuel 
https://github.com/openwrt/openwrt/blob/b57703264f0eba086f22894c5c2ef996e135a993/package/network/services/dnsmasq/files/dnsmasq.init#L590
I _think_ this bad range is inserted in the config because ipcalc is run is a subshell:
```
~ $ eval "$(false)"
~ $ echo $?
0
~ $ eval "false"
~ $ echo $?
1
```
https://github.com/openwrt/openwrt/blob/b57703264f0eba086f22894c5c2ef996e135a993/package/network/services/dnsmasq/files/dnsmasq.init#L593

---

22.03.4
```
ipcalc.sh 10.53.100.0 255.255.255.254 1 1
IP=10.53.100.0
NETMASK=255.255.255.254
BROADCAST=10.53.100.1
NETWORK=10.53.100.0
PREFIX=31
START=10.53.100.1
END=10.53.100.0
```

this would result in the dnsmasq.conf line (which ran fine):

```
dhcp-range=set:test,10.53.100.1,10.53.100.0,255.255.255.254,1h
```

### OpenWrt version

23.05.0-rc1

### OpenWrt target/subtarget

mt7621

### Device

760igs

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-06-16:
First and last address in broadcast network are not usable. Use 30 or bigger with static addressing since dhcpd end is not guaranteed to come up first always.

yogo1212, 2023-06-16:
> First and last address in broadcast network are not usable. Use 30 or bigger with static addressing since dhcpd end is not guaranteed to come up first always.

`/31` was altered subsequently. Both addresses are hosts - which isn't reflected in the patch.
I was sure I had done that. Looks like I didn't :see_no_evil: 
Let me fix that.

> Even if ipcalc rejects the inputs, the if statement still allows the `dhcp-range` statement to be added because the `eval` always returns true.

Shame on me!

Both should be addressed in https://github.com/openwrt/openwrt/pull/12925.

closed 2023-09-25 by openwrt-bot

john-tho, 2024-05-08:
In upgrading a device to 23.05.3, I came across this issue again, a /31 network crashing openwrt starting dnsmasq.

In the snapshot/main branch, it looks like the dnsmasq crash loop would be addressed by https://github.com/openwrt/openwrt/commit/6b23836071b171499f3983286a37a8d5c15eeab9, but this avoid `eval $(subshell)` functionality did not get backported to 23.05. Is this suitable to backport to 23.05 @yogo1212?

https://github.com/openwrt/openwrt/blob/ec8f647d168fa8f3b1eedd9b5fe665f793f3a659/package/network/services/dnsmasq/files/dnsmasq.init#L590

```
Example using network 10.53.100.0/31 for link_house

root@shed:~# logread -e dhcp
Wed May  8 11:28:49 2024 daemon.crit dnsmasq[1]: inconsistent DHCP range at line 43 of /var/etc/dnsmasq.conf.cfg01411c
Wed May  8 11:28:49 2024 daemon.crit dnsmasq[1]: FAILED to start up
Wed May  8 11:28:49 2024 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash

root@shed:~# tail -n+41 /var/etc/dnsmasq.conf.cfg01411c | head -n 4
dhcp-range=set:wifi,10.53.20.100,10.53.20.249,255.255.255.0,1h
dhcp-option=wifi,option:dns-server,0.0.0.0
dhcp-range=set:link_house,10.53.20.100,10.53.20.249,255.255.255.254,1h
dhcp-option=link_house,option:dns-server,0.0.0.0
```

---

It looks like the ipcalc awk /31 fixes did not get backported to 23.05, but the priority should be avoiding dnsmasq crash loop.
https://github.com/openwrt/openwrt/pull/12925
https://github.com/openwrt/openwrt/pull/13158

---

It looks like snapshot/main ipcalc shell rewrite does not handle /31 networks (rfc3021)
https://github.com/openwrt/openwrt/pull/13765

```
[john@john ~]$ busybox ash /tmp/owrt/ipcalc.sh.orig 10.53.100.0/31 0 1
IP=10.53.100.0
NETMASK=255.255.255.254
NETWORK=10.53.100.0
PREFIX=31
COUNT=2
network (10.53.100.0/31) too small
```

I believe this could be fixed using `upper=$(((network | hostmask))` for the prefix>30 condition @pprindeville?

https://github.com/openwrt/openwrt/blob/0d74b2a1e5f35d534fb5433ffcd85aa6b967cd47/package/base-files/files/bin/ipcalc.sh#L112-L115

yogo1212, 2024-05-09:
Yeah!

The code you mentioned is missing:

```
elif [ "$prefix" -eq 31 ]; then        
    upper=$((network | hostmask))
```

I will make a PR.

yogo1212, 2024-05-09:
There was another rewrite in shell after I resigned as a reviewer on that PR.
Technically, that brings ipcalc up to three rewrites.
When the rewrite in Rust comes, we'll make sure to remember this :cocktail:

commit bff0e1cf (2024-06-17): base-files: fix ipcalc bound calculation for /31 prefix

A small regress from the translation to shell.
Fixes #12921, originally fixed in #12925 (Github).

Link: https://github.com/openwrt/openwrt/pull/15430
PR #15430 (2024-06-17): base-files: fix ipcalc bound calculation for /31 prefix
A small regress from the translation to shell.
Fixes #12921, originally fixed in #12925 (Github).