openwrt/openwrt #12927: IPv4 static route is not set

micw, 2023-06-16:
### Describe the bug

I have defined the following static route:

```
config route
        option gateway '192.168.61.2'
        option target '10.200.0.0/16'
```

The config came from LUCI webui. I also tried it manually with a separate netmask

```
config route
        option gateway '192.168.61.2'
        option target '10.200.0.0'
        option netmask '255.255.0.0'
```

Both settings do not add a route:

`ip route show table main` Shows different routes but not the one above.

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link Archer C6 v3

### Image kind

Official downloaded image

### Steps to reproduce

Go to LUCI, Network -> Routing -> Static IPv4 route

Add a simple route. Save+Apply

### Actual behaviour

No route added

### Expected behaviour

Route should be added to table main

### Additional info

I have another static route configured which works:

```
config route
        option interface 'gw2'
        option target '0.0.0.0/0'
        option table '2'
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

micw, 2023-06-16:
I can manually add the route and it works (till reboot):

`ip route add 10.200.0.0/16 via 192.168.61.2`

ptpt52, 2023-06-16:
ack, I notice the issue.
ping @nbd168 @ynezz @hauke

ptpt52, 2023-06-16:
@micw 
but it looks like that if you set the correct `option interface 'xxx'` config line, it would work.

micw, 2023-06-17:
Affirmative. After I set the interface where the route is going to, it works.
Before I tried with the interface where the traffic comes wrong (which is obviously wrong). Then I tried the way above without an interface which seemed logical to me because
a) interface is not a required field in LUCI and
b) when seetting the route manually, no interface is required

john-tho, 2023-06-17:
If someone wants to dig into this, I think it is parsed by `interface_ip_add_route` in `netifd`:
https://lxr.openwrt.org/source/netifd/interface-ip.c#L402
For luci, it looks like there is an `o.optional = false;` setting used some places. Guessing this could make an option required. You might be able to add it to the Interface selection, but I do not know if an interface is required for every route config stanza: https://github.com/openwrt/luci/blob/4d631c34533960f092f7d1969fd848d31f57d2b9/modules/luci-mod-network/htdocs/luci-static/resources/view/network/routes.js#L41
Guessing netifd uses interface to determine the gateway address. Luci helptext seems to indicate this as well: https://github.com/openwrt/luci/blob/4d631c34533960f092f7d1969fd848d31f57d2b9/modules/luci-mod-network/htdocs/luci-static/resources/view/network/routes.js#L75

Ashus, 2023-08-06:
I also noticed this. I thought if was caused by interface being Wireguard VPN.
I can't use "Route Allowed IPs" option in LuCI tunnel configuration, because in my configuration it overwrote my local LAN subnet (which I need to be routed as well) making the device completely inaccessible from LAN.

ColonelThirtyTwo, 2024-11-09:
I'm having this issue as well, also on a Wireguard VPN, where I have my "Route Allowed IPs" inserted into a different routing table but want a few routes in the main one.