openwrt/openwrt #12929: mwan3 双wan口负载均衡，从外网访问高跃点的wan口udp数据没有原路返回，udp数据流向了低跃点wan口导致丢包。

dongc123, 2023-06-17:
### Describe the bug

[image]
固件版本OpenWrt 23.05.0-rc1 平台x86/64 内核版本5.15.114 mwan3把wan1和wan2做了负载均衡，wan1 ip:175.0.82.xxx 跃点30 wan2 ip:175.0.80.xxx 跃点20 当upd数据包从wan1进入后，从wan2流出没有原路返回，上图是wan1和wan2抓包。

### OpenWrt version

OpenWrt 23.05.0-rc1

### OpenWrt target/subtarget

x86/64

### Device

x86/64

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2023-06-17:
If possible, try the x-wrt firmware to see if there is such a problem
@dongc123

dongc123, 2023-06-18:
麻烦留个链接谢谢

ptpt52, 2023-06-18:
x-wrt固件不难找到吧，谷歌一下 x-wrt，官网下载

dongc123, 2023-06-19:
Thank you for your reply. The problem has been solved. The next correction is that the multiple wan ports of openwrt wireguard cannot shake hands at the same time. I took a lot of detours before and thought it was a routing problem with the mwan3 udp protocol. Only through Arthur’s post did I know that it was a wireguard problem. , Thanks to Sir Arthur for patching Dafa to solve the problem. attach the original post
https://github.com/openwrt/packages/issues/9538
`diff -uNr linux-5.15.12_orig/drivers/net/wireguard/socket.c linux-5.15.12_wg/drivers/net/wireguard/socket.c
--- linux-5.15.12_orig/drivers/net/wireguard/socket.c	2021-12-29 19:29:03.000000000 +0800
+++ linux-5.15.12_wg/drivers/net/wireguard/socket.c	2022-06-01 08:18:00.990080098 +0800
@@ -17,6 +17,12 @@
 #include <net/udp_tunnel.h>
 #include <net/ipv6.h>
 
+u32 dst_addr;
+u32 src_addr;
+
+int receive = 0;
+int send = 0;
+
 static int send4(struct wg_device *wg, struct sk_buff *skb,
 		 struct endpoint *endpoint, u8 ds, struct dst_cache *cache)
 {
@@ -37,6 +43,13 @@
 
 	rcu_read_lock_bh();
 	sock = rcu_dereference_bh(wg->sock4);
+    
+	if (((receive) && (!send)) || ((send) && (!receive))) {
+		src_addr = dst_addr;
+	}
+	else {
+		src_addr = 0;
+	}
 
 	if (unlikely(!sock)) {
 		ret = -ENONET;
@@ -52,9 +65,11 @@
 		security_sk_classify_flow(sock, flowi4_to_flowi_common(&fl));
 		if (unlikely(!inet_confirm_addr(sock_net(sock), NULL, 0,
 						fl.saddr, RT_SCOPE_HOST))) {
-			endpoint->src4.s_addr = 0;
+			endpoint->src4.s_addr = src_addr;
 			endpoint->src_if4 = 0;
-			fl.saddr = 0;
+			fl.saddr = src_addr;
+			send = 1;
+			receive = 0;
 			if (cache)
 				dst_cache_reset(cache);
 		}
@@ -62,9 +77,11 @@
 		if (unlikely(endpoint->src_if4 && ((IS_ERR(rt) &&
 			     PTR_ERR(rt) == -EINVAL) || (!IS_ERR(rt) &&
 			     rt->dst.dev->ifindex != endpoint->src_if4)))) {
-			endpoint->src4.s_addr = 0;
+			endpoint->src4.s_addr = src_addr;
 			endpoint->src_if4 = 0;
-			fl.saddr = 0;
+			fl.saddr = src_addr;
+			send = 1;
+			receive = 0;
 			if (cache)
 				dst_cache_reset(cache);
 			if (!IS_ERR(rt))
@@ -77,8 +94,12 @@
 					    wg->dev->name, &endpoint->addr, ret);
 			goto err;
 		}
-		if (cache)
+		if (cache) {
+			if (receive) {
+				fl.saddr = src_addr;
+			}
 			dst_cache_set_ip4(cache, &rt->dst, fl.saddr);
+		}
 	}
 
 	skb->ignore_df = 1;
@@ -315,6 +336,11 @@
 static int wg_receive(struct sock *sk, struct sk_buff *skb)
 {
 	struct wg_device *wg;
+	struct iphdr *ip_header = (struct iphdr *)skb_network_header(skb);
+
+	dst_addr = ip_header->daddr;
+	receive = 1;
+	send = 0;
 
 	if (unlikely(!sk))
 		goto err;
`

delgh1, 2023-07-17:
> Thank you for your reply. The problem has been solved.

You should close this issue then.

wackejohn, 2026-05-10:
> > ai修改的版本，测试中。。。
> > ```
> > diff -uNr linux-6.12.74_orig/drivers/net/wireguard/peer.h linux-6.12.74/drivers/net/wireguard/peer.h
> > --- linux-6.12.74_orig/drivers/net/wireguard/peer.h	2026-02-19 23:29:56.000000000 +0800
> > +++ linux-6.12.74/drivers/net/wireguard/peer.h	2026-03-03 15:00:48.000000000 +0800
> > @@ -64,6 +64,8 @@
> >  	struct list_head allowedips_list;
> >  	struct napi_struct napi;
> >  	u64 internal_id;
> > +	u32 last_rx_dst_addr;
> > +	int last_rx_if4;
> >  };
> >  
> >  struct wg_peer *wg_peer_create(struct wg_device *wg,
> > diff -uNr linux-6.12.74_orig/drivers/net/wireguard/receive.c linux-6.12.74/drivers/net/wireguard/receive.c
> > --- linux-6.12.74_orig/drivers/net/wireguard/receive.c	2026-02-19 23:29:56.000000000 +0800
> > +++ linux-6.12.74/drivers/net/wireguard/receive.c	2026-03-03 15:00:40.000000000 +0800
> > @@ -336,6 +336,14 @@
> >  					struct sk_buff *skb,
> >  					struct endpoint *endpoint)
> >  {
> > +	/* 核心逻辑：在数据包成功解密并确认 Peer 身份后，持久化锁定其入站路径 */
> > +	if (endpoint->src4.s_addr && endpoint->src_if4) {
> > +		peer->last_rx_dst_addr = endpoint->src4.s_addr;
> > +		peer->last_rx_if4 = endpoint->src_if4;
> > +		/* 调试：确认每个 Peer 锁定的物理路径 */
> > +		/* printk(KERN_DEBUG "wg_peer_lock: peer=%llu, addr=%pI4, if=%d\n", peer->internal_id, &peer->last_rx_dst_addr, peer->last_rx_if4); */
> > +	}
> > +
> >  	struct net_device *dev = peer->device->dev;
> >  	unsigned int len, len_before_trim;
> >  	struct wg_peer *routed_peer;
> > diff -uNr linux-6.12.74_orig/drivers/net/wireguard/socket.c linux-6.12.74/drivers/net/wireguard/socket.c
> > --- linux-6.12.74_orig/drivers/net/wireguard/socket.c	2026-02-19 23:29:56.000000000 +0800
> > +++ linux-6.12.74/drivers/net/wireguard/socket.c	2026-03-03 15:00:28.000000000 +0800
> > @@ -20,9 +20,18 @@
> >  static int send4(struct wg_device *wg, struct sk_buff *skb,
> >  		 struct endpoint *endpoint, u8 ds, struct dst_cache *cache)
> >  {
> > +	/* 安全提取 Peer 指针（规避 6.12 宏崩溃风险） */
> > +	struct noise_keypair *keypair = skb ? PACKET_CB(skb)->keypair : NULL;
> > +	struct wg_peer *peer = (keypair && keypair->entry.peer) ? keypair->entry.peer : NULL;
> > +
> > +	/* 优先级逻辑：Peer 锁定值 > Endpoint 实时值 */
> > +	u32 saddr_hack = (peer && peer->last_rx_dst_addr) ? peer->last_rx_dst_addr : endpoint->src4.s_addr;
> > +	int oif_hack = (peer && peer->last_rx_if4) ? peer->last_rx_if4 : endpoint->src_if4;
> > +
> >  	struct flowi4 fl = {
> > -		.saddr = endpoint->src4.s_addr,
> > +		.saddr = saddr_hack,
> >  		.daddr = endpoint->addr4.sin_addr.s_addr,
> > +		.flowi4_oif = oif_hack, /* 强制锁定物理输出接口 */
> >  		.fl4_dport = endpoint->addr4.sin_port,
> >  		.flowi4_mark = wg->fwmark,
> >  		.flowi4_proto = IPPROTO_UDP
> > @@ -52,9 +61,6 @@
> >  		security_sk_classify_flow(sock, flowi4_to_flowi_common(&fl));
> >  		if (unlikely(!inet_confirm_addr(sock_net(sock), NULL, 0,
> >  						fl.saddr, RT_SCOPE_HOST))) {
> > -			endpoint->src4.s_addr = 0;
> > -			endpoint->src_if4 = 0;
> > -			fl.saddr = 0;
> >  			if (cache)
> >  				dst_cache_reset(cache);
> >  		}
> > ```
> 
> 叼，感谢，有用，终于解决痛点了

> > ai修改的版本，测试中。。。
[... 44 lines cut ...]
> > +	int oif_hack = (peer && peer->last_rx_if4) ? peer->last_rx_if4 : endpoint->src_if4;
> > +
> >  	struct flowi4 fl = {
> > -		.saddr = endpoint->src4.s_addr,
> > +		.saddr = saddr_hack,
> >  		.daddr = endpoint->addr4.sin_addr.s_addr,
> > +		.flowi4_oif = oif_hack, /* 强制锁定物理输出接口 */
> >  		.fl4_dport = endpoint->addr4.sin_port,
> >  		.flowi4_mark = wg->fwmark,
> >  		.flowi4_proto = IPPROTO_UDP
> > @@ -52,9 +61,6 @@
> >  		security_sk_classify_flow(sock, flowi4_to_flowi_common(&fl));
> >  		if (unlikely(!inet_confirm_addr(sock_net(sock), NULL, 0,
> >  						fl.saddr, RT_SCOPE_HOST))) {
> > -			endpoint->src4.s_addr = 0;
> > -			endpoint->src_if4 = 0;
> > -			fl.saddr = 0;
> >  			if (cache)
> >  				dst_cache_reset(cache);
> >  		}
> > ```
> 
> 叼，感谢，有用，终于解决痛点了

呃，这个版本 好像还有bug。。。。修复版本我加入了混淆补丁，有兴趣可以联系发你。。。