openwrt/openwrt #12930: mwan3 dual wan port load balancing, accessing the udp data of the high-hop wan port from the external network does not return the original route, and the udp data flows to the low-hop wan port, resulting in packet loss.

dongc123, 2023-06-17:
### Describe the bug

[image]
Firmware version OpenWrt 23.05.0-rc1 platform x86/64 kernel version 5.15.114 mwan3 load balances wan1 and wan2, wan1 ip: 175.0.82.xxx hop 30 wan2 ip: 175.0.80.xxx hop 20 when After the upd data packet enters from wan1, it flows out from wan2 without going back the same way. The picture above shows the packets captured by wan1 and wan2.

### OpenWrt version

OpenWrt 23.05.0-rc1

### OpenWrt target/subtarget

x86/64

### Device

x86/64

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-06-18:
Does the packet data somehow encode source address besides normal UDP header?

ptpt52, 2023-06-18:
show:
```
ip route show table 1
ip route show table 2
ip route show table main
```

sunnyqeen, 2023-06-28:
maybe a bug in socket.c
fi.flowi4_oif is not set, should be fi.flowi4_oif = endpoint->src_if4 ??
`static int send4(struct wg_device *wg, struct sk_buff *skb,
		 struct endpoint *endpoint, u8 ds, struct dst_cache *cache)
{
	struct flowi4 fl = {
		.saddr = endpoint->src4.s_addr,
		.daddr = endpoint->addr4.sin_addr.s_addr,
		.fl4_dport = endpoint->addr4.sin_port,
		.flowi4_mark = wg->fwmark,
		.flowi4_proto = IPPROTO_UDP
	};`

brada4, 2023-06-28:
How fo you know wireguatd us relevant?