openwrt/openwrt #12934: libubox20230523 package is not installed by default on 23.05-rc1 on ipx807x target

fortellerq, 2023-06-17:
### Describe the bug

I discovered this issue when trying to install non-basic version of wpad on Xiaomi AX3600 router. Uninstalling the default wpad-basic-mbedtls package and installing it again (or any other wpad-* package) causes following errors to occur in log:
Error loading shared library libubox.so.20230523: No such file or directory (needed by /usr/sbin/hostapd)

WiFi stops working when this happens. Installing libubox20230523 resolves this issue.

As of now, the custom build cannot be performed on https://firmware-selector.openwrt.org/ page, as it fails with following error even if default list of packages is selected:
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libubox20220927 for mtd
 * pkg_hash_fetch_best_installation_candidate: Packages for mtd found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package mtd.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for mtd:
 *     libubox20220927
 * opkg_install_cmd: Cannot install package mtd.

### OpenWrt version

r23077-bb03069691

### OpenWrt target/subtarget

ipq807x/generic

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Steps to reproduce

opkg remove wpad-basic-mbedtls
opkg install wpad-basic-mbedtls

### Actual behaviour

WiFi stops working, following error is visible in log:
Error loading shared library libubox.so.20230523: No such file or directory (needed by /usr/sbin/hostapd)

### Expected behaviour

I can change wpad without breaking the wifi.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2023-06-17:
Duplicate of #12934

closed 2023-06-17 by ynezz

fortellerq, 2023-06-17:
Maybe I am not following something, but how this issue is an duplicate of itself?