openwrt/openwrt #12939: Canadian CPE510 v1.1 - Hardware not supported

mungewell, 2023-06-18:
### Describe the bug

I have been trying to flash a TP-Link CPE510v1.1, and getting the message 'hardware not supported'. The official FW identifies the unit as:
[image]

I am aware the the 'supported' list in the code here, which **would** suggest that it is possible....
https://git.archive.openwrt.org/?p=openwrt.git;a=blob;f=tools/firmware-utils/src/tplink-safeloader.c;h=77a894b237281f6211dd1123ead4b5b0f2f58370;hb=c8315df1affa8eadb2ccbfd421a1e5806a444f47#l133

However, I **was** able to flash an official FW, and when I look at the strings within that binary I see:
```
simon@the-void:~/Downloads$ strings Pharos_2.0-up-ver2-1-6-P1[20170908-rel45233].bin | grep CPE510
(CPE510(TP-LINK|UN|N300-5|00000000):2.0
CPE510(TP-LINK|UN|N300-5):1.0
CPE510(TP-LINK|UN|N300-5):1.1
CPE510(TP-LINK|US|N300-5):1.1
CPE510(TP-LINK|EU|N300-5):1.1
CPE510(TP-LINK|CA|N300-5):1.1 <------------------
CPE510(TP-LINK|UN|N300-5|00000000):2.0
CPE510(TP-LINK|US|N300-5|55530000):2.0
CPE510(TP-LINK|EU|N300-5|45550000):2.0
CPE510(TP-LINK|CA|N300-5|43410000):2.0
CPE510(TP-LINK|UN|N300-5|00000000):3.0
```

Perhaps this suggests that there is a separate ID for units sold in Canada. I am not set up to rebuild the OpenWRT code, but can perform simple testing if it would help.

Is it 'safe' to binary edit the bin, to change 'US' to 'CA' as a quick test whether this is the root cause of my problem?

### OpenWrt version

n/a

### OpenWrt target/subtarget

n/a

### Device

CPE510v1.1

### Image kind

Official downloaded image

### Steps to reproduce

Attempt to flash FW "openwrt-22.03.5-ath79-generic-tplink_cpe510-v1-squashfs-factory.bin"

"Hardware Not Supported"

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-06-18:
Shorten file name to something like factory.bin

mungewell, 2023-06-19:
@brada4 Thanks for the suggestion, unfortunately this does NOT improve situation. The `.bin` was not uploaded with same 'Unsuppoorted hardware" message.

I made the following change with a hex editor
```
 00002050:  7c 4e 33 30 30 2d 35 29  3a 31 2e 30 0d 0a 43 50  |N300-5):1.0..CP
 00002060:  45 35 31 30 28 54 50 2d  4c 49 4e 4b 7c 55 4e 7c  E510(TP-LINK|UN|  <--- Change 0x55,0x4e to 0x43,0x41
 00002070:  4e 33 30 30 2d 35 29 3a  31 2e 31 0d 0a 43 50 45  N300-5):1.1..CPE
```

Giving
```
 00002050:  7c 4e 33 30 30 2d 35 29  3a 31 2e 30 0d 0a 43 50  |N300-5):1.0..CP
 00002060:  45 35 31 30 28 54 50 2d  4c 49 4e 4b 7c 43 41 7c  E510(TP-LINK|CA|  <-----
 00002070:  4e 33 30 30 2d 35 29 3a  31 2e 31 0d 0a 43 50 45  N300-5):1.1..CPE
```

And that resulted in a different error message "The file is incorrect". So I guess there must be a checksum covering the file. I'll either have to set up a build enviroment or figure another hack to bypass/recompute the checksum.

brada4, 2023-06-19:
It is described here:
https://forum.openwrt.org/t/tp-link-router-firmware-checksum-recalculation/99588
tp-link support says Canada has US devices? Check all kings of barcodes and fcc-id-s to confirm it is same device with few bits difference.

mungewell, 2023-06-20:
After fighting with the build system for a while I was able to add the 'CA' device ID, and could then flash the resultant `openwrt-ath79-generic-tplink_cpe510-v1-squashfs-factory.bin` to the unit.

Seems like I didn't enable enough packages to get a fully working system, but I can SSH into it via ethernet interface.... now to figure that out!

brada4, 2023-06-20:
Once in OpenWRT you can use sysupgrade to get any version you like, preferably same you built to load.

If looking at "GPL code" from CPE510 v1.0 then it is same exact hardware globally with software US/EU flip switch.

mungewell, 2023-06-20:
I was able to use `opkg` to install the GUI and extra bits I needed.

I also believe/presume that the hardware is the same.... What I **do** know is that I was unable to flash the first code I built (without changes) and **WAS** able to flash the adjusted code.

If there is something that I can do to display the device ID from my device (or anything else to convince you) then just give me instructions. :-)

brada4, 2023-06-20:
I have to search it too... It has to be near safeloader, in file containing 1.0|EU 1.1|US etc, search the commit adding another CPE510/520 revisions.

Djfe, 2023-06-27:
This is were the code currently sits:
https://github.com/openwrt/firmware-utils/blob/master/src/tplink-safeloader.c#L400-L435 (GitHub mirror)

closed 2023-09-02 by openwrt-bot