openwrt/openwrt #12955: IPQ40xx Wifi doesn't initialise correctly at boot if option txantenna '1' is used

Codemonkey1973, 2023-06-21:
### Describe the bug

Adding the following to /etc/config/wireless causes radio initialisation to fail without any obvious error message:

option txantenna '1'

2.4G Radio doesn't appear to send or receive any packets (checked output with a spectrum analyser and no tx whatsoever)

Also in LuCI, under Network->Wireless, the following is shown for radio0:

Qualcomm Atheros IPQ4019 802.11b/g/n
Device is not active

However, if I subsequently type the following, the radio bursts into life

/sbin/wifi up

or even just:

/sbin/wifi

or click on restart in LuCI next to radio0

### OpenWrt version

r23408-008cc836fe

### OpenWrt target/subtarget

ipq40xx/generic

### Device

8devices Habanero DVK

### Image kind

Self-built image

### Steps to reproduce

Use the following wifi configuration and reboot. Device boots but there is no activity on the 2.4G WiFi

wait a bit then type /sbin/wifi up and the wifi bursts into life.

config wifi-device 'radio0'
        option type 'mac80211'
        option channel '1'
        option path 'platform/soc/a000000.wifi'
        option country 'GB'
        option htmode 'HT20'
        option hwmode '11g'
        option cell_density '1'
        option txpower '20'
        option distance '250'
        option diversity '0'
        option txantenna '1'
        option rxantenna '1'

config wifi-device 'radio1'
        option type 'mac80211'
        option channel '36'
        option hwmode '11a'
        option path 'platform/soc/a800000.wifi'
        option htmode 'VHT80'
        option disabled '1'

config wifi-iface 'wifinet1'
        option device 'radio0'
        option mode 'mesh'
[... 680 lines cut ...]
CONFIG_PACKAGE_zlib=y
CONFIG_PCRE_JIT_ENABLED=y
# CONFIG_WOLFSSL_HAS_ECC25519 is not set
# CONFIG_WOLFSSL_HAS_OPENVPN is not set
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
CONFIG_PACKAGE_kmod-crypto-crc32c=y
CONFIG_PACKAGE_kmod-lib-crc-ccitt=y
CONFIG_PACKAGE_kmod-lib-crc32c=y
# CONFIG_PACKAGE_kmod-mppe is not set
CONFIG_PACKAGE_kmod-ppp=y
# CONFIG_PACKAGE_kmod-ppp-synctty is not set
# CONFIG_PACKAGE_kmod-pppoa is not set
CONFIG_PACKAGE_kmod-pppoe=y
# CONFIG_PACKAGE_kmod-pppol2tp is not set
CONFIG_PACKAGE_kmod-pppox=y
# CONFIG_PACKAGE_kmod-pptp is not set
CONFIG_PACKAGE_kmod-slhc=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Codemonkey1973, 2023-06-22:
Ah, yes, there's this:

root@OpenWrt:/# logread -e hostapd:
Wed Jun 21 22:31:02 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy phy0-ap0) --> new PHY
Wed Jun 21 22:31:04 2023 daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Wed Jun 21 22:31:04 2023 daemon.err hostapd: Driver does not support configured HT capability [TX-STBC]
Wed Jun 21 22:31:04 2023 daemon.notice hostapd: phy0-ap0: interface state COUNTRY_UPDATE->DISABLED
Wed Jun 21 22:31:04 2023 daemon.notice hostapd: phy0-ap0: AP-DISABLED
Wed Jun 21 22:31:04 2023 daemon.err hostapd: phy0-ap0: Unable to setup interface.
Wed Jun 21 22:31:04 2023 daemon.notice hostapd: nl80211: deinit ifname=phy0-ap0 disabled_11b_rates=0
Wed Jun 21 22:31:04 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-TERMINATING
Wed Jun 21 22:31:04 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface phy0-ap0 wasn't started

brada4, 2023-06-22:
Check in `iw list` 
```
        Available Antennas: TX 0x7 RX 0x7
        Configured Antennas: TX 0x7 RX 0x7
```
without wifi being brought up and after failed bringup.

Codemonkey1973, 2023-06-22:
If I disable the interface and then boot, I get this:

```
root@OpenWrt:/# iw list | grep Ant
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
```
if I then enable it and run /sbin/wifi up (wifi doesn't work) I get this:

`root@OpenWrt:/# logread -e hostapd:
Wed Jun 21 22:46:45 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy phy0-ap0) --> new PHY
Wed Jun 21 22:46:46 2023 daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Wed Jun 21 22:46:46 2023 daemon.err hostapd: Driver does not support configured HT capability [TX-STBC]
Wed Jun 21 22:46:46 2023 daemon.notice hostapd: phy0-ap0: interface state COUNTRY_UPDATE->DISABLED
Wed Jun 21 22:46:46 2023 daemon.notice hostapd: phy0-ap0: AP-DISABLED
Wed Jun 21 22:46:46 2023 daemon.err hostapd: phy0-ap0: Unable to setup interface.
Wed Jun 21 22:46:46 2023 daemon.notice hostapd: nl80211: deinit ifname=phy0-ap0 disabled_11b_rates=0
Wed Jun 21 22:46:47 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-TERMINATING
Wed Jun 21 22:46:47 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface phy0-ap0 wasn't started
root@OpenWrt:/# iw list | grep Ant
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x1 RX 0x1
`

If I then run /sbin/wifi up again (wifi now starts working), I get this:

```
root@OpenWrt:/# logread -e hostapd:
Wed Jun 21 22:46:45 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy phy0-ap0) --> new PHY
Wed Jun 21 22:46:46 2023 daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Wed Jun 21 22:46:46 2023 daemon.err hostapd: Driver does not support configured HT capability [TX-STBC]
Wed Jun 21 22:46:46 2023 daemon.notice hostapd: phy0-ap0: interface state COUNTRY_UPDATE->DISABLED
Wed Jun 21 22:46:46 2023 daemon.notice hostapd: phy0-ap0: AP-DISABLED
Wed Jun 21 22:46:46 2023 daemon.err hostapd: phy0-ap0: Unable to setup interface.
Wed Jun 21 22:46:46 2023 daemon.notice hostapd: nl80211: deinit ifname=phy0-ap0 disabled_11b_rates=0
Wed Jun 21 22:46:47 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-TERMINATING
Wed Jun 21 22:46:47 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface phy0-ap0 wasn't started
Wed Jun 21 22:49:09 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy phy0-ap0) --> new PHY
Wed Jun 21 22:49:11 2023 daemon.notice hostapd: phy0-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Wed Jun 21 22:49:12 2023 daemon.notice hostapd: phy0-ap0: interface state COUNTRY_UPDATE->ENABLED
Wed Jun 21 22:49:12 2023 daemon.notice hostapd: phy0-ap0: AP-ENABLED
root@OpenWrt:/# iw list | grep Ant
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x1 RX 0x1

```

Codemonkey1973, 2023-06-22:
Thanks, adding the following to /etc/config/wireless gets things working (verified only transmitting on the first antenna with a spectrum analyser)

option htmode 'HT20|TX-STBC'

iw reg get / set 'GB' / get before bringing up radio had no effect.

brada4, 2023-06-22:
Probably worth bringing this to upstreams' attention, in the meantime complementing documentation that led you to legacy option.

john-tho, 2023-06-22:
I saw this mention of `TX-STBC` with `tx_chainmask` in the ath10k driver:
https://elixir.bootlin.com/linux/latest/source/drivers/net/wireless/ath/ath10k/mac.c#L4951
```
	if (ar->ht_cap_info & WMI_HT_CAP_TX_STBC && (ar->cfg_tx_chainmask > 1))
		ht_cap.cap |= IEEE80211_HT_CAP_TX_STBC;
```
It may be that the driver does not advertise this capability with `option txantenna '1'`?
OpenWrt's config generator checks capabilities here: https://github.com/openwrt/openwrt/blob/df8e4b6c2e6440bc9a31467de1739c69c328e391/package/kernel/mac80211/files/lib/netifd/wireless/mac80211.sh#L208

This should give `0x80` without the `txantenna` option:
```
printf "0x%x\n" $(( $(iw phy phy0 info | grep Capabilities: | cut -d: -f2) & 0x80 ))
```
Check it with the `option txantenna '1'`?

Also worth checking `/tmp/run/hostapd-phy*conf` to check that the config generator is not asking for something that cannot be done.
Cheers

brada4, 2023-06-22:
It permits same config on 2nd attempt, so something probably irrelevant was programmed and driver magically now supports antenna mask.