openwrt/openwrt #12962: [MT7621] WiFi performance is much worse compared to OpenWRT 21.02

rany2, 2023-06-22:
### Describe the bug

OpenWRT 21.02 gives much better WiFi speeds compared to snapshot.

I do not believe it's an issue with mac80211 or mt76 as I've backported mac80211 6.1 and the latest mt76 to OpenWRT 21.02 and it works flawless there. With the same mt76 as the one used in OpenWRT 21.02, snapshot gives inferior performance.

Note that SNAPSHOT seems to consistently have a high retry count but 21.02 never does.

## OpenWRT SNAPSHOT

<pre><font color="#26A269"><b>root@gw</b></font>:<font color="#12488B"><b>~</b></font># iperf3 -c rany-laptop-wifi 
Connecting to host rany-laptop-wifi, port 5201
[  5] local dd1c:2285:baee::1 port 50766 connected to dd1c:2285:baee::987c port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  50.1 MBytes   420 Mbits/sec    0    832 KBytes       
[  5]   1.00-2.00   sec  43.8 MBytes   367 Mbits/sec   62   1.60 MBytes       
[  5]   2.00-3.00   sec  53.8 MBytes   451 Mbits/sec  482   1.24 MBytes       
[  5]   3.00-4.00   sec  61.2 MBytes   514 Mbits/sec    0   1.02 MBytes       
[  5]   4.00-5.00   sec  60.0 MBytes   503 Mbits/sec    0    686 KBytes       
[  5]   5.00-6.00   sec  60.0 MBytes   503 Mbits/sec    0    642 KBytes       
[  5]   6.00-7.00   sec  53.8 MBytes   451 Mbits/sec    2    791 KBytes       
[  5]   7.00-8.00   sec  65.0 MBytes   545 Mbits/sec   34    739 KBytes       
[  5]   8.00-9.00   sec  61.2 MBytes   514 Mbits/sec   13   1.39 MBytes       
[  5]   9.00-10.00  sec  61.2 MBytes   514 Mbits/sec  194   7.46 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   570 MBytes   478 Mbits/sec  787             sender
[  5]   0.00-10.25  sec   569 MBytes   466 Mbits/sec                  receiver

iperf Done.
<font color="#26A269"><b>root@gw</b></font>:<font color="#12488B"><b>~</b></font># iperf3 -c rany-laptop-wifi -R
Connecting to host rany-laptop-wifi, port 5201
Reverse mode, remote host rany-laptop-wifi is sending
[  5] local dd1c:2285:baee::1 port 46628 connected to dd1c:2285:baee::987c port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  76.6 MBytes   643 Mbits/sec                  
[  5]   1.00-2.00   sec  84.1 MBytes   705 Mbits/sec                  
[  5]   2.00-3.00   sec  79.0 MBytes   663 Mbits/sec                  
[  5]   3.00-4.00   sec  84.4 MBytes   708 Mbits/sec                  
[  5]   4.00-5.00   sec  78.0 MBytes   655 Mbits/sec                  
[  5]   5.00-6.00   sec  77.4 MBytes   649 Mbits/sec                  
[  5]   6.00-7.00   sec  74.9 MBytes   628 Mbits/sec                  
[  5]   7.00-8.00   sec  86.2 MBytes   723 Mbits/sec                  
[  5]   8.00-9.00   sec  82.6 MBytes   693 Mbits/sec                  
[  5]   9.00-10.00  sec  71.4 MBytes   599 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.04  sec   798 MBytes   666 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   795 MBytes   667 Mbits/sec                  receiver

iperf Done.
<font color="#26A269"><b>root@gw</b></font>:<font color="#12488B"><b>~</b></font># 

</pre>

## OpenWRT 21.02

```
root@gw:~# iperf3 -c rany-laptop-wifi  
Connecting to host rany-laptop-wifi, port 5201
[  5] local dd1c:2285:baee::1 port 47136 connected to dd1c:2285:baee::987c port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  89.5 MBytes   751 Mbits/sec    0   1.44 MBytes       
[  5]   1.00-2.00   sec  96.2 MBytes   807 Mbits/sec    0   1.83 MBytes       
[  5]   2.00-3.00   sec  93.8 MBytes   787 Mbits/sec    0   1.81 MBytes       
[  5]   3.00-4.00   sec  92.5 MBytes   776 Mbits/sec    0   1.82 MBytes       
[  5]   4.00-5.00   sec  96.2 MBytes   807 Mbits/sec    0   1.87 MBytes       
[  5]   5.00-6.00   sec  96.2 MBytes   807 Mbits/sec    0   1.86 MBytes       
[  5]   6.00-7.00   sec  97.5 MBytes   818 Mbits/sec    0   1.83 MBytes       
[  5]   7.00-8.00   sec  97.5 MBytes   818 Mbits/sec    0   1.85 MBytes       
[  5]   8.00-9.00   sec  91.2 MBytes   765 Mbits/sec    0   1.83 MBytes       
[  5]   9.00-10.00  sec  91.2 MBytes   765 Mbits/sec    0   7.46 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   942 MBytes   790 Mbits/sec    0             sender
[  5]   0.00-10.04  sec   939 MBytes   785 Mbits/sec                  receiver

iperf Done.
root@gw:~# iperf3 -c rany-laptop-wifi -R
Connecting to host rany-laptop-wifi, port 5201
Reverse mode, remote host rany-laptop-wifi is sending
[  5] local dd1c:2285:baee::1 port 43780 connected to dd1c:2285:baee::987c port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  57.7 MBytes   484 Mbits/sec                  
[  5]   1.00-2.00   sec  69.5 MBytes   583 Mbits/sec                  
[  5]   2.00-3.00   sec  81.7 MBytes   685 Mbits/sec                  
[  5]   3.00-4.00   sec  96.4 MBytes   809 Mbits/sec                  
[  5]   4.00-5.00   sec  85.8 MBytes   720 Mbits/sec                  
[  5]   5.00-6.00   sec  76.6 MBytes   643 Mbits/sec                  
[  5]   6.00-7.00   sec  70.2 MBytes   589 Mbits/sec                  
[  5]   7.00-8.00   sec  74.9 MBytes   628 Mbits/sec                  
[  5]   8.00-9.00   sec  77.0 MBytes   646 Mbits/sec                  
[  5]   9.00-10.00  sec  83.7 MBytes   702 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.04  sec   777 MBytes   649 Mbits/sec    1             sender
[  5]   0.00-10.00  sec   774 MBytes   649 Mbits/sec                  receiver

iperf Done.
root@gw:~# 
```

### OpenWrt version

r23408+3-008cc836fe

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

rany2, 2023-06-22:
> OpenWRT 21.02 tree used: https://github.com/rany2/openwrt/tree/openwrt-21.02
> OpenWRT SNAPSHOT tree used: https://github.com/rany2/openwrt/tree/master
> Same MT76 used (for both 21.02 and snapshot): https://github.com/rany2/mt76

I should note that this occurs with the official images as well, but I just wanted to use the same mt76 to confirm it's not an issue with mt76.

brada4, 2023-06-22:
Any neighbors with brand new 5GHz router within range? It is still above 433Mbps so it is connected with 2T2R AC.

rany2, 2023-06-22:
@paraka Apologies for the mention but do you think this might be related to some of the recent refactoring work for mt7621? It can't be the wireless stack or the wireless driver as I've run the same code for both 21.02 and snapshot.

rany2, 2023-06-22:
> Any neighbors with brand new 5GHz router within range? It is still above 433Mbps so it is connected with 2T2R AC.

It's actually an 802.11ax chip (MT7905D).

brada4, 2023-06-22:
Is the same channel selected in both cases? Other access points in survey from AP?

rany2, 2023-06-22:
@brada4 Yep, its on the same channel in both tests. I don't use ACS and am not on a DFS channel. I really don't think it's an issue with the WiFi driver.

Also the hostapd config is identical in both tests and the hostapd/libnl version are the same as I've backported that as well for OpenWRT 21.02 to rule out an issue in that department as well.

paraka, 2023-06-23:
> @paraka Apologies for the mention but do you think this might be related to some of the recent refactoring work for mt7621? It can't be the wireless stack or the wireless driver as I've run the same code for both 21.02 and snapshot.

@rany2 I don't know, honestly. Are other devices affected also with this issue or is only present in yours? This is a difficult question to answer without bisecting.

rany2, 2023-06-23:
@paraka I don't really have other MT7621 devices with WiFi chips capable of reaching those speeds, so I can't tell.. :(

rany2, 2023-06-23:
The issue is that the speeds reached in snapshot are within the bounds of 802.11ac's capabilities; so it makes sense why nobody has noticed.

DragonBluep, 2023-06-23:
I'm not sure if it's related, packet steering is broken since February: https://github.com/openwrt/openwrt/issues/12095

And again, perhaps you should switch the firewall from nftables to iptables and test it again.

brada4, 2023-06-23:
> I'm not sure if it's related, packet steering is broken since February: #12095
> 
> And again, perhaps you should switch the firewall from nftables to iptables and test it again.

Only for mwlwifi driver in snapshot.

DragonBluep, 2023-06-23:
> > I'm not sure if it's related, packet steering is broken since February: #12095
> > And again, perhaps you should switch the firewall from nftables to iptables and test it again.
> 
> Only for mwlwifi driver in snapshot.

not really...

lukasz1992, 2023-06-23:
confirm, it works for mt7530 switch but does not for mt7915 driver.

rany2, 2023-06-23:
> I'm not sure if it's related, packet steering is broken since February: #12095
> 
> And again, perhaps you should switch the firewall from nftables to iptables and test it again.

No change when switching to iptables with offloading setup. Here is the output re packet steering on 21.02:

```
# find /sys/devices/platform/ | egrep '[xr]ps_cpus'
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/net/guest5ghz/queues/tx-0/xps_cpus
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/net/guest5ghz/queues/rx-0/rps_cpus
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/net/psk5ghz/queues/tx-0/xps_cpus
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/net/psk5ghz/queues/rx-0/rps_cpus
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/net/guest2ghz/queues/tx-0/xps_cpus
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/net/guest2ghz/queues/rx-0/rps_cpus
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/net/psk2ghz/queues/tx-0/xps_cpus
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/net/psk2ghz/queues/rx-0/rps_cpus
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/net/eap5ghz/queues/tx-0/xps_cpus
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/net/eap5ghz/queues/rx-0/rps_cpus
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/net/eap2ghz/queues/tx-0/xps_cpus
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/net/eap2ghz/queues/rx-0/rps_cpus
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/net/iot5ghz/queues/tx-0/xps_cpus
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/net/iot5ghz/queues/rx-0/rps_cpus
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/net/iot2ghz/queues/tx-0/xps_cpus
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/net/iot2ghz/queues/rx-0/rps_cpus
/sys/devices/platform/1e100000.ethernet/mdio_bus/mdio-bus/mdio-bus:1f/net/lan2/queues/tx-0/xps_cpus
/sys/devices/platform/1e100000.ethernet/mdio_bus/mdio-bus/mdio-bus:1f/net/lan2/queues/rx-0/rps_cpus
/sys/devices/platform/1e100000.ethernet/mdio_bus/mdio-bus/mdio-bus:1f/net/lan3/queues/tx-0/xps_cpus
/sys/devices/platform/1e100000.ethernet/mdio_bus/mdio-bus/mdio-bus:1f/net/lan3/queues/rx-0/rps_cpus
/sys/devices/platform/1e100000.ethernet/mdio_bus/mdio-bus/mdio-bus:1f/net/wan/queues/tx-0/xps_cpus
/sys/devices/platform/1e100000.ethernet/mdio_bus/mdio-bus/mdio-bus:1f/net/wan/queues/rx-0/rps_cpus
/sys/devices/platform/1e100000.ethernet/mdio_bus/mdio-bus/mdio-bus:1f/net/lan1/queues/tx-0/xps_cpus
/sys/devices/platform/1e100000.ethernet/mdio_bus/mdio-bus/mdio-bus:1f/net/lan1/queues/rx-0/rps_cpus
/sys/devices/platform/1e100000.ethernet/net/eth0/queues/tx-0/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/eth0/queues/rx-0/rps_cpus
```

brada4, 2023-06-23:
> not really...

It may cause IPI unnecessarily, thus warning it may be hit or miss. e.g. irqbalance's faraway sibling irqd (of SuSE) tries to also balance net queues (block queues etc) optimally besides IRQ-s

khanjui, 2023-06-24:
I don't know if it's related but there are lots of tx ques after 5.15 kernel update.

`/etc$ find /sys/devices/platform/ | egrep '[xr]ps_cpus'
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/net/wlan1/queues/tx-0/xps_cpus
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/net/wlan1/queues/rx-0/rps_cpus
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/net/phy0-ap0/queues/tx-0/xps_cpus
/sys/devices/platform/1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0/net/phy0-ap0/queues/rx-0/rps_cpus
/sys/devices/platform/1e100000.ethernet/mdio_bus/mdio-bus/mdio-bus:1f/net/lan1/queues/tx-0/xps_cpus
/sys/devices/platform/1e100000.ethernet/mdio_bus/mdio-bus/mdio-bus:1f/net/lan1/queues/rx-0/rps_cpus
/sys/devices/platform/1e100000.ethernet/mdio_bus/mdio-bus/mdio-bus:1f/net/lan2/queues/tx-0/xps_cpus
/sys/devices/platform/1e100000.ethernet/mdio_bus/mdio-bus/mdio-bus:1f/net/lan2/queues/rx-0/rps_cpus
/sys/devices/platform/1e100000.ethernet/net/eth0/queues/tx-6/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/eth0/queues/tx-14/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/eth0/queues/tx-4/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/eth0/queues/tx-12/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/eth0/queues/tx-2/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/eth0/queues/tx-10/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/eth0/queues/tx-0/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/eth0/queues/tx-9/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/eth0/queues/tx-7/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/eth0/queues/tx-15/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/eth0/queues/tx-5/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/eth0/queues/tx-13/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/eth0/queues/tx-3/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/eth0/queues/tx-11/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/eth0/queues/tx-1/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/eth0/queues/tx-8/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/eth0/queues/rx-0/rps_cpus
/sys/devices/platform/1e100000.ethernet/net/wan/queues/tx-6/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/wan/queues/tx-14/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/wan/queues/tx-4/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/wan/queues/tx-12/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/wan/queues/tx-2/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/wan/queues/tx-10/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/wan/queues/tx-0/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/wan/queues/tx-9/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/wan/queues/tx-7/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/wan/queues/tx-15/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/wan/queues/tx-5/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/wan/queues/tx-13/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/wan/queues/tx-3/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/wan/queues/tx-11/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/wan/queues/tx-1/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/wan/queues/tx-8/xps_cpus
/sys/devices/platform/1e100000.ethernet/net/wan/queues/rx-0/rps_cpus
`

brada4, 2023-06-25:
@khanjui you can find answers in kernel documentation or forums.

lukasz1992, 2023-06-25:
maybe driver started to support multiple queues?

Djfe, 2023-06-27:
While this is interesting, it's weird that we are giving support for a device that was never supported on official openwrt-21.02 ^^
Interesting, that it's not caused by the wifi driver but other changes.
Maybe there is a regression somewhere.

Have you tested openwrt-22.03 with the current mt76 driver, yet?

rany2, 2023-06-28:
@Djfe I have and its the same issue. Sorry forgot to update this thread on that, but the iperf3 stats are equivalent to the ones in snapshot.

rany2, 2023-06-28:
> While this is interesting, it's weird that we are giving support for a device that was never supported on official openwrt-21.02 ^^

Well, adding support for it in 21.02 wasn't too difficult to I just glossed over it but yes that device did not launch on 21.02; I just cherry picked the commits to add support for iy and make sure which was the last good version.

As usual in all my testing, I'm using the same mac80211, mt76 and hostapd/libnl whenever i tested 21.02, 22.03, and snapshot. So those could not have been the factors I believe.

rany2, 2023-06-28:
Something I noticed about 21.02 is that it is a lot more snappy than the later versions. For example, iwinfo would show output instantly on 21.02 whereas on later versions it takes noticeably longer. Same with other commands like dmesg, albeit less noticeable. I'm not sure if that's related or not but I thought i should mention it regardless.

Djfe, 2023-06-28:
ok since the issue was already present in 22-03, that means we can rule out patches like this one as the cause
https://forum.openwrt.org/t/2-gbps-wan-lan-nat-routing-on-ramips-mt7621-devices/131478 
(aka changes from snapshot)

lukasz1992, 2023-06-29:
Guys, could you share some CPU stats during tests?

Mine looks like this:

CPU:   0% usr  22% sys   0% nic  14% idle   0% io   0% irq  63% sirq
Load average: 1.10 0.27 0.10 3/111 13657
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
  696     2 root     RW       0   0%  18% [mt76-tx phy0]
  271     2 root     RW       0   0%  16% [napi/mtk_eth-6]
  689     2 root     RW       0   0%  13% [napi/phy0-9]
  690     2 root     SW       0   0%  10% [napi/phy0-10]
   12     2 root     RW       0   0%   7% [ksoftirqd/0]
   18     2 root     RW       0   0%   6% [ksoftirqd/1]
  270     2 root     SW       0   0%   5% [napi/mtk_eth-5]
   23     2 root     RW       0   0%   4% [ksoftirqd/2]
 1308  1305 network  S     3084   1%   1% /usr/sbin/hostapd -s -g /var/run/hostapd/global
   28     2 root     SW       0   0%   1% [ksoftirqd/3]

Condition: download speedtest on OpenWRT 23.05.0-rc1, RT-AX53U (means: mt7621 and mt7915 dbdc) on 5GHz channel HE80, WPA2+3.
My Internet connection is PPPoE; results are similar on iperf3 (client<--router) tests.
So it is saturating CPU (even if 14% is idle it is still far from ~800Mbps). Is this just maximum we can do or are there any ways to improve speeds? Receiving packets works faster, possible to achieve 500Mbps.

rany2, 2023-06-29:
@lukasz1992 sirq seems really high, on 21.02 with latest mt76 and mac80211: 

```
Mem: 58112K used, 193872K free, 56K shrd, 4496K buff, 11888K cached
CPU:   0% usr  39% sys   0% nic  41% idle   0% io   0% irq  19% sirq
Load average: 1.66 0.53 0.22 7/82 5431
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
  805     2 root     RW       0   0%  20% [mt76-tx phy0]
  390     2 root     SW       0   0%  15% [napi/mtk_eth-6]
  795     2 root     RW       0   0%  13% [napi/phy0-9]
  796     2 root     RW       0   0%   2% [napi/phy0-10]
   25     2 root     RW       0   0%   2% [ksoftirqd/3]
 1804     1 root     S     2756   1%   1% /usr/sbin/dawn
   20     2 root     SW       0   0%   1% [ksoftirqd/2]
    9     2 root     SW       0   0%   1% [ksoftirqd/0]
  389     2 root     SW       0   0%   1% [napi/mtk_eth-5]
 1270     1 root     S     5392   2%   0% /usr/sbin/hostapd -s -g /var/run/hostapd/global
   10     2 root     IW       0   0%   0% [rcu_sched]
 5393     2 root     IW       0   0%   0% [kworker/u8:0-ph]
 5431  5421 root     R     1260   0%   0% top
  793     2 root     SW       0   0%   0% [napi/phy0-7]
```

The download speed achieved was 880Mbit/s here.

lukasz1992, 2023-06-29:
I wonder if opposite direction: downgrade kernel on 23.05 (instead of upgrading mac80211/mt76)
would keep the results. Maybe some compiler flags do the magic degradation?
Example of such problem was described here: https://github.com/openwrt/openwrt/issues/8778

lukasz1992, 2023-06-29:
My idea was to check if that is about kernel regression (then you're right, too much effort needed), or some change in config that brings regression (could be much easier to spot). I think this is a second case, as my stats are somehow similar - and mean there is a slowdown in all threads.

Today I switched to oem software - mt76 patches developed by rany2 makes driver rock-stable for me, but speeds like 40% compared to OEM is too much.

rany2, 2023-06-29:
I didn't share this information but there is some speed up possible in SNAPSHOT, namely disabling CPU idle driver for mt7621 helped with CPU performance. With the idle driver disabled I could reach normal ~930Mbit/s speeds running iperf3 on the router itself for download and ~880Mbit/s for upload whereas it was only cracking under 600Mbit/s prior to disabling the CPU idle driver.

HOWEVER, the issue remains so its irrelevant to this issue; just worth mentioning so you know what I tried on the kernel config side.

lukasz1992, 2023-06-30:
Which config setting exactly is this?

rany2, 2023-06-30:
> Which config setting exactly is this?

If you're asking about disabling dynamic ticks, I am talking about `HZ_PERIODIC`

rany2, 2023-06-30:
@lukasz1992 I'm not sure if it's a coincidence or if nbd read this thread, but seems like the cpu idle issue got fixed just minutes ago: https://github.com/openwrt/openwrt/commit/dc38199b96ee3ef0ac52873893c42e28fa0564fa

nbd168, 2023-06-30:
I looked at this thread and ran some ethernet performance tests, where I could see a clear difference between between cpuidle enabled vs disabled.
Regarding the wifi performance issues - are you testing bridged or NATed traffic?

nbd168, 2023-06-30:
Please try installing the 'bridger' package and see if it helps

lukasz1992, 2023-06-30:
> I looked at this thread and ran some ethernet performance tests, where I could see a clear difference between between cpuidle enabled vs disabled. Regarding the wifi performance issues - are you testing bridged or NATed traffic?

I did tests:
1) to iperf3 locally installed on OpenWRT as a server
2) just speedtest - my WAN is PPPoE, maybe it makes difference
For both of them download does not exceed ~360Mbit (very rare exceptions to 400-450), and cpu is almost 90%

On OEM firmware wifi->internet (pppoe) traffic is being offloaded. But as I see - it works only on ARM on OpenWrt.

rany2, 2023-06-30:
> Please try installing the 'bridger' package and see if it helps

I already did, no difference.

nbd168, 2023-07-01:
Could you please test if using the ethernet driver from 21.02 on a newer build makes it faster?

rany2, 2023-07-01:
> Could you please test if using the ethernet driver from 21.02 on a newer build makes it faster?

I'll give this a try tomorrow or maybe later today, I'm too busy right now. I'll keep you posted regardless!~

rany2, 2023-07-02:
Can't get it to work yet, though running only with 5.4's DSA driver causes ethernet to stop working. Still need to get mtk_eth_soc to be the 5.4 version though, which isn't too simple.

DragonBluep, 2023-07-04:
Compared to the 5.4 kernel config (OpenWrt 21.x), I noticed some systick changes. I'm not sure if this patch will help improve performance.
```
--- a/target/linux/ramips/mt7621/config-5.15
+++ b/target/linux/ramips/mt7621/config-5.15
@@ -92,6 +92,10 @@ CONFIG_HAS_DMA=y
 CONFIG_HAS_IOMEM=y
 CONFIG_HAS_IOPORT_MAP=y
 CONFIG_HIGHMEM=y
+CONFIG_HZ=250
+# CONFIG_HZ_100 is not set
+CONFIG_HZ_250=y
+CONFIG_HZ_PERIODIC=y
 CONFIG_I2C=y
 CONFIG_I2C_ALGOBIT=y
 CONFIG_I2C_BOARDINFO=y
@@ -186,8 +190,6 @@ CONFIG_NET_SELFTESTS=y
 CONFIG_NET_SWITCHDEV=y
 CONFIG_NET_VENDOR_MEDIATEK=y
 # CONFIG_NET_VENDOR_RALINK is not set
-CONFIG_NO_HZ_COMMON=y
-CONFIG_NO_HZ_IDLE=y
 CONFIG_NR_CPUS=4
 CONFIG_NVMEM=y
 CONFIG_NVMEM_SYSFS=y
```

rany2, 2023-07-04:
@DragonBluep From quick testing a few days ago, it made no difference. I just tested iperf3 performance and it's the same as without 5.4's systick params (incl. setting timer to 250hz). Also the WiFi performance issue remains.

DragonBluep, 2023-07-13:
I did some simple tests with iperf3, main branch, mt7621 + mt7915d.
|wan -> wlan|wlan -> lan|
|-|-|
|446 Mbits/sec|745 Mbits/sec|

All network interrupts are loaded onto CPU0. It is very abnormal. It should be related to https://github.com/openwrt/openwrt/issues/12095 or https://github.com/openwrt/openwrt/commit/37dbe302158c7c567835ed64c118236dfc0425b2.

There are a lot of IPI call interrupts here, which are related to multi-core processing. Some commits since 2020 causes this problem?
https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/log/arch/mips/kernel/smp.c
```
root@OpenWrt:~# cat /proc/interrupts
           CPU0       CPU1       CPU2       CPU3
  8:      24403      27825      28217      26404  MIPS GIC Local   1  timer
  9:     364858          0          0          0  MIPS GIC  63  IPI call
 10:          0     264995          0          0  MIPS GIC  64  IPI call
 11:          0          0     325144          0  MIPS GIC  65  IPI call
 12:          0          0          0     408082  MIPS GIC  66  IPI call
 13:       6311          0          0          0  MIPS GIC  67  IPI resched
 14:          0       6517          0          0  MIPS GIC  68  IPI resched
 15:          0          0       2063          0  MIPS GIC  69  IPI resched
 16:          0          0          0       2383  MIPS GIC  70  IPI resched
 17:          0          0          0          0  MIPS GIC  19  1e000600.gpio-bank0, 1e000600.gpio-bank1, 1e000600.gpio-bank2
 19:         12          0          0          0  MIPS GIC  33  ttyS0
 20:     687821          0          0          0  MIPS GIC  10  1e100000.ethernet
 21:         16          0          0          0  MIPS GIC  30  mt7530
 22:         14          0          0          0    mt7530   1  mt7530-0:01
 23:          2          0          0          0    mt7530   2  mt7530-0:02
 24:          0          0          0          0    mt7530   3  mt7530-0:03
 25:          0          0          0          0  1e000600.gpio  18  keys
 26:          0          0          0          0  1e000600.gpio   4  keys
 27:      27888          0          0          0  MIPS GIC  11  mt7915e-hif
 28:     130803          0          0          0  MIPS GIC  31  mt7915e
ERR:        419

Mem: 66472K used, 183560K free, 19200K shrd, 0K buff, 19200K cached
CPU:   0% usr  42% sys   0% nic  13% idle   0% io   0% irq  43% sirq
Load average: 1.70 0.89 0.60 7/88 3939
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
  201     2 root     RW       0   0%  22% [napi/mtk_eth-6]
   28     2 root     RW       0   0%  20% [ksoftirqd/3]
  551     2 root     RW       0   0%  19% [mt76-tx phy0]
  541     2 root     RW       0   0%  14% [napi/phy0-9]
  542     2 root     SW       0   0%   5% [napi/phy0-10]
  200     2 root     SW       0   0%   3% [napi/mtk_eth-5]
```

rany2, 2023-07-13:
@DragonBluep Very interesting observation, do you plan on testing out which commit may have been the culprit or do you not have the time? I could try to revert to the 5.4 version to see if it's related if you can't ATM.

DragonBluep, 2023-07-13:
> @DragonBluep Very interesting observation, do you plan on testing out which commit may have been the culprit or do you not have the time? I could try to revert to the 5.4 version to see if it's related if you can't ATM.

I will try to check it out on the weekend. Have you seen these IPI interrupts in the 5.4 kernel?

rany2, 2023-07-13:
@DragonBluep This is what I'm getting on 5.4 with backported mac80211 and mt76 from snapshot:

```
root@garbagerouter:~# cat /proc/interrupts 
           CPU0       CPU1       CPU2       CPU3       
  8:    2390006    2390077    2390048    2390030  MIPS GIC Local   1  timer
  9:       9051          0          0          0  MIPS GIC  63  IPI call
 10:          0       2974          0          0  MIPS GIC  64  IPI call
 11:          0          0       9213          0  MIPS GIC  65  IPI call
 12:          0          0          0       2844  MIPS GIC  66  IPI call
 13:     884005          0          0          0  MIPS GIC  67  IPI resched
 14:          0     987212          0          0  MIPS GIC  68  IPI resched
 15:          0          0     532293          0  MIPS GIC  69  IPI resched
 16:          0          0          0     584005  MIPS GIC  70  IPI resched
 17:          0          0          0          0  MIPS GIC  19  1e000600.gpio-bank0, 1e000600.gpio-bank1, 1e000600.gpio-bank2
 19:         12          0          0          0  MIPS GIC  33  ttyS0
 20:          0          0          0          0  MIPS GIC  29  xhci-hcd:usb1
 21:     766761          0          0          0  MIPS GIC  10  1e100000.ethernet
 23:     303696          0          0          0  MIPS GIC  11  mt7915e-hif
 24:    1081549          0          0          0  MIPS GIC  31  mt7915e
 26:          0          0          0          0  1e000600.gpio  16  keys
 27:          0          0          0          0  1e000600.gpio  15  keys
ERR:      20824
root@garbagerouter:~# top -b -n 1
Mem: 54068K used, 197916K free, 56K shrd, 4172K buff, 11456K cached
CPU:   2% usr   0% sys   0% nic  95% idle   0% io   0% irq   2% sirq
Load average: 0.11 0.07 0.02 2/81 3148
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 3147  3124 root     R     1252   0%   2% top -b -n 1
  390     2 root     SW       0   0%   2% [napi/mtk_eth-6]
 1207     1 root     S     5276   2%   0% /usr/sbin/hostapd -s -g /var/run/hostapd/global
 1208     1 root     S     5136   2%   0% /usr/sbin/wpa_supplicant -n -s -g /var/run/wpa_supplicant/global
 2646     1 root     S     3988   2%   0% /usr/sbin/uhttpd -f -h /www -r garbagerouter -x /cgi-bin -t 60 -T 30 -k 20 -A 1 -n 3 -N 100 -R -p 10.146.109.9:80 -C /etc/uhttpd.crt -K /etc/uhttpd.key -s 10.146.109.9:443 -q
 1095     1 root     S     2012   1%   0% /sbin/rpcd -s /var/run/ubus/ubus.sock -t 30
  736     1 root     S     1864   1%   0% /usr/sbin/haveged -F -w 1024 -d 32 -i 32 -v 1
 1267     1 root     S     1840   1%   0% /sbin/netifd
    1     0 root     S     1632   1%   0% /sbin/procd
  657     1 ubus     S     1268   1%   0% /sbin/ubusd
 1029     1 logd     S     1264   1%   0% /sbin/logd -S 64
 3124  3123 root     S     1252   0%   0% -ash
 1727     1 root     S<    1248   0%   0% /usr/sbin/ntpd -n -N -S /usr/sbin/ntpd-hotplug -p 10.146.109.1
 1566     1 root     S     1192   0%   0% /usr/sbin/irqbalance -f -t 10
```

lukasz1992, 2023-07-13:
And @rany2 what speeds can you acchieve right now, during the load test?

@DragonBluep I had almost the same results as you.

I tried manually run these scripts (already removed in main) during testing, but it did not help.
It seems that radio interrupts are assigned to only core, because mac80211 stack supports one queue for tx.

But anyway cpu is loaded in 85% - which I think makes not important if it runs on one core only.
It means that we hit the cpu limit

rany2, 2023-07-13:
> I tried manually run these scripts (already removed in main) during testing, but it did not help.

You need to modify that script to have it apply to mt7915. At any rate, I already tried this before and it does not help.

>And @rany2 what speeds can you acchieve right now, during the load test?

WiFi speeds on 5.4 are averaging 790-880 Mbit/s both up and down according to iperf3. When under load, it's only using roughly 50% of the CPU on all cores, not 85% according to htop. 

>  It means that we hit the cpu limit

For 5.10 and later, there is a IRQ swarm issue I think, if you enable htop's "detailed mode" it shows that most of this load is softirq handling related but this does not occur on 5.4

lukasz1992, 2023-07-13:
@rany2 , yes I know - I added mt79* to the list, it did not help.

It means like 2x degradation on 5.10 and newer kernels.

I tried to check load with `perf `util, but no conclusions except mt76-worker is very busy.

DragonBluep, 2023-07-14:
https://github.com/openwrt/openwrt/issues/12962#issuecomment-1634478672 to be continue:
1. After backing to the old irq set API, CPU usage will be reduced by 10% - 25%. But it doesn't help with WiFi performance.
@lukasz1992 @rany2 Perhaps you are interested in testing whether this is a measurement error.
[0001-Revert-genirq-Remove-setup_irq-and-remove_irq.patch](https://github.com/openwrt/openwrt/files/12049952/0001-Revert-genirq-Remove-setup_irq-and-remove_irq.patch)
[0001-Revert-MIPS-Replace-setup_irq-by-request_irq.patch](https://github.com/openwrt/openwrt/files/12049954/0001-Revert-MIPS-Replace-setup_irq-by-request_irq.patch)
```
Mem: 70188K used, 179844K free, 19188K shrd, 0K buff, 19188K cached
CPU:   0% usr  34% sys   0% nic  31% idle   0% io   0% irq  33% sirq
Load average: 1.78 1.91 1.21 4/90 6172
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
  200     2 root     RW       0   0%  25% [napi/mtk_eth-6]
  552     2 root     SW       0   0%  16% [mt76-tx phy0]
  542     2 root     RW       0   0%  14% [napi/phy0-9]
  543     2 root     RW       0   0%   4% [napi/phy0-10]
  199     2 root     RW       0   0%   2% [napi/mtk_eth-5]
 6172  3625 root     R     1400   1%   1% top -d1

Mem: 74316K used, 175716K free, 19208K shrd, 0K buff, 19208K cached
CPU:   0% usr  31% sys   0% nic  39% idle   0% io   0% irq  27% sirq
Load average: 2.45 1.99 1.26 7/91 3736
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
  572     2 root     RW       0   0%  26% [napi/phy0-9]
  582     2 root     SW       0   0%  15% [mt76-tx phy0]
  230     2 root     RW       0   0%  13% [napi/mtk_eth-6]
  573     2 root     SW       0   0%   3% [napi/phy0-10]
 3735  3564 root     R     1392   1%   1% top -d1
  229     2 root     SW       0   0%   1% [napi/mtk_eth-5]
```

2. MT7621 is a 2C4T SoC, 25% CPU usage is the maximum value that a process can use. If I'm wrong please correct me. `napi/mtk_eth-6` process frequently reaches this maximum value. I believe this process limits the maximum WiFi speed. The question is whether mtk_eth-6 represents port 6. If so, it doesn't make sense. Because all WAN port packages flow from port 5 to the CPU via RGMII2. Friendly ping @arinc9

3. About why all interrupts are handled by CPU0, it is expected behavior. This is the hardware limit. Please refer to the [MT7621 ProgrammingGuide](https://github.com/DragonBluep/archive_openwrt/blob/main/Documents/MediaTek/MT7621/MT7621%20ProgrammingGuide.pdf) page 9-10.

rany2, 2023-07-18:
I just gave OpenWRT SNAPSHOT (58c498247b3d06c0ff1967d2f48ba84084a6cf25) another try and I've noticed something odd. Unlike my previous testing, it seems that now `CLIENT -> AP` is slower than `AP -> CLIENT` by 2x. Previously it was the opposite.

Client:

```
$ iwinfo wlan0
wlan0     IEEE 802.11  ESSID:"xxx"  
          Mode:Managed  Frequency:5.765 GHz  Access Point: x   
          Bit Rate=1.0208 Gb/s   Tx-Power=22 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=70/70  Signal level=-34 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:75   Missed beacon:0
```

Test on x86 gateway connected to router via lan3 port on RTAX53U (AP and lan3 bridged):

```
root@gw:~# iperf3 -c rany-laptop-wifi
Connecting to host rany-laptop-wifi, port 5201
[  5] local dd1c:2285:baee::1 port 50282 connected to dd1c:2285:baee::987c port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  96.9 MBytes   812 Mbits/sec    0   1.80 MBytes       
[  5]   1.00-2.00   sec  97.5 MBytes   818 Mbits/sec    0   2.00 MBytes       
[  5]   2.00-3.00   sec  96.2 MBytes   807 Mbits/sec    0   1.87 MBytes       
[  5]   3.00-4.00   sec  98.8 MBytes   829 Mbits/sec    0   1.92 MBytes       
[  5]   4.00-5.00   sec  98.8 MBytes   828 Mbits/sec    0   1.42 MBytes       
[  5]   5.00-6.00   sec  95.0 MBytes   797 Mbits/sec    0   1.96 MBytes       
[  5]   6.00-7.00   sec  98.8 MBytes   828 Mbits/sec    0   1.91 MBytes       
[  5]   7.00-8.00   sec  98.8 MBytes   828 Mbits/sec    0   1.92 MBytes       
[  5]   8.00-9.00   sec  96.2 MBytes   807 Mbits/sec    0   1.93 MBytes       
[  5]   9.00-10.00  sec  93.8 MBytes   786 Mbits/sec    0   7.61 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   971 MBytes   814 Mbits/sec    0             sender
[  5]   0.00-10.04  sec   967 MBytes   808 Mbits/sec                  receiver

iperf Done.
root@gw:~# iperf3 -c rany-laptop-wifi -R
Connecting to host rany-laptop-wifi, port 5201
Reverse mode, remote host rany-laptop-wifi is sending
[  5] local dd1c:2285:baee::1 port 43356 connected to dd1c:2285:baee::987c port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  45.4 MBytes   381 Mbits/sec                  
[  5]   1.00-2.00   sec  58.4 MBytes   490 Mbits/sec                  
[  5]   2.00-3.00   sec  57.1 MBytes   479 Mbits/sec                  
[  5]   3.00-4.00   sec  56.1 MBytes   471 Mbits/sec                  
[  5]   4.00-5.00   sec  55.9 MBytes   469 Mbits/sec                  
[  5]   5.00-6.00   sec  56.8 MBytes   477 Mbits/sec                  
[  5]   6.00-7.00   sec  57.2 MBytes   480 Mbits/sec                  
[  5]   7.00-8.00   sec  57.8 MBytes   485 Mbits/sec                  
[  5]   8.00-9.00   sec  57.0 MBytes   478 Mbits/sec                  
[  5]   9.00-10.00  sec  55.7 MBytes   467 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.07  sec   563 MBytes   469 Mbits/sec   56             sender
[  5]   0.00-10.00  sec   558 MBytes   468 Mbits/sec                  receiver

iperf Done.
```

The good news is that this is close to what I'm getting in 21.02, though (Client -> AP) is not comparable to 21.02 results. 

# Notes

- Hardware flow offloading was enabled
- Bridger was installed and running

rany2, 2023-07-18:
> After backing to the old irq set API, CPU usage will be reduced by 10% - 25%. But it doesn't help with WiFi performance.
> @lukasz1992 @rany2 Perhaps you are interested in testing whether this is a [...]

I see, I guess it's a measurement issue but I wonder if it has some hidden performance implications.

DragonBluep, 2023-07-19:
> OpenWrt SNAPSHOT on Unielec U7621-06
> ```
> Accepted connection from 192.168.2.106, port 46712
> [  5] local 192.168.2.2 port 5201 connected to 192.168.2.106 port 46718
> [ ID] Interval           Transfer     Bitrate
> [  5]   0.00-1.00   sec  75.8 MBytes   636 Mbits/sec                  
> [  5]   1.00-2.00   sec  77.7 MBytes   652 Mbits/sec                  
> [  5]   2.00-3.00   sec  78.8 MBytes   661 Mbits/sec                  
> [  5]   3.00-4.00   sec  78.1 MBytes   655 Mbits/sec                  
> [  5]   4.00-5.00   sec  77.1 MBytes   647 Mbits/sec                  
> [  5]   5.00-6.00   sec  77.0 MBytes   646 Mbits/sec                  
> [  5]   6.00-7.00   sec  77.1 MBytes   647 Mbits/sec                  
> [  5]   7.00-8.00   sec  78.2 MBytes   656 Mbits/sec                  
> [  5]   8.00-9.00   sec  78.5 MBytes   659 Mbits/sec                  
> [  5]   9.00-10.00  sec  77.9 MBytes   654 Mbits/sec                  
> [  5]  10.00-10.02  sec  2.05 MBytes   699 Mbits/sec                  
> - - - - - - - - - - - - - - - - - - - - - - - - -
> [ ID] Interval           Transfer     Bitrate
> [  5]   0.00-10.02  sec   778 MBytes   651 Mbits/sec                  receiver
> ```
> 

@arinc9 It seems that you are testing **LAN -> WLAN** and **WLAN -> LAN** speed.

It's true that we can always get full speed with directions **LAN -> WLAN**, **WLAN -> LAN** and even **WLAN -> WAN**. But **WAN -> WLAN** speed is really bad. That's why I only posted **WAN <--> WLAN** test results in https://github.com/openwrt/openwrt/issues/12962#issuecomment-1634478672

With iperf3 server running on host 192.168.137.1, my result:
```
PS D:\Applications\iperf-3.14-win64> .\iperf3.exe -c 192.168.137.1 -P 4 -t 60 -R
Connecting to host 192.168.137.1, port 5201
Reverse mode, remote host 192.168.137.1 is sending
[  5] local 192.168.1.104 port 11748 connected to 192.168.137.1 port 5201
[  7] local 192.168.1.104 port 11749 connected to 192.168.137.1 port 5201
[  9] local 192.168.1.104 port 11750 connected to 192.168.137.1 port 5201
[ 11] local 192.168.1.104 port 11751 connected to 192.168.137.1 port 5201
[ ID] Interval           Transfer     Bitrate
...
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-60.02  sec   786 MBytes   110 Mbits/sec                  sender
[  5]   0.00-60.00  sec   785 MBytes   110 Mbits/sec                  receiver
[  7]   0.00-60.02  sec   908 MBytes   127 Mbits/sec                  sender
[  7]   0.00-60.00  sec   907 MBytes   127 Mbits/sec                  receiver
[  9]   0.00-60.02  sec   800 MBytes   112 Mbits/sec                  sender
[  9]   0.00-60.00  sec   798 MBytes   112 Mbits/sec                  receiver
[ 11]   0.00-60.02  sec   806 MBytes   113 Mbits/sec                  sender
[ 11]   0.00-60.00  sec   805 MBytes   113 Mbits/sec                  receiver
[SUM]   0.00-60.02  sec  3.22 GBytes   461 Mbits/sec                  sender
[SUM]   0.00-60.00  sec  3.22 GBytes   461 Mbits/sec                  receiver
```

rany2, 2023-07-19:
Switching to a vlan aware/vlan filtering bridge (not sure what the term for it is) fixed this issue for me, I think this may have made it use offloading properly and consistently; before that restarting the router would sometimes cause this issue to not occur and other times it would occur. I guess this issue is fixed on my side.

Something worth nothing is that the raw CPU performance from the device is a lot worse on SNAPSHOT despite disabling cpuidle driver but at least this issue in particular no longer pops up. (iperf on the device nets me 400-500Mbit/s instead of nearing 800-900Mbit/s on 21.02, but that's another issue that I need to open)

This fix was tested on ASUS RT-AX53U on https://github.com/openwrt/openwrt/commit/9e86a96af5ecea6e73fa463d6cfa4408e2b49d35.

closed 2023-07-19 by rany2

rany2, 2023-07-19:
> Something worth nothing is that the raw CPU performance from the device is a lot worse on SNAPSHOT despite disabling cpuidle driver but at least this issue in particular no longer pops up. (iperf on [...]

Small update, this is not an issue either. I forgot that the server needs to be started on the device's site not client end to reduce the CPU load. It works fine on SNAPSHOT, the same as 21.02.

So all seems OK so far, but seeing that this issue is a bit inconsistent (sometimes occurs and other times doesn't, with it usually occurring consistently; there is a chance it is not actually fixed and I got exceptionally lucky for the past 20 reboots). 

Hopefully I won't re-open this issue.

bjo81, 2024-05-24:
I also got this issue, VLAN filtering is active, but no extra VLANs used.

A NUC sending to a Thinkpad with an Intel AX WiFi:
```

Reverse mode, remote host 192.168.x.y is sending
[  5] local 192.168.x.z port 34360 connected to 192.168.x.z port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  65.8 MBytes   551 Mbits/sec                  
[  5]   1.00-2.00   sec  67.1 MBytes   563 Mbits/sec                  
[  5]   2.00-3.00   sec  65.2 MBytes   547 Mbits/sec                  
[  5]   3.00-4.00   sec  67.5 MBytes   566 Mbits/sec                  
[  5]   4.00-5.00   sec  62.4 MBytes   523 Mbits/sec                  
[  5]   5.00-6.00   sec  66.6 MBytes   559 Mbits/sec                  
[  5]   6.00-7.00   sec  67.1 MBytes   563 Mbits/sec                  
[  5]   7.00-8.00   sec  67.2 MBytes   564 Mbits/sec                  
[  5]   8.00-9.00   sec  67.0 MBytes   562 Mbits/sec                  
[  5]   9.00-10.00  sec  65.9 MBytes   553 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.01  sec   676 MBytes   566 Mbits/sec  386             sender
[  5]   0.00-10.00  sec   662 MBytes   555 Mbits/sec                  receiver
```

while

```
wlp3s0    IEEE 802.11  ESSID:"ssid"            Mode:Managed  Frequency:5.18 GHz Access Point: 50:EB:F6:86:XX:YY   
Bit Rate=1.1342 Gb/s   Tx-Power=22 dBm             Retry short limit:7   RTS thr:off   Fragment thr:off
Power Management:on          Link Quality=65/70  Signal level=-45 dBm
```
SoftIRQ goes up to 15% on the RT-AX53U.