openwrt/openwrt #12964: TFTP Recovery CPE510v1.1 - Allocated memory for elf image failed

mungewell, 2023-06-23:
### Describe the bug

Shortly after getting my CPE510 accepting images, I (maybe mistakenly) flashed another OpenWRT image and seem to have corrupted flash. The unit will boot to safe loader and pull files via TFTP.

None of the 'recovery.bin' images I have tried function, all give `Incorrect File`. I am now trying a 'vmlinuz' and getting the error `Allocated memory for elf image failed`.

My file came from my build tree:
```
./build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.15.117/vmlinuz
```

Full log from serial 
```
TP-LINK SafeLoader (Build time: Jun 12 2015 - 09:49:53)
CPU: 560MHz AHB: 225MHz DDR: 64MB
Performing LED check..  PASS
Press CTRL+B to enter SafeLoader: 1
Flash Manufacturer: Unknown(0xc8)
Flash Device ID: Unknown(0x4017)
Data flash init failed.
open user-config failed.
open user-config failed.

enet0 port4 up

TFTP server address is 192.168.0.100; our address is 192.168.0.254
Get filename 'recovery.bin'.
#################################################################
######################################
Done.
Bytes transferred = 5229389, 2420 Kbytes/sec
Incorrect File.
Writting error.
##Warning: File is unidentiable. Try downloading

TFTP server address is 192.168.0.100; our address is 192.168.0.254
Get filename 'vmlinuz'.
#################################################################
########################################
Done.
Bytes transferred = 5338684, 2359 Kbytes/sec
Skipping non LOAD program header (type 0x70000003)
Allocated memory for elf image failed: addr: 0x81330000, size 0xa09330
image_load_hi is 0x81a00000
##ERROR: File is unidentiable.
Input Password:
```

I'm guessing that the unit is past recoverable, but the kernel should boot... right?

### OpenWrt version

a860e439edefc941b1b5e2d4adf3956b353f4f90

### OpenWrt target/subtarget

CONFIG_TARGET_ath79_generic_DEVICE_tplink_cpe510-v1=y

### Device

CPE510v1.1 - (CA)

### Image kind

Self-built image

### Diffconfig

```text
simon@the-void:~/openwrt-github$ ./scripts/diffconfig.sh 
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_tplink_cpe510-v1=y
CONFIG_TARGET_EXT4_BLOCKSIZE=4096
CONFIG_TARGET_EXT4_BLOCKSIZE_4K=y
CONFIG_TARGET_EXT4_RESERVED_PCT=0
CONFIG_TARGET_IMAGES_GZIP=y
CONFIG_TARGET_INITRAMFS_FORCE=y
CONFIG_TARGET_ROOTFS_EXT4FS=y
CONFIG_TARGET_ROOTFS_PARTSIZE=104
```
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mungewell, 2023-06-23:
I found another thread talking about recovering CPE210 (similar device).
https://forum.archive.openwrt.org/viewtopic.php?id=70244

In that thread the capture shows a vastly different kernel size of:
```
    Get filename 'vmlinuz'.
    #################################################################
    ########
    Done.
    Bytes transferred = 3732560, 927 Kbytes/sec
    Skipping non LOAD program header (type 0x70000003)
    Allocated memory for elf segment ok: addr: 0x806b3050, size 0x88df94 <-----
    Loading .text @ 0x806b3050 (128 bytes)
    Loading .text.error @ 0x806b30d0 (8 bytes)
```

mungewell, 2023-06-23:
I found some other kernels in the directory above... note these are ending in 'X' not 'Z'.
```
simon@the-void:~/openwrt-github/build_dir/target-mips_24kc_musl/linux-ath79_generic$ ls -l vmlinux*
-rwxr-xr-x 1 simon simon  7817088 Jun 22 02:14 vmlinux
-rwxr-xr-x 1 simon simon 37700580 Jun 22 01:59 vmlinux.debug
-rwxr-xr-x 1 simon simon  7822016 Jun 22 02:14 vmlinux.elf
-rwxr-xr-x 1 simon simon 19666816 Jun 22 02:14 vmlinux-initramfs
-rwxr-xr-x 1 simon simon 49550308 Jun 22 02:14 vmlinux-initramfs.debug
-rwxr-xr-x 1 simon simon 19671744 Jun 22 02:14 vmlinux-initramfs.elf
```

The 'elf's will boot, up-to a point when it declare `Failed to get CPU node`. Again this is quiet suprising, but I guess this could be related to a corrupted flash... maybe this router is just '_done for_'.
```
TP-LINK SafeLoader (Build time: Jun 12 2015 - 09:49:53)
CPU: 560MHz AHB: 225MHz DDR: 64MB
Performing LED check..  PASS
Press CTRL+B to enter SafeLoader: 1
Flash Manufacturer: Unknown(0xc8)
Flash Device ID: Unknown(0x4017)
Data flash init failed.
open user-config failed.
open user-config failed.

enet0 port4 up

TFTP server address is 192.168.0.100; our address is 192.168.0.254
Get filename 'recovery.bin'.
##############################################
Done.
Bytes transferred = 2323044, 2363 Kbytes/sec
Incorrect File.
Writting error.
##Warning: File is unidentiable. Try downloading

TFTP server address is 192.168.0.100; our address is 192.168.0.254
Get filename 'vmlinuz'.
#################################################################
#################################################################
#######################
Done.
Bytes transferred = 7822016, 2417 Kbytes/sec
Allocated memory for elf segment ok: addr: 0x80060000, size 0x8b3c1c
Skipping non LOAD program header (type 0x4)
Loading .text @ 0x80060000 (5791856 bytes)
Loading __ex_table @ 0x805e6070 (5520 bytes)
Loading .rodata @ 0x805e8000 (1066656 bytes)
Loading .pci_fixup @ 0x806ec6a0 (3664 bytes)
Loading __ksymtab @ 0x806ed4f0 (42408 bytes)
Loading __ksymtab_gpl @ 0x806f7a98 (35640 bytes)
Loading __ksymtab_strings @ 0x807005d0 (118601 bytes)
Loading __param @ 0x8071d51c (1280 bytes)
Loading .data @ 0x8071e000 (596704 bytes)
Loading .init.text @ 0x807b0000 (121320 bytes)
Loading .init.data @ 0x807cd9e8 (28036 bytes)
Loading .fill @ 0x807d476c (20 bytes)
Clearing .bss @ 0x808e0000 (211996 bytes)

Starting kernel

[    0.000000] Linux version 5.15.117 (simon@the-void) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23401-a860e439ed) 12.3.0, GN3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001974c (MIPS 74Kc)
[    0.000000] SoC: Atheros AR9344 rev 2
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] OF: fdt: No valid device tree found, continuing without
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 55952K/65536K available (5655K kernel code, 585K rwdata, 1240K rodata, 1216K init, 207K bss, 9584K reserved, 0)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] Failed to get CPU node
[    0.000000] sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 21474836475000000ns
```

ValdikSS, 2023-06-30:
Have you tried to boot the stock kernel from the official manufacturer image?

>OF: fdt: No valid device tree found, continuing without
>Failed to get CPU node

This is because the kernel image does not have devicetree.

mungewell, 2023-07-01:
I had tried using the 'official_update.bin' as the 'recovery.bin' that it downloaded via TFTP, but it did not like that.

I assume that you mean to extract kernel, after some googling I got to:
```
$ binwalk pharos-up-ver1-3-3-P9\[20160705-rel52453\].bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
8212          0x2014          ELF, 32-bit MSB MIPS64 executable, MIPS, version 1 (SYSV)
17428         0x4414          LZMA compressed data, properties: 0x6D, dictionary size: 1048576 bytes, uncompressed size: -1 bytes
844545        0xCE301         Squashfs filesystem, little endian, version 4.0, compression:lzma, size: 4090076 bytes, 814 inodes, blocksize: 131072 bytes, created: 2016-07-05 02:49:43

$ dd if=pharos-up-ver1-3-3-P9\[20160705-rel52453\].bin of=kenel_lz.elf bs=1 skip=8212 count=836333
$ sudo cp kenel_lz.elf /srv/tftp/vmlinuz
```
After which it would TFTP boot further, but then panic'ed as it couldn't find rootfs.

I also tried to see if this kernel had a device tree appended, it does not...
```
$ ~/extract-dtb-github/extract_dtb/extract_dtb.py kenel_lz.elf 
No appended dtbs found
```

Is it possible to use the official kernel blended with the openwrt squashfs?

ValdikSS, 2023-07-01:
Well, have you tried to load https://downloads.openwrt.org/releases/22.03.5/targets/ath79/generic/openwrt-22.03.5-ath79-generic-tplink_cpe510-v1-initramfs-kernel.bin as the vmlinux?

Or https://downloads.openwrt.org/releases/22.03.5/targets/ath79/generic/openwrt-22.03.5-ath79-generic-tplink_cpe510-v1-squashfs-factory.bin as recovery.bin?

Spudz76, 2023-07-01:
Some TFTP loaders have a dumb limit on memory.  Would guess anything smaller than `0xa00000` (10MB) works while your initial attempt was `0xa09330` and everything else is smaller and has worked.  ELF files may not use as much memory since there is no need for compressed buffer extracting to destination.

And yes even though the thing has 64MB, the bootloader might only be able to see like 16MB, or they just have the limit because dumb software.

So if you can craft a stripped kernel+initramfs recovery image with size under 10MB uncompressed it might boot.  And then be able to mtd/sysupgrade a full sized image from there or the OEM image.  Like strip all the wifi junk you don't need that to boot and get in through serial or net and use mtd stuff.

Those suggested images are small enough as-is compressed but I assume they unpack to worse than 10MB and if so I expect them to also not work.