openwrt/openwrt #12965: collectd-mod-cpufreq is not built on some valid targets

raenye, 2023-06-23:
### Describe the bug

`collectd-mod-cpufreq` collects statistics about CPU frequency, which is only relevant to targets supporting CPU scaling.
One such target is MT7622, and everything was working fine on my Redmi AX6S (running snapshot `r23254-a0d7193425`)

Some buildbot change dropped support for building this package on certain targets, so I cannot upgrade. Probably it can still be self-built, but this is a regression (assumingly unintentional).

https://forum.openwrt.org/t/where-is-collectd-mod-cpufreq-gone/162459 states the same problem exists for another `mediatek/mt7622` target (Belkin RT3200) as well.

Thank you.

### OpenWrt version

r23411-68ef2d1856 (problem started earlier, see below)

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Xiaomi Redmi Router AX6S

### Image kind

Official downloaded image

### Steps to reproduce

Try to upgrade r23254 to r23411 via attended sysupgrade (LuCI or `auc`, both fail).

### Actual behaviour

```
# opkg list-installed collectd-mod-cpufreq
collectd-mod-cpufreq - 5.12.0-43

# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   SNAPSHOT r23254-a0d7193425 on mediatek/mt7622 (xiaomi,redmi-router-ax6s)
Available: SNAPSHOT r23411-68ef2d1856
Requesting package lists...
[...]
 kmod-crypto-sha512: 5.15.114-1 -> 5.15.117-1
installed package collectd-mod-cpufreq cannot be found in remote list!
 kmod-mt76-connac: 5.15.114+2023-05-13-969b7b5e-1 -> 5.15.117+2023-05-13-969b7b5e-1
[...]
```

### Expected behaviour

`collectd-mod-cpufreq` package should be available through `opkg`.

### Additional info

This started circa June 5, since I upgraded to the currently installed snapshot r23254 (which works fine) on June 4, and the forum discussion about R3200 is from June 8.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

raenye, 2023-06-23:
This discussion from 2021 might also be relevant:
https://forum.openwrt.org/t/collectd-mod-cpufreq-unavailable/87140

rany2, 2023-06-23:
It should be getting built: https://github.com/openwrt/packages/blob/master/utils/collectd/Makefile#L462

raenye, 2023-06-23:
> It should be getting built: https://github.com/openwrt/packages/blob/master/utils/collectd/Makefile#L462 ... mhm

It is built only for `aarch64_generic` and `arm_cortex-a15_neon-vfpv4`; sadly not for `aarch64_cortex-a53`, `aarch64_cortex-a72` and `arm_cortex-a9_.*`.
Redmi AX6S is `aarch64_cortex-a53`.

raenye, 2023-06-26:
@BKPepe are you perhaps familiar with the treatment of arm targets in OpenWrt?

pencil, 2023-09-18:
Has anyone made on any progress on this?

hnyman, 2023-09-18:
The reason is probably that aarch64_cortex-a53 is currently using bcm27xx/bcm2710 SDK for the package compilation, so neither TARGET_armsr or TARGET_mediatek gets matched.
(see step 7 download SDK in https://buildbot.staging.openwrt.org/master/packages/#/builders/14/builds/72  )
```
rsync -4 -va 'downloads.openwrt.org::downloads/snapshots/targets/bcm27xx/bcm2710/openwrt-sdk-*.tar.xz' sdk.archive
 in dir /builder/aarch64_cortex-a53/build (timeout 1200 secs)
 watching logfiles {}
 argv: [b'rsync', b'-4', b'-va', b'downloads.openwrt.org::downloads/snapshots/targets/bcm27xx/bcm2710/openwrt-sdk-*.tar.xz', b'sdk.archive']
 using PTY: False
receiving incremental file list
openwrt-sdk-bcm27xx-bcm2710_gcc-12.3.0_musl.Linux-x86_64.tar.xz
```

The easiest cure is probably to remove the target dependency and just let the module get built for all targets, even if they do not actually support any frequency scaling.
I will do that.

pencil, 2023-09-18:
@hnyman Thanks! I appreciate you following up here and opening a PR to fix ❤️

hnyman, 2023-09-20:
This has been fixed by https://github.com/openwrt/packages/pull/22137 (and its backport to 23.05)

Please close.

closed 2023-09-20 by raenye