openwrt/openwrt #12966: Gettext-full: Fails to build – gl_array_list.c

flipreverse, 2023-06-23:
### Describe the bug

Hi folks!

I'm having trouble building gettext-full as a hostpkg.
I use the current HEAD of OpenWRT's github repo: [9519a62](https://github.com/openwrt/openwrt/commit/9519a625499a963508755b7fc08fb825b3d8cfde). My host system is Debian Testing (now Stable aka Bookworm).
Did anyone of you come across the same issue? Or does someone know how to fix this?

I already tried a `make distclean`.
`make package/lib/gettext-full/hostpkg/compile V=s` as well as `make V=s` return the following error:
```
gcc -DHAVE_CONFIG_H -I. -I..  -I. -I. -I.. -I.. -Iglib -DIN_LIBTEXTSTYLE -DLIBXML_STATIC    -I./libcroco  -DDEPENDS_ON_LIBICO
NV=1 -I/media/raid/software/openwrt/c2600/staging_dir/host/include -I/media/raid/software/openwrt/c2600/staging_dir/hostpkg/i
nclude -I/media/raid/software/openwrt/c2600/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/include  -O2 -I/media
/raid/software/openwrt/c2600/staging_dir/host/include -I/media/raid/software/openwrt/c2600/staging_dir/hostpkg/include -I/med
ia/raid/software/openwrt/c2600/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/include -DPIC -fPIC -c gl_array_li
st.c                                                                                                                         
In file included from gl_array_list.h:21,                                                                                    
                 from gl_array_list.c:21:                                                                                    
gl_list.h:114:15: error: expected declaration specifiers or '...' before '*' token                                           
  114 | typedef bool (*gl_listelement_equals_fn) (const void *elt1, const void *elt2);                                       
      |               ^                                       
gl_list.h:518:33: error: unknown type name 'gl_listelement_equals_fn'; did you mean 'gl_listelement_compar_fn'?              
  518 |                                 gl_listelement_equals_fn equals_fn,                                                  
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~                                                             
      |                                 gl_listelement_compar_fn                                                             
gl_list.h:521:33: error: unknown type name 'bool'                                                                            
  521 |                                 bool allow_duplicates);                                                              
      |                                 ^~~~                                                                                 
gl_list.h:22:1: note: 'bool' is defined in header '<stdbool.h>'; did you forget to '#include <stdbool.h>'?                   
   21 | #include <stddef.h>                                                                                                  
  +++ |+#include <stdbool.h>
```

Cheers,
Alex

### OpenWrt version

9519a62

### OpenWrt target/subtarget

ipq806x

### Device

TP-Link C2600

### Image kind

Self-built image

### Steps to reproduce

* Start with a clean checkout
*  Run `make package/lib/gettext-full/hostpkg/compile V=s`

### Actual behaviour

`gettext-full` fails to build.

### Expected behaviour

`gettext-full` successfully compiles.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-06-24:
Please take out 2 gnulib-s from Makefile and try again.

flipreverse, 2023-06-24:
> Please take out 2 gnulib-s from Makefile and try again.

In which Makefile? The one from gettext-full?

brada4, 2023-06-24:
This https://github.com/openwrt/openwrt/blob/acd9981b4ef750544202df9d9e2d0143a6dfd478/package/libs/gettext-full/Makefile#L37

flipreverse, 2023-06-25:
> This
> 
> https://github.com/openwrt/openwrt/blob/acd9981b4ef750544202df9d9e2d0143a6dfd478/package/libs/gettext-full/Makefile#L37

Nope. That didn't do the trick. I tried both removing `gnulib-lib` as well as `gnulib-lib` and `gnulib-local`. 
I also cleaned the build dir: `make package/gettext-full/host/clean V=s`

brada4, 2023-06-25:
And you get error building gnulib?

flipreverse, 2023-06-25:
> And you get error building gnulib?

Yes. As quoted above, gcc fails compiling `gl_arrayList.c`.

flipreverse, 2023-07-09:
I'll close this issue, because I'm able to  built OpenWRT for my AR750S in a different checkout on the same host.

closed 2023-07-09 by flipreverse