openwrt/openwrt #12970: Enabling 5 GHz (MT76) Soft Bricks the Router [SNAPSHOT]

khanjui, 2023-06-24:
### Describe the bug

Just an hour ago I've upgraded to latest snapshot using asu, my router started bootlooping. I've reseted it via the button on the back, it turned on fine, until I set up and enable 5 GHz.

### OpenWrt version

r23422-e0fb38f4ee

### OpenWrt target/subtarget

ramips/mt7621

### Device

Xiaomi Mi Router 4A Gigabit Edition

### Image kind

Official downloaded image

### Steps to reproduce

Enable 5 GHz (MT76x2E).

### Actual behaviour

Router bootloops.

### Expected behaviour

Should work fine.

### Additional info

2.4 GHz is working fine.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-06-25:
Backup configuration via LuCI for less typing after reset.

Disconnect anything involving cheap USB3.0 cables in proximity
Put up logread in other session
Then in 1st session spacing "apply" buttons one minute apart do:
Via luci set country code AND SAVE (It is crucial for 5GHz operation)
Run wireless survey from Luci/Status (or `iw scan` if on command line)
Set automatic channel and 40MHz bw AND SAVE (80GHz is fine in most areas, but stay conservative for starters)
Configure access point, like with same authentication as 2.4GHz and bring it up

Either it works or logread tells something interesting or it magically works. Even if it works reboot, confirm it works, then unpower and check if it works after.

It should not crash system, but in either case we extract useful info.

khanjui, 2023-06-25:
> Backup configuration via LuCI for less typing after reset.
>
> Disconnect anything involving cheap USB3.0 cables in proximity Put up logread in other session Then in 1st session spacing "apply" buttons [...]

I will try that but it was working fine with a snapshot from 2 days earlier. I guess latest mt76 changes broke something.

brada4, 2023-06-25:
Please find commiter in git log then.

raenye, 2023-06-25:
I'm running r23422 on two DAP-1620 devices (SoC=mt7621, WiFi=mt7615dn) without 5 GHz issues.

badulesia, 2023-06-25:
Hi.
I have described the same behavior here for a R6220.
https://forum.openwrt.org/t/encryption-causes-reboot-with-latest-driver-on-mt76-kernel-5-15-118/163927/6
Issue appears when the 5GHz has encryption.

MrHenry81, 2023-06-25:
The same result for me - Bootloop when 5 GHz with encryption is enabled on SK-WB8 (mt76/ MT7612E) and Cudy X6 (mt76/ MT7915E).

khanjui, 2023-06-25:
@nbd168

nbd168, 2023-06-25:
I reverted the mt76 update for now. It would be really helpful if somebody could use git bisect to figure out which commit in mt76.git broke it. I will look into this again next week. Thanks for pinging me and for reporting this.

edrikk, 2023-06-26:
Just as a data point, I’ve been running the build on the below devices without issue:

Xiaomi Redmi Router AC2100 (2.4GHz MT7603E disabled, 5GHz MT7615E enabled)

Xiaomi Redmi Router AX6S (2.4GHz MT7622  enabled, 5GHz MT7915E enabled).

All configured with WPA2 PSK (CCMP).

One thing to note in case it triggers thought, is that I’m running OpenSSL builds (libopenssl3, libustream-openssl20201210, wpad-openssl) instead of the current default of mbedtls.

raenye, 2023-06-26:
Incidentally, my devices also have openssl instead of embdtls. 
WPA3-SAE mesh and WPA2-PSK AP, both on 5GHz.

badulesia, 2023-06-26:
Hi.
Device R6220
I have just tried a build with openssl, and the result is the same. As soon as the 5GHz is set with encryption (WPA2), the device reboot and bootloop. It was worth trying.

badulesia, 2023-06-26:
This may solve the issue in the next build
https://github.com/openwrt/openwrt/commit/fc08989a3215b542f82d886a71437f059b147d3d

badulesia, 2023-06-26:
The commit is already availalble. I have just installed succesfully OpenWrt SNAPSHOT r23449-186976c963 / LuCI Master git-23.158.78004-23a246e. Wifi 5GHz works again. Thanks to @nbd168 .

cmonroe, 2023-06-26:
Just to add a 3rd data point (in addition to @edrikk and @raenye), I am using hostapd-openssl and have not observed any issues with a handful of radios (MT7615, MT7915D, MT7915E, MT7986) running the latest mt76.

Hostapd .config options:
```
# CONFIG_PACKAGE_hostapd is not set
# CONFIG_PACKAGE_hostapd-basic is not set
# CONFIG_PACKAGE_hostapd-basic-mbedtls is not set
# CONFIG_PACKAGE_hostapd-basic-openssl is not set
# CONFIG_PACKAGE_hostapd-basic-wolfssl is not set
CONFIG_PACKAGE_hostapd-common=y
# CONFIG_PACKAGE_hostapd-mbedtls is not set
# CONFIG_PACKAGE_hostapd-mini is not set
CONFIG_PACKAGE_hostapd-openssl=y
CONFIG_PACKAGE_hostapd-utils=y
# CONFIG_PACKAGE_hostapd-wolfssl is not set
```

And for SSL, I have libopenssl and mbedtls on the target while wolfssl is disabled:
```
CONFIG_PACKAGE_libmbedtls=y
CONFIG_PACKAGE_libopenssl=y
# CONFIG_PACKAGE_libwolfssl is not set
```
 
Lastly, my codebase is slightly out of date @ kernel 5.15.110 w/OpenSSL 3.0.8. I'll rebase to current master again today. Both mt76 and mac80211 are overridden from OpenWrt master (prior to the revert) so mt76-2023-06-22-29cfabbb.

khanjui, 2023-06-26:
I don't think it's related to channel or openssl.

> Hi. Device R6220 I have just tried a build with openssl, and the result is the same. As soon as the 5GHz is set with encryption (WPA2), the device reboot and bootloop. It was worth trying.

Same. I don't think it has anything to do with OpenSSL or channel number.

cmonroe, 2023-06-26:
Since it seems to be related to older devices, perhaps an issue with 40f6e433 ?

ptpt52, 2023-06-27:
@nbd168 
this may help.

crash log catch:
```
<6>[   85.496642] br-lan: port 2(phy1-ap0) entered blocking state
<6>[   85.502419] br-lan: port 2(phy1-ap0) entered forwarding state
<1>[   93.070934] CPU 0 Unable to handle kernel paging request at virtual address 00000060, epc == 82a608e8, ra == 82a608e8
<4>[   93.082007] Oops[#1]:
<4>[   93.084340] CPU: 0 PID: 1967 Comm: hostapd Not tainted 5.15.118 #0
<4>[   93.090657] $ 0   : 00000000 00000001 f405496a f405496a
<4>[   93.096019] $ 4   : 84069200 ffffffff 82c7f990 00000000
<4>[   93.101372] $ 8   : 00000001 afd8691c 23f475a2 64d1c55c
<4>[   93.106726] $12   : 00000010 0000000a fefefefe 00000018
<4>[   93.112078] $16   : 84069200 82b996a0 00000000 00000000
<4>[   93.117431] $20   : 00000001 82a1a4c0 82a1add0 00000000
<4>[   93.122782] $24   : 00000000 82a60800
<4>[   93.128133] $28   : 82c7e000 82c7f9c8 00000000 82a608e8
<4>[   93.133485] Hi    : 00000000
<4>[   93.136421] Lo    : 00000004
<4>[   93.139355] epc   : 82a608e8 mt76x02_set_key+0xe8/0x198 [mt76x02_lib]
<4>[   93.145979] ra    : 82a608e8 mt76x02_set_key+0xe8/0x198 [mt76x02_lib]
<4>[   93.152574] Status: 1100f403      KERNEL EXL IE
<4>[   93.156858] Cause : 40800008 (ExcCode 02)
<4>[   93.160949] BadVA : 00000060 
<4>[   93.163885] PrId  : 00019650 (MIPS 24KEc)
<4>[   93.167974] Modules linked in: rt2800soc rt2800mmio rt2800lib qcserial pppoe ppp_async option cdc_mbim wireguard usb_wwan sierra_net sierra rt2x00soc rt2x00mmio rt2x00lib rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic nft_fib_inet mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake r8152 poly1305_mips nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat
<4>[   93.168597]  nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c ipvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables crc_ccitt compat chacha_mips cdc_wdm br_netfilter natflow natcap nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ntfs3 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact xt_set
<4>[   93.258247]  ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 ksmbd asn1_decoder nfs msdos ip6_gre ip_gre gre ifb ip6_udp_tunnel udp_tunnel sit ip6_tunnel oid_registry tunnel6 tunnel4 ip_tunnel tun lockd sunrpc grace autofs4 dns_resolver nls_utf8 nls_iso8859_1 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 md4 hmac ecb des_generic libdes cmac arc4 uas usb_storage leds_gpio ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd gpio_button_hotplug vfat fat ext4 mbcache jbd2 exfat 
<4>[   93.347422]  usbcore nls_base usb_common mii crc32c_generic
<4>[   93.441629] Process hostapd (pid: 1967, threadinfo=6304fd70, task=f9a2fa72, tls=77e69dfc)
<4>[   93.449988] Stack : 8409c100 8409c100 ffffffff 80250304 84069000 00000000 82a1a4c0 82b431a0
<4>[   93.458557]         82b996a0 84353200 00000010 84100d00 84069200 00000000 84069000 00000000
<4>[   93.467128]         00000000 82b438e4 00000000 000fac04 84353000 802558c4 fff000ff 84353e00
<4>[   93.475699]         000fff00 84353824 00000010 84353e00 84069000 00000000 82a1a4c0 00000000
<4>[   93.484264]         00000000 82a1add0 00000000 80700000 00000003 82b44854 000fac04 80251c20
<4>[   93.492827]         ...
<4>[   93.495336] Call Trace:
<4>[   93.497829] [<82a608e8>] mt76x02_set_key+0xe8/0x198 [mt76x02_lib]
<4>[   93.504094] [<82b431a0>] ieee80211_key_mic_failure+0x4c4/0x1308 [mac80211]
<4>[   93.511426] 
<4>[   93.512944] Code: a0a200d4  0caf0dfe  02003025 <92450060> 8fbf001c  8fb20018  02003025  02202025  8fb00010
<4>[   93.522946] 
<4>[   93.524596] ---[ end trace 69e18666a4040d5c ]---
```

device is xiaomi-miwifi-r3 (mt7620 + mt7612)

DragonBluep, 2023-06-27:
> @nbd168 this may help.
>
> crash log catch:
>
> ```
> <6>[   85.496642] br-lan: port 2(phy1-ap0) entered blocking state
> <6>[   85.502419] br-lan: port 2(phy1-ap0) entered forwarding state
> <1>[   93.070934] [...]

`msta` may be a null pointer here, reverting https://github.com/openwrt/mt76/commit/40f6e433f747b01e7d088762c52c0d92111064df should be able to fix it.
https://github.com/openwrt/mt76/blob/2f851902d6b1284ba7c326a685c96f5ba88dae1d/mt76x02_util.c#L471

ptpt52, 2023-06-27:
@DragonBluep @nbd168 
the issue resolved with reverting the commit https://github.com/openwrt/mt76/commit/40f6e433f747b01e7d088762c52c0d92111064df

danpawlik, 2023-06-28:
@khanjui if new compilation is working, could you close issue? Thanks

closed 2023-06-29 by khanjui