openwrt/openwrt #12976: Fritz!Box 7520 VDSL PPPoE dial-up timeout: NETDEV WATCHDOG transmit queue timed out

eomanis, 2023-06-25:
### Describe the bug

#### Summary

I have two AVM FRITZ!Box 7520 devices that are visually identical, including their label, the latter with exceptions of
 - Default login passphrase
 - Default WPA2 passphrase
 - Serial number

For one of them, VDSL works as expected, for the other one it doesn't.

#### Build and platform information

OpenWrt version: `23.05.0-rc1`
OpenWrt build: `openwrt-23.05.0-rc1-ipq40xx-generic-avm_fritzbox-7530`
Keywords: `vrx518`
Table of hardware: [AVM FRITZ!Box 7520 / 7530](https://openwrt.org/toh/avm/avm_fritz_box_7530)

#### Good device with serial number `L112.719.30.XXX.XXX`:

VDSL works as expected

#### Bad device with serial number `N206.719.30.XXX.XXX`:

VDSL works with stock AVM firmware, yet is not functional in OpenWrt. 
VDSL appears to acquire sync, then the PPPoE login times out with NETDEV WATCHDOG involvement and `ppp` error message "Timeout waiting for PADO packets".
[Relevant forum post here.](https://forum.openwrt.org/t/openwrt-23-05-0-rc1-first-release-candidate/162544/268?u=eomanis)

#### Log files

Attached are two log files, one from the good `L112` device and one from the bad `N206` device, that show part of the initial boot-up and the initial VDSL connection attempt.
[2023-06-05.FB-7520.sn-N206.bad.log](https://github.com/openwrt/openwrt/files/11860235/2023-06-05.FB-7520.sn-N206.bad.log)
[2023-06-05.FB-7520.sn-L112.good.log](https://github.com/openwrt/openwrt/files/11860236/2023-06-05.FB-7520.sn-L112.good.log)

The log files are manually modified to facilitate easy side-by-side comparison.

### OpenWrt version

r23069-e2701e0f33

### OpenWrt target/subtarget

ipq40xx/generic

### Device

AVM FRITZ!Box 7530

### Image kind

Official downloaded image

### Steps to reproduce

- Have an AVM FRITZ!Box 7520 whose version number starts with `N...` (?) and a VDSL connection
- Install OpenWrt `23.05.0-rc1` on it according to the table of hardware: [table of hardware](https://openwrt.org/toh/avm/avm_fritz_box_7530), including the VDSL firmware files
- Configure it so that the required VDSL connection parameters are in place
- Attach the device to the VDSL endpoint and power it on

### Actual behaviour

The device syncs up the VDSL connection, then tries to dial in with PPPoE.
Dialing in times out because `ppp` gets no response.
Additionally, this linux kernel warning message with stack trace is printed to its log. Excerpt:

```
daemon.notice netifd: VLAN 'dsl0.7' link is up
daemon.notice netifd: Interface 'wan' has link connectivity
daemon.notice netifd: Interface 'wan' is setting up now
daemon.info pppd[XXXX]: Plugin pppoe.so loaded.
daemon.info pppd[XXXX]: PPPoE plugin from pppd 2.4.9
daemon.notice pppd[XXXX]: pppd 2.4.9 started by root, uid 0
kern.warn kernel: [   XX.XXXXXX] ------------[ cut here ]------------
kern.warn kernel: [   XX.XXXXXX] WARNING: CPU: 2 PID: 0 at net/sched/sch_generic.c:477 0xc08336fc
kern.info kernel: [   XX.XXXXXX] NETDEV WATCHDOG: dsl0 (vrx518): transmit queue 0 timed out
kern.warn kernel: [   XX.XXXXXX] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet ath10k_pci ath10k_core ath pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211 iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables slhc sch_cake nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat clip sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact drv_dsl_cpe_api drv_mei_cpe ifb br2684 vrx518_tc
kern.warn kernel: [   XX.XXXXXX]  atm vrx518 drv_ifxos sha512_arm md5 ghash_arm_ce cmac leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom gpio_button_hotplug crc32c_generic
kern.warn kernel: [   XX.XXXXXX] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 5.15.114 #0
kern.warn kernel: [   XX.XXXXXX] Hardware name: Generic DT based system
kern.warn kernel: [   XX.XXXXXX] Function entered at [<c030d3a0>] from [<c0309784>]
kern.warn kernel: [   XX.XXXXXX] Function entered at [<c0309784>] from [<c060c020>]
kern.warn kernel: [   XX.XXXXXX] Function entered at [<c060c020>] from [<c032266c>]
kern.warn kernel: [   XX.XXXXXX] Function entered at [<c032266c>] from [<c0322748>]
kern.warn kernel: [   XX.XXXXXX] Function entered at [<c0322748>] from [<c08336fc>]
kern.warn kernel: [   XX.XXXXXX] Function entered at [<c08336fc>] from [<c038ab08>]
kern.warn kernel: [   XX.XXXXXX] Function entered at [<c038ab08>] from [<c038b228>]
kern.warn kernel: [   XX.XXXXXX] Function entered at [<c038b228>] from [<c038b2cc>]
kern.warn kernel: [   XX.XXXXXX] Function entered at [<c038b2cc>] from [<c03012b4>]
kern.warn kernel: [   XX.XXXXXX] Function entered at [<c03012b4>] from [<c03271a8>]
kern.warn kernel: [   XX.XXXXXX] Function entered at [<c03271a8>] from [<c03703ec>]
kern.warn kernel: [   XX.XXXXXX] Function entered at [<c03703ec>] from [<c0624490>]
kern.warn kernel: [   XX.XXXXXX] Function entered at [<c0624490>] from [<c0300b3c>]
kern.warn kernel: [   XX.XXXXXX] Exception stack(0xc107bf58 to 0xc107bfa0)
kern.warn kernel: [   XX.XXXXXX] bf40:                                                       0000b63a 00000000
kern.warn kernel: [   XX.XXXXXX] bf60: 00000001 c0312e60 00000002 c0c04f28 c107a000 00000000 00000000 ffffe000
kern.warn kernel: [   XX.XXXXXX] bf80: 00000000 c0c04f5c 00000000 c107bfa8 c0306fdc c0306fe0 60000013 ffffffff
kern.warn kernel: [   XX.XXXXXX] Function entered at [<c0300b3c>] from [<c0306fe0>]
kern.warn kernel: [   XX.XXXXXX] Function entered at [<c0306fe0>] from [<c0351744>]
kern.warn kernel: [   XX.XXXXXX] Function entered at [<c0351744>] from [<c0351aa8>]
kern.warn kernel: [   XX.XXXXXX] Function entered at [<c0351aa8>] from [<80301510>]
kern.warn kernel: [   XX.XXXXXX] ---[ end trace 31539cc41415c45b ]---
daemon.warn pppd[XXXX]: Timeout waiting for PADO packets
daemon.err pppd[XXXX]: Unable to complete PPPoE Discovery
daemon.info pppd[XXXX]: Exit.
daemon.notice netifd: Interface 'wan' is now down
```

### Expected behaviour

The device syncs up the VDSL connection, then tries to dial in with PPPoE.
Dial-in succeeds; log excerpt:

```
daemon.notice netifd: VLAN 'dsl0.7' link is up
daemon.notice netifd: Interface 'wan' has link connectivity
daemon.notice netifd: Interface 'wan' is setting up now
kern.info kernel: [   XX.XXXXXX] IPv6: ADDRCONF(NETDEV_CHANGE): dsl0.7: link becomes ready
daemon.info pppd[XXXX]: Plugin pppoe.so loaded.
daemon.info pppd[XXXX]: PPPoE plugin from pppd 2.4.9
daemon.notice pppd[XXXX]: pppd 2.4.9 started by root, uid 0
daemon.warn odhcpd[XXXX]: No default route present, overriding ra_lifetime!
daemon.info pppd[XXXX]: PPP session is 149
daemon.warn pppd[XXXX]: Connected to XX:XX:XX:XX:XX:XX via interface dsl0.7
kern.info kernel: [   XX.XXXXXX] pppoe-wan: renamed from ppp0
daemon.info pppd[XXXX]: Renamed interface ppp0 to pppoe-wan
daemon.info pppd[XXXX]: Using interface pppoe-wan
daemon.notice pppd[XXXX]: Connect: pppoe-wan <--> dsl0.7
daemon.info pppd[XXXX]: Remote message: [XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX]
daemon.notice pppd[XXXX]: PAP authentication succeeded
```

### Additional info

Nothing comes to mind.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Djfe, 2023-07-01:
just a guess, this could be related to
https://patchwork.ozlabs.org/project/openwrt/patch/20230130224020.473703-1-jan@3e8.eu/
Still not having been merged.
If you know how to compile openwrt yourself, then you could test this patch :)
else you'll have to wait

eomanis, 2023-07-01:
> just a guess, this could be related to https://patchwork.ozlabs.org/project/openwrt/patch/20230130224020.473703-1-jan@3e8.eu/

Oh, ew, what a shit show of a bugged proprietary firmware or ROM that needs to be tricked into working in such an ugly way.

Well, at least we have that workaround, I guess we can thank the GPL for that.

Djfe, 2023-07-01:
well it's the kind of firmware that has been printed into silicon and cannot be changed by an update.

analogy:
you could throw the faulty piece of hardware away, or you dunk the light bulb into red paint and ignore that this stupid hack made it work all of a sudden :D

If the hw is fine but the identification is broken, why not ignore it for one board and still load the correct firmware to run on it.
Saves lots of money in the long run, because silicon bugs just happen and it prevents waste

eomanis, 2023-07-02:
So uh, the best time to merge that patch and backport it to the OpenWrt 23.05 branch was before doing release candidates. We missed that, but the second-best time to do it is now, before there is the first stable release.

Who do I bother about getting it merged, and how?

It *is* a really popular device in Germany, and VDSL is very much the most prevalent broadband technology here as of 2023, because we suck at proper Internet. 
Getting these things to work reliably will likely be a quality-of-life improvement for a substantial number of OpenWrt users.

Djfe, 2023-07-02:
I messaged hauke about it on irc. I bet this will get merged before release.
regardless: irc is a good way to get hold of devs (openwrt-devel)
I'm gonna try getting people's attention

octomike, 2023-10-23:
> I bet this will get merged before release.

@Djfe So, do you know if this was successful in a different way? Because 23.05 is finally here and I'm not sure the patch has been merged..

Djfe, 2023-10-23:
as you can see: it still says _new_ on patchwork
https://patchwork.ozlabs.org/project/openwrt/patch/20230130224020.473703-1-jan@3e8.eu/

sadly Noone seemed to care/nothing has changed. the patch wasn't picked up for release like I had hoped.

octomike, 2023-10-23:
I still had small hopes that a different approach was merged, but I guess it was wishful thinking.

basarbulut, 2023-11-17:
> ### Describe the bug
> #### Summary
> I have two AVM FRITZ!Box 7520 devices that are visually almost identical, including their type label, the latter with exceptions of
> 
> * Default login passphrase
> * Default WPA2 passphrase
> * Serial number
> 
> For one of them, VDSL works as expected, for the other one it doesn't. Both devices have article number `2000 2882`.
> 
> **EDIT** Just noticed: The print on the top of the devices differs.
> 
> * Good `L112` device: "FRITZ!Box 1&1 HomeServer"
> * Bad `N206` device:" FRITZ!Box 1&1 WLAN-Router"
> 
> To avoid confusion I have attached pictures at the bottom.
> 
> #### Build and platform information
> OpenWrt version: `23.05.0-rc1` OpenWrt build: `openwrt-23.05.0-rc1-ipq40xx-generic-avm_fritzbox-7530` Keywords: `vrx518` Table of hardware: [AVM FRITZ!Box 7520 / 7530](https://openwrt.org/toh/avm/avm_fritz_box_7530)
> 
> #### Good device with serial number `L112.719.30.XXX.XXX`:
> VDSL works as expected
> 
> #### Bad device with serial number `N206.719.30.XXX.XXX`:
> VDSL works with stock AVM firmware, yet is not functional in OpenWrt. VDSL appears to acquire sync, then the PPPoE login times out with NETDEV WATCHDOG involvement and `ppp` error message "Timeout waiting for PADO packets". [Relevant forum post here.](https://forum.openwrt.org/t/openwrt-23-05-0-rc1-first-release-candidate/162544/268?u=eomanis)
> 
> #### Log files
> Attached are two log files, one from the good `L112` device and one from the bad `N206` device, that show part of the initial boot-up and the initial VDSL connection attempt. [2023-06-05.FB-7520.sn-N206.bad.log](https://github.com/openwrt/openwrt/files/11860235/2023-06-05.FB-7520.sn-N206.bad.log) [2023-06-05.FB-7520.sn-L112.good.log](https://github.com/openwrt/openwrt/files/11860236/2023-06-05.FB-7520.sn-L112.good.log)
> 
> The log files are manually modified to facilitate easy side-by-side comparison.
> 
> ### OpenWrt version
> r23069-e2701e0f33
> 
> ### OpenWrt target/subtarget
> ipq40xx/generic
> 
> ### Device
> AVM FRITZ!Box 7530
> 
> ### Image kind
> Official downloaded image
> 
> ### Steps to reproduce
> * Have an AVM FRITZ!Box 7520 whose version number starts with `N...` (?) and a VDSL connection
> * Install OpenWrt `23.05.0-rc1` on it according to the [table of hardware](https://openwrt.org/toh/avm/avm_fritz_box_7530), including the VDSL firmware files
> * Configure it so that the required VDSL connection parameters are in place
> * Attach the device to the VDSL endpoint and power it on
> 
> ### Actual behaviour
> The device syncs up the VDSL connection, then tries to dial in with PPPoE. Dialing in times out because `ppp` gets no response. Additionally, this linux kernel warning message with stack trace is printed to its log. Excerpt:
> 
> ```
> daemon.notice netifd: VLAN 'dsl0.7' link is up
> daemon.notice netifd: Interface 'wan' has link connectivity
> daemon.notice netifd: Interface 'wan' is setting up now
> daemon.info pppd[XXXX]: Plugin pppoe.so loaded.
> daemon.info pppd[XXXX]: PPPoE plugin from pppd 2.4.9
> daemon.notice pppd[XXXX]: pppd 2.4.9 started by root, uid 0
> kern.warn kernel: [   XX.XXXXXX] ------------[ cut here ]------------
> kern.warn kernel: [   XX.XXXXXX] WARNING: CPU: 2 PID: 0 at net/sched/sch_generic.c:477 0xc08336fc
> kern.info kernel: [   XX.XXXXXX] NETDEV WATCHDOG: dsl0 (vrx518): transmit queue 0 timed out
> kern.warn kernel: [   XX.XXXXXX] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet ath10k_pci ath10k_core ath pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211 iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables slhc sch_cake nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c hwmon crc_ccitt compat clip sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact drv_dsl_cpe_api drv_mei_cpe ifb br2684 vrx518_tc
> kern.warn kernel: [   XX.XXXXXX]  atm vrx518 drv_ifxos sha512_arm md5 ghash_arm_ce cmac leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom gpio_button_hotplug crc32c_generic
> kern.warn kernel: [   XX.XXXXXX] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 5.15.114 #0
> kern.warn kernel: [   XX.XXXXXX] Hardware name: Generic DT based system
> kern.warn kernel: [   XX.XXXXXX] Function entered at [<c030d3a0>] from [<c0309784>]
> kern.warn kernel: [   XX.XXXXXX] Function entered at [<c0309784>] from [<c060c020>]
> kern.warn kernel: [   XX.XXXXXX] Function entered at [<c060c020>] from [<c032266c>]
> kern.warn kernel: [   XX.XXXXXX] Function entered at [<c032266c>] from [<c0322748>]
[... 36 lines cut ...]
> daemon.info pppd[XXXX]: PPP session is 149
> daemon.warn pppd[XXXX]: Connected to XX:XX:XX:XX:XX:XX via interface dsl0.7
> kern.info kernel: [   XX.XXXXXX] pppoe-wan: renamed from ppp0
> daemon.info pppd[XXXX]: Renamed interface ppp0 to pppoe-wan
> daemon.info pppd[XXXX]: Using interface pppoe-wan
> daemon.notice pppd[XXXX]: Connect: pppoe-wan <--> dsl0.7
> daemon.info pppd[XXXX]: Remote message: [XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX]
> daemon.notice pppd[XXXX]: PAP authentication succeeded
> ```
> 
> ### Additional info
> Nothing comes to mind.
> 
> ### Diffconfig
> _No response_
> 
> ### Terms
> * [x]  I am reporting an issue for OpenWrt, not an unsupported fork.
> 
> #### Pictures
> Good `L112` device: [image]
> 
> Bad `N206` device: [image]

Hi. My FB 7520 1&1 WLAN router and serial begins with N151. It runs OpenWrt 23.05.2 r2363 and VDSL is working perfect.

maurerle, 2025-01-20:
fixed with #17622 - should be closed.

eomanis, 2025-01-21:
I intend to at least test 24.10-rc6 that is supposed to get the fix for this via backport on my not-affected FRITZ!Box 7520, and if that works I'll close this issue.