openwrt/openwrt #12978: hostapd ignoring log verbosity configuration and spamming syslog

chrcoluk, 2023-06-25:
### Describe the bug

system log is flooded with informational level log entries from hostapd, I currently have the openwrt logging configured to warnings.

In addition I also added the following to the wireless config for radio0 which the spam is coming from.

`        option log_level '4'
        option log_wpa 0
        option log_driver 0
        option log_80211 0
        option log_8021x 0
        option log_radius 0
        option log_iapp 0
        option log_mlme 0`

I confirmed live configuration has generated this config.

`# cat /tmp/run/hostapd-wl0.conf
driver=nl80211
logger_syslog=0
logger_syslog_level=4
logger_stdout=0
logger_stdout_level=4`

But still no effect on the log spam.

### OpenWrt version

r23375-cdfcac6e24

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Netgear WAX206

### Image kind

Official downloaded image

### Steps to reproduce

Install OpenWRT, enable 2.4ghz radio, connect Oneplus6 phone running OOS9 (android 9) to the wifi.

WPA2-PSK security, all other options dont matter.

### Actual behaviour

system log is flooded with informational level log entries from hostapd.

example below.

`Sun Jun 25 21:13:02 2023 daemon.notice hostapd: wl0-ap0: STA-OPMODE-SMPS-MODE-CHANGED 64:a2:f9:4c:f4:4c off
Sun Jun 25 21:13:03 2023 daemon.notice hostapd: wl0-ap0: STA-OPMODE-SMPS-MODE-CHANGED 64:a2:f9:4c:f4:4c static
Sun Jun 25 21:13:03 2023 daemon.notice hostapd: wl0-ap0: STA-OPMODE-SMPS-MODE-CHANGED 64:a2:f9:4c:f4:4c off
Sun Jun 25 21:13:04 2023 daemon.notice hostapd: wl0-ap0: STA-OPMODE-SMPS-MODE-CHANGED 64:a2:f9:4c:f4:4c static`

### Expected behaviour

Expect the configured log level to be honoured.  No informational logging.

### Additional info

Image was downloaded from OpenWRT website, but was built using these packages built in on the firmware selector page.

'luci luci-app-attendedsysupgrade bash busybox ca-certificates nano ethtool'

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

geeksville, 2025-07-17:
FWIW - just confirming this bug still exists.

geeksville, 2025-07-17:
I think this can be closed as a dup of #26280.