openwrt/openwrt #12981: ath9k: qca956x distance No effect

zhangsd20, 2023-06-26:
### Describe the bug

I tested the QCA956X and found that setting Distance didn't actually work.
(The same method tests QCA9531/AR9344 to set the distance has obvious results.)

I tried reading the SLOT/ACK/CTS register after setting the distance and found that the register value changed as the distance changed, but it didn't work in the long distance test (signal:-60dBm).

How to solve this problem?

### OpenWrt version

22.03/19.07/18.06

### OpenWrt target/subtarget

ar71xx

### Device

Qxwlan e1700ac

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-06-26:
Test with v22 and report supported version tags.

zhangsd20, 2023-06-27:
Is there a fix for v22 for QCA956X Distance? 
If it's just an update iteration, such testing should be meaningless.

brada4, 2023-06-27:
distance attribute can be set via iw.? 
Please confirm that latest version does not work as intended.

zhangsd20, 2023-06-27:
The distance parameter is set in /etc/config/wireless (Ultimately, of course, is through the IW setup) 
option distance '10000'

AR9344/QCA9531 works fine,Only the QCA956X works abnormally, and it seems to have no effect.

I tested several versions with the same results, and I speculate that it should have nothing to do with the version, and the most recent version should have the same problem

brada4, 2023-06-27:
Should not be a problem to produce precise versipn/hardware names editing op.

closed 2023-07-13 by zhangsd20