openwrt/openwrt #12991: Any wpad other than basic prevents wifi authentication in client mode

bassmannate, 2023-06-27:
### Describe the bug

Removing wpad-basic-mbedtls and installing any other wpad package more than basic (wpad-mesh-openssl, wpad-openssl, wpad-mesh-wolfssl, wpad-wolfssl, wpad-mesh-mbedtls, etc.) appears to prevent wireless authentication in client mode on 2.4 and 5ghz bands.

Wifi can scan and select the ssid but will not connect and behaves like it does when you don't have the correct encryption key.

### OpenWrt version

23.05-RC1

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Linksys MR8300 (Dallas)

### Image kind

Official downloaded image

### Steps to reproduce

From fresh install, remove wpad-basic-mbedtls and install wpad-mbedtls or any other package listed above. Reboot. Connect to wifi with wpa2 authentication.

### Actual behaviour

client mode will no longer authenticate and finish connecting. It does not matter if the client was connected before wpad package was changed before or after initially connecting.

### Expected behaviour

It is expected that the wifi client mode would connect and authenticate, allowing connectivity to the host network.

### Additional info

I have not tested with this in master mode to see if clients can wirelessly connect with the full version of wpad.

Some info isn't available as I've rolled back to 22.03 for functionality. I can switch back over to grab a few things if needed since this router is a dual firmware model.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-06-29:
Please clarify - does it hang when enabling master and station mode both at once or jjst station is enough?

bassmannate, 2023-06-29:
Here's what I'm getting on my wireless. The client has WPA2 encryption. The master on the second 5G radio has no encryption so it looks like I was wrong. Wifi will not come completely up with anything other than basic wpad. In this particular instance, I have wpad-mesh-wolfssl installed. Also, the client is not automatically changing to the correct channel like it normally does. That access point is actually on channel 120, I believe. Even manually changing it to match does not work.
[image]

closed 2023-06-29 by bassmannate

brada4, 2023-06-29:
Thanks, i was just curious aboit setup sice i have similar extender setup working like now.