openwrt/openwrt #12995: PPPoE disconnected by itself

Karmylr, 2023-06-28:
### Describe the bug

After running for a few days, it will suddenly disconnect from the network. Found that PPPoE is disconnected

### OpenWrt version

r23227-106c83a1ea

### OpenWrt target/subtarget

x86/64

### Device

J4125,VMware ESXi

### Image kind

Self-built image

### Steps to reproduce

Create a wan interface and set up PPPoE

### Actual behaviour

disconnects itself occasionally

### Expected behaviour

no longer disconnect

### Additional info

logs:
```
Tue Jun 20 10:20:52 2023 daemon.info pppd[1845]: No response to 5 echo-requests
Tue Jun 20 10:20:52 2023 daemon.notice pppd[1845]: Serial link appears to be disconnected.
Tue Jun 20 10:20:52 2023 daemon.info pppd[1845]: Connect time 11091.4 minutes.
Tue Jun 20 10:20:52 2023 daemon.info pppd[1845]: Sent 3644444638 bytes, received 331321140 bytes.
Tue Jun 20 10:20:52 2023 daemon.err odhcp6c[3577]: Failed to send RS (Network unreachable)
Tue Jun 20 10:20:52 2023 daemon.notice netifd: Network device 'pppoe-wan' link is down
Tue Jun 20 10:20:52 2023 daemon.notice netifd: Network alias 'pppoe-wan' link is down
Tue Jun 20 10:20:52 2023 daemon.err odhcp6c[3577]: Failed to send RELEASE message to ff02::1:2 (Network unreachable)
Tue Jun 20 10:20:52 2023 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Tue Jun 20 10:20:52 2023 daemon.notice netifd: Interface 'wan' has lost the connection
Tue Jun 20 10:20:52 2023 daemon.notice netifd: Interface 'wan_6' is disabled
Tue Jun 20 10:20:52 2023 daemon.notice netifd: wan_6 (3577): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan_6" } (Permission denied)
Tue Jun 20 10:20:52 2023 daemon.err odhcp6c[3577]: Failed to send SOLICIT message to ff02::1:2 (Network unreachable)
Tue Jun 20 10:20:52 2023 daemon.notice netifd: Interface 'wan_6' is now down
Tue Jun 20 10:20:52 2023 daemon.info dnsmasq[1]: read /etc/hosts - 13 names
Tue Jun 20 10:20:52 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Tue Jun 20 10:20:52 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 names
Tue Jun 20 10:20:52 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Tue Jun 20 10:20:55 2023 daemon.notice pppd[1845]: Connection terminated.
Tue Jun 20 10:20:55 2023 daemon.info pppd[1845]: Connect time 11091.4 minutes.
Tue Jun 20 10:20:55 2023 daemon.info pppd[1845]: Sent 3644444638 bytes, received 331321140 bytes.
Tue Jun 20 10:20:55 2023 daemon.info pppd[1845]: Exit.
Tue Jun 20 10:20:55 2023 daemon.notice netifd: Interface 'wan' is now down
Tue Jun 20 10:20:55 2023 daemon.notice netifd: Interface 'wan' is disabled
Tue Jun 20 10:20:55 2023 daemon.notice netifd: Interface 'wan' is enabled
Tue Jun 20 10:20:55 2023 daemon.notice netifd: Interface 'wan' is setting up now
Tue Jun 20 10:20:55 2023 kern.info kernel: [665431.758212] vmxnet3 0000:13:00.0 eth1: intr type 3, mode 0, 5 vectors allocated
Tue Jun 20 10:20:55 2023 kern.info kernel: [665431.759064] vmxnet3 0000:13:00.0 eth1: NIC Link is Up 10000 Mbps
Tue Jun 20 10:20:55 2023 kern.info kernel: [665431.759308] 8021q: adding VLAN 0 to HW filter on device eth1
Tue Jun 20 10:20:55 2023 daemon.info pppd[15375]: Plugin pppoe.so loaded.
Tue Jun 20 10:20:55 2023 daemon.info pppd[15375]: PPPoE plugin from pppd 2.4.9
Tue Jun 20 10:20:55 2023 daemon.notice pppd[15375]: pppd 2.4.9 started by root, uid 0
Tue Jun 20 10:20:55 2023 daemon.info pppd[15375]: PPP session is 18779
Tue Jun 20 10:20:55 2023 daemon.warn pppd[15375]: Connected to cc:cc:81:eb:7c:a7 via interface eth1
[... 334 lines cut ...]
Tue Jun 20 21:28:38 2023 daemon.notice pppd[23494]: Connection terminated.
Tue Jun 20 21:28:38 2023 daemon.info pppd[23494]: Exit.
Tue Jun 20 21:28:38 2023 daemon.notice netifd: Interface 'wan' is now down
Tue Jun 20 21:28:38 2023 daemon.notice netifd: Interface 'wan' is disabled
Tue Jun 20 21:28:38 2023 daemon.notice netifd: Interface 'wan' is enabled
Tue Jun 20 21:28:38 2023 daemon.notice netifd: Interface 'wan' is setting up now
Tue Jun 20 21:28:38 2023 kern.info kernel: [705491.679018] vmxnet3 0000:13:00.0 eth1: intr type 3, mode 0, 5 vectors allocated
Tue Jun 20 21:28:38 2023 kern.info kernel: [705491.679585] vmxnet3 0000:13:00.0 eth1: NIC Link is Up 10000 Mbps
Tue Jun 20 21:28:38 2023 kern.info kernel: [705491.679799] 8021q: adding VLAN 0 to HW filter on device eth1
Tue Jun 20 21:28:38 2023 daemon.info pppd[23641]: Plugin pppoe.so loaded.
Tue Jun 20 21:28:38 2023 daemon.info pppd[23641]: PPPoE plugin from pppd 2.4.9
Tue Jun 20 21:28:38 2023 daemon.notice pppd[23641]: pppd 2.4.9 started by root, uid 0
Tue Jun 20 21:28:38 2023 daemon.info pppd[23641]: PPP session is 29619
Tue Jun 20 21:28:38 2023 daemon.warn pppd[23641]: Connected to cc:cc:81:eb:7c:a7 via interface eth1
Tue Jun 20 21:28:38 2023 kern.info kernel: [705491.881462] pppoe-wan: renamed from ppp0
Tue Jun 20 21:28:39 2023 daemon.info pppd[23641]: Renamed interface ppp0 to pppoe-wan
Tue Jun 20 21:28:39 2023 daemon.info pppd[23641]: Using interface pppoe-wan
Tue Jun 20 21:28:39 2023 daemon.notice pppd[23641]: Connect: pppoe-wan <--> eth1
Tue Jun 20 21:28:39 2023 kern.info kernel: [705492.741404] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
Tue Jun 20 21:28:56 2023 daemon.err uhttpd[1880]: [info] luci: accepted login on / for root from 192.168.1.105
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mrkiko, 2023-06-28:
Hi!
Thanks for your report.
Would you mind adding some more details allowing to clarify why you think this problem is relatred to OpenWrt and not other factors.
For example - will reverting back to an older version fix this issue?
If so, you might "git bisect".

Thanks.

Karmylr, 2023-06-28:
I found that after the network was disconnected immediately with millet ac2100 PPPoE, can successfully dial

brada4, 2023-06-29:
There is ubus notify permission denied. Normally it should reconnect on LCP ping detecting other end down like on 1st line of log.

closed 2023-11-24 by Karmylr