openwrt/openwrt #13003: mt76: Device in STA mode only negotiates basic 6mbit/s speed

VA1DER, 2023-06-29:
### Describe the bug

An mt7915 or mt7986/mt7975 WiFi in STA mode willonly ever negotiate the basic 6mbit/s 20MHz connection, regardless of distance, signal strength, or signal-to-noise ratio:

[image]

Sometimes a proper RX bitrate will be negotiated after some time spent at 6mbit/s.  This depends on the remote AP connected to.  However the TX rate never does negotiate to a "normal" rate, and the RX rate will only go to something expected after a long delay.

The breaking change occurs sometime after 22257-166ab6f90e (which works).  The first build which demonstrates the problem is r23401-a860e439ed, approximately a week ago.  I don't update my STA machine often, so I can't narrow it down any further.

When booting up an identical configuration under 22257-166ab6f90e, expected connection rates are achieved immediately.  The following screen shot was taken about two minutes after the above in the exact same device after swapping sd cards:
[image]

[Today's large mt76 commit](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=01885bc6a33dbfa6f3c9e97778fd8f4f60e2514f) was tested to see if it resolved the issue and it did not.

Problem occurs on:
- discreet mt7915 (AsiaRF AWD7915-NP1 mPCIe card in a BPI-R64 device)
- SoC mt7986 (with mt7975p AFE) on BPI-R3

### OpenWrt version

r23453-5af7d47cd7

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Bananapi BPI-R64

### Image kind

Self-built image

### Steps to reproduce

1. Connect as STA in to an AP in HE80 or similar AX mode under r23401-a860e439ed
2. Observe connection rate

### Actual behaviour

Connection rate remains 6mbit/s

### Expected behaviour

A connection rate commensurate with the signal strength and signal to noise ratio.

### Diffconfig

```text
# On BPI-R64 device:
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_mt7622=y
CONFIG_TARGET_mediatek_mt7622_DEVICE_bananapi_bpi-r64=y
CONFIG_ALL=y
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_BUILDBOT=y
CONFIG_BUILD_PATENTED=y
CONFIG_COLLECT_KERNEL_DEBUG=y
CONFIG_GST1_LIBAV_DECODER_aac=y
CONFIG_GST1_LIBAV_DECODER_atrac3=y
CONFIG_GST1_LIBAV_DECODER_h264=y
CONFIG_GST1_LIBAV_DECODER_mpeg4=y
CONFIG_GST1_LIBAV_DECODER_mpeg4aac=y
CONFIG_GST1_LIBAV_DECODER_wmav1=y
CONFIG_GST1_LIBAV_DECODER_wmav2=y
CONFIG_GST1_LIBAV_DEMUXER_h264=y
CONFIG_GST1_LIBAV_PARSER_aac=y
CONFIG_GST1_LIBAV_PARSER_h264=y
CONFIG_GST1_LIBAV_PATENTED=y
CONFIG_IB=y
CONFIG_INCLUDE_CONFIG=y
CONFIG_KERNEL_BUILD_DOMAIN="buildhost"
CONFIG_KERNEL_BUILD_USER="builder"
CONFIG_LIBDVBCSA_NEON=y
CONFIG_MAKE_TOOLCHAIN=y
CONFIG_PACKAGE_attendedsysupgrade-common=y
CONFIG_PACKAGE_blkid=y
CONFIG_PACKAGE_cgi-io=y
CONFIG_PACKAGE_e2fsprogs=m
CONFIG_PACKAGE_f2fs-tools=y
CONFIG_PACKAGE_fdisk=y
# CONFIG_PACKAGE_gpiod-tools is not set
CONFIG_PACKAGE_kmod-ata-ahci-mtk=m
CONFIG_PACKAGE_kmod-ata-core=m
CONFIG_PACKAGE_kmod-bluetooth=m
CONFIG_PACKAGE_kmod-btmtkuart=m
CONFIG_PACKAGE_kmod-crypto-crc32=y
CONFIG_PACKAGE_kmod-crypto-ecb=m
CONFIG_PACKAGE_kmod-crypto-ecdh=m
CONFIG_PACKAGE_kmod-crypto-kpp=m
CONFIG_PACKAGE_kmod-fs-ext4=y
CONFIG_PACKAGE_kmod-fs-f2fs=y
CONFIG_PACKAGE_kmod-fs-ntfs3=y
CONFIG_PACKAGE_kmod-hid=m
CONFIG_PACKAGE_kmod-input-core=m
CONFIG_PACKAGE_kmod-input-evdev=m
CONFIG_PACKAGE_kmod-mt7615-firmware=y
CONFIG_PACKAGE_kmod-mt7915-firmware=y
CONFIG_PACKAGE_kmod-mt7915e=y
CONFIG_PACKAGE_kmod-mt7916-firmware=y
CONFIG_PACKAGE_kmod-regmap-core=m
CONFIG_PACKAGE_kmod-thermal=y
CONFIG_PACKAGE_kmod-usb-storage=y
CONFIG_PACKAGE_kmod-usb-storage-uas=y
CONFIG_PACKAGE_libcomerr=m
CONFIG_PACKAGE_libdvbcsa=m
CONFIG_PACKAGE_libext2fs=m
CONFIG_PACKAGE_libfdisk=y
# CONFIG_PACKAGE_libgpiod is not set
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libmount=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libparted=y
CONFIG_PACKAGE_libreadline=y
CONFIG_PACKAGE_libsmartcols=y
CONFIG_PACKAGE_libss=m
CONFIG_PACKAGE_libwolfsslcpu-crypto=y
CONFIG_PACKAGE_libx264=m
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-attendedsysupgrade=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-ssl=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_mount-utils=y
CONFIG_PACKAGE_mt7622bt-firmware=m
# CONFIG_PACKAGE_openocd is not set
CONFIG_PACKAGE_parted=y
CONFIG_PACKAGE_px5g-mbedtls=y
# CONFIG_PACKAGE_python3-awscli is not set
# CONFIG_PACKAGE_python3-boto3 is not set
# CONFIG_PACKAGE_python3-botocore is not set
# CONFIG_PACKAGE_python3-docutils is not set
# CONFIG_PACKAGE_python3-gpiod is not set
# CONFIG_PACKAGE_python3-numpy is not set
# CONFIG_PACKAGE_python3-s3transfer is not set
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rpcsys=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-basic-wolfssl=y
CONFIG_REPRODUCIBLE_DEBUG_INFO=y
CONFIG_SDK=y
CONFIG_STRIP_ARGS="--strip-all"
CONFIG_TARGET_ROOTFS_PARTSIZE=2048
# CONFIG_USE_SSTRIP is not set
CONFIG_USE_STRIP=y
CONFIG_WOLFSSL_HAS_ECC448=y
# CONFIG_BPF_TOOLCHAIN_BUILD_LLVM is not set
CONFIG_BPF_TOOLCHAIN_PREBUILT=y
CONFIG_USE_LLVM_PREBUILT=y

# On BPI-R3 device:
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r3=y
CONFIG_ALL=y
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_BUILDBOT=y
CONFIG_COLLECT_KERNEL_DEBUG=y
CONFIG_IB=y
CONFIG_INCLUDE_CONFIG=y
CONFIG_KERNEL_BUILD_DOMAIN="buildhost"
CONFIG_KERNEL_BUILD_USER="builder"
CONFIG_MAKE_TOOLCHAIN=y
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
CONFIG_PACKAGE_cgi-io=y
# CONFIG_PACKAGE_cjdns is not set
# CONFIG_PACKAGE_cjdns-tests is not set
# CONFIG_PACKAGE_cloudflared is not set
CONFIG_PACKAGE_f2fs-tools=y
CONFIG_PACKAGE_fdisk=y
# CONFIG_PACKAGE_gpiod-tools is not set
# CONFIG_PACKAGE_ipfs-http-client-tests is not set
CONFIG_PACKAGE_joe=y
CONFIG_PACKAGE_joe-extras=y
CONFIG_PACKAGE_kmod-fs-ext4=y
CONFIG_PACKAGE_kmod-lib-crc16=y
CONFIG_PACKAGE_kmod-scsi-core=y
CONFIG_PACKAGE_kmod-usb-storage=y
CONFIG_PACKAGE_kmod-usb-storage-uas=y
CONFIG_PACKAGE_libfdisk=y
# CONFIG_PACKAGE_libgpiod is not set
# CONFIG_PACKAGE_libipfs-http-client is not set
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_liblucihttp=y
CONFIG_PACKAGE_liblucihttp-lua=y
CONFIG_PACKAGE_liblucihttp-ucode=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libsmartcols=y
CONFIG_PACKAGE_libwolfsslcpu-crypto=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
# CONFIG_PACKAGE_luci-app-cjdns is not set
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-light=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-ssl=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
# CONFIG_PACKAGE_ooniprobe is not set
# CONFIG_PACKAGE_openocd is not set
CONFIG_PACKAGE_px5g-mbedtls=y
# CONFIG_PACKAGE_python3-gpiod is not set
# CONFIG_PACKAGE_python3-numpy is not set
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-file=y
CONFIG_PACKAGE_rpcd-mod-iwinfo=y
CONFIG_PACKAGE_rpcd-mod-luci=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_rpcd-mod-ucode=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_ucode-mod-html=y
CONFIG_PACKAGE_ucode-mod-math=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_wpad-basic-mbedtls=m
CONFIG_PACKAGE_wpad-basic-wolfssl=y
CONFIG_REPRODUCIBLE_DEBUG_INFO=y
CONFIG_SDK=y
CONFIG_STRIP_ARGS="--strip-all"
CONFIG_TARGET_ROOTFS_PARTSIZE=2048
# CONFIG_USE_SSTRIP is not set
CONFIG_USE_STRIP=y
# CONFIG_BPF_TOOLCHAIN_BUILD_LLVM is not set
CONFIG_BPF_TOOLCHAIN_PREBUILT=y
CONFIG_USE_LLVM_PREBUILT=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2023-06-29:
same issue?
https://github.com/openwrt/mt76/issues/792

VA1DER, 2023-06-29:
> [openwrt/mt76#792](https://github.com/openwrt/mt76/issues/792)

Probably the same issue, but it's hard to know what "broken" means in that report.

The same underlying problem is also likely causing #12905, and these forum reports seem the same:
1. https://forum.openwrt.org/t/wireless-client-mode-broken-on-wax206-post-r22866/162406
2. https://forum.openwrt.org/t/mt7621-wds-with-channel-width-20-mhz-and-ax-not-working-in-newer-snapshots/161942
3. https://forum.openwrt.org/t/latest-snapshot-degrades-mt7915/164306

(the last one is mine for this issue)

The forum report at 1 above has identified commit [f7665a0](https://github.com/openwrt/openwrt/commit/f7665a0f1a2da63e2046119c0c3de578f16651ce) as offending change.

VA1DER, 2023-06-29:
I have identified the [April 17th mt76](https://github.com/openwrt/mt76/commit/577fbd4751529e05c17cdc59ccd5624eaed60507) commit as the offending change.  I am trying to identify the specific commit.

nbd168, 2023-06-29:
Thanks for your work on tracking this down. Please test the fix that I just pushed to mt76.git

VA1DER, 2023-06-29:
@nbd168 Device is no longer locked at 6mbit/20MHz.

That said, the RX rate seems to revert to 6mbit/s regularly and then renegotiate an HE mode.  This was normal on the pre-April-17th driver, but would only happen once or twice on a new connection and then it would settle on an HE mode and stay there-ish.

Right now it's spending about 60 seconds on an HE rate, reverting to 6mbit/20MHz for 4-5 seconds, then negotiating an HE rate again and repeating.  I can time it on a stopwatch and the above doesn't vary by more than a few seconds.

When it is spending the 60 seconds on the HE rate, it will fluctuate occasionally between different HE rates, as I expect on WiFi.  So during those 60 seconds, rate renegotiations are happening normally.  But at the end of the 60 seconds the RX rate is **always** reverting to 6mbit/s for 4-5 seconds.

soxrok2212, 2023-06-30:
> Thanks for your work on tracking this down. Please test the fix that I just pushed to mt76.git

I did test this and I am still unable to connect with 802.11s. I did not try STA mode.

Additionally, as mentioned in https://github.com/openwrt/openwrt/issues/12905, when the radio hosting the 802.11s node also has VAPs enabled, the wireless speeds slow to < 1/3 and I am still experiencing that.

nbd168, 2023-06-30:
I added some fixes for mesh mode to mt76.git, please test the latest version

soxrok2212, 2023-06-30:
> I added some fixes for mesh mode to mt76.git, please test the latest version

Looks to have fixed 802.11s! The second issue previously mention still exists though where a radio hosting both a mesh node and a VAP has very poor throughput on the VAP. 

On a 2x2 802.11ax client in 80MHz, I could previously see >700mbit/s. Now, no matter what the signal quality or strength, it maxes out at 170mbit/s.

VA1DER, 2023-06-30:
> please test the latest version

I haven't tested that yet.  But what I have done is:
- Went back to the driver as per June 23rd
- Reverted only April 17's  [wifi: mt76: mt7915: remove mt7915_mcu_beacon_check_caps()](https://github.com/openwrt/mt76/commit/4046b59183234a503e8fb017549c6d66576e941d) 

That is rock solid stable in STA mode.  No RX rate fluctuations between 6MBit/20MHz and HE rates.

I'm building to test today's fixes now.

VA1DER, 2023-07-01:
I'm pretty certain the RX fluctuations I reported were artifacts caused by my own test bench by having two devices in STA mode too close to each other.

I apologize, I should have caught that earlier.

Djfe, 2023-07-01:
so you can close this now, right? glad you caught this bug :)

closed 2023-07-01 by VA1DER