openwrt/openwrt #13005: Switch does not work on Asus AX53U

kot-begemot-uk, 2023-06-29:
### Describe the bug

DSA does not work.

Trying to configure vlans both tagged or untagged - all connectivity is lost.

Nothing informative in dmesg, nothing informative in syslog.

### OpenWrt version

23.05-SNAPSHOT, r23227-106c83a1ea

### OpenWrt target/subtarget

ramips mt7621

### Device

Asus AX53U

### Image kind

Self-built image

### Steps to reproduce

Build current snapshot, install try to configure vlans on the switch.

### Actual behaviour

No connectivity. Trying to configure vlans and DSA only one port succeeds, traffic is visible via tcpdump, but ignored by the router due to not stripping the vlan tag. Untagged in my setup fails because the MAC on all ports is the same resulting in issues with arp resolution.

### Expected behaviour

correct vlan config

### Diffconfig

```text
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_asus_rt-ax53u=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_OPENVPN_openssl_ENABLE_DEF_AUTH=y
CONFIG_OPENVPN_openssl_ENABLE_FRAGMENT=y
CONFIG_OPENVPN_openssl_ENABLE_LZ4=y
CONFIG_OPENVPN_openssl_ENABLE_LZO=y
CONFIG_OPENVPN_openssl_ENABLE_MULTIHOME=y
CONFIG_OPENVPN_openssl_ENABLE_PF=y
CONFIG_OPENVPN_openssl_ENABLE_PORT_SHARE=y
CONFIG_OPENVPN_openssl_ENABLE_SMALL=y
CONFIG_PACKAGE_kmod-tun=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libevdev=y
CONFIG_PACKAGE_liblzo=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libpcap=m
CONFIG_PACKAGE_librt=y
CONFIG_PACKAGE_libudev-zero=y
CONFIG_PACKAGE_libusb-1.0=y
CONFIG_PACKAGE_openvpn-openssl=y
CONFIG_PACKAGE_tcpdump=m
CONFIG_PACKAGE_usb-modeswitch=y
CONFIG_PACKAGE_usbutils=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

xserrat, 2024-01-28:
I'm in the same situation @kot-begemot-uk . I have a ASUS RT-AX53U and the "Switch" section does not appear so I cannot tag traffic to a specific VLAN.

I've flashed this firmware version:
```
OpenWrt 22.03.5 r20134-5f15225c1e / LuCI openwrt-22.03 branch git-23.093.57104-ce20b4a
```

kot-begemot-uk, 2024-01-28:
It started working for me in the latest 23.05-SNAPSHOT

It is a good idea to clear /etc/config/networks before upgrading to it so it creates the expected DSA sections.

xserrat, 2024-02-04:
Thanks for your reply @kot-begemot-uk , I finally found the way doing the following:
1. Create a Bridge Device (br-wan) pointing to the bridge port `wan`.
2. In the Bridge VLAN filtering, define each VLAN ID and setting the "tagging" mode:
[image]
3. Create each interface setting the corresponding device like here:
[image]
4. Create the SSID in the wireless section setting the interface defined in the step 3:
[image]
Thanks!!