openwrt/openwrt #13007: min_tx_power and/or acs_exclude_dfs don't work properly

ascendbeing, 2023-06-29:
### Describe the bug

when you have both min_tx_power and acs_exclude_dfs as set options in an AP config, they share the same line, meaning either ACS is read or it isn't, and min tx isn't respected, because either it's being interpreted as the value for ACS or it's being ignored because there's a space delimiter between the hostapd conf file that is generated. when you have both like I will show here, you can get UNI I channels despite specifying a higher min Tx than UNI I (24 vs the 29 I am using here, which is 1 higher than DFS). The solution would be to add a line break or delimiter. I have given guidance on how to do this further on; I wrote this part last, in post, so although I'm being a bit thin on detail here it's cause I already wrote it below so read on please.

### OpenWrt version

snapshot 2nd latest to newest 

### OpenWrt target/subtarget

r23453-5af7d47cd7

### Device

various (issue isn't device dependent, it's universal)

### Image kind

Official downloaded image

### Steps to reproduce

in /etc/config/wireless try with option min_tx_power 29 and option acs_exclude_dfs 1 in wifi_device section

### Actual behaviour

cat /tmp/run/*f|grep min_
acs_exclude_dfs=1 min_tx_power=29

The ACS and the min are on the same line so if anything only the ACS takes effect. So you will get channel 40, 36, etc.

### Expected behaviour

You don't get the two parameters on the same line. Which currently you do and it's been this way for at least 6 months possibly ever since you added these options they have been broken like this. In effect you shouldn't be put on channel 36, 40, etc with this specific setup but you can be. The ACS line should have a line break after but it doesn't. So either ACS is read or it's not, but min is ignored for sure because min 29 would exclude the min 24 36, 40 etc.

### Additional info

Relevant code is in https://github.com/openwrt/openwrt/blob/master/package/kernel/mac80211/files/lib/netifd/wireless/mac80211.sh lines 141-150. What I suspect is happening is it's appending the option workout a line break or the normal way options are delimited in the shell file. I haven't looked really into it but that's the best I can explain it. See below:

cat /tmp/run/*f|grep -e min_ -e acs
min_tx_power=29                                                 acs_exclude_dfs=1
channel=acs_survey                                              acs_exclude_dfs=1 min_tx_power=29

Above I only showed some of the output, plus my config has "list hostapd_options 'min_tx_power=29 acs_exclude_dfs=1'". So clearly the line with both isn't my hostapd options. I have both the individual options and the hostapd options configured cause i was seeing if I could remove the hapd options. As you see the two options per line isn't my hapd options because my hapd options are in reverse order. Hapd options afaik are not added in reverse order and are always added separately whether you have them space delimited or separate list entry with any number of space delimiters.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ascendbeing, 2023-06-29:
I appended a solution in bold. This problem i think effects every openwrt branch since version 20 or 21 where this was added, unless it just wasn't fixed in main snapshot branch. The fix is adding 5 characters to the min_tx_power line in Mac80211.sh

ascendbeing, 2023-06-30:
Last clarification: for main and 23.05, add `"$N"`, as the final argument passed to `append base_cfg`, on the same line as the last addition line of https://github.com/openwrt/openwrt/commit/d2d5aabdbb947443b0d5f75c3847754c7c38a117 which there is line 147 but currently is line 150 on main . Looks like it's been broken the whole time but it's only on 23.05 and main so it's not super bad. I think if you use min_ at all, it puts it on a neighboring line, whether or not you have the acs option, it would stick to whatever is next if not that because it's lacking the $N.

danpawlik, 2023-07-11:
@ascendbeing could you amend your git commit message in the PullRequest and we can move forward?

ascendbeing, 2023-07-16:
> @ascendbeing could you amend your git commit message in the PullRequest and we can move forward?
I would be glad to. I'm just new to GitHub and I'm operating on a mobile phone (i have a computer but I haven't assembled it) so I've been very diminished in my ability to competently do things. So if you can please elucidate what you mean I will go ahead and do so

danpawlik, 2023-07-16:
Just change your commit message by using:
```
git commit --amend
```
And now change your commit message:

```
Update mac80211.sh

The min tx isn't respected, because it doesn't have a newline character 
on the end of it's line so it's being ignored or added onto the adjoining 
options value.

This fixes #13007

```

ascendbeing, 2023-07-16:
Thank you. I will be back before too long (<1h) and see if I can't do something along those lines when I am back.

danpawlik, 2023-07-21:
Hi,
Could you update the commit?

stintel, 2024-11-24:
This should be fixed by 3aec71a176b43da5c4f1901ad423bacfb258f444?