openwrt/openwrt #13009: Conntrack helpers management by UCI is broken

fingolfin0, 2023-06-29:
### Describe the bug

# Short description
I tried to use "auto_helper" option (documented [here](https://openwrt.org/docs/guide-user/firewall/firewall_configuration#options1)) and I tried manual assigning helpers with "helper" option, both are broken. I solved my problem with custom nftables rule.

# Detailed description
### Using automatic conntrack helpers assignment
There is configuration of "wan" zone:
```
config zone 'wan'
        option name 'wan'
        option output 'ACCEPT'
        option forward 'DROP'
        list network 'WAN'
        list network 'ArtX'
        option masq '1'
        option input 'REJECT'
```
Note that the default value of "auto_helper" option is "0" due to enabled masquerade on this zone.
The output of `nft list ruleset` does not, as expected, contain rules for the "wan" zone:
<details>
  <summary> (<i>click to show</i>)</summary>

table inet fw4 {
	ct helper amanda {
		type "amanda" protocol udp
		l3proto inet
	}

	ct helper RAS {
		type "RAS" protocol udp
		l3proto inet
	}

	ct helper Q.931 {
		type "Q.931" protocol tcp
		l3proto inet
	}

	ct helper irc {
		type "irc" protocol tcp
		l3proto ip
	}

	ct helper pptp {
		type "pptp" protocol tcp
		l3proto ip
	}

	ct helper sip {
		type "sip" protocol udp
		l3proto inet
	}

	ct helper snmp {
		type "snmp" protocol udp
		l3proto ip
	}

	ct helper tftp {
		type "tftp" protocol udp
		l3proto inet
	}

	chain input {
		type filter hook input priority filter; policy accept;
		iifname "lo" accept comment "!fw4: Accept traffic from loopback"
		ct state established,related accept comment "!fw4: Allow inbound established and related flows"
		iifname "eth0" jump input_wan comment "!fw4: Handle wan IPv4/IPv6 input traffic"
		iifname "eth1" jump input_clearnet comment "!fw4: Handle clearnet IPv4/IPv6 input traffic"
[... 351 lines cut ...]

# Summary
1. Lucy must consider the masquerade status when displaying the checkbox for controlling auto-assigning of helpers.
2. Enabled "auto_helper" option should work.
3. Rules for helpers (like `iifname "eth0" jump helper_wan`) must be added to both chains "output" and "prerouting"

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (i440FX + PIIX, 1996)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

MisterW-git, 2025-07-01:
This would appear to still be a problem. I had a PPTP VPN working fine on my TPlink ER605 running 23.05.04. Upgraded to 24.10 and it wouldn't connect, giving 'timeout sending LCP config requests' error. Strange thing is that the VPN could be connected from devices on the LAN but not from the router itself.
Eventually found this bug and adding the suggested entry to /etc/nftables.d/10-custom-filter-chains.nft fixed the problem!.

NikitaKorneev, 2025-09-01:
As of today and as of LuCI openwrt-22.03 branch git-24.208.65546-18f3914 the issue still exists.

The fix is described above:

kmod-nf-nathelper and kmod-nf-nathelper-extra should be installed.

**create file like /etc/nftables.d/20-pptp-conntrack-helper.nft with the following content:**

```
chain user_pre_output {
    type filter hook output priority -1; policy accept;
    tcp dport 1723 ct helper set "pptp"
}
```