openwrt/openwrt #13016: Compile errors in fortify-headers stdio.h with -Werror=format-nonliteral

ksperling-apple, 2023-06-30:
### Describe the bug

When compiling any source file that uses stdio.h with FORTIFY_SOURCE enabled (the default) and `-Werror=format-nonliteral`, format-nonliteral errors occur in `..._musl/include/fortify/stdio.h`. This header is part of the fortify-headers toolchain package and should compile cleanly with these compile settings.

### OpenWrt version

22.03.5

### OpenWrt target/subtarget

ramips/mt7621

### Device

GL.iNet GL-MT1300

### Image kind

Self-built image

### Steps to reproduce

1. Unpack [hello.tar.gz](https://github.com/openwrt/openwrt/files/11915040/hello.tar.gz) into e.g. packages/util
2. `echo CONFIG_PACKAGE_hello=y >> .config && make defconfig`
3. `make package/hello/compile V=s`

### Actual behaviour

Multiple `format-nonliteral` errors in `toolchain-mipsel_24kc_gcc-11.2.0_musl/include/fortify/stdio.h`:

```
In file included from hello.c:1:
/home/karsten/workspace/openwrt-repro/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/include/fortify/stdio.h: In function 'snprintf':
/home/karsten/workspace/openwrt-repro/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/include/fortify/stdio.h:101:9: error: format not a string literal, argument types not checked [-Werror=format-nonliteral]
  101 |         return __orig_snprintf(__s, __n, __f, __builtin_va_arg_pack());
      |         ^~~~~~
/home/karsten/workspace/openwrt-repro/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/include/fortify/stdio.h: In function 'sprintf':
/home/karsten/workspace/openwrt-repro/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/include/fortify/stdio.h:110:17: error: format not a string literal, argument types not checked [-Werror=format-nonliteral]
  110 |                 __r = __orig_snprintf(__s, __b, __f, __builtin_va_arg_pack());
      |                 ^~~
/home/karsten/workspace/openwrt-repro/staging_dir/toolchain-mipsel_24kc_gcc-11.2.0_musl/include/fortify/stdio.h:114:17: error: format not a string literal, argument types not checked [-Werror=format-nonliteral]
  114 |                 __r = __orig_sprintf(__s, __f, __builtin_va_arg_pack());
      |                 ^~~
cc1: some warnings being treated as errors
```

Full log: [hello.log](https://github.com/openwrt/openwrt/files/11915441/hello.log)

### Expected behaviour

The package should compile cleanly

### Additional info

This is a synthetic example to reproduce the issue mentioned in https://github.com/openwrt/openwrt/pull/13001

### Diffconfig

```text
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_glinet_gl-mt1300=y
CONFIG_PACKAGE_hello=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-07-02:
Why package should compile cleanly? 1-argument printf is duly reported.

ksperling-apple, 2023-07-02:
> Why package should compile cleanly? 1-argument printf is duly reported.

There is nothing wrong with 1-arg printf from a security point of view as long as the format string is a string literal, as it is here. The warning/error does not occur in hello.c, it occurs in the fortify-headers wrappers of `snprintf` and `sprintf` (which  the example isn't even calling).

The issue reproduces just the same with `printf("%s", "hello\n");` or in fact without the `printf` statement. Including stdio.h is all it takes.

The error goes away with `PKG_FORTIFY_SOURCE:=0` which makes sense since the problem is in the fortify-headers stdio.h.

brada4, 2023-07-02:
But you want to suppress warnings when the argument is anything. Would be more considerate to define 1-arg printf as %s 1-arg one?

ksperling-apple, 2023-07-03:
> But you want to suppress warnings when the argument is anything. Would be more considerate to define 1-arg printf as %s 1-arg one?

You mean in my PR (https://github.com/openwrt/openwrt/pull/13001) ? I want to suppress the warning within the implementation of the fortified wrapper functions only. The warnings will still apply to all other code, e.g. with the PR applied the following still warn / error:

```
hello.c: In function 'main':
hello.c:5:5: error: format not a string literal, argument types not checked [-Werror=format-nonliteral]
    5 |     printf(argv[0], "");
      |     ^~~~~~
```

```
hello.c: In function 'main':
hello.c:5:5: error: format not a string literal and no format arguments [-Werror=format-security]
    5 |     printf(argv[0]);
      |     ^~~~~~
```

I think the things that are involved in the bug are
- gcc seems to be making certain assumptions about std C functions like `sprintf` (and we're not telling it not to via `-ffreestanding`)
- fortify-headers wraps and redefines these functions
- gcc seems to understand that the `orig_...` names `_FORTIFY_FN` creates reference the standard functions, or at least carries across `__attribute__((format))` correctly, and applies the relevant checks
- but somehow it doesn't understand (nor can it be made to via `__attribute__((format))`) that the wrappers themselves have the same argument semantics.

So I'd say it's either a gcc bug, or the way fortify-headers wraps these functions is somehow "incorrect" from a gcc point of view.

ksperling-apple, 2023-07-03:
Looks like this is the relevant gcc bug, filed 10 years ago 😬 https://gcc.gnu.org/bugzilla/show_bug.cgi?id=57137

brada4, 2023-07-03:
My idea is that hardening headers may bend standard to harden code vs force porter to patch harmless warnings like
```
#define printf(zzzz) (printf("%s",zzzz))
```

ksperling-apple, 2023-07-03:
> My idea is that hardening headers may bend standard to harden code vs force porter to patch harmless warnings

I'm not sure I understand what you mean exactly.

The workaround that's been applied to the packages that have hit this issue so far (e.g. glib2) has been to disable fortify-source for the whole package via `PKG_FORTIFY_SOURCE:=0`.

brada4, 2023-07-03:
I mean single-arg printf is all over books on C, just prepend %s for single-arg case as to decode intent of programmer and harden it.

ksperling-apple, 2023-07-04:
Sure, but single-arg printf isn't what this issue report is about. My example has a single-arg printf (with a literal string), but as I pointed out above, the issue reproduces even when the printf statement is removed entirely from hello.c.

ksperling-apple, 2023-07-17:
Maybe the actual gcc bug here is a little more specific actually, since it's hitting exactly those invocations that use `__builtin_va_arg_pack()`.

ksperling-apple, 2023-07-17:
Compiling with `-fno-builtin` or `-ffreestanding` make the problem go away. These flags don't seem like they should have to be specified in a package makefile though to avoid warnings in a system header.

It's not clear to me how the fortify-headers wrappers are intended to interact with GCC builtins, or if maybe `-fno-builtin` should be added by default if fortify is enabled... but the impact of such a change go would way beyond this format-nonliteral warning.

ksperling-apple, 2023-07-27:
I've filed a gcc bug here: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=110834

jvoisin, 2023-12-06:
> Alternatively, it looks like fortify-headers itself is receiving some maintenance over at https://github.com/jvoisin/fortify-headers now, so maybe a fix / workaround could be applied there, and the OpenWrt package updated to point to that version.

I just got (yesterday!) the commit bit in the original repo, I'll push my changes there soon, and will try to do a release at some point.