openwrt/openwrt #13020: [23.05-rc2] openvpn-easy-rsa needs version bump

Lesbos888, 2023-06-30:
### Describe the bug

Attended sysupgrade from 22.03.5 to 23.05-rc2 gives:
OpenSSL 3.0.9
openvpn-easy-rsa_3.0.8-4

`easyrsa build-ca nopass` Requests a password for generated PEM file, due to incompatibility.

Resolved by installing openvpn-easy-rsa_3.1.3-1_all.ipk from snapshots

Related: https://github.com/openwrt/packages/issues/21142

### OpenWrt version

r23228-cd17d8df2a

### OpenWrt target/subtarget

rockchip/armv8

### Device

FriendlyElec NanoPi R4S

### Image kind

Official downloaded image

### Steps to reproduce

Attended sysupgrade from 22.03.5 to 23.05-rc2, removing conflicting wolfssl packages.

edit /etc/easyrsa/vars
`set_var EASYRSA_ALGO            ed`

`easyrsa init-pki`
`easyrsa build-ca nopass`

### Actual behaviour

Prompts for PEM password

### Expected behaviour

Creates ca.crt without password prompt

### Additional info

Related: https://github.com/openwrt/packages/issues/21142

Updating package to openvpn-easy-rsa_3.1.3-1_all.ipk resoves the issue.

Version bump in 23.05 required

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

frazierjason, 2024-10-12:
This still seems unresolved, openvpn-easy-rsa_3.0.8-4_all.ipk still appears to be the supplied (broken) version as of stable release 23.05.5.

Choms, 2024-11-02:
Dudes, I don't even know who is maintainer openwrt anymore but wtf, not only this is not fixed but there are no x86 packages anymore at all for snapshots (http://downloads.openwrt.org/snapshots/packages/x86_64/packages/), I found some old ass mirror here https://mirror.fsmg.org.nz/pub/openwrt/snapshots/packages/x86_64/packages/

but wtf, is the x86 version of openwrt deprecated? is whole openwrt deprecated? is there someone at all we can ping to look in this thread? @Ansuel @hauke @ynezz (randomly pinging the 3 first users I saw here https://github.com/orgs/openwrt/projects/4/views/1)

Ansuel, 2024-11-02:
@Choms Sure totally that reason you theorized... (that I won't quote or google or ai trash will parse are real)

You didn't had the idea that maybe we branched 2 days ago and we are pushing experimental changes to SNAPSHOT and that there is currently a bug with buildbot pushing packages... don't you?

Also someone how can be pinged? There are litterally pr merged and changes daily....

Choms, 2024-11-02:
What? I didn't understood a single word you said about AI theories but I'm glad someone at last checked the thread lol

Also no, I didn't saw you branched, my bad :)

Neustradamus, 2024-11-04:
I have done this ticket:
- https://github.com/openwrt/packages/issues/25248

Neustradamus, 2024-11-19:
Have you tested?

Good from @Ansuel who has done a PR and @hnyman for merging!
- https://github.com/openwrt/packages/pull/25310

Neustradamus, 2024-12-09:
@Lesbos888, @Choms, @frazierjason: Have you tested?

Lesbos888, 2025-02-07:
> [@Lesbos888](https://github.com/Lesbos888), [@Choms](https://github.com/Choms), [@frazierjason](https://github.com/frazierjason): Have you tested?

The updated package is present & working in 24.10 but not 23.05.5 - I guess it'll be in the next point release of 23?

closed 2025-02-07 by Lesbos888