openwrt/openwrt #13027: mvebu, clients lose DHCP address

Jim-Wright, 2023-07-01:
### Describe the bug

Issue noticed after upgrading from 21.02.7 to 22.03.2, WRT1900AC v1.  Clients connected to Wifi would periodically lose networking and required wifi to turn off then back on to correct.  Observed that clients had 169.254.x.x address at this point.  No DHCP clients on wired network to test this with.  Assuming bug in WRT1900AC wireless, set up a separate access point and disabled radios in WRT1900AC, issue continued to be present.

Upgraded router to 23.05.0-rc1, which had newer kernel and supposedly fixed many issues with mvebu systems, but issue persisted.  Finally reverted router back to 21.02.7 and issue hasn't recurred.

Version field with this report is from 21.02.7 as I had already reverted the firmware to resolve this.

### OpenWrt version

r16847-f8282da11e

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT1900AC v1

### Image kind

Official downloaded image

### Steps to reproduce

See above.  Clients may be connected for minutes or up to an hour before observing the issue.

### Actual behaviour

DHCP address changed to 169, no networking available until wifi disabled.

### Expected behaviour

Clients expected to keep assigned DHCP address.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-07-02:
Try to extract information related to single affected client via `logread -e '12:34:56:78:9a:bc'`

Jim-Wright, 2023-07-06:
I've installed 23.05.0-rc2 since that was available, a bit over 24 hours now and I've not run into this bug again.  Will continue to monitor.

Jim-Wright, 2023-07-11:
Still working fine, so presumably whatever the issue was is now fixed, I'd feel better if I actually knew what the root cause was to better track this, but it may have been some combination of things.  Fingers crossed that it won't come back.

brada4, 2023-07-11:
You can try to shorten dhcp lease time from default 12h to like 5min  in luci/network/interfaces/lan/dhcp server/general to try to trigger the problem if it still lurks around.

Jim-Wright, 2023-07-11:
It was actually happening before the usual lease expired.  I was seeing it sometimes within minutes of rebooting OpenWRT.  Didn't appear related to lease time at all.

brada4, 2023-07-11:
The lease file has absolute time and if uplink does not come up quickly openwrt may start services with a/ manufacture time b/root fs time c/ (optimally) ntp time , so clock time may run so fast dhcp server expires its side of leases, clients keep relative time and refresh on 1/2 3/4 etc more towards lease expiry.