openwrt/openwrt #13034: RT acrh17 wireless cant use

stvber, 2023-07-02:
### Describe the bug

IEEE 802.11: deauthenticated due to inactivity

### OpenWrt version

master and openwrt-22.03

### OpenWrt target/subtarget

ipq40xx/generic

### Device

ASUS RT-AC42U

### Image kind

Self-built image

### Steps to reproduce

about 30 seconds disconnect device and reconnect again

### Additional info

official downloaded image, master  and openwrt-22.03 image .the result is the same

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-07-02:
Is there any problem showing? It is default behaviour to free 5min idle station's ressources.
https://w1.fi/cgit/hostap/tree/hostapd/hostapd.conf#n517

stvber, 2023-07-02:
Sun Jul  2 18:46:20 2023 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx
Sun Jul  2 18:46:20 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disassociated
Sun Jul  2 18:46:20 2023 kern.info kernel: [196984.111129] ath10k_pci 0000:01:00.0: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
Sun Jul  2 18:46:21 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Jul  2 18:46:52 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
Sun Jul  2 18:46:52 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 1)
Sun Jul  2 18:46:52 2023 daemon.notice hostapd: wlan0: AP-STA-CONNECTED xx:xx:xx:xx:xx:xx
Sun Jul  2 18:46:52 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Sun Jul  2 18:46:52 2023 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED xx:xx:xx:xx:xx:xx
Sun Jul  2 18:46:52 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 172.16.2.70 xx:xx:xx:xx:xx:xx
Sun Jul  2 18:46:52 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 172.16.2.70 xx:xx:xx:xx:xx:xx amlogic
Sun Jul  2 18:47:20 2023 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx
Sun Jul  2 18:47:20 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disassociated
Sun Jul  2 18:47:21 2023 kern.info kernel: [197044.169557] ath10k_pci 0000:01:00.0: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
Sun Jul  2 18:47:22 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Jul  2 18:47:52 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
Sun Jul  2 18:47:52 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 1)
Sun Jul  2 18:47:52 2023 daemon.notice hostapd: wlan0: AP-STA-CONNECTED xx:xx:xx:xx:xx:xx
Sun Jul  2 18:47:52 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Sun Jul  2 18:47:52 2023 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED xx:xx:xx:xx:xx:xx
Sun Jul  2 18:47:52 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 172.16.2.70 xx:xx:xx:xx:xx:xx
Sun Jul  2 18:47:52 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 172.16.2.70 xx:xx:xx:xx:xx:xx amlogic
Sun Jul  2 18:48:14 2023 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx
Sun Jul  2 18:48:14 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disassociated
Sun Jul  2 18:48:14 2023 kern.info kernel: [197097.288298] ath10k_pci 0000:01:00.0: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
Sun Jul  2 18:48:15 2023 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'soc/40000000.pci/pci0000:00/0000:00:00.0/0000:01:00.0'
        option band '5g'
        option htmode 'VHT80'
        option channel '149'
        option country 'AU'
        option txpower '30'
        option cell_density '1'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option key '1234567890'
        option encryption 'psk2+ccmp'
        option disassoc_low_ack '0'
        option disabled '1'

config wifi-device 'radio1'
        option type 'mac80211'
        option path 'platform/soc/a000000.wifi'
        option channel '1'
        option band '2g'
        option htmode 'HT40'
        option cell_density '0'
        option disabled '1'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option ssid 'OpenWrt'
        option key '1234567890'
        option disabled '1'
        option mode 'monitor'

config wifi-iface 'wifinet2'
        option device 'radio0'
        option mode 'ap'
        option ssid 'OpenWrt-5G'
        option encryption 'psk2+ccmp'
        option disassoc_low_ack '0'
        option key '1234567890'
        option network 'lan'

2.4G and 5G ,the result is the same

brada4, 2023-07-02:
What is the issue at hand outside normal-looking  log entries?

stvber, 2023-07-04:
Because of my wireless terminal has done a scheduled task to reconnect to other ssid

closed 2023-07-04 by stvber