openwrt/openwrt #13035: Xiaomi AX 3600 - Unable to Operate at 160 MHz with Country Code PA

starraiderx, 2023-07-02:
### Describe the bug

Xiaomi AX 3600 is unable to operate at 160 MHz when the country code is set to PA (Panama). Despite the wireless-regdb allowing 160 MHz operation in this region, all devices fail to connect to the router.

### OpenWrt version

r23459-a0ae7a50e0

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Steps to reproduce

1.     Set the router's country code to PA.
2.     Attempt to operate the router at 160 MHz.
3.     Attempt to connect any device to the router.
4.     See error (devices fail to connect).

### Expected behaviour

Given that the wireless-regdb allows 160 MHz operation in Panama, it is expected that the router should function properly at this frequency and allow devices to connect.

### Additional info

Here is the wireless-regdb entry for Panama:

`country PA: DFS-FCC
	(2400 - 2483.5 @ 40), (36)
	(5150 - 5250 @ 80), (36), AUTO-BW
	(5250 - 5350 @ 80), (30), AUTO-BW
	(5470 - 5725 @ 160), (30)
	(5725 - 5850 @ 80), (36)
	(57000 - 64000 @ 2160), (43)
`

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-07-02:
You can only use channel 149 for 160MHz operation in Panamá according to [regulatory database ](https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt#n1349)

Device has 2 5GHz radios - one ac 1x1 other ax 160 2x2 xor 80 4x4. Are you configuring the correct one (should be if it allows 160MHz and other not)

starraiderx, 2023-07-02:
Channel 149 operates at 5745 MHz, placing it outside the regulated frequency band of Panama. Channels within the range of 96 to 144 should function correctly. Despite [Wikipedia](https://en.wikipedia.org/wiki/List_of_WLAN_channels) indicating that Channel 114 corresponds to 160 MHz, I'm unable to select it using Luci. Additionally, attempting to set it in the configuration results in the Access Point becoming inaccessible.

brada4, 2023-07-02:
I mixed up 2 text lines, it is CH100 (of 36 and 149 being other alternatives in other places). You have to select first channel and in 5GHz they are not overlapping like 2.4GHz.
Regulatory document (CH 11.5) matches regdb with absent antennas , though link is not accurate.
https://www.asep.gob.pa/wp-content/uploads/telecomunicaciones/pnaf/pnaf-dic2015.pdf

starraiderx, 2023-07-02:
I attempted to use Channel 100, but unfortunately, it was unsuccessful. Shouldn't the 'auto' mode automatically configure the permissible frequencies, even if there's only one channel allowed, as in this case? For testing purposes, I've temporarily disabled the 2.4 GHz frequency.

brada4, 2023-07-02:
Yes, auto should auto-select only permissible frequency i.e CH100.
I read [global specs](https://www.mi.com/global/product/mi-aiot-router-ax3600/specs) of that router and it does not mention 160MHz - could there be two models or at least calibration data versions out there? i.e 80MHz only? I am not sure.

starraiderx, 2023-07-02:
When setting the router's country code to either the United States or any European Union nation, the 160 MHz frequency functions properly.

robimarko, 2023-07-02:
What does `iw reg get` print once the PA code was set?

brada4, 2023-07-02:
It seems that ath11k will not enable 160MHz when there is no DFS scan performed. Which is rather unfair.
Also worth comparing logread between US and PA bringup (PA should have no DFS and come up instantly)

starraiderx, 2023-07-02:
> What does `iw reg get` print once the PA code was set?

Here's what's printed:
`global
country PA: DFS-FCC
        (2400 - 2483 @ 40), (N/A, 36), (N/A)
        (5150 - 5250 @ 80), (N/A, 36), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5470 - 5725 @ 160), (N/A, 30), (N/A)
        (5725 - 5850 @ 80), (N/A, 36), (N/A)
        (57000 - 64000 @ 2160), (N/A, 43), (N/A)

phy#2 (self-managed)
country PA: DFS-UNSET
        (2402 - 2472 @ 40), (6, 36), (N/A)
        (5170 - 5250 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 24), (N/A), AUTO-BW
        (5490 - 5730 @ 160), (N/A, 24), (N/A), AUTO-BW
        (5735 - 5835 @ 80), (N/A, 30), (N/A), AUTO-BW

phy#1 (self-managed)
country PA: DFS-UNSET
        (2402 - 2472 @ 40), (6, 36), (N/A)
        (5170 - 5250 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 24), (N/A), AUTO-BW
        (5490 - 5730 @ 160), (N/A, 24), (N/A), AUTO-BW
        (5735 - 5835 @ 80), (N/A, 30), (N/A), AUTO-BW

phy#0
country US: DFS-FCC
        (902 - 904 @ 2), (N/A, 30), (N/A)
        (904 - 920 @ 16), (N/A, 30), (N/A)
        (920 - 928 @ 8), (N/A, 30), (N/A)
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)
`

There appear to be discrepancies in the database entries. For instance, the transmission power for 160 should be 30dbm, not 24 in PA. It should be noted that all antennas are set to PA.

brada4, 2023-07-02:
You need to set region for each antenna . i.e. in /etc/config/wireless each wifi-device needs to have own option country PA. Last seems not there.

starraiderx, 2023-07-02:
As previously mentioned, all antennas are configured to PA.

```

config wifi-device 'radio0'
	option type 'mac80211'
	option path 'soc/20000000.pci/pci0000:00/0000:00:00.0/0000:01:00.0'
	option band '5g'
	option htmode 'VHT80'
	option disabled '1'
	option country 'PA'
	option cell_density '0'

config wifi-iface 'default_radio0'
	"removed"

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc/c000000.wifi'
	option channel 'auto'
	option band '5g'
	option htmode 'HE160'
	option cell_density '0'
	option country 'PA'

config wifi-iface 'default_radio1'
	"removed"

config wifi-device 'radio2'
	option type 'mac80211'
	option path 'platform/soc/c000000.wifi+1'
	option channel 'auto'
	option band '2g'
	option htmode 'HE40'
	option txpower '30'
	option country 'PA'
	option cell_density '0'
	option noscan '1'

config wifi-iface 'default_radio2'
	"removed"

```

brada4, 2023-07-03:
Your setup is solid, but card does not take it.
Try setting `iw reg set` on online (but down) interfaces jumping via EU/US
like
```
iw reg set 'US'
iw reg set 'PA'
iw reg get
iw reg set 'ES'
....
```
Very little likely to work, but if it does it pinpoints some bug in driver/card.

starraiderx, 2023-07-03:
I attempted to adjust the 'iw reg set' on online (but down) interfaces by toggling between EU/US as you instructed, but it had no effect. The output remains unchanged from the previous one.

```
phy#0
country US: DFS-FCC
```

brada4, 2023-07-03:
Does it clamp to US (not changing to EU or PA) with 80MHz? It is still usable under US to some limited degree until problem is figured out.

starraiderx, 2023-07-03:
Indeed, it seems to consistently remain set to the US. When I set the country to DE (Germany), despite 'iw reg get' indicating that phy#0 is still set to the US, I can successfully connect to the device. I suspect this is due to the overlap of frequencies between the US and Germany. However, when I set it to PA (Panama), I am unable to connect to the device, even though phy#0 continues to report the US setting.

robimarko, 2023-07-04:
Whatever frequency restrictions end up being enforced are enforced by the FW itself and the kernel is just getting a list from the FW.

All of the BDF-s were updated to the latest regulatory info we have, maybe it was fixed in even later version

starraiderx, 2023-07-05:
@robimarko, if I'm understanding you correctly, there's nothing we can currently do about this because it's an issue originating upstream?

robimarko, 2023-07-05:
The issue is most likely in the ath11k FW itself, there is nothing that we can do about that.

rany2, 2023-07-05:
@starraiderx ath11k has its own regdb, it doesn't care about Linux's regdb

starraiderx, 2023-07-05:
Alright, thank you for all the information provided.

closed 2023-07-05 by starraiderx