openwrt/openwrt #13036: Zyxel 3301-PLUS - UDP MTU hang/crash with QMI LTE

KiralyCraft, 2023-07-02:
### Describe the bug

When attempting to connect to a WireGuard server from my laptop (running Linux with KDE, and Wireguard in NetworkManager) via the router, the uplink dies and does not recover. There are no logs in dmesg.

I've installed OpenWRT snapshot yesterday on this device, and have configured it's Quectel EG506 LTE modem as the one and only uplink. Given that it set it's MTU to 1500, I lowered it down to 1400 because I've read for LTE connections, it should be at most 1436, and thus playing it safe with 1400. 

For the Wireguard server, I have two clients - My laptop and my phone. I am accessing a website through the VPN which is rather large, so I would assume it fills quite a few packets to their max. 

The laptop has an MTU of 1500 for the WLAN interface, and 1420 for Wireguard. When I access the page with these settings, the uplink dies immediately. No more traffic flows until i cut the power to the modem and turn it on again.

I've tested on my phone as well, it has an MTU of 1500 for the LTE uplink (unlike my 1400 which I've set manually), and 1280 for Wireguard. It also has an MTU of 1500 for the WLAN connection, but it still keeps it at 1280 for Wireguard. This is why I can access the same site on my phone, through the same Wireguard server and the same LTE router without issues. 

It seems that if using UDP packets with a size greater than the one of the MTU of the LTE uplink, it causes the modem to hang. I'm not sure where the problem is exactly here. I've also enabled "MSS Clamping" for the uplink. I have also applied [this](https://forum.openwrt.org/t/solved-lte-qmi-troubles/28379/26) workaround for hanging QMI. I also have packet steering enabled, but no software flow offloading (or hardware for that matter)

When the MTU for LTE was 1500, there was significant packet loss especially when uploading.

### OpenWrt version

r23459-a0ae7a50e0

### OpenWrt target/subtarget

ramips/mt7621

### Device

ZyXEL LTE3301-Plus

### Image kind

Official downloaded image

### Steps to reproduce

1. Configure LTE with QMI and OpenWRT
2. Set LTE's interface MTU to 1400. Enable MSS clamping
3. Connect via an interface with MTU 1500 and create UDP traffic with a size larger than 1400 from a device in the network, which traverses the 
4. Observe the uplink becoming unresponsive 

### Actual behaviour

The behaviour is described in "steps to reproduce"

### Expected behaviour

The uplink should not become unresponsive, but in the worst case packets that are unable to be split should be dropped. 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-07-02:
You can manually apply patch from that PR on device, in current state it clamps MTU to input ethernet/wifi i.e. actually does not change packet be it pppoe or wireguard tunnel on internet side.

brada4, 2023-07-02:
(probably you need to remove all overriden mtu/mtu6 settings from /etc/config/*)

KiralyCraft, 2023-07-02:
Wow, that was quick! I applied the patch manually and removed my custom MTUs for the link. It no longer hangs where it always hung before, thank you!

Also the speed is now significantly better as well, 57 from 50 mbps - May be due to other factors, but I'd like to think it's because of this

brada4, 2023-07-02:
you could let @jow- close this

brada4, 2023-07-02:
4g comes in 1.2mbps slots, its not managed from cpe.

KiralyCraft, 2023-07-02:
> you could let @jow- close this

Reopened it, thanks for letting me know.

brada4, 2023-07-03:
It is about PMTU blackhole - dropping DF packets that dont fit in next wire in place of sending frag-needed ICMP.
Check major CDN-s - they use MSS in range 1200..1300 to address exactly this problem. While if you have website at your work connected to ethernet and connect it from faraway places PMTU blackhole is very likely if interface MTU is below least of client (adjusted by mssfix) and server (1500).
Quackery-quality methods to address this is to reduce internet-facing MTU until happy aka matching random CDN, or striping DF bit (not trivial). In either case it just improves your success, there will be blackhole with even smaller MTU, or dropping packet instead of fragmenting it in half.

brada4, 2023-07-04:
Another option from big CDN:
Shortcut from: https://blog.cloudflare.com/path-mtu-discovery-in-practice/
```
sysctl net.ipv4.tcp_mtu_probing=1 >> /etc/sysctl.conf
```

KiralyCraft, 2023-07-08:
I've enabled MTU probing as suggested, although it may still be a while until I find out a result. I believe that this might be more of an in depth problem, as when the connection fails, if I restart the QMI interface in OpenWRT it will not receive an IP. The only way to recover in this scenario is to turn off the modem and turn it back on (via a GPIO pin, but still).

While before the patch this behavior happened sooner, now it only happens every now and then. I've also lowered the MTU of the interface to 1400 and made sure that it applied, but it still seems to be happening. If packets going out the wire are now enforced to be 1400 bytes long, I have to believe this is a firmwre issue. I noticed that when this happened, I also got a stack trace in dmesg:

```[ 2447.026107] ------------[ cut here ]------------
[ 2447.035356] WARNING: CPU: 2 PID: 0 at net/sched/sch_generic.c:477 dev_watchdog+0x2e4/0x2ec
[ 2447.051845] NETDEV WATCHDOG: wwan0 (qmi_wwan): transmit queue 0 timed out
[ 2447.065339] Modules linked in: pppoe ppp_async option nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet wireguard usb_wwan qmi_wwan pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7615e mt7615_common mt76_connac_lib mt76 mac80211 libchacha20poly1305 cfg80211 usbserial usbnet slhc poly1305_mips nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c hwmon crc_ccitt compat chacha_mips cdc_wdm ledtrig_usbport ip6_udp_tunnel udp_tunnel sha512_generic seqiv jitterentropy_rng drbg kpp hmac cmac leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore nls_base usb_common mii crc32c_generic
[ 2447.229128] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 5.15.118 #0
[ 2447.241261] Stack : 809672b8 80083e68 00000000 00000004 00000000 00000000 81411d84 80a70000
[ 2447.257938]         808b0000 807d68a0 81454a18 808b2ea3 00000000 00000001 81411d30 81489320
[ 2447.274591]         00000000 00000000 807d68a0 81411bd0 ffffefff 00000000 ffffffea 00000000
[ 2447.291233]         81411bdc 00000135 808b8af8 ffffffff 807d68a0 00000001 00000000 8058ab64
[ 2447.307876]         00000009 00000002 ffffffff 80920000 00000018 80415acc 00000008 80a70008
[ 2447.324519]         ...
[ 2447.329377] Call Trace:
[ 2447.334226] [<80007f34>] show_stack+0x28/0xf0
[ 2447.342922] [<803878d0>] dump_stack_lvl+0x60/0x80
[ 2447.352296] [<8002e13c>] __warn+0x9c/0x124
[ 2447.360465] [<8002e250>] warn_slowpath_fmt+0x8c/0xac
[ 2447.370349] [<8058ab64>] dev_watchdog+0x2e4/0x2ec
[ 2447.379712] [<800a2b9c>] call_timer_fn.constprop.0+0x1c/0x8c
[ 2447.390965] [<800a38f0>] __run_timers.part.0+0x224/0x284
[ 2447.401527] [<800a39a4>] run_timer_softirq+0x54/0xa4
[ 2447.411396] [<80725a60>] __do_softirq+0x150/0x334
[ 2447.420781] [<800332d0>] __irq_exit_rcu+0xe4/0x128
[ 2447.430313] [<80033414>] irq_exit+0x10/0x1c
[ 2447.438637] [<803a3e7c>] plat_irq_dispatch+0x98/0xcc
[ 2447.448530] [<80003568>] except_vec_vi_end+0xb8/0xc4
[ 2447.458444] [<80725160>] r4k_wait_irqoff+0x1c/0x24
[ 2447.467993] [<8072522c>] default_idle_call+0x28/0x4c
[ 2447.477866] [<80061cc4>] do_idle+0xd0/0x174
[ 2447.486184] [<80062020>] cpu_startup_entry+0x24/0x34
[ 2447.496053] [<800130b4>] not_nmi+0x8c/0x1d8
[ 2447.504379]
[ 2447.507404] ---[ end trace fdef0031c07a5862 ]---
```

I think this issue may be caused by the MTU among other causes

brada4, 2023-07-08:
You can go lower like 1280 or 1024 as mentioned in one or another RFC  **IF IT HELPS**, the TCP/UDP woes are because somewhere close to you is specific type of blackhole that drops incoming frag-needed ICMP and you have to guess around it.

Reading stack trace upside down - hardware interrupt from mt76 wifi device (not NMI) called softirq like nftables which timed out.
It should not have caused any adverse effect past individual packets, or in worst case device queue to be dicarded causing TCP to adapt by slowing down slightly.

I dont quite understand involvement of swapper there - are you using a lot of memory under idle conditions, like any userland process crashed outside that IRQ problem?

brada4, 2023-07-08:
continuing ->
You have multi-core device - you can balance interrupts across cores with you guessed it - irqbalance for more time allowed for independent devices as opposed to all running IRQ routines on the first core.

brada4, 2023-07-28:
Disable flowtable offload for watchdog (?)
It is different issue.
For mtu - try lpwer and lower, 46bytes at time

KiralyCraft, 2023-07-28:
If you are referring to the "Software flow offload" which I believe affects netfilter's flowtable, then it was disabled since the beginning. I've also upgraded the firmware of the modem to the latest there is, EP06ELAR04A20M4G, built on the 18th of January 2023, and the problem is still happening. 

I've gone with the MTU as low as 1200, and unfortunately it's still happening. It seems to also happen for TCP connections, not just UDP.

I've also moved to ModemManager, which seems to have improved things a bit - The crashes don't happen so often now, but it may be placebo;

KiralyCraft, 2023-07-29:
I can't express my gratitude for your assistance, thank you so much. I went back to using this device in a "raw" QMI mode, with the newest firmware. It seems like ModemManager and LUCI don't play vrey nicely together, for example the MTU setting wasn't applying. 

Now with the newest firmware, a MTU of 1200 and the "QMI Celullar" protocol in LUCI, it seems to be stable for web browsing. 

With a Wireguard tunnel with a MTU of 1300 however, the module dies and no connection can be made until I reset it via the GPIO pins (cut power to it and bring it back up). I've also tried PPP over /dev/cdc-wdm0 and the serial AT ports, but I can't get a connection. 

Indeed, the packet steering might not affect the LTE issue I'm having too much.

I found the original firmware of the device, which also seems to be running OpenWRT, but with modifications applied.  They seem to be using iptables, I've uploaded a CPIO archive here: https://1fichier.com/?f30jmx1vyqr3shc190ly

brada4, 2023-07-29:
Iptables fw3 had same problem , patch is in that other thread

brada4, 2023-07-29:
You can look for modem config in vendor fw using binwalk. Likely mimicking it is best way to go

KiralyCraft, 2023-08-03:
Looks like I made it work for the most part, I've updated the firmware to the latest there is (EP06ELAR04A20M4G) and moved to using "QMI Celullar", with a forced MTU of 1200. It's been rock solid since, but it still dies when doing a Microsoft Teams call (from a browser), and the Wireguard problem (with bigger MTU than WAN) persists. 

Will look at the vendor fw as well

KiralyCraft, 2023-08-03:
I may have found a workaround. I switched to MBIM mode, and installed luci-proto-mbim. I switched the modem to MBIM mode by doing connecting to /dev/ttyUSB2 and typing 

`AT+QCFG="usbnet",2`

And then configured the MBIM connection identically to the QMI one. I did not notice a decrease in performance, Teams calls work, and Wireguard with higher MTU then WAN also works. I will continue testing

brada4, 2023-08-03:
Is automatic MTU now 1500?

KiralyCraft, 2023-08-03:
It seems to be 1460, configured automatically

brada4, 2023-08-03:
Leave it 1460, after the patch openwrt obeys that.