openwrt/openwrt #13045: [23.05-rc2] ipq806x: wireguard performance is much worse compared to 22.03.5

csharper2005, 2023-07-03:
### Describe the bug

Wireguard performance is much worse compared to 22.03.5.

Internet connection - PPPoE 
Internet speed - 200/200 MBit

Wireguard speed  (22.03.5) - 200/200 MBit
Wireguard speed  (23.05.0-rc2) - 118/100 MBit

### OpenWrt version

r23228-cd17d8df2a

### OpenWrt target/subtarget

ipq806x/generic

### Device

ZyXEL NBG6817

### Image kind

Official downloaded image

### Steps to reproduce

1. Set up Wireguard connection
2. Run https://www.speedtest.net

### Actual behaviour

Low wireguard speed  (23.05.0-rc2) - 118/100 MBit

### Expected behaviour

Good enough Wireguard speed  (22.03.5) - 200/200 MBit or more depending on you internet connection.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

luochongjun, 2023-07-04:
In fact, wireguard performance in 22.03 is also much lower than in 19.07, and I currently suspect that nft is causing the problem.

csharper2005, 2023-07-04:
> In fact, wireguard performance in 22.03 is also much lower than in 19.07, and I currently suspect that nft is causing the problem.

On which target did you notice this problem?

I also suspected nft. At the same time in my case both 22.03 and 23.05 contain nft. And I use both nft + iptables because I need mwan3, which depends on iptables.

neheb, 2023-07-04:
Is this with only ipq806x or others as well?

csharper2005, 2023-07-04:
> Is this with only ipq806x or others as well?

The same config on mt7621:
23.05.0-rc2 -105/76
22.03.5 - 118/76

The difference is small.

neheb, 2023-07-04:
so IIRC @Ansuel did some changes to ipq806x having to do with power saving and voltage and whatnot. Might be what causes the slowdown.

Ansuel, 2023-07-04:
@neheb the thing was reverted and now the router just runs always at full speed and I just checked rc2 have the change in it so i'm confused about the perf regression and we should check also other target

luochongjun, 2023-07-05:
> 

I used the target of ATH79 and analyzed it with the perf tool, and found that the usage of nft was normal, but I found that hostapd at the application layer would produce a large number of soft interrupts, resulting in a decline in network performance
the kernel perf result:
[image]
the hostapd perf result
[image]

Obviously, there is code in hostapd that repeatedly alloc memory and occupy sirq. I am trying to find and improve this part of the code. Progress will be updated here

csharper2005, 2023-07-08:
@luochongjun what tool have you used to collect the statistic?

csharper2005, 2023-07-09:
Here is the difference between 5.10 and 5.15 config:
```
-CONFIG_CLKDEV_LOOKUP=y
+CONFIG_CRC8=y
+CONFIG_CRYPTO_GF128MUL=y
+CONFIG_CRYPTO_NULL2=y
+CONFIG_CRYPTO_SHA512=y
+CONFIG_DEVFREQ_GOV_PASSIVE=y
-CONFIG_ETHERNET_PACKET_MANGLE=y
+CONFIG_FWNODE_MDIO=y
+CONFIG_GENERIC_LIB_DEVMEM_IS_ALLOWED=y
-CONFIG_GPIOLIB=y
+CONFIG_GPIO_CDEV=y
-CONFIG_HIGHPTE=y
+CONFIG_KMAP_LOCAL=y
+CONFIG_LLD_VERSION=0
+CONFIG_NET_SELFTESTS=y
+CONFIG_PTP_1588_CLOCK_OPTIONAL=y
+CONFIG_RCU_CPU_STALL_TIMEOUT=21
+CONFIG_SOCK_RX_QUEUE_MAPPING=y
```
I paid attention on:
```
-CONFIG_CLKDEV_LOOKUP=y
+CONFIG_DEVFREQ_GOV_PASSIVE=y
-CONFIG_ETHERNET_PACKET_MANGLE=y
+CONFIG_RCU_CPU_STALL_TIMEOUT=21
```
@Ansuel сan any of these config options reduce performance in your opinion?

csharper2005, 2023-07-09:
https://openwrt.org/docs/guide-user/perf_and_log/benchmark.openssl

1. We can notice md5, sha1, sha256, sha512 degradation (up to 4 times) in OpenWrt 23.05.0-rc2. Can it correlate with wg performance?
2. des-cbc doesn't work
3. Other ciphers aren't impacted ->looks like overall CPU performance is ok 

OpenWrt 22.03.5
```
type             16 bytes     64 bytes    256 bytes   1024 bytes   8192 bytes  16384 bytes
md5              27187.59k    70584.59k   132361.42k   164911.94k   184874.41k   185535.06k
sha1             29958.03k    85160.87k   174107.74k   236847.79k   266836.53k   269197.92k
des cbc          20864.52k    21596.81k    21859.24k    21931.35k    21207.84k    21918.39k
des ede3          8056.96k     8211.24k     8266.66k     8261.63k     7928.90k     8186.46k
aes-128 cbc      54433.19k    59145.14k    60442.97k    59456.66k    60626.30k    60598.95k
aes-192 cbc      49429.58k    53849.07k    55339.61k    55219.72k    54779.90k    55159.47k
aes-256 cbc      45949.63k    48969.21k    50531.17k    50735.26k    50724.86k    50763.09k
sha256           21223.87k    53554.11k   100986.50k   130904.19k   142873.94k   144712.50k
sha512            7851.37k    31696.71k    47754.15k    67216.18k    75745.96k    76507.82k
                  sign    verify    sign/s verify/s
rsa 2048 bits 0.007378s 0.000161s    135.5   6230.1
                  sign    verify    sign/s verify/s
dsa 2048 bits 0.002032s 0.001682s    492.1    594.6
```

OpenWrt 23.05.0-rc2
```
type             16 bytes     64 bytes    256 bytes   1024 bytes   8192 bytes  16384 bytes
md5               6329.45k    23324.06k    68525.38k   131731.80k   179500.37k   184421.37k
sha1              6767.63k    25585.04k    78837.93k   168814.85k   251661.6CE4F5B6:error:0308010C:digital envelope routines:inner_evp_generic_fetch:unsupported:crypto/evp/evp_fetch.c:373:Global default library context, Algorithm (DES-CBC : 18), Properties ()
53k   259387.76k
sha256            6158.45k    20952.62k    58805.10k   105885.35k   138802.17k   141093.55k
sha512            3335.81k    13322.45k    32600.62k    57507.50k    74001.07k    75954.68k
des-cbc              0.00         0.00         0.00         0.00         0.00         0.00
des-ede3          7985.85k     8294.10k     8402.30k     8388.27k     8396.80k     8394.75k
aes-128-cbc      55982.99k    73697.00k    81670.49k    83920.21k    83577.51k    83388.53k
aes-192-cbc      50625.69k    65038.53k    71136.68k    72922.50k    72575.66k    72330.70k
aes-256-cbc      45812.93k    58413.76k    63220.31k    64597.67k    64588.46k    64297.61k
                  sign    verify    sign/s verify/s
rsa 2048 bits 0.007292s 0.000149s    137.1   6695.4
                  sign    verify    sign/s verify/s
dsa 2048 bits 0.002023s 0.001662s    494.2    601.5
```

csharper2005, 2023-07-09:
mt7621 too: md5, sha1, sha256, sha512 degradation (up to 4 times, OpenWrt 23.05.0-rc2)

OpenWrt 22.03.5 (mt7621)
-----------------------------
```
type             16 bytes     64 bytes    256 bytes   1024 bytes   8192 bytes  16384 bytes
md5               9985.13k    27244.06k    55413.28k    74876.51k    82872.56k    83781.57k
sha1              7813.99k    19827.56k    37264.92k    47802.77k    52197.46k    52521.33k
des cbc           6399.34k     6703.54k     6782.04k     6792.76k     6812.15k     6803.99k
des ede3          2357.06k     2398.02k     2407.59k     2410.65k     2414.05k     2411.33k
aes-128 cbc      10527.32k    11761.99k    12141.20k    12220.64k    12258.06k    12263.51k
aes-192 cbc       9266.30k    10213.97k    10477.12k    10560.81k    10578.84k    10581.56k
aes-256 cbc       8271.48k     9008.01k     9231.48k     9289.14k     9305.13k     9296.97k
sha256            4270.55k     9804.08k    17078.35k    21042.35k    22542.97k    22665.44k
sha512             865.84k     3455.53k     4780.06k     6453.58k     7187.73k     7244.89k
                  sign    verify    sign/s verify/s
rsa 2048 bits 0.102041s 0.002679s      9.8    373.3
                  sign    verify    sign/s verify/s
dsa 2048 bits 0.035714s 0.034947s     28.0     28.6
```

OpenWrt 23.05.0-rc2 (mt7621)
-----------------------------
```
type             16 bytes     64 bytes    256 bytes   1024 bytes   8192 bytes  16384 bytes
md5               2113.02k     7835.30k    24560.10k    52510.11k    78474.46k    80891.24k
sha1              2190.85k 8C8DE077:error:0308010C:digital envelope routines:inner_evp_generic_fetch:unsupported:crypto/evp/evp_fetch.c:373:Global default library context, Algorithm (DES-CBC : 18), Properties ()
    7663.78k    21333.30k    38557.51k    50311.40k    51576.83k
sha256            1677.65k     5187.74k    12309.01k    18733.08k    22126.56k    22387.84k
sha512             549.08k     2197.73k     4270.27k     6648.52k     7936.17k     8039.59k
des-cbc              0.00         0.00         0.00         0.00         0.00         0.00
des-ede3          2256.98k     2366.13k     2398.07k     2405.55k     2408.61k     2405.89k
aes-128-cbc      10474.14k    13647.29k    14754.70k    15044.98k    15145.67k    15148.40k
aes-192-cbc       9394.23k    11874.50k    12658.65k    12877.57k    12922.13k    12938.46k
aes-256-cbc       8491.93k    10437.36k    11079.19k    11250.73k    11283.73k    11283.73k
                  sign    verify    sign/s verify/s
rsa 2048 bits 0.095429s 0.002415s     10.5    414.0
                  sign    verify    sign/s verify/s
dsa 2048 bits 0.031943s 0.029970s     31.3     33.4
```

DragonBluep, 2023-07-10:
> 1. We can notice md5, sha1, sha256, sha512 degradation (up to 4 times) in OpenWrt 23.05.0-rc2.
>    **Can it be correlated with wg performance?**

@csharper2005 I think, yes. The [Official introduction](https://www.wireguard.com/performance/) mentioned SHA and AES. 

Perhaps you need to remove all mbedtls/wolfssl related packages before installing openssl?

luochongjun, 2023-07-10:
The issue was resolved when we replaced the built-in wireguard in the kernel with the wireguard version in 19.07.
https://github.com/gl-inet/gl-infra-builder/commit/81f46803e989349f6f8cd1fcc6976cbc1e635d29

luochongjun, 2023-07-10:
> @luochongjun what tool have you used to collect the statistic?

collect the statistic with perf tool

systemcrash, 2023-07-18:
> The issue was resolved when we replaced the built-in wireguard in the kernel with the wireguard version in 19.07. [gl-inet/gl-infra-builder@81f4680](https://github.com/gl-inet/gl-infra-builder/commit/81f46803e989349f6f8cd1fcc6976cbc1e635d29)

How about the version in 22? (same?)

olemmela, 2023-08-21:
@csharper2005 could you test if #13323 fixes wireguard performance issues too. This could be related to #11676

csharper2005, 2023-08-21:
> @csharper2005 could you test if #13323 fixes wireguard performance issues too. This could be related to #11676

Thanks for your investigation and fix. It looks like the root cause is the same. Couldn't make a relevant test right now, because currently my internet speed is less than 100 mbit/s.