openwrt/openwrt #13050: 23.05-rc bricks WNDR4300v2

mb555444, 2023-07-04:
### Describe the bug

Router ends up with a blinking green light.
Responds to ping at 192.168.1.1.
Does not allow web or ssh access.

See thread at https://forum.openwrt.org/t/23-05-sysupgrade-questions-wndr4300v2/160959/60

This happens in all of the following cases:
- flash via sysupgrade image from earlier working version of OpenWrt
- flash openwrt-23.05.0-rc2-ath79-nand-netgear_wndr4300-v2-squashfs-factory.img from the Netgear Web UI
- flash openwrt-23.05.0-rc2-ath79-nand-netgear_wndr4300-v2-squashfs-factory.img via nmrpflash.exe

### OpenWrt version

23.05-rc2

### OpenWrt target/subtarget

ath79/nand

### Device

WNDR4300v2

### Image kind

Official downloaded image

### Steps to reproduce

1. flash openwrt-23.05.0-rc2-ath79-nand-netgear_wndr4300-v2-squashfs-factory.img from the Netgear Web UI
2. reboot router

### Actual behaviour

Only responds to ping, no ssh/web access

### Expected behaviour

Should boot OpenWRT and display web UI

### Additional info

See https://forum.openwrt.org/t/23-05-sysupgrade-questions-wndr4300v2/160959/60

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

yousaf465, 2023-07-05:
https://forum.openwrt.org/t/23-05-sysupgrade-questions-wndr4300v2/160959/64 

logs from my router can be found at this link

mb555444, 2023-07-06:
@DragonBluep I am no expert but saw reent changes in https://github.com/openwrt/openwrt/pull/11395 - could this be the cause of the issue?

DragonBluep, 2023-07-07:
I have checked the u-boot source code and it should be fine. Have you tried the snapshot version?

DragonBluep, 2023-07-07:
Maybe you can try this image.
[openwrt-ath79-nand-netgear_wndr4300-v2-squashfs-factory.zip](https://github.com/openwrt/openwrt/files/11973720/openwrt-ath79-nand-netgear_wndr4300-v2-squashfs-factory.zip)

yousaf465, 2023-07-07:
> Maybe you can try this image. [openwrt-ath79-nand-netgear_wndr4300-v2-squashfs-factory.zip](https://github.com/openwrt/openwrt/files/11973720/openwrt-ath79-nand-netgear_wndr4300-v2-squashfs-factory.zip)

is it fixed in this snapshot?

DragonBluep, 2023-07-07:
> > Maybe you can try this image. [openwrt-ath79-nand-netgear_wndr4300-v2-squashfs-factory.zip](https://github.com/openwrt/openwrt/files/11973720/openwrt-ath79-nand-netgear_wndr4300-v2-squashfs-factory.zip)
> 
> is it fixed in this snapshot?

I do not know. Tester needed.

mb555444, 2023-07-07:
I will give it a try this evening and report back.

mb555444, 2023-07-07:
I tested the attached file. This firmware is quite remarkable, as it not only solves the current issue but also an older issue that I always had with 21.x and 22.x
- the router boots up after flashing the image with nmrpflash.exe (ssh/http is reachable)
- it also boots up again after the first reboot, even without the sysupgrade flash workaround from https://github.com/openwrt/openwrt/issues/8878 that I always needed in the past

yousaf465, 2023-07-07:
I will try in morning. if it found working this should be merged.

DragonBluep, 2023-07-08:
Can you try this version: [wndr4300-v2.tar.gz](https://github.com/openwrt/openwrt/files/11987156/wndr4300-v2.tar.gz)

DragonBluep, 2023-07-08:
> I tested the attached file. This firmware is quite remarkable, as it not only solves the current issue but also an older issue that I always had with 21.x and 22.x
> 
> * the router boots up after flashing the image with nmrpflash.exe (ssh/http is reachable)
> * it also boots up again after the first reboot, even without the sysupgrade flash workaround from [FS#3853 - ath79: NETGEAR WNDR4300v2 Bricked on Power Cycle After Flash #8878](https://github.com/openwrt/openwrt/issues/8878) that I always needed in the past
> * the following error in the kernel log (there was also a stack trace) is now also gone:
>   `ubi0 warning: 0x8046a910: error -5 while erasing PEB 991, retry`
> 
> What are the differences (seems like partition size is at least different)? Was it a local build?

It's a local build. Yes, partition size has changed. https://github.com/openwrt/openwrt/issues/8878 should be a flash driver issue. Your NAND Flash block 1023 is bad but driver does not skip it.

yousaf465, 2023-07-08:
> Can you try this version: [wndr4300-v2.tar.gz](https://github.com/openwrt/openwrt/files/11987156/wndr4300-v2.tar.gz)

this resulted in green flashing power led. 

> Maybe you can try this image. [openwrt-ath79-nand-netgear_wndr4300-v2-squashfs-factory.zip](https://github.com/openwrt/openwrt/files/11973720/openwrt-ath79-nand-netgear_wndr4300-v2-squashfs-factory.zip)

this worked like charm. 

[image]

DragonBluep, 2023-07-08:
Can you help test which of the following images brick your device?
[1] [1-netgear_wndr4300-v2-lzma.tar.gz](https://github.com/openwrt/openwrt/files/11988821/1-netgear_wndr4300-v2-lzma.tar.gz)
[2] [2-netgear_wndr4300-v2-0xff.tar.gz](https://github.com/openwrt/openwrt/files/11988822/2-netgear_wndr4300-v2-0xff.tar.gz)
[3] [3-netgear_wndr4300-v2-shrink.tar.gz](https://github.com/openwrt/openwrt/files/11988823/3-netgear_wndr4300-v2-shrink.tar.gz)

yousaf465, 2023-07-08:
> Can you help test which of the following images brick your device? [1] [1-netgear_wndr4300-v2-lzma.tar.gz](https://github.com/openwrt/openwrt/files/11988821/1-netgear_wndr4300-v2-lzma.tar.gz) [2] [...]

all to b flashed via nmrp?

DragonBluep, 2023-07-08:
> all to b flashed via nmrp?

Yes, you can also use some tftp server like `tftpd64` to upload them.

yousaf465, 2023-07-08:
> Yes, you can also use some tftp server like `tftpd64` to upload them.

1.Power led stable green. Both 2.4 and 5GHz stable. ethernet orange. doesn't respond to ping. same after restart. 
2. Green blinking power led. but responds to ping at 192.168.1.1
3. after flashing 2. had to use tftp to flash 3. Power led stable green. Both 2.4 and 5GHz stable. ethernet orange. but not responding to ping. boot looping after restart.

DragonBluep, 2023-07-08:
> > > > Can you help test which of the following images brick your device? [1] [1-netgear_wndr4300-v2-lzma.tar.gz](https://github.com/openwrt/openwrt/files/11988821/1-netgear_wndr4300-v2-lzma.tar.gz) [2] [2-netgear_wndr4300-v2-0xff.tar.gz](https://github.com/openwrt/openwrt/files/11988822/2-netgear_wndr4300-v2-0xff.tar.gz) [3] [3-netgear_wndr4300-v2-shrink.tar.gz](https://github.com/openwrt/openwrt/files/11988823/3-netgear_wndr4300-v2-shrink.tar.gz)
> > > 
> > > 
> > > all to b flashed via nmrp?
> > 
> > 
> > Yes, you can also use some tftp server like `tftpd64` to upload them.
> 
> @DragonBluep 1.Power led stable green. Both 2.4 and 5GHz stable. ethernet orange. doesn't respond to ping. same after restart. but somehow now bootlooping. 2. Green blinking power led. but responds to ping at 192.168.1.1 3. after flashing 2. had to use tftp to flash 3. Power led stable green. Both 2.4 and 5GHz stable. ethernet orange. but not responding to ping. boot looping after restart.

[1] and [3] should fine as wifi works. Reboot issue is related to https://github.com/openwrt/openwrt/issues/8878

Can you double check [2] [2-netgear_wndr4300-v2-0xff.tar.gz](https://github.com/openwrt/openwrt/files/11988822/2-netgear_wndr4300-v2-0xff.tar.gz) and previous:

> Can you try this version: [wndr4300-v2.tar.gz](https://github.com/openwrt/openwrt/files/11987156/wndr4300-v2.tar.gz)

It doesn't make sense that they can't work.

yousaf465, 2023-07-08:
> [1] and [3] should fine as wifi works. Reboot issue is related to #8878
>
> Can you double check [2] [...]

1. after connecting via WIFI it worked. 

[image]

2. flashed via nmrp, green power led blinking. flashed again via tftp same result. 
3.  working via wifi 
[image]

4. [wndr4300-v2.tar.gz](https://github.com/openwrt/openwrt/files/11987156/wndr4300-v2.tar.gz) flashed via tftp after 2. responds to ping but green power led blinking.

DragonBluep, 2023-07-08:
Thanks for your report && test.

This time should be the final test. If it can boot, I will send the patch to the mainline.
[netgear_wndr4300-final.tar.gz](https://github.com/openwrt/openwrt/files/11990353/netgear_wndr4300-final.tar.gz)

> 1.Power led stable green. Both 2.4 and 5GHz stable. ethernet orange. doesn't respond to ping. same after restart. but somehow now bootlooping.

You have to change your computer IP from static to dynamic/auto after using nmrp/tftp. And re-plug net cable may necessary.

As for reboot issue, there is a workaround for you. You can patch and local build your own image. Since it would brick the normal running devices, so I cannot upstream it.
```
--- a/target/linux/ath79/dts/qca9563_netgear_wndr.dtsi
+++ b/target/linux/ath79/dts/qca9563_netgear_wndr.dtsi
@@ -167,8 +167,11 @@
 
                        partition@400000 {
                                label = "ubi";
-                               reg = <0x400000 0x7c00000>;
+                               reg = <0x400000 0x7800000>;
                        };
+
+                       /* The final block #1023 is a bad block on some devices.
+                          Reserve 4 MiB space as redundancy to workaround it. */
                };
        };
 };
```

yousaf465, 2023-07-08:
> Thanks for your report && test.
>
> This time should be the final test. If it can boot, I will send the patch to the mainline. [...]

it is actually who should thank you for fixing this almost unfixable issue with my router. 
At least I am able to run 23 via this.

I am using a cheap USB to ethernet adapter to flash. 

let me test this before posting.

yousaf465, 2023-07-08:
[image]

[netgear_wndr4300-final.tar.gz](https://github.com/openwrt/openwrt/files/11990353/netgear_wndr4300-final.tar.gz)

1.Power led stable green. Both 2 wifi leds working. ping not responding on ethernet (no wan connected). 
[image]

yousaf465, 2023-07-08:
new issue, I can access the router page, but can't access the internet. Ip is assigned automatically, even though DHCP server ISP's modem

DragonBluep, 2023-07-08:
> new issue, I can access the router page, but can't access the internet. Ip isn't assigned automatically, even though DHCP server ISP's modem. If I assign an Ip manually I can only router the homepage at its IP 192.168.18.24.

Maybe some of my local commits break it. It will surely be fixed on official release. Can you try if this version fixes the net issue and the reboot issue? Please show me the system log.
[netgear_wndr4300-nandfix.tar.gz](https://github.com/openwrt/openwrt/files/11990546/netgear_wndr4300-nandfix.tar.gz)

yousaf465, 2023-07-08:
> Maybe some of my local commits break it. It will surely be fixed on official release. Can you try if this version fixes the net issue and the reboot issue? Please show me the system log. [...]

love the dashboard but no internet.
[image]

had to access it after setting a static IP on the laptop for the wifi card. 

how do i send the logs?

yousaf465, 2023-07-08:
https://gist.github.com/yousaf465/b987830d7007cb9851646652f94d82fd 

system startup log.

DragonBluep, 2023-07-08:
> https://gist.github.com/yousaf465/b987830d7007cb9851646652f94d82fd
> 
> system startup log.

It seems that you link the router wan port to your computer? Please use lan port. 

Does it have the reboot failure issue? If you want, you can provide your name and email to me, I can add them to the commit messages. These issues should be able to be fixed in the next official release.

yousaf465, 2023-07-08:
> It seems that you link the router wan port to your computer? Please use lan port.
>
> Does it have the reboot failure issue? If you want, you can provide your name and email to me, I can add them to [...]

reboot failure issue is no more there.

I am connecting via LAN port, tried all 4.  
Yousaf 
yousaf465@gmail.com

DragonBluep, 2023-07-08:
Patches have been sent to the mailinglist:
https://patchwork.ozlabs.org/project/openwrt/patch/OSYP286MB0312AE84EA07594488BD85C0BC32A@OSYP286MB0312.JPNP286.PROD.OUTLOOK.COM/
https://patchwork.ozlabs.org/project/openwrt/patch/OSYP286MB0312844FCADC602A4BA8EBC5BC32A@OSYP286MB0312.JPNP286.PROD.OUTLOOK.COM/

DragonBluep, 2023-07-08:
> > > https://gist.github.com/yousaf465/b987830d7007cb9851646652f94d82fd
> > > system startup log.
> > 
> > 
> > It seems that you link the router wan port to your computer? Please use lan port.
> > Does it have the reboot failure issue? If you want, you can provide your name and email to me, I can add them to the commit messages. These issues should be able to be fixed in the next official release.
> 
> reboot failure issue is no more there.
> 
> I am connecting via LAN port, tried all 4. Yousaf [yousaf465@gmail.com](mailto:yousaf465@gmail.com)

As these patches will not affect internet function. Perhaps you can temporarily use openwrt 22.03 as a workaround. Or you can get some help from the forum.

yousaf465, 2023-07-08:
> /s

That is documented here.
https://forum.openwrt.org/t/connect-to-huwaei-sun2000-inverter-using-fiber-modem/104635/23?u=koa

I think the firewall might need reconfiguring.

yousaf465, 2023-07-08:
> > > > https://gist.github.com/yousaf465/b987830d7007cb9851646652f94d82fd
> > > > system startup log.
> > > 
> > > 
> > > It seems that you link the router wan port to your computer? Please use lan port.
> > > Does it have the reboot failure issue? If you want, you can provide your name and email to me, I can add them to the commit messages. These issues should be able to be fixed in the next official release.
> > 
> > 
> > reboot failure issue is no more there.
> > I am connecting via LAN port, tried all 4. Yousaf [yousaf465@gmail.com](mailto:yousaf465@gmail.com)
> 
> As these patches will not affect internet function. Perhaps you can temporarily use openwrt 22.03 as a workaround. Or you can get some help from the forum.
> 
> It is very strange your lan ip is 192.168.18.x but your wifi ip is 192.168.200.x. Reset and re-config it should be a good choice.
> 
> ```
> Sat Jul  8 17:27:07 2023 daemon.notice netifd: wlan (2329): udhcpc: broadcasting select for 192.168.200.2, server 192.168.200.1
> Sat Jul  8 17:30:01 2023 daemon.err uhttpd[1454]: [info] luci: accepted login on /admin/system/reboot for root from 192.168.18.126
> ```

https://github.com/openwrt/openwrt/issues/13050#issuecomment-1624801861 This version was working fine, one of these three also worked I think 

https://github.com/openwrt/openwrt/issues/13050#issuecomment-1626896355

mb555444, 2023-07-08:
Hi - thanks for the efforts put into the issue(s) on these routers. Which image is the latest one implementing the patches sent to be merged to the main branch? I would like to test the final version.

DragonBluep, 2023-07-09:
@mb555444 You can try this one. https://github.com/openwrt/openwrt/issues/13050#issuecomment-1627168801

mb555444, 2023-07-09:
> @mb555444 You can try this one. [#13050 (comment)](https://github.com/openwrt/openwrt/issues/13050#issuecomment-1627168801)

Thanks. With the "nandfix" image, router boots up and also comes up after the first reboot.
I had connectivity issues and the "Switch" menu item under "Network" was missing. Also new status page. I assume this is due to a different overall version of the source and may be unrelated to the partition/nand issues.

DragonBluep, 2023-07-09:
Update: add the missing package, fix net issue.
[netgear_wndr4300-v2.tar.gz](https://github.com/openwrt/openwrt/files/11995271/netgear_wndr4300-v2.tar.gz)

yousaf465, 2023-07-09:
> nnectivity issues and the "Switch" menu item under "Network" was missing. Also new status page. I assume this is due to a different overall version of the s

will test it soon.

mb555444, 2023-07-09:
> Update: add the missing package, fix net issue. [netgear_wndr4300-v2.tar.gz](https://github.com/openwrt/openwrt/files/11995271/netgear_wndr4300-v2.tar.gz)

Thanks, net issue is fixed and overall works well.

yousaf465, 2023-07-10:
> Update: add the missing package, fix net issue. [netgear_wndr4300-v2.tar.gz](https://github.com/openwrt/openwrt/files/11995271/netgear_wndr4300-v2.tar.gz)

@mb555444 does it include the new dashboard? I love that concept.

DragonBluep, 2023-07-10:
> > Update: add the missing package, fix net issue. [netgear_wndr4300-v2.tar.gz](https://github.com/openwrt/openwrt/files/11995271/netgear_wndr4300-v2.tar.gz)
> 
> @mb555444 does it include the new dashboard? I love that concept.

Yes. It's an official support package.

yousaf465, 2023-07-11:
> > > Update: add the missing package, fix net issue. [netgear_wndr4300-v2.tar.gz](https://github.com/openwrt/openwrt/files/11995271/netgear_wndr4300-v2.tar.gz)
> > 
> > 
> > @mb555444 does it include the new dashboard? I love that concept.
> 
> Yes. It's an official support package.
[image]

it is confirmed as working with a restored backup of config. 

it was just missing a few hostnames, which were present in one of the above versions. 
[image]

yousaf465, 2023-07-12:
> Patches have been sent to the mailinglist: https://patchwork.ozlabs.org/project/openwrt/patch/OSYP286MB0312AE84EA07594488BD85C0BC32A@OSYP286MB0312.JPNP286.PROD.OUTLOOK.COM/ [...]

how will we know when these have been merged so that we can do a sysupgrade to the latest snapshots?

mb555444, 2023-07-29:
> how will we know when these have been merged so that we can do a sysupgrade to the latest snapshots?

I would also like to know that. Would it help to open a PR here on github?

DragonBluep, 2023-07-29:
@blocktrron Hi! Can you help to review & pull these two patches to the main/23.05 branch? 
https://patchwork.ozlabs.org/project/openwrt/list/?series=363021

The fixes are already verified by @mb555444 and @yousaf465. We may need some higher priority to unbrick these devices. 

Thank you for your efforts in the community.

yousaf465, 2023-07-31:
> @blocktrron Hi! Can you help to review & pull these two patches to the main/23.05 branch? https://patchwork.ozlabs.org/project/openwrt/list/?series=363021
>
> The fixes are already verified by [...]

with these patches at least we could use the latest snapshot without any issues.

yousaf465, 2023-08-23:
fixed in OpenWrt 23.05.0-rc3
 or not?

mb555444, 2023-08-23:
> fixed in OpenWrt 23.05.0-rc3 or not?

I'm afraid not. Patches have been submitted to https://patchwork.ozlabs.org where it seems they take a long time to be processed. In fact, I see many patches still in state "NEW" after 1-2 years.

DragonBluep, 2023-08-23:
@hauke @Ansuel Hi! Can you help to have a review on this patch? It has been here for a long time.
https://patchwork.ozlabs.org/project/openwrt/list/?series=363021

Regards
Shiji Yang

closed 2023-08-23 by openwrt-bot

yousaf465, 2023-08-24:
commit is done? so now we can use the image builder and official snapshots?

DragonBluep, 2023-08-24:
> commit is done? so now we can use the image builder and official snapshots?

@yousaf465 Yes, perhaps you can get the usable snapshot image tomorrow. Backport is still required for branch 23.05.

yousaf465, 2023-08-24:
I am okay with a snapshot but it will not work with 23.05rc3. for rc needs to wait for rc4?

blocktrron, 2023-08-24:
Backported to the 23.05 release branch.

yousaf465, 2023-08-29:
@mb555444 were you able to use the latest snapshot? I will flash once I get free time.