openwrt/openwrt #13059: DHCP option 121 makes OpenWrt get no default GW when connecting as dhcp client

ngrcld, 2023-07-05:
### Describe the bug

When the DHCP server sends option 121 to a OpenWrt configured as dhcp client, OpenWrt get no default GW

Bug existing at least since OpenWrt 18.06.5: see https://github.com/openwrt/openwrt/issues/8519

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

ramips/mt76x8

### Device

Xiaomi Mi Router 4A (100M Edition)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jmv2009, 2023-07-06:
Able to get a default gw just fine. Actually always have to disable default gw on this iptv wan to not get one in the routing table. In my normal operation, udchpc does request the 121 option, and it is getting the 121 option route, and it routes over this wan iptv wan just fine without default gw.

ngrcld, 2023-07-07:
My option 121 is specified like this, in Advanced Settings / DHCP-Options in OpenWrt used as dhcp server:
121,192.168.100.0/24,192.168.100.1,192.168.150.0/24,192.168.100.1,192.168.200.0/24,192.168.100.1

In OpenWrt used as dhcp client, not only default gateway is missing, but also the "scope link" route is missing.

If I remove option 121 from dhcp server (or I use option 249 for Windows dhcp clients only), in OpenWrt used as dhcp client, the "scope link" route is ok, and the default gateway is ok.

In issue #8519 there is the output of "ip r" showing the problem.

jmv2009, 2023-07-08:
So your test environment consists of using one openwrt system as a server and one system as a client?

ngrcld, 2023-07-08:
> So your test environment consists of using one openwrt system as a server and one system as a client?

Yes

jmv2009, 2023-07-08:
Hmm, can't owrt have accept the static routes yet from an owrt server. Maybe the static address and range need to be consistent? Please let us know more details about the config you are trying to test, that may help.

ngrcld, 2023-07-09:
I started from scratch with a test environment with two OpenWrt routers, this is what I did:

- on both routers: fresh install of OpenWrt 22.03.5

- on both routers: set password, hostname, timezone

- on both routers: set wan interface with static IP (each router in a different subnet), and dhcp server, and firewall with input accept for wan zone (so I can connect from pc, as the routers have only one port for wan and only one port for lan)

- connected my pc alternatively to one of the wan ports of the two routers

- on routerA, leaved lan interface with static IP 192.168.1.1/24 with dhcp server enabled

- on routerB, set lan interface as dhcp client, and disabled dhcp server

- connected the two routers via their lan port

- now the status is:

routerA
LAN static, ip 192.168.1.1/24 (with dhcp server enabled)

```
root@routerA:~# ip r
192.168.1.0/24 dev br-lan scope link  src 192.168.1.1
192.168.2.0/24 dev eth0.2 scope link  src 192.168.2.1
```

routerB
LAN dhcp client, ip 192.168.1.114/24

```
root@routerB:~# ip r
default via 192.168.1.1 dev br-lan  src 192.168.1.114
192.168.1.0/24 dev br-lan scope link  src 192.168.1.114
192.168.3.0/24 dev eth0.2 scope link  src 192.168.3.1
```

(so everithing is ok for now)

- now the test begins: I connected to routerA, and added the option in: interface lan -> DHCP Server -> Advanced Settings -> DHCP-Options:
121,192.168.1.0/24,192.168.1.1,192.168.20.0/24,192.168.1.1,192.168.30.0/24,192.168.1.1

- I connected to routerB, and restarted the lan interface

- now the status is:

```
root@routerB:~# ip r
192.168.1.0/24 via 192.168.1.1 dev br-lan  src 192.168.1.114
192.168.3.0/24 dev eth0.2 scope link  src 192.168.3.1
```

(so there is no default gateway, and no "scope link" route for br-lan)

I spent nearly 8 hours of tests because initially the test did not succeed (everything was fine), than I found the problem: this bug occurs only if in option 121 there is specified a route that is similar to the default gateway sent by dhcp server, in this case the first route sent: 192.168.1.0/24,192.168.1.1

So, in your test, I think you should see the problem setting:
121,10.10.0.0/16,10.10.0.1

May seem strange sending a route with option 121 similar to the default gateway, but in my normal (non-test) scenario (network with also Windows PC as clients), route sent with option 121 overrides a less specific route set by a VPN software.

jmv2009, 2023-07-09:
Hmm. applying 121,10.10.0.0/16,10.10.0.1 still generated a default route for me (client is a new snapshot x86 vm)

ngrcld, 2023-07-10:
I'll use option 249 to solve my specific problem, however neither Linux nor Windows follow what RFC3442 says, and the missing "scope link" route suggests more of a bug than a deliberate choice. Thanks for your time.