openwrt/openwrt #13063: /etc/board.json no longer accurate about network devices

Steve-Newcomb, 2023-07-06:
### Describe the bug

In the past, this info in `/etc/board.json` was very helpful in scripted configuration tweaks:
```
...
    'network': { 
        'lan': {
            'device': 'eth0.1',
            'protocol': 'static',
        },
        'wan': {
            'device': 'eth0.2',
            'macaddr': '50:c7:bf:78:37:fd',
            'protocol': 'dhcp',
        },
    },
...
```
Now in `r23479` (presumably nearly equivalent to the current 22.03xx), there's only this, which is incomplete and inaccurate:
```
...
    'network': {
        'lan': {
            'device': 'eth0',
            'protocol': 'static',
        },
    },
...
```
I gather that, with sufficient patience, I can still derive accurate, complete information from `/etc/board.d/02-network`, but it seems a pity to lose the single centralized self-description that used to be contained in `/etc/board.json`.

### OpenWrt version

r23479-f24c9b9d86

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer A7 v5

### Image kind

Self-built image

### Steps to reproduce

(1) Build the image and install it.
(2) Examine /etc/board.json.

### Actual behaviour

The information in /etc/board.json is wrong and incomplete with respect to the actual network devices.

### Expected behaviour

I expected to see the accurate, complete information that was present in previous versions of OpenWRT.

### Diffconfig

```text
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_tplink_archer-a7-v5=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_ATH10K_LEDS=y
CONFIG_BATMAN_ADV_BATMAN_V=y
CONFIG_BATMAN_ADV_BLA=y
CONFIG_BATMAN_ADV_DAT=y
CONFIG_BATMAN_ADV_MCAST=y
CONFIG_BUSYBOX_CONFIG_DIFF=y
CONFIG_BUSYBOX_CONFIG_FEATURE_EDITING_SAVEHISTORY=y
CONFIG_BUSYBOX_CONFIG_HOSTNAME=y
CONFIG_BUSYBOX_CONFIG_IONICE=y
# CONFIG_BUSYBOX_CONFIG_KILLALL is not set
CONFIG_BUSYBOX_CONFIG_PKILL=y
# CONFIG_BUSYBOX_CONFIG_VI is not set
CONFIG_KERNEL_BUILD_DOMAIN="coolheads.com"
CONFIG_KERNEL_BUILD_USER="srn"
CONFIG_KERNEL_DETECT_HUNG_TASK=y
CONFIG_KERNEL_HARDLOCKUP_DETECTOR=y
CONFIG_KERNEL_SOFTLOCKUP_DETECTOR=y
CONFIG_KERNEL_WQ_WATCHDOG=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_ath10k-firmware-qca988x=y
# CONFIG_PACKAGE_ath10k-firmware-qca988x-ct is not set
CONFIG_PACKAGE_batctl-full=y
CONFIG_PACKAGE_block-mount=y
CONFIG_PACKAGE_e2fsprogs=y
CONFIG_PACKAGE_iptables-mod-extra=y
CONFIG_PACKAGE_iptables-mod-ipopt=y
CONFIG_PACKAGE_iptables-mod-nat-extra=y
CONFIG_PACKAGE_kmod-ath10k=y
# CONFIG_PACKAGE_kmod-ath10k-ct is not set
CONFIG_PACKAGE_kmod-batman-adv=y
CONFIG_PACKAGE_kmod-fs-ext4=y
CONFIG_PACKAGE_kmod-ipt-core=y
CONFIG_PACKAGE_kmod-ipt-extra=y
CONFIG_PACKAGE_kmod-ipt-ipopt=y
CONFIG_PACKAGE_kmod-ipt-nat=y
CONFIG_PACKAGE_kmod-ipt-nat-extra=y
CONFIG_PACKAGE_kmod-lib-crc16=y
CONFIG_PACKAGE_kmod-nf-ipt=y
CONFIG_PACKAGE_kmod-scsi-core=y
CONFIG_PACKAGE_kmod-usb-storage=y
CONFIG_PACKAGE_kmod-usb-storage-uas=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libblkid=y
CONFIG_PACKAGE_libbz2=y
CONFIG_PACKAGE_libcomerr=y
CONFIG_PACKAGE_libext2fs=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libparted=y
CONFIG_PACKAGE_libpcre=y
CONFIG_PACKAGE_libpopt=y
CONFIG_PACKAGE_libpython3=y
CONFIG_PACKAGE_libreadline=y
CONFIG_PACKAGE_librt=y
CONFIG_PACKAGE_libss=y
CONFIG_PACKAGE_libuuid=y
CONFIG_PACKAGE_libxtables=y
CONFIG_PACKAGE_mailsend=y
CONFIG_PACKAGE_nano=y
CONFIG_PACKAGE_openssh-client=y
CONFIG_PACKAGE_openssh-keygen=y
CONFIG_PACKAGE_openssh-server=y
CONFIG_PACKAGE_parted=y
CONFIG_PACKAGE_procps-ng=y
CONFIG_PACKAGE_procps-ng-ps=y
CONFIG_PACKAGE_python3-asyncio=y
CONFIG_PACKAGE_python3-base=y
CONFIG_PACKAGE_python3-codecs=y
CONFIG_PACKAGE_python3-idna=y
CONFIG_PACKAGE_python3-light=y
CONFIG_PACKAGE_rsync=y
CONFIG_PACKAGE_screen=y
CONFIG_PACKAGE_sudo=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_wget-ssl=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-mesh-mbedtls=y
CONFIG_PACKAGE_zlib=y
CONFIG_PARTED_READLINE=y
# CONFIG_OPENSSH_LIBFIDO2 is not set
CONFIG_PACKAGE_kmod-hwmon-core=y
# CONFIG_PACKAGE_libcbor is not set
# CONFIG_PACKAGE_libevdev is not set
# CONFIG_PACKAGE_libfido2 is not set
# CONFIG_PACKAGE_libudev-zero is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-07-06:
It is overridden MAC address from /etc/config/network which is no longer the case.

Steve-Newcomb, 2023-07-07:
My point is unrelated to mac addresses.  My point is that it is still true that the lan device is eth0.1, not eth0 as reported in board.json, and the wan device is still eth0.2, and it exists, although it is not mentioned in board.json.  

I don't understand your comment about the mac address.

john-tho, 2023-07-07:
Likely the forums are a better place to help you address this.

> lose the single centralized self-description

`/etc/board.json` is only used to store the initial (first boot default) configuration needed for `uci`, and is not meant to be an exhaustive or stable interface.

If it was wrong and incomplete, you would have issues with the default network configuration (are LAN or WAN broken, or using a wrong MAC addresses then?). If so, please detail and help fix.

Have a look at the OpenWrt utilities:`ubus`, `uci`, `swconfig`, or the lower level utilities `ip`, `brctl`, `bridge`.

brada4, 2023-07-07:
> ? MAC addresses

The MAC address in board.json is no more. It used to be MAC on WAN overriden by same hardware MAC before (the only difference I  spotted in your outputs). 
It is worth asking in forums on how to summarize HW info akin board.json for your management system/scripts.

Steve-Newcomb, 2023-07-07:
Thanks, John-tho, for your tip: "Have a look at the OpenWrt utilities: ubus, uci, swconfig, or the lower level utilities ip, brctl, bridge." 

[I wonder what, if anything, `board.json` can be "relied upon" **for**.](https://openwrt.org/releases/21.02/notes-21.02.0-rc3#new_network_configuration_syntax_and_boardjson_change)  Perhaps I should just ignore it.

closed 2023-07-07 by Steve-Newcomb

Steve-Newcomb, 2023-07-08:
[I eventually found exactly what I needed.](https://openwrt.org/docs/guide-developer/network-scripting#examples)