openwrt/openwrt #13065: ath10k: failed to flush transmit queue with buggy client

nazar-pc, 2023-07-07:
### Describe the bug

This is a follow-up to https://github.com/openwrt/openwrt/issues/13053 where I incorrectly identified the source of the issue.

The reason seems to be [Govee TV Backlight T2](https://www.amazon.com/gp/product/B0BCQ9YQYW) when I block its access to the Internet on firewall level.

Here are logs of that device mac address filtered by mac address (I'm saving full logs in case they are requested):
<details>

```
Fri Jul  7 06:07:42 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:07:47 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:08:03 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:08:03 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:03 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: associated (aid 5)
Fri Jul  7 06:08:03 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: associated (aid 5)
Fri Jul  7 06:08:03 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: associated (aid 5)
Fri Jul  7 06:08:18 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:18 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:18 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:18 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:18 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:23 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:23 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:23 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:23 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:23 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:28 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:33 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: associated (aid 0)
Fri Jul  7 06:08:33 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:33 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:08:33 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:08:33 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:08:33 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:08:39 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:08:39 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:08:44 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:08:44 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:08:44 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:08:44 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:08:49 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:49 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:49 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:49 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:49 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:08:49 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:08:49 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:08:49 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:08:54 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:08:54 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:54 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:54 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:08:59 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:09:20 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:09:20 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:09:20 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:09:20 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:09:25 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: associated (aid 5)
Fri Jul  7 06:09:25 2023 daemon.err hostapd: nl80211: NL80211_ATTR_STA_VLAN (addr=d4:ad:fc:2c:f8:a8 ifname=wlan1-1 vlan_id=0) failed: -2 (No such file or directory)
Fri Jul  7 06:09:25 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: associated (aid 5)
Fri Jul  7 06:09:25 2023 daemon.err hostapd: nl80211: NL80211_ATTR_STA_VLAN (addr=d4:ad:fc:2c:f8:a8 ifname=wlan1-1 vlan_id=0) failed: -2 (No such file or directory)
Fri Jul  7 06:09:30 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:09:30 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:09:30 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:09:35 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:09:35 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:09:35 2023 daemon.notice hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: did not acknowledge authentication response
Fri Jul  7 06:09:45 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:09:45 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
Fri Jul  7 06:09:45 2023 daemon.info hostapd: wlan1-1: STA d4:ad:fc:2c:f8:a8 IEEE 802.11: authenticated
[... 163 lines cut ...]

* Have OpenWRT as a dumb Access Point (no DHCP, no DSN, just Wi-Fi bridged to LAN).
* Use Compex WLE900VX (QCA9880) wireless adapter in 2.4 GHz mode with ath10k driver (I installed ath10k-board-qca988x, ath10k-firmware-qca988x and kmod-ath10k packages for this)
* Have Govee TV Backlight T2 connected to Wi-Fi
* Block Govee TV Backlight T2 access to Internet on the firewall level (such that it connects to Wi-Fi, gets IP address and everything, but all of the packets going to the Internet are dropped)

### Actual behaviour

Observe Wi-Fi issues as described above

### Expected behaviour

No Wi-Fi issues

### Additional info

[... cut ...] 84, using 5GHz), which when issues happen on 2.4 GHz is also affected for some reason.

Might be a bug in the upstream driver that affects all adapters using the driver or the whole OS, not sure.

When reboot is attempted, it is slow to shut down, after reboot it fails to initialize the card and soon after locks up the whole host machine completely (this is the first time I see PCIe card in a VM locking up the host).

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-07-07:
non-candela firmware seems to not implement per-client queue flush aka CVE-2022-47522 fix. Try -ct firmware and kmod while waiting for qca firmware/driver upgrade, which is default on almost all applicable embedded devices.

nazar-pc, 2023-07-07:
Interesting, are you aware of upstream or downstream issue to follow for stock ath10k driver?

brada4, 2023-07-07:
Your firmware/driver are mainline ones from kernel.org. Probably something pops up when you search ath10k. Please try candela driver.
same: ath10k-board-qca988x 
replace: ath10k-firmware-qca988x-ct kmod-ath10k-ct

nazar-pc, 2023-07-07:
`-ct` looked promising at first, I didn't observe such issues on 2.4 GHz, but then I tried transferring some data over 5.8 GHz with QCA9984 and after ~220 MiB Wi-Fi was becoming almost completely dead, not fast enough to render `htop` in terminal in real time. Had to revert to non-ct for now and it is fast again.

brada4, 2023-07-07:
No matter driver ot goes slow? So it meams somethinh changed around PCI passthrough - interrupts/powersave/bar remap

nazar-pc, 2023-07-08:
The issue is only with kmod-ath10k-ct, regular driver works fine.

brada4, 2023-07-08:
I cannot determine which of these is "works fine":
> What I get then is extremely unresponsive Wi-Fi (hard to connect, frequent disconnections, etc.) and this in kernel logs:
> not fast enough to render htop

Please carefully examine guest and host and virtualisation logs around PCI virtualisation, or spend 10min with OpenWRT booted directly on the hardware.

brada4, 2023-07-08:
both ath10k_pci have following parameter that I suspect was flipped:
parm:           irq_mode:0: auto, 1: legacy, 2: msi (default: 0) (uint)

So try:
rmmod ath10k_pci
modprobe ath10k_pci irq_mode=1
(or 2) and check stability.

nazar-pc, 2023-07-08:
Thanks for suggestions, I'll be away from the machine for a week, will experiment when I get back

brada4, 2023-07-08:
Two parameters:
```
parm:           irq_mode:0: auto, 1: legacy, 2: msi (default: 0) (uint)
parm:           reset_mode:0: auto, 1: warm only (default: 0) (uint)
```
If you know how to edit KVM domain to mask or bend these parameters to one or other side, try that after confirming it is stable. First can cause guest to glitch due to atypical latency characteristics of device virtualisation, 2nd may reset unwillingly PCI hub/bus in place of PCI slot alone.Either rhymes with your observations.

nazar-pc, 2023-07-29:
Okay, the issue with `-ck` and QCA9984 happens when I'm sending data from one Wi-Fi device to another. Just tried to copy 1G file full of zeroes and it hanged shortly after like I mentioned above.
When this happened following was printed by the driver, maybe a coincidence:
```
[ 1064.080419] ath10k_pci 0000:01:00.0: htt tx: fixing invalid VHT TX rate code 0xff
```

When copying from wired Ethernet to device that is connected via Wi-Fi this doesn't happen, doing concurrent speedtest on two devices connected to QCA9984 I have not seen any issues either. From that I make a conclusion if it only happens during client-client communication through QCA9984 (client isolation is disabled on this SSID).

`modprobe ath10k_pci irq_mode=1` helped indeed, here are logs (system boot, having issues, unloading/loading driver after that, restarting devices from LuCI since SSIDs were down otherwise):
<details>

```
[    0.000000] Linux version 5.10.176 (builder@buildhost) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20134-5f15225c1e) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 SMP Thu Apr 27 20:28:15 2023
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=bf7a017b-02 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001ffd7fff] usable
[    0.000000] BIOS-e820: [mem 0x000000001ffd8000-0x000000001fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000b0000000-0x00000000bfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (Q35 + ICH9, 2009), BIOS rel-1.16.2-0-gea1b7a073390-prebuilt.qemu.org 04/01/2014
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 26c6001, primary cpu clock
[    0.000000] kvm-clock: using sched offset of 473629682975 cycles
[    0.000002] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000004] tsc: Detected 3192.052 MHz processor
[    0.000827] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000829] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000834] last_pfn = 0x1ffd8 max_arch_pfn = 0x400000000
[    0.000878] MTRR default type: write-back
[    0.000878] MTRR fixed ranges enabled:
[    0.000879]   00000-9FFFF write-back
[    0.000880]   A0000-BFFFF uncachable
[    0.000881]   C0000-FFFFF write-protect
[    0.000881] MTRR variable ranges enabled:
[    0.000882]   0 base 00C0000000 mask FFC0000000 uncachable
[    0.000883]   1 disabled
[    0.000883]   2 disabled
[    0.000883]   3 disabled
[    0.000884]   4 disabled
[    0.000884]   5 disabled
[    0.000885]   6 disabled
[    0.000885]   7 disabled
[    0.000895] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.011414] found SMP MP-table at [mem 0x000f5b60-0x000f5b6f]
[    0.011524] Using GB pages for direct mapping
[    0.011660] ACPI: Early table checksum verification disabled
[    0.011663] ACPI: RSDP 0x00000000000F5990 000014 (v00 BOCHS )
[    0.011668] ACPI: RSDT 0x000000001FFE30E0 000034 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.011673] ACPI: FACP 0x000000001FFE2F08 0000F4 (v03 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.011678] ACPI: DSDT 0x000000001FFE0040 002EC8 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.011682] ACPI: FACS 0x000000001FFE0000 000040
[    0.011686] ACPI: APIC 0x000000001FFE2FFC 000080 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.011689] ACPI: MCFG 0x000000001FFE307C 00003C (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.011693] ACPI: WAET 0x000000001FFE30B8 000028 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.011696] ACPI: Reserving FACP table memory at [mem 0x1ffe2f08-0x1ffe2ffb]
[    0.011697] ACPI: Reserving DSDT table memory at [mem 0x1ffe0040-0x1ffe2f07]
[    0.011698] ACPI: Reserving FACS table memory at [mem 0x1ffe0000-0x1ffe003f]
[... 738 lines cut ...]
[ 1573.587184] br-guest: port 2(wlan3-1) entered disabled state
[ 1573.588148] device wlan3-1 entered promiscuous mode
[ 1573.590539] ath10k_pci 0000:04:00.0: rts threshold -1
[ 1573.591680] br-guest: port 2(wlan3-1) entered blocking state
[ 1573.592586] br-guest: port 2(wlan3-1) entered forwarding state
[ 1573.600637] IPv6: ADDRCONF(NETDEV_CHANGE): wlan3-1: link becomes ready
[ 1579.212555] IPv6: ADDRCONF(NETDEV_CHANGE): wlan2: link becomes ready
[ 1579.213851] br-lan: port 3(wlan2) entered blocking state
[... cut ...]
[ 1579.391118] ath10k_pci 0000:01:00.0: wmi: fixing invalid VHT TX rate code 0xff
[ 1584.311711] ath10k_pci 0000:01:00.0: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
```

</details>

Not sure what do I do regarding that IQR in a VM, it worked for as far as I remember with other PCIe devices (Wi-Fi cards, GPUs, USB controllers) without any special tweaks.

Annoyingly, after loading driver with `modprobe ath10k_pci irq_mode=1` once card doesn't hang even if I stop/start VM again, as if something remains in Wi-Fi card's memory :thinking:

brada4, 2023-07-29:
More likely check kvm xml  if it stores some pcie port 
 parameter.
I have seen full host crashes wirh device reset pulling down 2-nic pcie cards

nazar-pc, 2023-07-29:
I have Wi-Fi card bound to vfio-pci driver on the host if that is what you're asking, but driver is still fully unloaded when VM shuts down. So `-ck` driver is a potential solution, but why is it not liking default IRQ mode (and I don't see it changing in logs, while it clearly made a difference)?

brada4, 2023-07-29:
-ct is default in openwrt
Frankly a mystery how irq fixed itself.

nazar-pc, 2023-07-29:
Any idea why it might only happen in client-client communication?

Also for QCA9984 there are actually 3 `-ct` packages:
* ath10k-firmware-qca9984-ct
* ath10k-firmware-qca9984-ct-full-htt
* ath10k-firmware-qca9984-ct-htt

No clear docs on what is the meaningful difference, should I try second or third instead?

brada4, 2023-07-29:
If you browse to the url in either package's description...

nazar-pc, 2023-07-29:
Been there, looks like it is written for experts in the field, not more casual users.

nazar-pc, 2023-08-27:
As a persistent workaround for -ct I have added this to `/etc/rc.local`:
```bash
rmmod ath10k_pci
modprobe ath10k_pci irq_mode=1

wifi down
wifi up
```

brada4, 2023-08-27:
You can pass parameters in modules.d file containing ath10k_pci
```diff
---ath10k_pci
+++ath10k_pci irq_mode=1
```

nazar-pc, 2023-08-27:
I want things to work across upgrades. Editing files that come from packages means I need to go and manually edit them after upgrade. At the same time `/etc/rc.local` is persisted and appears in generated backups by default. It is not ideal, but it is less ongoing maintenance.

brada4, 2023-08-27:
You can add changed "system" files in /etc/sysupgrade.conf and check with `sysupgrade -L` if they are set to be preserved.

brada4, 2023-08-27:
(sysupgrade and auc does not work on x86_64, this way marked files also get saved in "config backup" from LuCI web)

nazar-pc, 2023-08-27:
That is a better workaround indeed, I'll switch to it, thanks!

> (sysupgrade and auc does not work on x86_64, this way marked files also get saved in "config backup" from LuCI web)

I used sysupgrade before from CLI a few times, but mostly from LuCI. I suspect "not" is a typo then, it does work fine on x86-64.

brada4, 2023-08-27:
Following snapshots not so well... You can try `auc` to avoid installing packages too.

Aleksanaa, 2023-08-28:
I'm using 6.1.44 kernel, not with `-ct` driver, but the bugs behaves the same. After running two speedtests, the wifi connection disappears, and dmesg shows

```
[ 4199.585084] ath10k_pci 0000:06:00.0: pdev param 0 not supported by firmware
[ 4199.602875] br-lan: port 1(wlan0) entered blocking state
[ 4199.602883] br-lan: port 1(wlan0) entered disabled state
[ 4199.602959] device wlan0 entered promiscuous mode
[ 4260.479096] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 4260.479165] br-lan: port 1(wlan0) entered blocking state
[ 4260.479169] br-lan: port 1(wlan0) entered forwarding state
[ 4260.479296] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[ 4537.946093] device wlan0 left promiscuous mode
[ 4537.946174] br-lan: port 1(wlan0) entered disabled state
[ 4538.048022] ath10k_pci 0000:06:00.0: DFS region 0x0 not supported, will trigger radar for every pulse
[ 4539.310485] ath10k_pci 0000:06:00.0: pdev param 0 not supported by firmware
[ 4539.330987] br-lan: port 1(wlan0) entered blocking state
[ 4539.330996] br-lan: port 1(wlan0) entered disabled state
[ 4539.331095] device wlan0 entered promiscuous mode
```

Using `ath10k_pci.irq_mode=1` also fixed the issue, but sacrificing about 25% of max bandwidth.

brada4, 2023-08-28:
>DFS region 0x0

Not related to this issue at all. It is a driver limitation, ct or not.
https://wireless.wiki.kernel.org/en/users/drivers/ath10k/configuration#building_hostapd (end of that paragraph)