openwrt/openwrt #13070: Changing country code has no effect, when there's more than one radio device in the system.

sspphheerraa, 2023-07-07:
### Describe the bug

I have get usb wifi adapter, which is turnes on by kmod-ath9k-htc module. 
Here's regulatory values, when usb adapter is physically disconnected: 
```
 root@openwrt:/# iw reg get
    global
    country TW: DFS-FCC
    (2400 - 2483 @ 40), (N/A, 30), (N/A)
    (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
    (5250 - 5350 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
    (5470 - 5730 @ 160), (N/A, 23), (0 ms), DFS
    (5725 - 5850 @ 80), (N/A, 30), (N/A)
    (57000 - 66000 @ 2160), (N/A, 40), (N/A)

    phy#0
    country US: DFS-FCC
    (902 - 904 @ 2), (N/A, 30), (N/A)
    (904 - 920 @ 16), (N/A, 30), (N/A)
    (920 - 928 @ 8), (N/A, 30), (N/A)
    (2400 - 2472 @ 40), (N/A, 30), (N/A)
    (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
    (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
    (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
    (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
    (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
    (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
    (57240 - 71000 @ 2160), (N/A, 40), (N/A)
```
If adapter physically connected, than, after reboot I can't set right country code. On both radio devices (main router - radio0, and wifi adapter - radio1) it sets "98" country code values (in LuCI).
But really each device works with self values, which, I guess, it takes from EEPROMs of them.

```
    root@router:/# iw reg get
    global
    country 98: DFS-UNSET
    (2400 - 2472 @ 40), (N/A, 20), (N/A)
    (5150 - 5250 @ 100), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
    (5250 - 5350 @ 100), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
    (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
    (5725 - 5730 @ 5), (N/A, 23), (0 ms), NO-OUTDOOR, DFS
    (5730 - 5850 @ 80), (N/A, 23), (N/A), NO-OUTDOOR
    (5925 - 6425 @ 160), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
    (57240 - 71000 @ 2160), (N/A, 40), (N/A)

    phy#1
    country GB: DFS-ETSI
    (2400 - 2483 @ 40), (N/A, 20), (N/A)
    (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
    (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
    (5470 - 5730 @ 160), (N/A, 26), (0 ms), DFS
    (5725 - 5850 @ 80), (N/A, 23), (N/A), NO-OUTDOOR
    (5925 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
    (57000 - 71000 @ 2160), (N/A, 40), (N/A)

    phy#0
    country US: DFS-FCC
    (902 - 904 @ 2), (N/A, 30), (N/A)
    (904 - 920 @ 16), (N/A, 30), (N/A)
    (920 - 928 @ 8), (N/A, 30), (N/A)
    (2400 - 2472 @ 40), (N/A, 30), (N/A)
    (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
    (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
    (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
    (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
    (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
    (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
    (57240 - 71000 @ 2160), (N/A, 40), (N/A)
```
As far as I can see, radio0 is working with 'US' country, and radio1 - with 'GB'. And, respectively, main radio doesn't see 12,13 channel stations any more, and at usb adapter - wrong tx power values.
Here is two screenshots of LuCI (first is radio0/wlan0, second - radio1/wlan1):
https://forum.openwrt.org/uploads/default/original/3X/7/2/72b13f133678fd290ad2bdfb72ee1e549130a515.png
https://forum.openwrt.org/uploads/default/original/3X/3/f/3f9f846996a1c13626201a241d154a7b517dcdfc.png

So, attaching additional wifi device makes 'hard set' country code values from its EEPROM ones.

### OpenWrt version

r20134-5f15225c1e

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link TL-WR1043ND v2

### Image kind

Official downloaded image

### Steps to reproduce

1. Connect to the router additional radio device, for example, usb wifi adapter, which is supported by kmod-ath9k-htc module
2. opkg update && opkg install kmod-ath9k-htc
3. reboot

### Actual behaviour

After reboot, changing country code has no effect. Each radio device is working on self country values (from EEPROM). No way to change it.

### Expected behaviour

It must be possibility to set the right country code option for all radio devices working simultaneously. All radio devices must use it, and not self EEPROM values.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-07-07:
It is a per-device setting, in /etc/config/wireless each radio can have own country code.

sspphheerraa, 2023-07-07:
> It is a per-device setting, in /etc/config/wireless each radio can have own country code.

There is no effect from that, 'iw reg get' and LuCI always shows '98' country. But really radio devices work as "US" and "GB".

brada4, 2023-07-07:
do `iw reg set 'TW'` and check channels avail.

sspphheerraa, 2023-07-08:
> do `iw reg set 'TW'` and check channels avail.

```
root@router:~# iw reg get
global
country 98: DFS-UNSET
        (2400 - 2472 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 100), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 100), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5725 - 5730 @ 5), (N/A, 23), (0 ms), NO-OUTDOOR, DFS
        (5730 - 5850 @ 80), (N/A, 23), (N/A), NO-OUTDOOR
        (5925 - 6425 @ 160), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#1
country GB: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 26), (0 ms), DFS
        (5725 - 5850 @ 80), (N/A, 23), (N/A), NO-OUTDOOR
        (5925 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
        (57000 - 71000 @ 2160), (N/A, 40), (N/A)

phy#0
country US: DFS-FCC
        (902 - 904 @ 2), (N/A, 30), (N/A)
        (904 - 920 @ 16), (N/A, 30), (N/A)
        (920 - 928 @ 8), (N/A, 30), (N/A)
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)

root@router:~# iw reg set 'TW'
root@router:~# iw reg get
global
country 98: DFS-UNSET
        (2400 - 2472 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 100), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 100), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5725 - 5730 @ 5), (N/A, 23), (0 ms), NO-OUTDOOR, DFS
        (5730 - 5850 @ 80), (N/A, 23), (N/A), NO-OUTDOOR
        (5925 - 6425 @ 160), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)

phy#1
country GB: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 26), (0 ms), DFS
        (5725 - 5850 @ 80), (N/A, 23), (N/A), NO-OUTDOOR
        (5925 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR
        (57000 - 71000 @ 2160), (N/A, 40), (N/A)

phy#0
country US: DFS-FCC
        (902 - 904 @ 2), (N/A, 30), (N/A)
        (904 - 920 @ 16), (N/A, 30), (N/A)
        (920 - 928 @ 8), (N/A, 30), (N/A)
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)
```
Nothing changes. And no 12,13 channels at radio0

brada4, 2023-07-08:
So basically plugging USB ath9k plug you lose CH12-13 on ath9k PCI formally unrelated onboard device?
What does logread say regarding regdom-s etc? normal boot vs plugging adapter vs boot with adapter connected?

sspphheerraa, 2023-07-08:
Here is dmesg, when pluggedin:
```
root@router:~# dmesg
[    0.000000] Linux version 5.10.176 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20134-5f15225c1e) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 Thu Apr 27 20:28:15 2023
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is TP-Link TL-WR1043ND v2
[    0.000000] SoC: Qualcomm Atheros QCA9558 ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] On node 0 totalpages: 16384
[    0.000000]   Normal zone: 144 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 16384 pages, LIFO batch:3
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 56176K/65536K available (5884K kernel code, 605K rwdata, 748K rodata, 1200K init, 210K bss, 9360K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 720.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 5309056796 ns
[    0.000009] sched_clock: 32 bits at 360MHz, resolution 2ns, wraps every 5965232126ns
[    0.008370] Calibrating delay loop... 358.80 BogoMIPS (lpj=1794048)
[    0.075044] pid_max: default: 32768 minimum: 301
[    0.080104] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.087887] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.098193] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.109031] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.119537] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.126879] pinctrl core: initialized pinctrl subsystem
[    0.136581] NET: Registered protocol family 16
[    0.142093] thermal_sys: Registered thermal governor 'step_wise'
[    0.193343] clocksource: Switched to clocksource MIPS
[    0.206339] NET: Registered protocol family 2
[    0.211199] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.219683] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.228685] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.236863] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.244381] TCP: Hash tables configured (established 1024 bind 1024)
[    0.251274] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.258285] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.266001] NET: Registered protocol family 1
[    0.270667] PCI: CLS 0 bytes, default 32
[    0.280802] workingset: timestamp_bits=14 max_order=14 bucket_order=0
[    0.291527] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.297796] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.309972] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 251)
[    0.320684] pinctrl-single 1804002c.pinmux: 544 pins, size 68
[    0.327361] gpio-export gpio-export: 1 gpio(s) exported
[    0.333799] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
[    0.342786] printk: console [ttyS0] disabled
[    0.347453] 18020000.uart: ttyS0 at MMIO 0x18020000 (irq = 9, base_baud = 2500000) is a 16550A
[    0.356660] printk: console [ttyS0] enabled
[    0.365566] printk: bootconsole [early0] disabled
[    0.392326] spi-nor spi0.0: s25fl064k (8192 Kbytes)
[    0.397392] 3 fixed-partitions partitions found on MTD device spi0.0
[... 117 lines cut ...]
[   36.978307] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 2 is up
[   36.984731] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 4 is up
[   36.992352] eth0: link up (1000Mbps/Full duplex)
[   37.112885] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   37.853444] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   44.107088] br-lan: port 2(wlan0) entered blocking state
[   44.112621] br-lan: port 2(wlan0) entered disabled state
[   44.118404] device wlan0 entered promiscuous mode
[   44.123437] br-lan: port 2(wlan0) entered blocking state
[   44.128836] br-lan: port 2(wlan0) entered forwarding state
[   44.333656] br-lan: port 2(wlan0) entered disabled state
[   46.524543] wlan1: authenticate with 18:d6:c7:af:74:44
[... cut ...]
[   47.259003] wlan1: send auth to 18:d6:c7:af:74:44 (try 1/3)
[   47.335382] wlan1: authenticated
[   47.343654] wlan1: associate with 18:d6:c7:af:74:44 (try 1/3)
[   47.357621] wlan1: RX AssocResp from 18:d6:c7:af:74:44 (capab=0x431 status=0 aid=1)
[   47.372023] wlan1: associated
[   48.000029] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   49.354418] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   49.361262] br-lan: port 2(wlan0) entered blocking state
[   49.366716] br-lan: port 2(wlan0) entered forwarding state

```

sspphheerraa, 2023-07-08:
Here is boot when unplugged:
```
root@router:~# dmesg
[    0.000000] Linux version 5.10.176 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20134-5f15225c1e) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 Thu Apr 27 20:28:15 2023
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is TP-Link TL-WR1043ND v2
[    0.000000] SoC: Qualcomm Atheros QCA9558 ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] On node 0 totalpages: 16384
[    0.000000]   Normal zone: 144 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 16384 pages, LIFO batch:3
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 56176K/65536K available (5884K kernel code, 605K rwdata, 748K rodata, 1200K init, 210K bss, 9360K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 720.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 5309056796 ns
[    0.000008] sched_clock: 32 bits at 360MHz, resolution 2ns, wraps every 5965232126ns
[    0.008369] Calibrating delay loop... 358.80 BogoMIPS (lpj=1794048)
[    0.075045] pid_max: default: 32768 minimum: 301
[    0.080104] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.087879] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.098180] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.109014] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.119521] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.126862] pinctrl core: initialized pinctrl subsystem
[    0.136562] NET: Registered protocol family 16
[    0.142075] thermal_sys: Registered thermal governor 'step_wise'
[    0.193285] clocksource: Switched to clocksource MIPS
[    0.206289] NET: Registered protocol family 2
[    0.211146] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.219633] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.228635] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.236813] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.244331] TCP: Hash tables configured (established 1024 bind 1024)
[    0.251223] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.258235] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.265951] NET: Registered protocol family 1
[    0.270617] PCI: CLS 0 bytes, default 32
[    0.280749] workingset: timestamp_bits=14 max_order=14 bucket_order=0
[    0.291477] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.297746] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.309927] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 251)
[    0.320631] pinctrl-single 1804002c.pinmux: 544 pins, size 68
[    0.327313] gpio-export gpio-export: 1 gpio(s) exported
[    0.333750] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
[    0.342738] printk: console [ttyS0] disabled
[    0.347401] 18020000.uart: ttyS0 at MMIO 0x18020000 (irq = 9, base_baud = 2500000) is a 16550A
[    0.356602] printk: console [ttyS0] enabled
[    0.365506] printk: bootconsole [early0] disabled
[    0.392277] spi-nor spi0.0: s25fl064k (8192 Kbytes)
[    0.397341] 3 fixed-partitions partitions found on MTD device spi0.0
[... 98 lines cut ...]
[   35.798008] device eth1.1 entered promiscuous mode
[   35.802874] device eth1 entered promiscuous mode
[   35.811341] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   35.846039] br-lan: port 1(eth1.1) entered blocking state
[   35.851534] br-lan: port 1(eth1.1) entered forwarding state
[   35.956177] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 2 is up
[   35.962542] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 4 is up
[   35.970220] eth0: link up (1000Mbps/Full duplex)
[   36.103388] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   36.813375] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   39.821239] ath: EEPROM regdomain: 0x8324
[   39.825362] ath: EEPROM indicates we should expect a country code
[... cut ...]
[   39.841809] ath: Country alpha2 being used: UA
[   39.846325] ath: Regpair used: 0x3e
[   39.849861] ath: regdomain 0x8324 dynamically updated by user
[   42.057656] br-lan: port 2(wlan0) entered blocking state
[   42.063169] br-lan: port 2(wlan0) entered disabled state
[   42.068921] device wlan0 entered promiscuous mode
[   42.328021] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   42.334837] br-lan: port 2(wlan0) entered blocking state
[   42.340247] br-lan: port 2(wlan0) entered forwarding state

```

sspphheerraa, 2023-07-08:
and here is additional dmesg tail when plugning in after "unplugged boot":
```
[  262.033357] usb 1-1: new full-speed USB device number 2 using ehci-platform
[  262.563361] usb 1-1: new high-speed USB device number 3 using ehci-platform
[  262.787299] usb 1-1: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[  263.101597] usb 1-1: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[  263.352342] ath9k_htc 1-1:1.0: ath9k_htc: HTC initialized with 33 credits
[  263.592336] ath9k_htc 1-1:1.0: ath9k_htc: FW Version: 1.4
[  263.597951] ath9k_htc 1-1:1.0: FW RMW support: On
[  263.602835] ath: EEPROM regdomain: 0x833a
[  263.606984] ath: EEPROM indicates we should expect a country code
[  263.613196] ath: doing EEPROM country->regdmn map search
[  263.618615] ath: country maps to regdmn code: 0x37
[  263.623490] ath: Country alpha2 being used: GB
[  263.627994] ath: Regpair used: 0x37
[  263.636191] ath: EEPROM regdomain: 0x8324
[  263.640287] ath: EEPROM indicates we should expect a country code
[  263.646718] ath: doing EEPROM country->regdmn map search
[  263.652124] ath: country maps to regdmn code: 0x3e
[  263.657019] ath: Country alpha2 being used: UA
[  263.661522] ath: Regpair used: 0x3e
[  263.665077] ath: regdomain 0x8324 dynamically updated by user
[  263.672555] ieee80211 phy1: Atheros AR9271 Rev:1
```
and now all is fine... until reboot (and boot when "pluggedin").

brada4, 2023-07-08:
Edit /etc/config/wireless and add `option country 'TW'` to each wifi-device section so that each has only one. 
`grep -e wifi-device -e country /etc/config/wireless`
I trust you did it right at first try, but check twice, it looks unbelievable that random domains got applied from nowhere.

sspphheerraa, 2023-07-08:
Done. Here is it:

```
root@router:~# grep -e wifi-device -e country /etc/config/wireless
config wifi-device 'radio0'
        option country 'TW'
config wifi-device 'radio1'
        option country 'TW'
```

Nothing changes.

ps The thing is, that `ath: regdomain XXXXXX dynamically updated by user` doesn't apear when 'pluggedin' boot. And the EEPROM values are used.

rany2, 2023-07-08:
@sspphheerraa Do you have a WiFi client setup on that router? If you do, then it will automatically update the country code after connecting to an AP with a different country code.

rany2, 2023-07-08:
Anyway it's kind of odd you're having this issue with ath9k, I think only ath10k+ have issues changing regulatory domains. Perhaps your other radio device is ath10k or higher and when it updates its regdomain it changes it for the ath9k chip as well?

sspphheerraa, 2023-07-08:
> Anyway it's kind of odd you're having this issue with ath9k, I think only ath10k+ have issues changing regulatory domains. Perhaps your other radio device is ath10k or higher and when it updates its regdomain it changes it for the ath9k chip as well?

No.
As I understand, there's some _service_ in the system, that reads country codes from /etc/config/wireless (each wifi-device section) and applies that codes to devices.
When "usb is unplugged" boot, that _service_ starts and makes it's job. If I plug in usb adapter just after, - _service_ make it's job for usb adapter too. All is fine. Country codes are 'changable' and 'appliable'.
Next. Reboot.
So now, the boot is "usb is pluggedin". That _service_ doesn't (start?) do it's job like before... I don't know what... But 'global' country code value sets to '98', and both radio devices use their 'EEPROM' country code values. And this is unchangable from software level.
Sorry if confusingly.

brada4, 2023-07-08:
service is called wpad. but seems problem is that at start it chokes somehow as opposed to hot-add.

sspphheerraa, 2023-07-12:
Well, interesting things.
I set up the module `ath9k-htc` to not be loaded by system boot.
Reboot.
Next, "pluggedin boot", all is fine. Country code can be changed at LuCI.
Next, I do:
`modprobe ath9k-htc`
dmesg (tail) shows this:
```
[   79.165597] usb 1-1: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[   79.173223] usbcore: registered new interface driver ath9k_htc
[   79.494857] usb 1-1: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[   79.745489] ath9k_htc 1-1:1.0: ath9k_htc: HTC initialized with 33 credits
[   79.987110] ath9k_htc 1-1:1.0: ath9k_htc: FW Version: 1.4
[   79.992708] ath9k_htc 1-1:1.0: FW RMW support: On
[   79.997530] ath: EEPROM regdomain: 0x833a
[   80.001589] ath: EEPROM indicates we should expect a country code
[   80.007798] ath: doing EEPROM country->regdmn map search
[   80.013192] ath: country maps to regdmn code: 0x37
[   80.018063] ath: Country alpha2 being used: GB
[   80.022612] ath: Regpair used: 0x37
[   80.030596] ath: EEPROM regdomain: 0x809e
[   80.034850] ath: EEPROM indicates we should expect a country code
[   80.041150] ath: doing EEPROM country->regdmn map search
[   80.046571] ath: country maps to regdmn code: 0x50
[   80.051438] ath: Country alpha2 being used: TW
[   80.055955] ath: Regpair used: 0x50
[   80.059488] ath: regdomain 0x809e dynamically updated by user
[   80.066985] ieee80211 phy1: Atheros AR9271 Rev:1
```
and now
```
root@router:~# iw reg get
global
country TW: DFS-FCC
        (2400 - 2483 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5725 - 5850 @ 80), (N/A, 30), (N/A)
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#1
country TW: DFS-FCC
        (2400 - 2483 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5725 - 5850 @ 80), (N/A, 30), (N/A)
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#0
country US: DFS-FCC
        (902 - 904 @ 2), (N/A, 30), (N/A)
        (904 - 920 @ 16), (N/A, 30), (N/A)
        (920 - 928 @ 8), (N/A, 30), (N/A)
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)
```
So, I havn't see that before, now country code changes not only on "global" section, but also at "phy#1" one. 
I do:
`wifi up radio1`
Well, country code is changable at LuCI on both radios. And it works.
But if I try to change country code by "iw reg set UA" (for example, Ukraine), - it changes only "global" section.
```
root@router:~# iw reg set UA
root@router:~# iw reg get
global
country UA: DFS-ETSI
        (2400 - 2483 @ 40), (N/A, 20), (N/A), NO-OUTDOOR
        (5150 - 5250 @ 80), (N/A, 20), (N/A), NO-OUTDOOR, AUTO-BW
        (5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
        (5470 - 5725 @ 160), (N/A, 20), (0 ms), NO-OUTDOOR, DFS
        (5725 - 5850 @ 80), (N/A, 20), (N/A), NO-OUTDOOR
        (57000 - 66000 @ 2160), (N/A, 16), (N/A), NO-OUTDOOR

phy#1
country TW: DFS-FCC
        (2400 - 2483 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5725 - 5850 @ 80), (N/A, 30), (N/A)
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#0
country US: DFS-FCC
        (902 - 904 @ 2), (N/A, 30), (N/A)
        (904 - 920 @ 16), (N/A, 30), (N/A)
        (920 - 928 @ 8), (N/A, 30), (N/A)
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)
```
So, the conclusion is, country code values for specific radio define at kernel module initialization (or, probably, kernel initialization itself; like as radio0). But when all radios are initialized at boot 'wpad' (this service?) can change nothing. "Global" section can be changed if there's only one radio initialization at boot.
It's not normal and need to be fixed.

ps
Temporary workaround, is to remove `ath9k-htc` from `/etc/modules.d/` and add the following lines to `/etc/rc.local`
```
sleep 5
modprobe ath9k-htc

sleep 5
wifi up radio1
```

brada4, 2023-07-12:
You need to examine logread, not dmesg, to see hostapd traces.
Is it just when adapter is connected and kmod initializes it or when it is brought up by hostapd?

brada4, 2023-07-12:
Radio setup failed phy1... 
If it is permissible bring disable builtin radio reboot and check if you get same warning/error regarding usb radio.

brada4, 2023-07-12:
It could be as romantic as insufficient usb power. I had same but v1 and it glitched with any usb3-capable device.

sspphheerraa, 2023-07-12:
> Radio setup failed phy1... 

Where?

Here is logread when builtin radio device is disabled:
```
root@router:~# logread 
Wed Jul 12 15:16:06 2023 kern.notice kernel: [    0.000000] Linux version 5.10.176 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20134-5f15225c1e) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 Thu Apr 27 20:28:15 2023
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] MIPS: machine is TP-Link TL-WR1043ND v2
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] SoC: Qualcomm Atheros QCA9558 ver 1 rev 0
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] Zone ranges:
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] Movable zone start for each node
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] Early memory node ranges
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
Wed Jul 12 15:16:06 2023 kern.debug kernel: [    0.000000] On node 0 totalpages: 16384
Wed Jul 12 15:16:06 2023 kern.debug kernel: [    0.000000]   Normal zone: 144 pages used for memmap
Wed Jul 12 15:16:06 2023 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Wed Jul 12 15:16:06 2023 kern.debug kernel: [    0.000000]   Normal zone: 16384 pages, LIFO batch:3
Wed Jul 12 15:16:06 2023 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Wed Jul 12 15:16:06 2023 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
Wed Jul 12 15:16:06 2023 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] Writing ErrCtl register=00000000
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] Readback ErrCtl register=00000000
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] Memory: 56176K/65536K available (5884K kernel code, 605K rwdata, 748K rodata, 1200K init, 210K bss, 9360K reserved, 0K cma-reserved)
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] NR_IRQS: 51
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] CPU clock: 720.000 MHz
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 5309056796 ns
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.000008] sched_clock: 32 bits at 360MHz, resolution 2ns, wraps every 5965232126ns
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.008370] Calibrating delay loop... 358.80 BogoMIPS (lpj=1794048)
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.075044] pid_max: default: 32768 minimum: 301
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.080103] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.087879] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.098182] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.109023] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.119529] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.126871] pinctrl core: initialized pinctrl subsystem
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.136575] NET: Registered protocol family 16
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.142085] thermal_sys: Registered thermal governor 'step_wise'
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.193331] clocksource: Switched to clocksource MIPS
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.206326] NET: Registered protocol family 2
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.211190] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
Wed Jul 12 15:16:06 2023 kern.info kernel: [    0.219673] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[... cut ...]
Wed Jul 12 15:16:39 2023 daemon.info dnsmasq[1]: read /etc/hosts - 4 addresses
Wed Jul 12 15:16:39 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 10 addresses
Wed Jul 12 15:16:39 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 0 addresses
Wed Jul 12 15:16:39 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Wed Jul 12 15:17:25 2023 daemon.warn odhcpd[1463]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
```

sspphheerraa, 2023-07-12:
> It could be as romantic as insufficient usb power.

Insufficient usb power during boot, but enough after? Are you kidding?

sspphheerraa, 2023-07-22:
Well, the workaround I'm mentioned above is only for a few hours.
Some time later 12,13th channels become unavailable. Country sets to "98".
dmesg:
```
[    0.000000] Linux version 5.10.176 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20134-5f15225c1e) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 Thu Apr 27 20:28:15 2023
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is TP-Link TL-WR1043ND v2
[    0.000000] SoC: Qualcomm Atheros QCA9558 ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] On node 0 totalpages: 16384
[    0.000000]   Normal zone: 144 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 16384 pages, LIFO batch:3
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 56176K/65536K available (5884K kernel code, 605K rwdata, 748K rodata, 1200K init, 210K bss, 9360K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 720.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 5309056796 ns
[    0.000008] sched_clock: 32 bits at 360MHz, resolution 2ns, wraps every 5965232126ns
[    0.008370] Calibrating delay loop... 358.80 BogoMIPS (lpj=1794048)
[    0.075044] pid_max: default: 32768 minimum: 301
[    0.080105] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.087887] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.098189] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.109032] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.119537] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.126879] pinctrl core: initialized pinctrl subsystem
[    0.136583] NET: Registered protocol family 16
[    0.142097] thermal_sys: Registered thermal governor 'step_wise'
[    0.193302] clocksource: Switched to clocksource MIPS
[    0.206296] NET: Registered protocol family 2
[    0.211157] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.219643] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.228644] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.236821] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.244339] TCP: Hash tables configured (established 1024 bind 1024)
[    0.251232] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.258242] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.265959] NET: Registered protocol family 1
[    0.270625] PCI: CLS 0 bytes, default 32
[    0.280759] workingset: timestamp_bits=14 max_order=14 bucket_order=0
[    0.291487] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.297754] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.309935] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 251)
[    0.320638] pinctrl-single 1804002c.pinmux: 544 pins, size 68
[    0.327315] gpio-export gpio-export: 1 gpio(s) exported
[    0.333747] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
[    0.342742] printk: console [ttyS0] disabled
[    0.347409] 18020000.uart: ttyS0 at MMIO 0x18020000 (irq = 9, base_baud = 2500000) is a 16550A
[    0.356610] printk: console [ttyS0] enabled
[    0.365516] printk: bootconsole [early0] disabled
[    0.392271] spi-nor spi0.0: s25fl064k (8192 Kbytes)
[... 943 lines cut ...]
Thu Jul 20 11:56:15 2023 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Thu Jul 20 11:56:15 2023 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Thu Jul 20 11:56:15 2023 daemon.info dnsmasq[1]: using only locally-known addresses for lan40
Thu Jul 20 11:56:15 2023 daemon.info dnsmasq[1]: read /etc/hosts - 4 addresses
Thu Jul 20 11:56:15 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 10 addresses
Thu Jul 20 11:56:15 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 2 addresses
Thu Jul 20 11:56:15 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Thu Jul 20 11:56:15 2023 daemon.info dnsmasq[1]: read /etc/hosts - 4 addresses
[... cut ...]
Thu Jul 20 11:56:26 2023 kern.info kernel: [260791.076054] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Thu Jul 20 11:56:26 2023 kern.info kernel: [260791.082918] br-lan: port 2(wlan0) entered blocking state
Thu Jul 20 11:56:26 2023 kern.info kernel: [260791.088564] br-lan: port 2(wlan0) entered forwarding state
Thu Jul 20 11:56:26 2023 daemon.notice netifd: Network device 'wlan0' link is up
Thu Jul 20 11:56:26 2023 daemon.notice hostapd: wlan0: interface state ACS->ENABLED
Thu Jul 20 11:56:26 2023 daemon.notice hostapd: wlan0: AP-ENABLED
```
Looks like the kernel bug, no?

brada4, 2023-07-22:
Looks like rate and country got overwritten in kernel either by sibling device driver due to missing lock at simultaneous init, or by some minor overflow into adjacent structures. 
Do devices fix if you unplug usb, then plug back in waiting dozen seconds? I.e if some global struct is corrupt or just USB dev related.

sspphheerraa, 2023-07-23:
"Pluggedin" boot, then I unplug adapter, and after some minutes plug it back:
dmesg
```
[  401.801775] usb 1-1: USB disconnect, device number 2
[  401.927485] ath: phy1: RX failed to go idle in 10 ms RXSM=0x2000000
[  401.947217] ath: phy1: Failed to wakeup in 500us
[  402.053255] ath: phy1: RX failed to go idle in 10 ms RXSM=0x2000000
[  402.267103] ath: phy1: RX failed to go idle in 10 ms RXSM=0x2000000
[  402.483544] ath: phy1: RX failed to go idle in 10 ms RXSM=0x2000000
[  402.906246] usb 1-1: ath9k_htc: USB layer deinitialized
[  811.002045] usb 1-1: new high-speed USB device number 3 using ehci-platform
[  811.225794] usb 1-1: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[  811.515821] usb 1-1: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[  811.766708] ath9k_htc 1-1:1.0: ath9k_htc: HTC initialized with 33 credits
[  812.007335] ath9k_htc 1-1:1.0: ath9k_htc: FW Version: 1.4
[  812.012949] ath9k_htc 1-1:1.0: FW RMW support: On
[  812.017739] ath: EEPROM regdomain: 0x833a
[  812.021797] ath: EEPROM indicates we should expect a country code
[  812.028022] ath: doing EEPROM country->regdmn map search
[  812.033525] ath: country maps to regdmn code: 0x37
[  812.038387] ath: Country alpha2 being used: GB
[  812.042964] ath: Regpair used: 0x37
[  812.052964] ieee80211 phy2: Atheros AR9271 Rev:1
```
logread:
```
Sun Jul 23 14:03:37 2023 kern.info kernel: [  811.002045] usb 1-1: new high-speed USB device number 3 using ehci-platform
Sun Jul 23 14:03:38 2023 kern.info kernel: [  811.225794] usb 1-1: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
Sun Jul 23 14:03:38 2023 kern.info kernel: [  811.515821] usb 1-1: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
Sun Jul 23 14:03:38 2023 kern.info kernel: [  811.766708] ath9k_htc 1-1:1.0: ath9k_htc: HTC initialized with 33 credits
Sun Jul 23 14:03:38 2023 kern.info kernel: [  812.007335] ath9k_htc 1-1:1.0: ath9k_htc: FW Version: 1.4
Sun Jul 23 14:03:38 2023 kern.info kernel: [  812.012949] ath9k_htc 1-1:1.0: FW RMW support: On
Sun Jul 23 14:03:38 2023 kern.debug kernel: [  812.017739] ath: EEPROM regdomain: 0x833a
Sun Jul 23 14:03:38 2023 kern.debug kernel: [  812.021797] ath: EEPROM indicates we should expect a country code
Sun Jul 23 14:03:38 2023 kern.debug kernel: [  812.028022] ath: doing EEPROM country->regdmn map search
Sun Jul 23 14:03:38 2023 kern.debug kernel: [  812.033525] ath: country maps to regdmn code: 0x37
Sun Jul 23 14:03:38 2023 kern.debug kernel: [  812.038387] ath: Country alpha2 being used: GB
Sun Jul 23 14:03:38 2023 kern.debug kernel: [  812.042964] ath: Regpair used: 0x37
Sun Jul 23 14:03:38 2023 kern.info kernel: [  812.052964] ieee80211 phy2: Atheros AR9271 Rev:1
Sun Jul 23 14:03:39 2023 daemon.notice wpa_supplicant[1366]: wlan1: CTRL-EVENT-REGDOM-CHANGE init=DRIVER type=INTERSECTION
Sun Jul 23 14:03:41 2023 daemon.notice netifd: Network device 'wlan1' link is up
Sun Jul 23 14:03:41 2023 daemon.notice netifd: Interface 'wwan' has link connectivity
Sun Jul 23 14:03:41 2023 daemon.notice netifd: Interface 'wwan' is setting up now
Sun Jul 23 14:03:41 2023 daemon.notice netifd: Interface 'wwan6' has link connectivity
Sun Jul 23 14:03:41 2023 daemon.notice netifd: Interface 'wwan6' is setting up now
Sun Jul 23 14:03:41 2023 kern.info kernel: [  814.565788] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Sun Jul 23 14:03:41 2023 daemon.notice wpa_supplicant[1366]: wlan1: Unknown event 37
Sun Jul 23 14:03:41 2023 daemon.notice netifd: wwan (3239): udhcpc: started, v1.35.0
Sun Jul 23 14:03:41 2023 daemon.err odhcp6c[3238]: Failed to send RS (Address not available)
Sun Jul 23 14:03:41 2023 daemon.notice netifd: wwan (3239): udhcpc: broadcasting discover
Sun Jul 23 14:03:41 2023 daemon.notice netifd: wwan (3239): udhcpc: broadcasting select for 192.168.1.10, server 192.168.1.1
Sun Jul 23 14:03:41 2023 daemon.notice netifd: wwan (3239): udhcpc: lease of 192.168.1.10 obtained from 192.168.1.1, lease time 43200
Sun Jul 23 14:03:42 2023 daemon.notice netifd: Interface 'wwan' is now up
Sun Jul 23 14:03:42 2023 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Sun Jul 23 14:03:42 2023 daemon.info dnsmasq[1]: using nameserver 192.168.1.1#53
Sun Jul 23 14:03:42 2023 daemon.info dnsmasq[1]: using only locally-known addresses for test
Sun Jul 23 14:03:42 2023 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Sun Jul 23 14:03:42 2023 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Sun Jul 23 14:03:42 2023 daemon.info dnsmasq[1]: using only locally-known addresses for local
Sun Jul 23 14:03:42 2023 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Sun Jul 23 14:03:42 2023 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Sun Jul 23 14:03:42 2023 daemon.info dnsmasq[1]: using only locally-known addresses for lan40
Sun Jul 23 14:03:42 2023 daemon.err odhcp6c[3238]: Failed to send SOLICIT message to ff02::1:2 (Address not available)
Sun Jul 23 14:03:42 2023 user.notice firewall: Reloading firewall due to ifup of wwan (wlan1)
Sun Jul 23 14:03:46 2023 daemon.notice netifd: Interface 'wwan6' is now up
Sun Jul 23 14:03:46 2023 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Sun Jul 23 14:03:46 2023 daemon.info dnsmasq[1]: using nameserver 192.168.1.1#53
Sun Jul 23 14:03:46 2023 daemon.info dnsmasq[1]: using nameserver fd88:6eb7:edba::1#53
[... cut ...]
Sun Jul 23 14:03:46 2023 daemon.info dnsmasq[1]: using only locally-known addresses for lan40
Sun Jul 23 14:03:46 2023 user.notice firewall: Reloading firewall due to ifup of wwan6 (wlan1)
Sun Jul 23 14:03:47 2023 daemon.warn odhcpd[1461]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
Sun Jul 23 14:05:12 2023 daemon.warn odhcpd[1461]: A default route is present but there is no public prefix on lan thus we don't announce a default route!
```
Nothing changes.

brada4, 2023-07-23:
But reg domain is permanently corrupt after boot with USB radio plugged?

sspphheerraa, 2023-07-24:
> But reg domain is permanently corrupt after boot with USB radio plugged?

Looks like so.

sspphheerraa, 2023-07-24:
> workaround (not elegant)
> 
> a) stop usb driver from loading at boot by truncating/commenting respective file in /etc/modules.d b) in rc.local add `(sleep 10 ; modprobe driver) &`
> 
> Dont close this in hope it gets fixed some day.

Did you miss my [comment](https://github.com/openwrt/openwrt/issues/13070#issuecomment-1632131109) where the workaround I already mentioned?
And that it's useful only for a few hours?

brada4, 2023-07-24:
If you boot with device connected but no -htc driver loaded it fails after some time?

sspphheerraa, 2023-07-26:
> If you boot with device connected but no -htc driver loaded it fails after some time?

No, two hours later, it's ok.

brada4, 2023-07-26:
Go figure how long you have to add sleep for -htc load to avoid race condition corrupting memory.

sspphheerraa, 2023-07-27:
> Go figure how long you have to add sleep for -htc load to avoid race condition corrupting memory.

You mean the time ('sleep' command) before 'modprobe' at /etc/rc.local?
Here's it:
```
sleep 3
modprobe ath9k-htc
```
is ok,
```
sleep 2
modprobe ath9k-htc
```
leads to "device is not active" at LuCi. Power indicator on adapter is not lighting. Revert to "sleep 3" doesn't help. Only unplug-plug is the way to up it.
And the dmesg:
```
[    0.000000] Linux version 5.10.176 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20134-5f15225c1e) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 Thu Apr 27 20:28:15 2023
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is TP-Link TL-WR1043ND v2
[    0.000000] SoC: Qualcomm Atheros QCA9558 ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] On node 0 totalpages: 16384
[    0.000000]   Normal zone: 144 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 16384 pages, LIFO batch:3
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 56176K/65536K available (5884K kernel code, 605K rwdata, 748K rodata, 1200K init, 210K bss, 9360K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 720.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 5309056796 ns
[    0.000008] sched_clock: 32 bits at 360MHz, resolution 2ns, wraps every 5965232126ns
[    0.008370] Calibrating delay loop... 358.80 BogoMIPS (lpj=1794048)
[    0.075044] pid_max: default: 32768 minimum: 301
[    0.080104] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.087887] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.098190] dyndbg: Ignore empty _ddebug table in a CONFIG_DYNAMIC_DEBUG_CORE build
[    0.109030] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.119537] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.126878] pinctrl core: initialized pinctrl subsystem
[    0.136575] NET: Registered protocol family 16
[    0.142088] thermal_sys: Registered thermal governor 'step_wise'
[    0.193315] clocksource: Switched to clocksource MIPS
[    0.206315] NET: Registered protocol family 2
[    0.211173] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.219656] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.228660] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.236837] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.244356] TCP: Hash tables configured (established 1024 bind 1024)
[    0.251249] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.258259] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.265977] NET: Registered protocol family 1
[    0.270642] PCI: CLS 0 bytes, default 32
[... 122 lines cut ...]
[   35.989024] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   36.733413] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   37.401066] usbcore: registered new interface driver ath9k_htc
[   38.972612] ath: EEPROM regdomain: 0x809e
[   38.976737] ath: EEPROM indicates we should expect a country code
[   38.982912] ath: doing EEPROM country->regdmn map search
[   38.988322] ath: country maps to regdmn code: 0x50
[   38.993182] ath: Country alpha2 being used: TW
[   38.997696] ath: Regpair used: 0x50
[   39.001233] ath: regdomain 0x809e dynamically updated by user
[   41.207954] br-lan: port 2(wlan0) entered blocking state
[   41.213420] br-lan: port 2(wlan0) entered disabled state
[   41.219136] device wlan0 entered promiscuous mode
[   41.477818] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   41.484539] br-lan: port 2(wlan0) entered blocking state
[   41.489963] br-lan: port 2(wlan0) entered forwarding state
[   45.843326] usb 1-1: device descriptor read/64, error -145
[   45.963373] usb usb1-port1: attempt power cycle
[   46.433345] usb 1-1: new high-speed USB device number 4 using ehci-platform
[   56.953454] usb 1-1: device not accepting address 4, error -145
[   57.113402] usb 1-1: new high-speed USB device number 5 using ehci-platform
[   67.673343] usb 1-1: device not accepting address 5, error -145
[   67.679422] usb usb1-port1: unable to enumerate USB device
```
Main wifi (radio0) country code value can be changed normaly in all these cases.