openwrt/openwrt #13077: E8450 reboot request just shuts down and does not restart. Must power cycle to complete reboot

U251, 2023-07-09:
### Describe the bug

Latest Snapshot
Model	Linksys E8450 (UBI)
Architecture	ARMv8 Processor rev 4
Target Platform	mediatek/mt7622
Firmware Version	OpenWrt SNAPSHOT r23545-98e3f82c3f / LuCI Master git-23.158.78004-23a246e

Reboot request either thru luci or thru Bash command line produces a shutdown without a restart.
Must power cycle to complete a successful reboot.

Happening on a  client(WDS) unit.

### OpenWrt version

DISTRIB_REVISION='r23545-98e3f82c3f'

### OpenWrt target/subtarget

DISTRIB_TARGET='mediatek/mt7622'

### Device

Linksys E8450 (UBI)

### Image kind

Official downloaded image

### Steps to reproduce

Calling for a unit reboot either thru luci interface or via bash command line "reboot" produces a unit shut down without a following restart. Unit must be power cycled to complete a successful reboot sequence.

Unit is in a (WDS)client configuration.

### Additional info

make image PROFILE="linksys_e8450-ubi" FILES="files" PACKAGES="block-mount blockd collectd collectd-mod-cpu collectd-mod-irq collectd-mod-memory collectd-mod-network collectd-mod-thermal collectd-mod-write-graphite collectd-mod-write-http exfat-fsck exfat-mkfs htop ip-full irqbalance kmod-fs-exfat kmod-usb-storage kmod-usb-storage-extras kmod-usb-storage-uas luci luci-app-attendedsysupgrade luci-app-commands luci-app-hd-idle luci-app-samba4 luci-app-statistics luci-app-wifischedule luci-theme-openwrt luci-theme-openwrt-2020 luci-proto-ppp luci-proto-relay mount-utils nano-full netdata ntfs-3g ntfs-3g-low ntfs-3g-utils openssh-sftp-client openssh-sftp-server relayd speedtestpp"

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

mrkiko, 2023-07-10:
Hi!!

This looks like a cpufreq-related issue.
Have you tried setting the performance governor or leaving cpufreq in it's default state?

Thanks!

mrkiko, 2023-07-10:
Oh - and if you are using an USB storage device I would try rebooting without it plugged in.

U251, 2023-07-10:
OK here is where I am so far.......

To try and eliminate anything I have messed up I did this...

1. Downloaded a fresh copy of https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/openwrt-mediatek-mt7622-linksys_e8450-ubi-squashfs-sysupgrade.itb

2. Removed any USB flashdrive and booted into Recovery mode  "Power OFF -- Hold down RESET -- Power ON"

3.Flashed Above .itb file from running recovery interface.

4.Unit did reboot on its own after completing the successful firmware flash and booted up to the latest snapshot. ......545 I think.

5. Correctly ..Luci is not installed from this stock snapshot firmware.

6. ssh into router.....Execute a "reboot" command from bash terminal window. Router shuts down and does NOT restart.

7.Power OFF and Power ON boots router and all is good.

8.Again reboot into recovery and then flash a 22.03.5 firmware file that has been constructed from imagebuilder on a linux machine.

9.From this point on the router reboots normally from luci or bash command line.

U251, 2023-07-10:
I have left the cpu governor at ondemand which I believe is the default setting

U251, 2023-07-20:
To: @mrkiko

THANK YOU so much
After some  trial and error I added

`echo schedutil > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor
`

to etc/rc.local and that seems to have corrected the problem.

I don't understand why only one of three routers need this edit but I guessing there is a hardware weakness somewhere.

Again..... a sincere thank you.

I know I never would have figured this one out.

closed 2023-08-03 by U251