openwrt/openwrt #13082: Trying to install anything that has libnl-tiny as dependency fails

mikk150, 2023-07-09:
### Describe the bug

since libnl-tiny package name includes package version for whatever name, opkg is not considering just upgrading libnl-tiny, rather just stating that file that has been provided by libnl-tiny is being overwritten by file that is being provided by libnl-tiny

### OpenWrt version

r23228-cd17d8df2a

### OpenWrt target/subtarget

ipq40xx/generic

### Device

ASUS RT-AC58U

### Image kind

Official downloaded image

### Steps to reproduce

1. flash 23.05.0-rc2
2. try to install usteer

### Actual behaviour

```
root@OpenWrt:~# opkg install usteer
Installing usteer (2022-08-18-7d2b17c9-1) to root...
Downloading https://downloads.openwrt.org/releases/23.05.0-rc2/packages/arm_cortex-a7_neon-vfpv4/packages/usteer_2022-08-18-7d2b17c9-1_arm_cortex-a7_neon-vfpv4.ipk
Installing libnl-tiny2023-07-01 (2023-07-01-d433990c-1) to root...
Downloading https://downloads.openwrt.org/releases/23.05.0-rc2/packages/arm_cortex-a7_neon-vfpv4/base/libnl-tiny2023-07-01_2023-07-01-d433990c-1_arm_cortex-a7_neon-vfpv4.ipk
Collected errors:
 * check_data_file_clashes: Package libnl-tiny2023-07-01 wants to install file /usr/lib/libnl-tiny.so
	But that file is already provided by package  * libnl-tiny2023-04-02
 * opkg_install_cmd: Cannot install package usteer.
```

### Expected behaviour

I should be able to install usteer

### Additional info

I think that commit b5b2ba09c4f1 caused this issue?

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ascendbeing, 2023-07-10:
When the ASU was down, I actually force overwrote the libl-tiny with the newer one, and when I would opkg files both versions, the old one no longer had files associated, however it had packages depending on it (old libnl-tiny)
So I recommend you try to ascertain whether the tiny lib and the build both have actually gotten everything up to date, as it can take like 3-7ish days after a tiny lib update for both to become synchronized properly.
Whether the force options are entirely safe I can't say for sure. It is ABI incongruence that would be the most concerning. I didn't personally find the April and July tiny lib to have any show stoppers but I was blindly updating it and eventually rebuilding a snapshot when ASU came back online. I didn't actually inspect whether there was ABI breakage, so my being able to use my stuff is no indication your assuredly different use case means force is safe. As a general rule if you look at the libnl changeliog between the two dates, if there's no major restructuring wet ABI the worst that can happen is a workaround for an existent bug in the old version could theoretically mean a break with a newer version for a package which has a workaround

mikk150, 2023-07-12:
> EDIT: You might want to try a 23.05-SNAPSHOT build. Cause I wrote below assuming you were on a snapshot. 23.05-SNAPSHOT would probably be less likely to have this issue.
> 

Don't really care, I wanted to test RC-which to my knowledge stands for Release Candidate.. and I found it to be not usable.. So therefore I reported a bug on RC

brada4, 2023-07-13:
https://github.com/openwrt/openwrt/issues/13106 duplicate

tmn505, 2023-07-14:
This is a issue with current infrastructure. The "external" repositories (packages, luci, routing, telephony) are NOT built for each tagged version of OpenWrt (while core and base repositories are), those are built with usage of 23.05-SNAPSHOT SDK, continuously. The 23.05-SNAPSHOT SDK is built from openwrt-23.05 branch in https://github.com/openwrt/openwrt, any commit bumping package version in that branch will mean that each package in "external" repositories, will have new version dependency.

To summarize, the firmware version You have installed points to fixed state of core and base repositories, and at the same time, points to varying state of packages, luci, routing and telephony repositories, which are built against varying state of core and base repositories. If not careful, this has to break at some point.

cc'ing people knowing more about it than I do: @ynezz @jow- and author of the commit: @hauke

mikma, 2023-07-14:
> The 23.05-SNAPSHOT SDK is built from openwrt-23.05 branch in https://github.com/openwrt/openwrt, any commit bumping package version in that branch will mean that each package in "external" repositories, will have new version dependency.

In my case I tried to install "ss" which is part of the base packages in OpenWrt. The base packages have been rebuilt (and contain libnl-tiny2023-07-01) but the images are the same as before (and contain libnl-tiny2023-04-02) . Which means "ss" depends on the new libnl-tiny, but that package can't be installed since both packages contain libraries with the same soname.

tmn505, 2023-07-24:
So this looks to be worse than it seemed to me. Maybe the buildbots were not in sync or something in buildbots is broken right now. Can't judge where is the issue as I'm not familiar with that matter.

brada4, 2023-07-24:
One can remove package metadata and replace .so via installation of new version. At least dropbear and busybox does not need it, so you have a way to reflash if it goes foul.

closed 2023-07-27 by openwrt-bot

jow-, 2023-07-27:
The fix will take one or two days to propagate to the package repos, afterwards all should work as expected.

mikk150, 2023-07-27:
I don't think it will be fixed until new RC comes out, as current images are already compiled with `libnl-tiny2023-04-02`, and now new `libnl-tiny` should be `libnl-tiny1` if I am not mistaken but I take this as a fix, as it will evenually fix the issue

jow-, 2023-07-27:
> I don't think it will be fixed until new RC comes out, [...]

The `libnl-tiny1` package will ship `/usr/lib/libnl-tiny.so.1` instead of `/usr/lib/libnl-tiny.so`, so it can be installed alongside the shipped version on older images.

commit 4af0a72a (2023-07-27): libnl-tiny: update to latest Git HEAD

8667347 build: allow passing SOVERSION value for dynamic library

Also adjust packaging of the library to only ship the SOVERSION
suffixed library object, to allow for concurrent installation of
ABI-incompible versions in the future.

Fixes: #13082
commit 77b8ce64 (2023-07-27): libnl-tiny: update to latest Git HEAD

8667347 build: allow passing SOVERSION value for dynamic library

Also adjust packaging of the library to only ship the SOVERSION
suffixed library object, to allow for concurrent installation of
ABI-incompible versions in the future.

Fixes: #13082

(cherry picked from commit 4af0a72a65d7c92ed4e7c2455090f695f424903d)