openwrt/openwrt #13090: misleading upgrade warning

thomas374b, 2023-07-11:
### Describe the bug

during upgrade my Mikrotik RB-951G-2HnD I got the following warning

"Device rb-951g-2hnd not supported by this image Supported devices: mikrotik,routerboard-951g-2hnd rb-951g-2hnd - Image version mismatch: image 1.1, device 1.0. Please wipe config during upgrade (force required) or reinstall. Reason: NAND images switched to yafut. If running older image, reinstall from initramfs. Invalid sysupgrade file. Image check failed. "

It states that my device is not supported but later in the text it states that it is supported

On the label of the device it is written  "RB951G-2HnD"

The image states that it is for
      routerboard-951g-2hnd  rb-951g-2hnd

Where is the error ? I guess I have the right image nevertheless I got this warning

I dont know, should I continue or not ?

### OpenWrt version

SNAPSHOT via firmware builder

### OpenWrt target/subtarget

ar71xx/mikrotik

### Device

MikroTik RouterBOARD 951G-2HnD

### Image kind

Official downloaded image

### Steps to reproduce

- netboot    openwrt-19.07.10-ar71xx-mikrotik-vmlinux-initramfs.elf     (this is the last image to find for RB951G-2HnD)
- login 192.168.1.1  System->Backup/Flash Firmware    flash image from 
        https://sysupgrade.openwrt.org/store/cb259f7c45a4d931fcae6ad6e8d53bde/
  

### Actual behaviour

A modal dialog box with a warning appears that the firmware image does not match.

### Expected behaviour

Upgrade continues without warning

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

thomas374b, 2023-07-11:
If one does a 
     cat /tmp/sysinfo/model
and convert everything to lowercase there would be a space char instead of comma compared to the supported devices

supported_devices 
0 | "mikrotik,routerboard-951g-2hnd"
1 | "rb-951g-2hnd"

brada4, 2023-07-11:
It is a way to force to wipe configuration for flash format change. Same happens with ar71->ath79 architecture change, or swconfig-> dsa changes.
This is important part of message that same hardware but different image format is being used. It correctly leads to reset configuration as there is bo way to maintain old config.

> Image version mismatch: image 1.1, device 1.0

john-tho, 2023-07-11:
Not a bug.

> Reason: NAND images switched to yafut. If running older image, reinstall from initramfs.

Do this (install from initramfs image). If you don't, you will not be able to boot the installed kernel.

See the yafut PR if you want the explanation: https://github.com/openwrt/openwrt/pull/12225#issuecomment-1517535996

thomas374b, 2023-07-11:
So how do I install this router ?

I can netboot/tftp
     openwrt-19.07.10-ar71xx-mikrotik-vmlinux-initramfs.elf
I can't
     sysupgrade openwrt-19.07.10-ar71xx-mikrotik-nand-large-squashfs-sysupgrade.bin

I can netboot/tftp
    openwrt-ath79-mikrotik-mikrotik_routerboard-951g-2hnd-initramfs-kernel.bin  (SNAPSHOT)
I can't
   sysupgrade openwrt-ath79-mikrotik-mikrotik_routerboard-951g-2hnd-squashfs-sysupgrade.bin

Both methods leave me with an unresponsive router (bricked until next netboot/tftp restart)

The sysupgrade program kicks me out of ssh and gives no extra clues.  I tried also

   sysupgrade -n -i -v image
  
When booting with 19.x version I can see 3 MTD partitions with cat /proc/mtd.
With the snapshot version I see more i.e. the partitions are there..

thomas374b, 2023-07-11:
I got to know that most probably I have crappy hardware. I cant update the flash. I can only boot via netboot/tftp.

When I run '''dmesg'''  I get a lot Bad eraseblock warnings and a segfault/kernel panic

Any help with getting this device back to life is appreciated

[indices.txt](https://github.com/openwrt/openwrt/files/12022262/indices.txt)

Djfe, 2023-07-13:
@john-tho 
still reading on this, but if a full reinstall is required, why was it not a mayor compat increase then (prevent sysupgrade with config reset)
Do you agree?
I'll read the rest now

thomas374b, 2023-07-15:
I tried to install this device several times. I'm not sure if I ticked the "keep configurations" checkbox the first or the second time.
I tried all combinations with latest snapshot version and version 19.x

I think the main reason for this install to fail are 22 bad eraseblocks laying exactly at the kernel location starting at 0x40000
and covering this partition fully.  The kernel cant get written, thus the device cant boot.
I'm wondering if this is a new measure of the manufacturer to avoid getting "foreign" OS installed to their devices.  The router was brand new and I cant believe, that there was the incidence for bad eraseblocks to cover the kernel partition exactly and fully.
If I read this partition there is something that looks like a kernel image, but i cant check if all bits are valid.

In order to fix this device, one would need to modify the mikrotik bootloader to change the mtdpart kernel argument to point to a different region. Enough space is there (it has 128M flash and RAM). I could compile a custom kernel and "netboot" from it, but the device would still not reboot on its own if the bootloader is not also modified.

Djfe, 2023-07-15:
So you have tried runnning `sysupgrade -n imagename` from netboot then?

thomas374b, 2023-07-18:
Yes

thomas374b, 2023-07-18:
The device was sent back.
I gave up getting this device running.