openwrt/openwrt #13102: patchelf failed to compile with gcc-10.2

thomas374b, 2023-07-12:
### Describe the bug

latest snapshot cant be built

Solution: step back to patchelf-0.17 (see additional info)

### OpenWrt version

SNAPSHOT

### OpenWrt target/subtarget

ar71xx/mikrotik

### Device

MikroTik RouterBOARD 951G-2HnD

### Image kind

Official downloaded image

### Steps to reproduce

- be on a Devuan Chimaera distribution (maybe equivalent to Debian Buster)

- cd src/OpenWrt
- git pull
- make -j1 -V=sc

### Actual behaviour

build fails when building patchelf

### Expected behaviour

build continues

### Additional info

the build continues with the following changes 

--- tools/patchelf/Makefile-orig	2023-07-12 15:50:06.000000000 +0200
+++ tools/patchelf/Makefile	2023-07-12 17:22:09.000000000 +0200
@@ -7,11 +7,11 @@
 include $(TOPDIR)/rules.mk
 
 PKG_NAME:=patchelf
-PKG_VERSION:=0.18.0
+PKG_VERSION:=0.17.0
 
 PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.bz2
 PKG_SOURCE_URL:=https://github.com/NixOS/patchelf/releases/download/$(PKG_VERSION)
-PKG_HASH:=1952b2a782ba576279c211ee942e341748fdb44997f704dd53def46cd055470b
+PKG_HASH:=45d76f4a31688a523718ec512f31650b1f35d1affec3eafeb3feeb5448d341e1
 
 HOST_BUILD_PARALLEL:=1
 HOST_FIXUP:=autoreconf

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ynezz, 2023-07-13:
> build fails when building patchelf

Would it be possible to include actual build failure, relevant output from `make -j1 -V=sc` should be good enough.

thomas374b, 2023-07-18:
Sorry, unfortunately I cant reproduce this error.  In between I did a
     apt-get upgrade

followed by
    cd src/openwrt
    rm -rf staging_dir
    rm -rf build_dir

Now compilation hangs on
   build_dir/target-mips_24kc_musl/acl-2.3.1

tools/chacl.c: In function 'walk_dir':
tools/chacl.c:335:21: warning: implicit declaration of function 'readdir64'; did you mean 'readdir_r'? [-Wimplicit-function-declaration]
  335 |         while ((d = readdir64(dir)) != NULL) {
      |                     ^~~~~~~~~
      |                     readdir_r
tools/chacl.c:335:19: warning: assignment to 'struct dirent64 *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
  335 |         while ((d = readdir64(dir)) != NULL) {

KanjiMonster, 2023-08-04:
https://github.com/NixOS/patchelf/issues/506 seems to be the appropriate upstream issue. At a first guess it is using a GCC / C++ feature that was only implemented in newer GCC versions (maybe a c++20 feature?).

lb1425, 2023-08-04:
The compilation error takes place in the patchelf tool as mentioned by @KanjiMonster . In the main/HEAD version (commit 1eb67cb0704a61f49d9be9f88aecbce1d881f8c9), I fixed it by using and older version of patchelf. Here is the git diff:

diff --git a/tools/patchelf/Makefile b/tools/patchelf/Makefile
index bbda3a9e40..4cf039c407 100644
--- a/tools/patchelf/Makefile
+++ b/tools/patchelf/Makefile
@@ -7,11 +7,12 @@
 include $(TOPDIR)/rules.mk
 
 PKG_NAME:=patchelf
-PKG_VERSION:=0.18.0
 
-PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.bz2
-PKG_SOURCE_URL:=https://github.com/NixOS/patchelf/releases/download/$(PKG_VERSION)
-PKG_HASH:=1952b2a782ba576279c211ee942e341748fdb44997f704dd53def46cd055470b
+PKG_SOURCE_PROTO:=git
+PKG_SOURCE_URL:=https://github.com/NixOS/patchelf.git
+PKG_SOURCE_VERSION:=f34751b88bd07d7f44f5cd3200fb4122bf916c7e
+PKG_SOURCE_DATE:=2020-12-07
+PKG_MIRROR_HASH:=ac746930b919b97da40f259cfc9ab7bbd48a0c9cbf2eebd8cee5ae19a94356fd
 
 HOST_BUILD_PARALLEL:=1
 HOST_FIXUP:=autoreconf

Regards

KanjiMonster, 2023-08-05:
I gave it a small with diferent GCC versions:

GCC 6 (our current minimum according to include/prereq-build.mk does not even pass the configure step:

```
checking dependency style of gcc-6... none
checking whether g++-6 supports C++17 features with -std=c++17... no
checking whether g++-6 supports C++17 features with +std=c++17... no
checking whether g++-6 supports C++17 features with -h std=c++17... no
checking whether g++-6 supports C++17 features with -std=c++1z... no
checking whether g++-6 supports C++17 features with +std=c++1z... no
checking whether g++-6 supports C++17 features with -h std=c++1z... no
configure: error: *** A compiler with support for C++17 language features is required.
```

GCC 7 fails with the above error.

GCC 8 works.

Going back patchelf versions:

patchelf 0.17.x was the last one compiling with GCC 7.

patchelf 0.13.x was the last one compiling with GCC 6.

So unless we want to go back to patchelf 0.13 we should probably bump our GCC requirement.

neheb, 2023-08-12:
Fixed with https://github.com/NixOS/patchelf/pull/513

neheb, 2024-06-15:
This is fixed. @aparcar please close

commit f147569f (2023-08-15): patchelf: Revert "tools/patchelf: update to 0.18.0"

This reverts commit ec6bcda8e49815efeee845da50defc85ae068b79.

patchelf 0.18.0 requires GCC 8, it does not compile using GCC 7 any
more. Downgrade to patchelf 0.17.2 which still supports GCC 7.

Downgrade patchelf to 0.17.2 only in the OpenWrt 23.05 branch and
require GCC 8 in the master branch.

Ubuntu 18.04 uses GCC 7 by default.

Fixes: #13102
PR #13276 (2023-08-15): patchelf: Revert "tools/patchelf: update to 0.18.0"
This reverts commit ec6bcda8e49815efeee845da50defc85ae068b79.

patchelf 0.18.0 requires GCC 8, it does not compile using GCC 7 any more. Downgrade to patchelf 0.17.2 which still supports GCC 7.

Downgrade patchelf to 0.17.2 only in the OpenWrt 23.05 branch and require GCC 8 in the master branch.

Ubuntu 18.04 uses GCC 7 by default.

Fixed: #13102 

We should require GCC 8 in master, see #12188
closed 2024-06-17 by aparcar